ghaerr/elks #886: Failed install with sys

toncho11, 2020-11-29:
The sys command does not complete the installation: no files are copied to /bin and /etc folders. 
The cp command does not give a reason for the failure or provides an error code - nothing. 
It happens at the end of the copy process, because it takes time (proportional to the file I think) before it fails. 
The system can boot after that even without the files in /bin with some errors.
Both fat16 and fat32 are affected.

ghaerr, 2020-11-29:
Hi @toncho11,

Thanks for the bug report. This all used to work, I wonder if I broke something with the `cp -R` fix I added afterwards. I'll check into it.

To repeat this, one just boots from FAT, then uses `sys /dev/hda1` to copy the system to the newly created partition, right? I just want to be sure you have nothing mounted on /mnt, this is done directly after booting, correct?

Thank you!

ghaerr, 2020-11-30:
@toncho11,

I've duplicated your problem, and fixed it in #887. Sorry!

Thank you!

toncho11, 2020-11-30:
Yes, it works.

PR #887 (2020-11-30): [cmds] Fix cp command
Fixes #886. 

So much for a perfect v0.4.0. Looks like my `cp -R` enhancement broke cp without -R.
closed 2020-11-30 by toncho11