ghaerr/elks #892: cron

toncho11, 2020-12-03:


ghaerr, 2020-12-03:
There are two programs associated with cron: `cron` and `crontab`. Can you give specifics about exactly how you started either program? The automatic cron doesn't run at boot by default. This is a contributed program by @georgp24, I don't recall exactly how it is supposed to work, but it did work successfully last time I played with it.

georgp24, 2020-12-04:
Hi Greg,
I have not used it for quite some time but I wrote documentation files, I hope these help:
elks/Documentation/html/user/cron.html
elks/elkscmd/cron/cron.8 
Georg

From: Gregory Haerr 
Sent: Friday, December 4, 2020 12:42 AM
To: jbruchon/elks 
Cc: georgp24 ; Mention 
Subject: Re: [jbruchon/elks] cron (#892)

There are two programs associated with cron: cron and crontab. Can you give specifics about exactly how you started either program? The automatic cron doesn't run at boot by default. This is a contributed program by @georgp24, I don't recall exactly how it is supposed to work, but it did work successfully last time I played with it.

toncho11, 2020-12-04:
Thanks!

I have noticed a problem though. The kernel is reporting something about slots:

It says "Only 0 slots" or "Only 1 slots".

I think it is here in elks/kernel/fork.c
```
if (task_slots_unused <= 1) {
   printk("Only %d slots\n", task_slots_unused);

```

@georgp24 so one needs to increase the POLL_MAX in the limits.h according to your documentatiuon. I am having a single job every minute. This should be possible and should not cause an error of any sort, neither it should require a kernel recompile. In cron.log I see complaints about "job fork() try again - ". 

This is the first time I am using ELKS on QEMU for Windows.

toncho11, 2020-12-04:
If I do `ps -l` I see mutiple instances of cron. These were not created by me incidentally calling cron. It seems it is multiplying until it reaches the limit? After 11 crons it starts to complain about the slots. And the process limit is 16?

ghaerr, 2020-12-04:
Hello @toncho11,

Well, you are doing a good job at continuing to find ELKS programs that don't work, as usual.

> I have noticed a problem though. The kernel is reporting something about slots:
It says "Only 0 slots" or "Only 1 slots".

This somewhat confusing kernel message is trying to say that almost all the task slots are in use. I'll update it to say "task slots" to be better understood.

> If I do ps -l I see mutiple instances of cron. These were not created by me calling cron. It seems it is multiplying until it reaches the limit?

This is the actual problem. It seems that a previously reported with cron is still not fixed. Cron forks processes to do its work and there is a problem with the kernel `wait` function, or in cron, to get their termination exit status. The kernel keeps the task slot in-use until the wait is executed, which isn't happening, until no more task slots are available. I'll take another look at it and fix it.

> so one needs to increase the POLL_MAX in the limits.h according to your documentation.

This is incorrect, there is no need to increase POLL_MAX nor recompile the kernel.

Thank you!

toncho11, 2020-12-04:
@georgp24 

I am looking into runjob.c. On line 71 there is another fork() on some condition? I mean the previous one at line 54 was to execute the user's command. Still in parent process we do another fork on line 71. Why is that ? What is the "input" field for?

georgp24, 2020-12-05:
I am currently working on a different project and do not have time for cron. I do not remember the details of the code anymore. It took me quite a while to understand the code because I had not written code that opened and closed processes before. What I did was to port an old version of cron to ELKS, tested and documented it. Then wait.c was changed and it did not work anymore. This was very frustrating for me.

toncho11, 2020-12-05:
@georgp24 You are referring to #609. 

So the potential problem is in:
elks/elks/kernel/exit.c in 

`int sys_wait4(pid_t pid, int *status, int options)`

toncho11, 2020-12-05:
There are also some comments in do_exit() which suggest that something was left unimplemented - because there were no signals. But now there are signals in ELKS, right? It is on line 131.

toncho11, 2020-12-06:
Is vfork() implemented?

`The vfork() function is the same as fork() except that it does not make a copy of the address space. The memory is shared reducing the overhead of spawning a new process with a unique copy of all the memory. This is typically used when using fork() to exec() a process and terminate. The vfork() function also executes the child process first and resumes the parent process when the child terminates.`

ssieb, 2020-12-06:
There's no MMU, so if it exists, it will be the same as fork() because there's no way to do COW.

ghaerr, 2020-12-06:
Hello @toncho11,

> So the potential problem is in:
elks/elks/kernel/exit.c in
int sys_wait4(pid_t pid, int *status, int options)

Yes. I have made progress fixing `sys_wait4`, but there are still problems being exposed due to the large number of forks/processes cron uses to do anything, which are in turn exposing other problems in cron and ELKS.

> Is vfork() implemented? It can be used in cron before executing the scheduled command.

@ssieb is right, since there's no hardware copy on write, vfork will be the same as fork. Vfork was added back in early BSD when it was realized that lots of CPU cycles were being wasted with the oft-used fork/exec UNIX paradigm copying entire data segments only to exec another file shortly thereafter. In order to improve performance, vfork also forced the child to run first, since the parent was typically going to just `wait` anyways. ELKS implemented this latter mechanism, but its been commented out for quite a while. I've been looking at that and I think the reason it's commented out is that it implements a truly shared data segment for the time between fork and exec, *without* copy on write, which means the parent process local data and stack are also changed - too dangerous for non-aware ELKS applications.

Thank you!

toncho11, 2020-12-06:
I think this is important. It is related to ELKS main advantage - being multitasking by design.
I also want to use ELSK and my Amstrad 1640 to control some things in my home. For example I would like to use cron to power on my Christmas lights at 18:00 until 22:00 :). I will use the parallel port for the communication. 

Thank you @ghaerr !!!

ghaerr, 2020-12-07:
Hello @toncho11,

I've fixed the `wait` problem with cron in #895. However, I'm still unsure as to whether cron's usage of `sleep` to time job execution actually works properly. It definitely does not on QEMU. You might play with cron a bit more on real hardware, now that I know the kernel works properly with fork/wait.

Good luck with your Christmas tree lights!!

Thank you.

toncho11, 2020-12-07:
Thank you @ghaerr I see it was a lot of work!

I am testing now. I have the impression that cron is not starting at all or it is exiting immediately.

ghaerr, 2020-12-07:
@toncho11,

Make sure you're running the fetched and checked-out version of #895. I have just tested it on QEMU (graphics console only), and it is working fine there. This could be further debugged on your end by setting TEST=1 in cron's config.h file on line 8, which will then run commands every two seconds, and display that on the console.

If /var/cron/root does not already exist when starting cron, it will be created with default values, which run a job every minute and separately, a job every 10 minutes.

toncho11, 2020-12-08:
I am using hd.bin to test. I have set manually the commands including cron.
The .bin image is converted to .vdi and executed in VirtualBox. 

So I have TEST = 1. When it it starts it creates the root file in /var/log. With TEST=1 it created a "0-59 * ... Message 1 by cron >> cron.out"

But still I do not see cron with ps -l . Where is cron.out supposed to be generated? There were no other messages.

ghaerr, 2020-12-08:
> But still I do not see cron with ps -l . Where is cron.out supposed to be generated? There were no other messages.

cron.out is generated in /root (root's home directory).
Directly after boot, running cron should result in creating /var/cron/root (cat it or use crontab -l to see it). Ps should show cron as a task.

> I have also set the CPU execution cap to 1%, but there was no difference for cron. ELKS boots slower though.

I cannot recommend setting any CPU limits on execution, this will definitely affect system timing, of which cron depends. Turn that off for testing cron. 

> In QEMU is the same.
@Mellvik Can you test please? Does cron stays and execute tasks?

@Mellvik has confirmed cron works on QEMU https://github.com/jbruchon/elks/pull/895#issuecomment-740573129.

You are adding a number of variable using cron on hd.bin, I assume you're using a FAT filesystem? So far, I have only run cron on fd1440.bin using Minix filesystem and floppy disk. You might check that the /var/cron directory exists beforehand, I would bet no one has run cron on FAT, but could be other issues as well.

ghaerr, 2020-12-08:
@toncho11,

I have just confirmed cron doesn't run on FAT filesystems. For a temporary fix, comment out line 172 in cron.c: it quits when seeing that /var is group writable, which is always the case for FAT. Commenting out that line will allow cron to run for your testing:
```
    //if ( sb.st_mode & (S_IWGRP|S_IWOTH) ) fatal("%s: group or world writable", part);
```
I found this by running cron on FAT, then examining /var/cron/cron.log - it gave this very error.

Mellvik, 2020-12-08:
@ghaerr,
I'm unsure why the behaviour of cron is different on physical hardware - it must have to do with speed. Anyway, what I'm seeing is running out of memory and tasks all the time, even though there are only 10 processes running.

And - I stand corrected – I now get the same delays as in qemu, this time in a telnet connection. I've added some more lines in crontab (root), and while there may be bugs in cron (seems like nested forks), this seems/feels like something else.

-M

-----------------------------------------------------------------
cron: sleep start 2
cron: running 'ls'
Only 1 task slots
/bin/sh/bin/sh: : /bin/sh: Out of memory Out of memory 

Out of memory 
cron: sleep return 2
cron: sleep start 2
cron: sleep return 0
cron: running 'echo Message 1 by cron >>cron.out'
cron: running 'date > /dev/console'
cron: sleep start 2
cron: running 'ls'
Only 1 task slots
Only 1 task slots
Tue Dec 08 18:12:33 2020
cron.out  
cron: sleep return -1
cron: running 'echo Message 1 by cron >>cron.out'
cron: sleep start 2
cron: running 'date > /dev/console'
cron: running 'ls'
Only 1 task slots
Only 0 task slots
Cannot fork
cron.out  
cron: sleep return -1
cron: running 'echo Message 1 by cron >>cron.out'
cron: sleep start 2
cron: running 'date > /dev/console'
cron: running 'ls'
Only 1 task slots
Tue Dec 08 18:12:41 2020
Only 1 task slots
cron.out  
cron: sleep return -2
cron: running 'echo Message 1 by cron >>cron.out'
cron: running 'date > /dev/console'
cron: sleep start 2
Only 1 task slots
cron: running 'ls'
Only 1 task slots
Tue Dec 08 18:12:47 2020
cron.out  
cron: sleep return 0
cron: running 'echo Message 1 by cron >>cron.out'
cron: running 'date > /dev/console'
cron: sleep start 2
cron: running 'ls'
Only 1 task slots
Only 0 task slots
Cannot fork
/bin/sh/bin/sh: : Out of memory Out of memory 

cron: sleep return 0
cron: running 'echo Message 1 by cron >>cron.out'
cron: sleep start 2
cron: running 'date > /dev/console'
cron: running 'ls'

toncho11, 2020-12-08:
I am currenly using:

- VirtualBox on 100% (no execution cap)
- cron with TEST disabled 
- on fat with commented group checking
- with line: */2 * * * * clock >> cron.out

And it works. No multiple crons and the timing is OK. Every 2 minutes I get a clock message. I checked with my local clock and it seems OK.

toncho11, 2020-12-08:
Hi again,

I am running it on my Amstrad 1640 and it looks OK.

- cron with TEST disabled
- on fat with commented group checking
- with line: */2 * * * * clock >> cron.out

The output of clock in cron.out is every 2 minutes.

ghaerr, 2020-12-09:
@Mellvik,

> I'm unsure why the behaviour of cron is different on physical hardware - it must have to do with speed. 
> I've added some more lines in crontab (root)

Well, when one makes changes to the software mid-test by adding crontab entries, whilst also changing platforms, we are no longer comparing apples to apples, but apples to oranges, by adding new variables to the equation...

> Anyway, what I'm seeing is running out of memory and tasks all the time, even though there are only 10 processes running.

> (I'm cross-post responding from this comment in #895):
> BTW - I just noticed - cron is a quite large process(38k), and I am getting sh: out of memory messages for periods.

Since there are now added variables, I am pulling out my multi-variable-analysis sliderule to solve the new multi-variable equation... and what does it come up with? 

I will take a guess at the solution (and should I be right, someone can perhaps buy me lunch 😊 :) 

Looking at all the comments and information on this thread, the answer is obvious: cron is working fine but the added lines to the crontab table are causing ELKS to run out of processes and memory!

How could this be? Well, there are only 16 tasks total, which is 5 left before the "Only 1 tasks left" messages start coming. Cron checks the crontab file every 2 seconds and, for every line in the crontab file, it:
- forks once to allow the original, master cron to keep running, which immediately reads the next crontab line entry;
- forks again to produce a worker task which will run the command, while the parent waits for completion;
- execs "/bin/sh -c command >> output" to parse the command arguments for execl/execv and process file redirection;
- /bin/sh forks again to actually run the command once the arguments are parsed and redirection setup, while waiting for the command to complete.

Each fork copies the caller's data segment; cron is already requesting a 0x3800-sized heap in its Makefile (for some reason, I suspect because it was running out of memory processing too quickly), looking at `ps` shows cron has an 18k (heap+free) data segment; and /bin/sh has a 10k data segment.

That's 3 forks and @ 18k + 10k = 64k per crontab line plus memory for whatever the cron job is. With networking running, we start with only 243K free.

So - starting with 6 extra task slots and 243K, with cron running multiple crontab lines and all of them loaded and trying to execute at the same time, I am not surprised.

This `cron` was written with the idea that tasks are almost free, low-cost. And they are on Linux. On ELKS, memory and tasks are in short supply. Should my guess be correct, cron could be modified so that each job doesn't fork just so cron can keep running, but instead run each job run in-sequence. Of course, should a job take a while or not complete, this will cause problems (which it would anyways). For those that would like to experiment, the code is at the bottom of runjob.c.

ghaerr, 2020-12-09:
> And - I stand corrected – I now get the same delays as in qemu, this time in a telnet connection.

Can you be more specific as to what the telnet connection is, and when exactly the delays occur?

I am treating this as completely seperate from cron issues - what we are talking about is the ELKS timer is seemingly happening slower, or being inhibited, or simply not occurring for some other reason, which directly affects the "jiffies" (100 HZ clock tick) kernel time value, which affects the cron, since it calls `sleep`. In QEMU, mouse-down on the window frame will immediately cause a clock tick and jiffies update, which is strange since there is no new I/O occurring within ELKS.

> there may be bugs in cron (seems like nested forks), this seems/feels like something else.

I don't think cron is buggy, explanation in previous post. The question is whether the timer issue is in ELKS, or just QEMU.

ghaerr, 2020-12-09:
@toncho11,

Thanks for your testing on VB and real hardware. Should my guess above be correct, I'll commit PR #895 and deal with the QEMU slowdown issue separately. Let me know when you have your Christmas lights working!

toncho11, 2020-12-09:
Ok, so to avoid problems with cron and new entries one needs to:

- kill cron
- modify the crontab entries
- start cron again

This will still not be perfect because if it is a task "every 2 minutes" then such task will affected by the time it takes to modify the file. Still it will avoid all of the above problems.

Another possibility is to add:
`cron -reload`
to avoid manual stopping (killing) of cron which can work:
- identifying that this is a second instance cron
- sending signal to first one to die and we continue with the new one
- sending a signal to first one to reset - wait for all forks to complete, read crontab file, start waiting

ghaerr, 2020-12-09:
@toncho11,

`crontab <file>| -l | -e` should already do what you're suggesting above. See the man page and crontab.c in the cron source directory for details.

toncho11, 2020-12-09:
So I need to use `crontab` -e because it takes care of reloading the crontab file.
If I use kilo and edit it manually I will need to restart cron myself. Also manual editing might lead to cron multiplying processes?
Is that correct?

Mellvik, 2020-12-10:
> 
> And - I stand corrected – I now get the same delays as in qemu, this time in a telnet connection.
> 
> Can you be more specific as to what the telnet connection is, and when exactly the delays occur?
> 
As far as I can tell (more testing as soon as I can), the behavior is exactly as with the serial console in QEMU. 
> I am treating this as completely seperate from cron issues
> 
I agree, this has nothing to do with cron.
> - what we are talking about is the ELKS timer is seemingly happening slower, or being inhibited, or simply not occurring for some other reason, which directly affects the "jiffies" (100 HZ clock [...]
We need to create a completely repeatable scenatrio for this and take it from there. I believe the problem has been there for a while, I have ignored it - attributing it to QEMU and never seen it in a physical setting - until now.

—M

Mellvik, 2020-12-10:
> I'm unsure why the behaviour of cron is different on physical hardware - it must have to do with speed.
> I've added some more lines in crontab (root)
>
> Well, when one makes changes to the software [...]
Just different approaches AFAIK - making changes in order to investigate an issue, provoke some behavior, look for patterns etc. is my normal approach, in particular when I don’t know what i’m looking for.
> Anyway, what I'm seeing is running out of memory and tasks all the time, even though there are only 10 processes running.
>
> (I'm cross-post responding from this comment in #895 [...]
I like your slide rule approach - I did the same and got a different result because my primary number was wrong. Don’t know why I had the impression that ELKS had 20 process slots. I messed up. Lunch at first opportunity regardless!
> This cron was written with the idea that tasks are almost free, low-cost. And they are on Linux. On ELKS, memory and tasks are in short supply. Should my guess be correct, cron could be modified so [...]
From my perspective, cron is fine. Meaning it’s there and works if someone needs it. To me, getting an update() daemon running would be much more important. I’ll submit a PR for that when I’ve tested it. Just a few lines of code to sync() every 30 seconds.

—Mellvik

ghaerr, 2020-12-10:
>  To me, getting an update() daemon running would be much more important. I’ll submit a PR for that when I’ve tested it. Just a few lines of code to sync() every 30 seconds.

I'm not sure its worth wasting a task slot for this, since they're in such short supply. Perhaps a better idea would be for task 0, the idle task, to do it, using some sort of algorithm (or a way to turn it on/off).

Mellvik, 2020-12-10:
> To me, getting an update() daemon running would be much more important. I’ll submit a PR for that when I’ve tested it. Just a few lines of code to sync() every 30 seconds.
>
> I'm not sure its worth [...]
Very valid point, @ghaerr. If task 0 can handle signals it should be easy. Otherwise maybe init can do it …

From BSD:

dosync() {
        sync();
        signal(SIGALRM, dosync);
        alarm(30);
}

—Mellvik

toncho11, 2020-12-11:
Ok I closing this as now we have a new improved version of cron thanks to @ghaerr .

PR #895 (2020-12-13): [kernel] Rewrite wait system call, fixes cron
Fixes #892.

There was a race condition between reaping process exit status and re-using unused task structures, which caused `cron` to fail. `sys_wait4` was rewritten to eliminate this problem. Handling of orphans is now performed at process exit time, rather than in wait, which fixes the race problem. Also previously, orphaned zombies were reparented to init, for reaping exit status. Now, the task structure entry is marked unused, for better use of the limited task structure during fast task creation periods.

Fixed a number of kernel problems wh [... cut]
closed 2020-12-11 by toncho11