ghaerr/elks #896: vi and hidden files

toncho11, 2020-12-10:
kilo is too slow be of any use on 8088.

I am trying to use vi, but backspace does not do what it is supposed to do. 

One solution is to: 

`vi .vimrc`
and then these commands

```
$ set nocompatible
$ set backspace=2
```

but then ... I can not save the file. Presumably because it is starting with "." and I am on FAT.

What are your thoughts? Using FAT helps a lot when using Windows.

ghaerr, 2020-12-11:
> Presumably because it is starting with "." and I am on FAT.

Yep.

The proper and best solution is to add support for long filenames in FAT on ELKS. There are two areas within ELKS filesystem code where long filename issues arise: in directory listings, and in file open/create. The directory listings are implemented, which is why long filenames like /root/.profile or /etc/resolv.conf can be built and seen, but not opened or created at runtime. I've been meaning to implement the remainder of FAT long filenames, but its complicated.

Another quick solution would be to hack vi and change .vimrc to vim.rc. That might be the best short term option.

A third thought would be to have you put "stty erase ^?" in /etc/profile and see whether that does anything. I don't think vi reads the termios values for backspace but it could.

> Or would you agree that we set BACKSPACE to delete characters by default?

No, we can't change the default permanently without affecting lots of other things. Although one could provide this as a special custom entry in one of the Direct Console keyboard tables.

toncho11, 2020-12-13:
So vi comes from the Elvis project?
I searched for  .vimrc in the source, but I could not find it. So it is not possible to configure such functionality?

ghaerr, 2020-12-13:
> OK ... it is called ".exrc". And the line for the backspace is: :map! ^? ^H

Is that the line that should fix this problem? I know it won't work on FAT, because of the filename.

It looks like vi supports the environment variable EXINIT; can you test using `export EXINIT=":map! ^? ^H"` in the shell? That will determine whether this fix will work for you. Afterwards, we can change the name of .exrc to ex.rc (or vim.rc or vi.rc), at least until I find time to fix the FAT long filename open problem.

toncho11, 2020-12-13:
I am trying different commands: 

https://unix.stackexchange.com/questions/23200/switch-vi-to-use-backspace-key-instead-of-controlh-for-backspace

I could not make it work. I tried typing the mapping commands above.

Mellvik, 2020-12-13:
Another comment from the sideline: 

I believe it’s a good idea to rename the ex-origin files. Unless I’m mistaken, ex is not part of this version of vi (and nobody misses it), so .virc (or vi.rc, wow do I hate FAT) would be good, along with a rename of the environment variables.

BTW - now that we’re on the vi subject, maybe it’s a good opporutiny to fix the vi lock file problem:
If vi bombs, it leaves a lock file behind, preventing a reopen of that file with a new instance of vi. I’ve har this on my list for a while, and I’ve been assuming the open routine in vi, where the lock file is detected, could check if the process owning the lock file still runs. Or something like that.

—Mellvik

ghaerr, 2020-12-13:
> I could not make it work. I tried typing the mapping commands above and your suggestion.

How about renaming changing the following line in the ANY_UNIX section of vi's config.h:
```
#  define EXRC      "ex.rc"     /* init file in current directory */
```
recompiling, then setting the following in ex.rc in the /root directory:
```
$ set nocompatible
$ set backspace=2
```
Lets get this resolved before we rename .exrc to probably vi.rc.

> wow do I hate FAT

@Mellvik, LOL!!! :) This is actually an ELKS problem, FAT does support long filenames! :)

I'll look more into the lock file problem, we can beat up vi for awhile now that we've finished getting cron working.

ghaerr, 2020-12-13:
@Mellvik,

I'm looking at the source code, vi creates a temporary file that is required for the file editing process in this version. I'm not sure if this is the "lock file" you speak of, though. What does vi display when the lock file is present at startup? More details will help me figure this out.

Thanks!

ghaerr, 2020-12-13:
@Mellvik,

Try commenting out this section of code in elvis/tmp.c:
```
    /* make sure nobody else is editing the same file */
    if (access(tmpname, 0) == 0)
    {
        if (*origname)
        {
            msg("\"%s\" is busy", filename);
            return tmpstart("");
        }
        FAIL("\"%s\" is busy", filename);
    }
```
That should fix the problem, if this is the error that you are getting.

You may need to add `unlink(tmpname);` directly after the commented out block in order for things to work, as a creat call is executed following that, which might fail.

Mellvik, 2020-12-13:
Thanks @ghaerr,
This is definitely it.

The question is whether to just delete the file or actually check for validity. How much of an effort is it to scan the process table from an application?

- Mellvik

toncho11, 2020-12-13:
Loading the file works now, but vi says that it does not recognize the commands. Remapping of the BS is not working. I am giving up ... How can you make such a bloated editor with so many options and ... no backspace. This is stupid.

Mellvik, 2020-12-13:
FWIW - vi works fine for me.
And - for a screen editor - it's definitely not bloated.

-M

toncho11, 2020-12-13:
I see.

ghaerr, 2020-12-14:
> Remapping of the BS is not working.

@toncho11, can you explain to me a bit more about exactly what you're trying to get the Delete key to do? In looking up "set nocompatible; mouse=2" some people talk about it, but it is not explained. Is it that you want it to continue backing up over previous text, the same way ^H does with Linux/macOS vim? With this version of elvis vi, when I use either Delete (which is top right on my macOS laptop keyboard), **or** ^H, both backspace only to the beginning of the line; that is, they seem to operate identically.

> How about renaming changing the following line in the ANY_UNIX section of vi's config.h:
> \#  define EXRC      "ex.rc"     /* init file in current directory */

I still think we should at least address the issue of .exrc not working on FAT - that is, change it to ex.rc or vi.rc.

toncho11, 2020-12-14:
@ghaerr I just wanted to make it as regular backspace (delete character and move left) like in MS Word, but it is not possible because you need to use commands, config files, etc and in the end something that simple is not possible.

I think it is up to you to decide whether it is better to add support for hidden files (read/write) on FAT or just rename the file and the documentation.

Mellvik, 2020-12-14:
@toncho11, @ghaerr,
FWIW I did some more testing on vi, here’s what it looks like from my end (on physical hardware, serial console):

- when in insert mode, erase moves the cursor back but does not visibly delete the character (until esc is pushed). This is standard vi behavior. Vim is different, and will physically delete the char from the screen using ^H<space>^H. I don’t know anything about the origins of ELKS vi, but suspect it is a descendant of ‘real’ vi, which is a lot simpler (and smaller) than vim. The ‘echoe’ (echo erase) behaviour is most likely a minor fix in vi.

- stty determines the delete character. If you want backspace to erase, set it with stty erase ^H, similar with the delete key. Vi reflects the change at next start.

- Bloat: @toncho11, I take my comment about bloating partly back. While not code wise particularly bloated, vi is bloated in terms of memory consumption. My init 6 command starts an extra getty in addition to what the distribution init 6 does. meminfo says I have 184k available memory when starting vi. Still it fails, not complaining about memory but with ‘vi: No such file or directory’. This is vi started with no arguments.

Other things we have discussed these past few days:
- All references to EX - files and environ) should be changed to ‘vi’.
- init-file: I suggest we change vi to first look for .virc, then for vi.rc. Slightly bloated, but effectively covering both cases without removing the advantage of using a real file system (minix in this case).
- ‘lock file’ - The fix suggested by @ghaerr will work, what about the risk of having several instances of vi editing the same file?

—mellvik

toncho11, 2020-12-14:
- `stty erase ^H` from the ELKS console does not change backspace in vi?
- @Mellvik can you please elaborate on the ‘echoe’ (echo erase) behavior?
- has anybody used the elvis editor? Does it compile?

ghaerr, 2020-12-14:
With regards to @tonch11's request for "backspace to do what its supposed to do", and that backspace "delete characters by default":

- This version of vi always treats "backspace" and "delete" characters identically, so stty has nothing to do with the solution.
- As pointed out by @Mellvik, this version of vi operates different than vim and doesn't change displayed text with BS/DEL until ESC is pressed. This is true, but isn't what I think @toncho11 is asking for.
- This version of vi does not support the "set nocompatible; mouse=2" special non-standard vim mode that @toncho11 is asking for. This mode will delete characters (by pressing Delete or ^H) the same way MS Word operates - which is the way most modern-day "visual" editors (like kilo) work. Vi does not work that way, it requires learning its commands to edit a file.
- The "elvis" editor is the editor we call vi in ELKS. It is renamed from elvis to vi on installation.
- There is source for two other vi "clones" in the elkscmd/. Both are smaller but quite buggy, we used and tested those about a year ago and both were rejected due to not accurately being able to edit a file properly. There is no "vim" port to ELKS; I have no idea whether such a vi clone would be small enough to run in 16 bits.

I ported kilo so that people like @toncho11 would have a visual editor that didn't require learning any commands was available on ELKS. Unfortunately, it has proven to be too large in memory usage, and too slow, to be of much use for ELKS. I suggest finding another very small screen editor for users not already knowledgeable of vi. I took a quick look six months ago, and found nano, but it seems to have grown to be extremely bloated. Perhaps a very early version, if we can find it, would work.

ghaerr, 2020-12-14:
> ‘lock file’ - The fix suggested by @ghaerr will work, what about the risk of having several instances of vi editing the same file?

For ELKS, I highly doubt that will ever happen. ELKS isn't a multi-user replacement for Linux (on older machines). Currently, this "feature" is causing more problems than is solves, so I suggested simply removing it. We just saw the same problem in cron - what started out nice and simple became bloated, as the natural evolution of programs and operating systems is to please more and more users, and the result is complexity and confusion. ELKS should (and must, for size reasons in many cases) remain simple. It doesn't do everything!

Mellvik, 2020-12-14:
> ‘lock file’ - The fix suggested by @ghaerr <https://github.com/ghaerr> will work, what about the risk of having several instances of vi editing the same file?
>
> For ELKS, I highly doubt that will [...]
Agreed!
—M

ghaerr, 2020-12-14:
> stty determines the delete character. If you want backspace to erase, set it with stty erase ^H, similar with the delete key. Vi reflects the change at next start.

@Mellvik: I didn't test this, but I believe you and stand corrected. So Delete and Backspace keystrokes are treated slightly differently, depending on the stty erase setting. However, this still doesn't solve @toncho11's problem.

toncho11, 2020-12-14:
Indeed I have always used nano, since it came out. That is the "problem". Nano is perfect for somebody who does not use Linux as his primary desktop and for work.

Actually I am working on a solution and I might need your help at some point :).

Mellvik, 2020-12-14:
> stty determines the delete character. If you want backspace to erase, set it with stty erase ^H, similar with the delete key. Vi reflects the change at next start.
>
> @Mellvik [...]
@ghaerr,
yes, as far as the VI effect is concerned, I stand corrected. I was too fast concluding that one.

For stty, I’m not sure what you mean is wrong. I did test this and it (still) works as expected (the unix way) and AFAIK DEL and BS work identically when configured to erase (it does not work with linenoise though, so the testing ground is sash). I’ve yet to find a stty setting that works for me - ironically it insists on deleting the current char instead of the previous char. That’s a different story.

-M

closed 2020-12-17 by ghaerr