ghaerr/elks #898: Run ELKS in a browser

ghaerr, 2020-12-29:
Preliminary support has been added to EMU86 that allows an ELKS image to run within a browser. While mostly for play and learning, this could allow testing and running ELKS from a web page, which would work from almost anywhere. 

This is a proof of concept for evaluation and understanding of ELKS dependencies, see https://github.com/mfld-fr/emu86/pull/27 for more details. The current ELKS codebase is such that its dependencies are now well known and organized, allowing ports of ELKS to newer platforms more easily.

The current setup runs the ROM version of ELKS, but running on an internal 1.44M floppy image should not be a big deal (except that the floppy image is internal to the browser, so no filesystem changes would be persistent after exiting the page).

[image]

toncho11, 2021-01-05:
I tested ELKS with hd image here: https://copy.sh/v86/ . Scroll down to select and upload an ELKS image file. This is a javascript x86 emulator and it works. We could put it on the front page so that people can try ELKS ... which will most likely discourage them from testing it on a real hardware ...

ghaerr, 2021-01-08:
@toncho11, interesting differences between the emulators. That version apparently uses data transfer across the internet to get requested ELKS BIOS HD/FD data from the local browser, while my version uses a special Emscripten capability allowing the filesystem to be embedded in the server web page.

pmp-p, 2021-01-31:
>  That version apparently uses data transfer across the internet to get requested ELKS BIOS HD/FD data from the local browser

that relies on setting an very quite internet bridge somewhere https://github.com/benjamincburns/websockproxy ( jor1k )

-> opens doors to all kind of abuse outside the lab and/or people starting thinking that is some new kind of web-virus.

> my version uses a special Emscripten capability allowing the filesystem to be embedded in the server web page.

2) Embedding filesystem costs ram / web archive size and may lead to bad user experience on first load.

-> is extremely usefull for getting started so thank you very much !

but adding using host9p  https://github.com/copy/v86/issues/41 ( v86)  or  https://vfsync.org  (jslinux)  support would allow for **much faster startup and usefullness**.

ghaerr, 2021-01-31:
Hello @pmp-p,

Thank you for your comments and links to possible sync methods! I am not familiar with them but will take a deeper look.

> Embedding filesystem costs ram / web archive size and may lead to bad user experience on first load.
-> but is extremely usefull for getting started so thank you very much !

Yes, this first take was just an attempt at getting the EMU86 interpreter compiled and running on WASM, rather than trying to build a really useful web emulator. However, I did make some changes to allow the asynchronous Emscripten read of the embedded filesystem (.data file) to work well on the first load, as initially I was seeing a race/timing issue with the emulator starting before the read completed.

> or (jslinux) support that would allow for much faster startup and usefullness.

Agreed! At this point, my focus has been on showing proof of concept of getting our ELKS Linux runable on smaller-footprint physical or virtual machines. I started looking at jslinux and got sidetracked into some very interesting implementations of javascript interpreters, but I suppose that's another conversation :)

Thank you!

pmp-p, 2021-04-27:
>  I did make some changes to allow the asynchronous Emscripten read of the embedded filesystem (.data file) to work well on the first load, as initially I was seeing a race/timing issue with the emulator starting before the read completed.

That's very interesting, cold start time and .data late arrival is a really annoying problem for numerous apps eg recently https://github.com/pyodide/pyodide/issues/646#issuecomment-827641595,  would you mind to give more info/what you did to fix that ?

mfld-fr, 2021-04-30:
EMU86 part now tracked by mfld-fr/emu86#32

ghaerr, 2021-05-05:
Hello @pmp-p,

> That's very interesting, cold start time and .data late arrival is a really annoying problem for numerous apps eg recently pyodide/pyodide#646 (comment), would you mind to give more info/what you did to fix that ?

When I initially ported EMU86 (the 8086 CPU emulator) to Emscripten, I was running into late arrival problems as described in your link. However, in looking at the code, it seems all I'm doing is using the `--preload-file=` to get the .data file loaded (supposedly) prior to execution start. Also, the program opens the file, but doesn't try to do a read until actually required, which is always likely quite a bit later, after having returned to control to the browser via `emscripten_sleep(1)` a number of times. EMU86 is operating without errors now, but the .data file is quite small, 64K or so.

You're probably already doing all this. The emscripten [docs](https://emscripten.org/docs/porting/files/packaging_files.html) suggest a file-packaging tool can be used with specialized JS code to load the .data file prior to execution, have you tried that?

Thank you!

EdoaLive, 2021-07-13:
Hello,
I casually came to this topic and I thought about this in the past.
I don't know if you already tried or talked about this, but I think the best browser emulator is [PCjs](https://www.pcjs.org/). 
This really got me impressed with his ability to run the original Olivetti M24 bios ROM: https://www.pcjs.org/machines/pcx86/att/6300/cga/. 
It also seems to have some debug capabilities!
What do you think?
Preparing a configuration that runs ELKS on a browser (eg. directly from the website, with the latest released images) should be pretty easy.

ghaerr, 2021-07-30:
Hello @EdoaLive,

> I don't know if you already tried or talked about this, but I think the best browser emulator is PCjs.
It also seems to have some debug capabilities!
What do you think?

I agree, PCjs is a super way to run almost any code that boots on a PC. At it's core, its an 8086 emulator that interprets both processor instructions as well as emulates the PCjr environment. This will certainly allow ELKS to be booted and run, without modification.

Alternatively, the ELKS kernel can be re-compiled using the Emscripten compiler, which translates the C code into WebAssembly (WASM), which is then interpreted within most browsers using an internal WASM interpreter, with console I/O specially redirected to a web drawing canvas. The filesystem isn't interpreted as a BIOS device, but instead a ROM filesystem is made accessible via Emscripten's file I/O subsystem within the browser.

So, two entirely different ways of accomplishing somewhat the same thing.

> Preparing a configuration that runs ELKS on a browser (eg. directly from the website, with the latest released images) should be pretty easy.

Agreed, that would be a nice thing to do - add the ability to boot and run the ELKS v0.4.0 boot image from the PCjr website.

Thank you!

closed 2022-02-08 by ghaerr