ghaerr/elks #900: Hardware Enablement, NE1000, and Turbo buttons

NCommander, 2021-02-07:
I've been working to get ELKS fully working on a VTech Laser XTSL with a D-Link DE-220C. This machine has a Intel 8086 processor, and there is a DE-220C installed, an NE2000 card that's been put in an 8-bit slot.

ELKS sorta worked as is, but it misidentifies the processor as a V30, and the NIC problem is a bit more problemic. I'm not sure how to fix this issue, so I'm just going to report it.

With some patching, I got the card to work in NE1000 mode by using byte mode and changing how DMA works. This patch is a little hackish since I just force byte transfer mode, and then do two in/out operations, but the card appears to be working properly with this tweak. There is a fair bit of TCP/IP transmissions but telnetd and httpd both work. Here's what I did for NE1000

```
diff --git a/elks/arch/i86/drivers/net/ne2k-asm.S b/elks/arch/i86/drivers/net/ne2k-asm.S
index d06196db..98ea41df 100644
--- a/elks/arch/i86/drivers/net/ne2k-asm.S
+++ b/elks/arch/i86/drivers/net/ne2k-asm.S
@@ -188,8 +188,12 @@ dma_write:
 
 emw_loop:
 
-       lodsw
-       out     %ax,%dx
+       lodsb
+       outb     %al,%dx
+
+       lodsb
+       outb     %al,%dx
+
        loop    emw_loop
 
        // wait for DMA completed
@@ -299,8 +303,12 @@ buf_local:
        mov     $io_ne2k_data_io,%dx
        cld
 emr_loop:
-       in      %dx,%ax
-       stosw
+       inb      %dx,%al
+        stosb
+
+       inb      %dx,%al
+        stosb
+
        loop    emr_loop
 
        // wait for DMA to complete
@@ -692,7 +700,7 @@ ne2k_base_init:
        // data I/O in words for PC/AT and higher
 
        mov     $io_ne2k_data_conf,%dx
-       mov     $0x49,%al       // set word access
+       mov     $0x48,%al       // set word access
        out     %al,%dx
 
        // clear DMA length 
```

I used NetBSD as a reference on how to do 8-bit NEx000 operations, and this is basically the same thing it does if NE_IS_16BIT is false. As a second issue, I also had to change the NE2000 location because on this machine the RTC is at 0x300. This was just changing the hardcoding but maybe move this into menuconfig?:

I'd be welcome to ideas on how to more cleanly implement this; if a runtime or compile time option.

```
diff --git a/elks/include/arch/ports.h b/elks/include/arch/ports.h
index d2c216ef..3f67259c 100644
--- a/elks/include/arch/ports.h
+++ b/elks/include/arch/ports.h
@@ -98,8 +98,10 @@
 
 /* ne2k, ne2k.c */
 //#define NE2K_IRQ     9               /* ensure CONFIG_NEED_IRQxx set properly above*/
-#define NE2K_IRQ       12
-#define NE2K_PORT      0x300
+//#define NE2K_IRQ     12
+//#define NE2K_PORT    0x300
+#define NE2K_IRQ       3
+#define NE2K_PORT      0x260
```

A bigger issue is maybe a mechansim for engaging Turbo. I dropped the following assembly in setup.S to engage Turbo, but obviously its a hack, and I wonder if some "turbo enable" command might be a good idea.

```
diff --git a/elks/arch/i86/boot/setup.S b/elks/arch/i86/boot/setup.S
index 0da4e1d2..f5f2a85a 100644
--- a/elks/arch/i86/boot/setup.S
+++ b/elks/arch/i86/boot/setup.S
@@ -212,6 +212,14 @@ no_blob:
 // Entry point for EMU86 ROM emulation
 
 start_os:
+       push %ax
+       push %dx
+       mov $0x80, %al
+       mov $0x01F0, %dx
+       out %al, %dx
+       pop %dx
+       pop %ax
+
        push    %cs
        pop     %ds
        mov     $hello_mess,%si
```

I'll send pull requests once I have an idea what these patches should look like.

Mellvik, 2021-02-07:
Hi Michael,

This is some great work you're doing. You've obviously already gained a lot of understanding of some of the lower level parts of ELKS.

I'm surprised that the NE2K card actually works in a 8bit slot at all - so this is all good news. You need to change the byte counter in the dma-setup routines too, maybe you did that already. 
When this works reliably, which seems to be close, it may become a config option to the NE2k config in ELKS. I'll leave to @ghaerr to comment on that - and the turbo issue.

Thank you.

--Mellvik

NCommander, 2021-02-07:
For the byte counter; I'm handling it as words so I didn't have to change all the DMA code so the two in/outs get handled the same .This is how NetBSD handles NE1000 8-bit cards. Quite a few NE2000 cards will work in 8-bit slots as an intended feature, but the only real difference between the NE1k and 2k is the faster data rate.

It's pretty interesting to see how ELKS compares to say Venix or Xenix/86, although I do wish it was possible to use EMS/XMS :/

We actually did get TCP/IP working over the Internet, so  it was stable enough to manage that. The machine does seem to have some scheduling ooph as it kinda grinds to a halt when we tried to do any real multitasking.

For the Turbo button, I don't think there's any sane way to autodetect the Turbo enable mechanism on machines that need it, so I'm thinking a simple freestanding executable is the sanest way to go here ..

Mellvik, 2021-02-07:
OK; I get it - sounds good to me. 

Question - if I remember correctly, 'turbo' never had a consistent definition back in the day (or now for that matter). What does turbo mean in your machine? Is it just a switch between clock frequencies (my compaqs have such a thing - a soft switch via the keyboard to slow them down to accommodate old floppies ).

Also I'm always happy to hear someone mentioning Venix. I'd be interested in hearing more about what you're doing with Venix. This is not the right venue for that, please contact me directly.

Thank you.

—Mellvik

ghaerr, 2021-02-07:
Hello Michael,

Thanks for your report and nice work! It's nice to see ELKS working on more hardware.

It seems that adding a compile-time option of something like \#define NE2K_BYTEACCESS in `elks/include/arch/ports.h` should allow the NIC to be configured for either word or byte access, with ifdef'd driver code accordingly. We haven't yet moved all of the IO/IRQ parameters from ports.h into menuconfig, so a hand edit of ports.h is still necessary.

For the turbo mode, ELKS already supports a high-speed configuration for Compaq computers using CONFIG_COMPAQ_FAST in menuconfig, found in `elks/config.in` and `elks/arch/i86/defconfig`. This ends up sending a byte out a special port in `elks/arch/i18/kernel/system.c`. For now, perhaps follow the same path and add a CONFIG_VTECH_FAST similar option?

> It's pretty interesting to see how ELKS compares to say Venix or Xenix/86, although I do wish it was possible to use EMS/XMS :/

I'm interested in your findings - how does ELKS compare? 

With regards to implementing EMS/XMS, there has been some discussion. This would only allow at most 64k of additional memory, but not to an individual process, only to the main memory available for additional processes, since ELKS runs only in real mode. We have EMS working at 0xD000 for ram disk, and could make that available as main memory with a change to support non-contiguous memory to the main memory allocator.

> We actually did get TCP/IP working over the Internet, so it was stable enough to manage that. The machine does seem to have some scheduling ooph as it kinda grinds to a halt when we tried to do any real multitasking.

What particular issues are you seeing? The scheduling is strict round-robin forced by hardware timer or any system call, and processes don't have separate priorities (yet), so tasks may at times receive unequal time slots. On older machines, it seems that CPU speed is a bigger problem.

Let us know if you need any help for your upcoming PR.

Thank you!

ghaerr, 2021-02-07:
> ELKS sorta worked as is, but it misidentifies the processor as a V30.

Forgot about this: agreed the get processor type asm code seems to think a number of processors are always NEC versions, incorrectly. The original reason the processor is identified in ELKS, other than "nice to see" on startup, is to enable automatically using PC/AT "advanced" hardware functions, like BIOS int 13h get drive parameters, keyboard LEDs, HD IDE probing and IRQ 2->9 auto-mapping. If the processor is deemed an 80286 or better ("arch_cpu > 5"), then these features are turned on. This was recently changed, using the elks/include/config.h SYS_CAPS define, to allow ELKS to be built and run on 8086 simulators which also include some of those features.

NCommander, 2021-02-07:
The V30 detection seems to just be because of the order, and I think its a
case of just needing some tweaks. It looks if a processor is a 8086, and
then it checks if the prefetch queue is bigger than a certain size, and
falls through. The code looks wrong to me since it feels like it will end
up being an V30 as a "default" case if the prefetch queue is larger but
none of the other criteria are met, but I'm not an expert enough in the
specific behaviors of 8086 behavior to know for sure. I wonder if a more
reliable metric would be to calculate something like BogoMIPS, and use the
PIT to figure out how fast we're running to try and narrow down the
potential CPU.

misterblack1, 2021-02-15:
NCommander was actually helping me get ELKS running on my Laser XT SL Turbo machine -- and with regards to how turbo works, it's similar to the Compaq in that is starts out in 4.77mhz mode at power-on. Enabling "turbo" changes the clock speed to 10mhz and turns on the Turbo LED on the front panel. There is a keyboard shortcut to do this, but once ELKS starts up the combo no longer works. I disassembled the DOS turbo utility for the machine to find the IO port that it writes to and NCommander added the code and compiled ELKS to run at full speed on the machine. 

While this is a Laser XT SL machine machine that I was able to test on, I have a feeling it will work the same on all Laser XT Turbo machines (They seem to have made at least 3 different models.) I think the DOS utility is the same for all the machines (and the disassembled code shows it only writing to one IO port.) 

I put some pictures of this machine here if anyone is interested:
https://imgur.com/gallery/rIctEUb

Also, for CPU detection, don't assume that a 286 always has 286 hardware like (IRQ > 7) because machines like the Tandy 1000TL are 286 XT class machines. It is more like a 8086 computer where they replaced the 8086 with a 80286 but didn't change anything else. So it still has 8-bit ISA slots, no high DMA or IRQ, etc. CheckIt v3 seems to be able to identify 286-XT class machines -- not sure what mechanism it's using though.

NCommander, 2021-02-15:
> Also, for CPU detection, don't assume that a 286 always has 286 hardware like (IRQ > 7) because machines like the Tandy 1000TL are 286 XT class machines. It is more like a 8086 computer where they [...]

Identifying the 8086 vs 286 could be done by checking if the 286 specific math opcodes are present, or by seeing if you can enter protected mode and then reset back to real mode. If you check for that and the IRQ, its probably the most reliable way of seeing if you have an XT 286 or an AT 286 based system.

I've been a bit tied up to get a patch-set together to change the ne2000 drive to use byte mode so it works w/ 8-bit cards (or genuine NE1k cards)

ghaerr, 2021-02-15:
> Identifying the 8086 vs 286 could be done by checking if the 286 specific math opcodes are present, or by seeing if you can enter protected mode and then reset back to real mode. If you check for [...]

ELKS doesn't really need to know what the CPU is, the system uses only 8086 opcodes. I stated earlier that, by default and historically, ELKS uses the CPU type to determine whether some extended functionality like AT BIOS calls and IRQs are available. This can now be overridden by changing a config.h file, which is useful for emulators which may or may not have various BIOS calls and IRQs emulated. I added it so that ELKS was not dependent on only the ancient IBM PC model architecture, as there is interest in running ELKS on newer non-IBM-compatible SBCs.

> Also, for CPU detection, don't assume that a 286 always has 286 hardware like (IRQ > 7) because machines like the Tandy 1000TL are 286 XT class machines. It is more like a 8086 computer where they replaced the 8086 with a 80286 but didn't change anything else.

Yes, exactly. These machines would currently require a config.h override, but can definitely be handled (meaning that even though they have an 80286 processor, ELKS can be told not to make the assumption of calling AT BIOS routines or IRQ vectors).

Magicrafter13, 2021-09-30:
> Also, for CPU detection, don't assume that a 286 always has 286 hardware like (IRQ > 7) because machines like the Tandy 1000TL are 286 XT class machines. It is more like a 8086 computer where they [...]

By any chance, is this why ELKS doesn't seem to work on my Tandy 1000 TL/2, or should I open an issue for it?

ghaerr, 2021-09-30:
Hello @Magicrafter13,

> By any chance, is this why ELKS doesn't seem to work on my Tandy 1000 TL/2, or should I open an issue for it?

It might be - can you describe the problem in more detail or take a picture of the boot screen? ELKS can be configured in elks/include/linuxmt/config.h to override various assumptions made regarding IRQ > 7 based on perceived processor type, by modifying the CAP_PC_AT and other defines. If this gets complicated, we can open a separate issue for it.

Thank you!

closed 2022-02-08 by ghaerr