ghaerr/elks #908: Self-decompressing exomizer binaries for ELKS

ghaerr, 2021-04-02:
Hello @tkchia (and others interested in compressed executables),

The recent conversation in [#902](https://github.com/jbruchon/elks/pull/902#issuecomment-811309170) as well as your previous research on Exomizer and comments in [#226](https://github.com/jbruchon/elks/issues/226#issuecomment-467067893) got me thinking about a way to implement self-contained, self-decompressing userland binaries on ELKS. I believe this could now be implemented with not too much trouble, while maintaining a maximum of backwards compatibility with the current ELKS a.out header. Here is the idea:

- Add a `--exomize` or `--compress` option to `ia16-elf-gcc` that runs an additional backend tool `elks-exomize` (or elks-compress) after `elf2elks`, which reads an ELKS a.out binary and produces a self-decompressing a.out binary.
- The `elks-exomize` tool will produce an a.out using an ELKS supplemental header of all zeros, except the two currently unused, reserved fields (esh_reserved1, esh_reserved2), which will store the compressed application text/data sizes (including the compressor text/data also). The reserved fields could be renamed 'esh_filetseg' and 'esh_filedseg'.
- The original a.out (non-supplemental) header will remain the same, and will describe the required actual, after-decompression text and data segment sizes. The supplemental header will be all zeros, and use the two reserved fields as the exomizer text and data segment sizes within the new binary.
- This structure will allow minimal changes to the ELKS executable loader: the main memory text and data segments will be allocated by the kernel with no changes, except that the kernel f_op->read() result will be compared with the reserved field text/data sizes, rather than the a.out header text/data sizes, when the reserved field text/data sizes are non-zero.

The above will allow the ELKS kernel to load a compressed binary, built by `elks-exomizer`, which allocates the (larger) after-decompression text and data segments in memory, but actually only reads the (smaller) text and data from the binary file. The kernel does not "know" about compressed executables, only about a different amount of text and data to read (and bss, see below).

The compressed binary will contain the exomizer decoder in the text segment, and the fully compressed userland application's text, data and original header in the compressed binary data segment. When the "application" is given control by the kernel, exomizer will start, and decompress the data into the application text and data segments.

Complications and restrictions:
- It may not be possible to actually decompress the executing text segment in-place. In this likely case, the exomizer text segment size will be added to the original a.out tseg field, and the exomizer decompressor will decompress the text segment into a paragraph-aligned segment just beyond its own, and reset CS to that by far-jumping to the application entry point.
- The combined compressed text and data may exceed the original a.out's data segment size. In this case, the decompressor can make a (new) system call to decrease the allocated data segment size just before far-jumping to the application entry point.
- The kernel can't clear .bss on load, so that will have to be done after decompression, and the original a.out header bss field will likely have to be modified to be the decompressor bss value.
- Should `elks-exomize` not be able to produce a ELF executable with both smaller text and data segments than the original application's a.out header, it will just leave the input file without modification (i.e. not compressed).
- For now, medium-model binaries (or any application with a non-zero supplemental header) will not be compressible. These are problematic anyways unless the relocation data is passed to the decompressor.

The above issues require that the original a.out header (or at least dseg and bseg) be included in the exomizer data segment (along with the compressed application text and data).

The `elks-exomize` tool would produce a new ELF binary by essentially duplicating a pre-built exomizer decompressor binary (ELF text and data segments) and expand the ELF data segment by the compressed application text and data, and likely stuff the old a.out header into a known offset in its data segment.

This could all be tested without kernel modifications nor the `elks-exomize` tool by just building an exomize-decompress application with a large static buffer for the compressed data, and having the compressor write data buffer and the supplemental reserved fields. (We would need a forced supplemental header and tseg forced to 64k).

The benefits of this approach would be very minimal modifications to the loader in the ELKS kernel and `elksemu`. Understanding the exact nature of how the decompression occurs, something I've not done yet, is necessary to design the structure of the decompressor text/data layout. But, the `elks-exomize` tool could be complicated, depending on the capabilities of the ELF module library.

What do you think?

Thank you!

Mellvik, 2021-04-02:
Great work @ghaerr, and very interesting indeed. 

Admittedly, you lost me on the technicalities in the second bullet list, but IMHO - if the effect of the compression qualifies  the effort , this is a very desirable improvement.

SO - I guess the first step is to test compression algorithms - effectiveness and size (decompression), starting with the one Exomizer is using? I haven't found any references to stats about the actual compression ratio for Exomizer or its siblings.

BTW It may be OK to use 'compress' instead of exomize since the latter is a product with specific properties, and possibly creates expectations (if only for the few surviving enthusiasts ...) .

Thank you!
--Mellvik

ghaerr, 2021-04-02:
Hello @Mellvik,

> I guess the first step is to test compression algorithms - effectiveness and size (decompression), starting with the one Exomizer is using? I haven't found any references to stats about the actual compression ratio for Exomizer or its siblings.

The results of Exomizer is discussed in #226, it is better than compress/gzip and seems to average 33-50% compression.

> It may be OK to use 'compress' instead of exomize since the latter is a product with specific properties, and possibly creates expectations (if only for the few surviving enthusiasts ...) .

The reason that Exomizer is such a good choice is that the decompressor is small, fast, and designed to run well on 8-bit systems. In Exomizer v3.0, the 8086 decompressor is only 212 bytes long and only requires 156 bytes of data segment. For reasons discussed above, the decompressor may have to remain in memory and be included in each compressed executable, so this is a big issue.

Thank you!

Mellvik, 2021-04-02:
Agreed, @ghaerr - sounds like you got the issues covered. By the way, my point was not to not use exomizer code and concept, but that using the name may cause confusion. 

How do we proceed from here? As usual, my primary contribution will probably be testing on physical hardware … 

—Mellvik

tkchia, 2021-04-02:
Hello @ghaerr,

Interesting ideas, thanks!  I am thinking that we might be able to develop on your proposed scheme a bit.  Since the decompressor is expected to be in userland, perhaps it is worth thinking about whether we can get the ELKS kernel to support self-modifying code in a more general way.

Thinking aloud:
  * Perhaps, instead of defining a "on-disk text size" field, we can have an "uninitialized text size" field (like BSS, except for a text segment, if that makes sense).  A value of 0 in this field will have the same semantics as before, i.e. all text is initialized and read from disk.
  * The current kernel assumes that text segments are read-only --- in particular, that two processes that run the same program can share their text sections in memory.  Obviously an `a.out` unpacker routine will break this.  How can an `a.out` program signal to the kernel that its text segments are meant to be writable?  Is it enough for the kernel to check "uninitialized text size" ≠ 0 --- or should we also somehow allow programs with no "uninitialized text" to be self-modifying?  Should there be any provisions for running self-modifying programs in 286+ protected mode?

Thank you!

ghaerr, 2021-04-03:
Hello @tkchia,

Thank you for your comments! I had not thought of the problem of shared text segments, and I now see my first design won't work for that case, which is a show-stopper. The primary reason it won't work isn't to do with self-modifying code, but that the self-decompressing binary's text segment wouldn't execute (to decompress even the data section), since the shared code segment is already loaded, and thus not loaded nor executed again.

> Should there be any provisions for running self-modifying programs in 286+ protected mode?

All code dealing with any kind of protected mode was removed from the kernel about a year and a half ago. ELKS concerns itself with real mode operation, only, and it seems likely to remain that way.

Rather than going down the path of worrying about the kernel-to-application communication issues with self-modifying text (and data) segments, I've realized a simpler design can work: When the OS kernel can be modified (as it would have to be for the earlier design anyways), it actually isn't necessary to go to the trouble of self-decompressing executables; it's far simpler for the kernel to decompress the executable instead. Going this route solves the shared text problem, since the kernel decides whether to decompress a text section or just increment the reference count on an already loaded copy, and doesn't require any self-modifying code. In addition, the decompressor doesn't have to be carried within each binary on the filesystem, reducing space requirements. 

The updated design has one copy of the decompressor in the kernel (could be as small as 212 bytes, but for reasons described below will likely be at least 2000 bytes or more). In addition to removing the need for self-modifying code and allowing shared text segments, it will also support compressed medium-model programs as well.

**Compressed executables: Design Two**
- Similar to first design, except all decompression done by kernel at time of reading binary executable.
- Binaries will be compressed using `elks-compress`, which will create a new binary using the larger supplemental header. Any `text`, `fartext` and `data` sections will be compressed. All other sections (e.g. relocation data) will remain uncompressed.
- Three 16-bit short words, out of the remaining four reserved 16-bit supplemental fields, will represent the text, fartext and data segments as compressed when non-zero, which will indicate their compressed length.
- The kernel will read in each section in order, as it does now, into allocated program memory, and that section decompressed using the Exomizer backwards in-place option. After decompressing, uncompressed relocation data will be read and applied, same as now.

I have been studying the Exomizer internals and could produce a PR shortly as a proof of concept. This would implement a standalone `elks-compress` that runs `exomizer -C -b` on the text and data segments of the input file, and produce an output file with those segments compressed. The kernel would include a C version of the decompressor initially, with a later asm optimization for space if desired.

**Problems found with backwards in-place decompressor**

This should all work fine, *except* there seems to be a potential problem with regards to exactly how the Exomizer decompressor works, which I have found through testing. This has to do with decompressing input data within the same buffer used for output: the idea is that the input buffer is contained within the output buffer, at an address less than the output buffer by a *safety offset* margin. This offset allows for a backwards in-place copy to be used without overwriting the input buffer or requiring additional space for it.

What got me interested in all of this was a comment made in https://github.com/jbruchon/elks/issues/226#issuecomment-467067893: the *safety offset*, usually 2-3 bytes, is used as the (negative) offset of the input buffer end from the end of the output buffer. My testing shows this is not quite true: instead, the full difference between the uncompressed data size minus the compressed data size *minus* the safety offset must be used. That is, if the uncompressed data is 5000 bytes, and the compressed data is 4000 bytes, and the safety offset is 3, then there needs to be 1003 bytes between the buffer ends or the input data gets clobbered.

What this means is that the effective "load address" of the input buffer is decreased by the safety margin. In our case, this requires the allocated code or data segment to have 2-3 bytes available *before* the allocated segment in order for the decompression to happen into the correct addresses, which we can't do. Adding 2-3 bytes to everything won't work either, as then the data is then placed 2-3 bytes too high. Hopefully this all makes sense!

This can be solved with a separately-available "sliding window" decompressor, but it's quite a bit larger. (See rawdecrs/exodecr.c vs rawdecrs/exodecrunch.c for comparison in Exomizer 3.1, available at https://bitbucket.org/magli143/exomizer/wiki/downloads/exomizer-3.1.0.zip). This decompressor also requires an additional 64K memory for the sliding window, which the kernel would have to allocate. This could be a problem in low-memory situations, effectively removing another 64K useful memory to run applications.

Thank you!

tkchia, 2021-04-03:
Hello @ghaerr,

> I had not thought of the problem of shared text segments, and I now see my first design won't work for that case, which is a show-stopper.

Actually that may well not be a huge obstacle.  The kernel code responsible for sharing text segments is mainly just this:

```
int sys_execve(char *filename, char *sptr, size_t slen)
{
    ...
    /* Look for the binary in memory */
    seg_code = 0;
    currentp = &task[0];
    do {
	if ((currentp->state <= TASK_STOPPED) && (currentp->t_inode == inode)) {
	    debug("EXEC found copy\n");
	    seg_code = currentp->mm.seg_code;
	    break;
	}
    } while (++currentp < &task[MAX_TASKS]);

    /* Read the header */
    ...
    if (!seg_code) {
	...
    } else {
	seg_get (seg_code);
	filp->f_pos += (size_t)mh.tseg;
#ifdef CONFIG_EXEC_MMODEL
	filp->f_pos += (size_t)esuph.esh_ftseg;
	need_reloc_code = 0;
#endif
    }
    ...
#ifdef CONFIG_EXEC_MMODEL
    if (need_reloc_code) {
	...
    } else {
	/* If reusing existing text segments, no need to re-relocate */
	filp->f_pos += esuph.msh_trsize;
	filp->f_pos += esuph.esh_ftrsize;
    }
    ...
#endif
    ...
}
```

So to inhibit the sharing, we mainly just need to nuke `seg_code` if, after reading the `a.out` header, we discover that the executable's text segments cannot be shared across processes.

Thank you!

ghaerr, 2021-04-03:
Hello @tkchia,

Thank you, I am aware of how the kernel shares text segments. I had called that issue a show-stopper because I felt that introducing compressed binaries should not cause the system to operate differently after decompression. For this shared text case in point, /bin/sh would surely be compressed, as it is one of the larger binaries, and it is also fork/exec'd frequently; should its text segment not be shared, 46000 bytes extra RAM would be required on every subsequent invocation, greatly decreasing memory available, especially in multi-user modes:
```
# chmem /bin/sh
 TEXT  FTEXT   DATA    BSS   HEAP  STACK  TOTDATA  TOTAL
46000      0   4928   5344   8192   2048    20512  66512 /bin/sh
```

My analysis of the actual operation of the Exomizer decompressor showing the input buffer containment issue is also a problem in the self-decompressing case, as more memory than allocated is needed for the input buffer. Did you find a way to get around this when you were researching it?

Thank you!

tkchia, 2021-04-03:
Hello @ghaerr,

> I had called that issue a show-stopper because I felt that introducing compressed binaries should not cause the system to operate differently after decompression.

Thanks, I get what you mean now.

> Did you find a way to get around this when you were researching it?

The case with "safety offset" = 2 should be easy to work around.  (The Exomizer output signals the end of a compressed stream --- which will appear at the front of the compressed output --- using a bit sequence `0 0000 0000 0000 0000 1`.  We can throw these bits away if we know the uncompressed data size.)  The cases with "safety offset" &gt; 2 might be trickier to deal with though.  Of course, all this is moot if we end up using the sliding window decompressor instead.

> This [sliding window] decompressor also requires an additional 64K memory for the sliding window, which the kernel would have to allocate. This could be a problem in low-memory situations, effectively removing another 64K useful memory to run applications.

If I understand the code correctly, the sliding window basically just comprises the most recently uncompressed data at any point in time?

Thank you!

ghaerr, 2021-04-03:
Hello @tkchia,

> If I understand the code correctly, the sliding window basically just comprises the most recently uncompressed data at any point in time?

Yes, it seems so, up to MAX_OFFSET (<= 64 KiB), thank you for the confirmation. Thus, no separate sliding window buffer is needed using backwards decompress-in-place, as the non-sliding window decompresser can directly access the uncompressed data in the output buffer instead.

I played around with setting various MAX_OFFSET values less than 64 KiB in the sliding window decompressor, such as 4096, which also has to match the `exomizer -m` compressor command line. This was in the hope of using a much smaller value for the sliding window buffer in the kernel. I found the results quite less than optimal, a 46k file only being compressed to 34k for max offset 4096 vs 24k using max offset 64k. I decided against using sliding window since smaller windows pretty much negate exomizer's superior compression results.

> (The Exomizer output signals the end of a compressed stream --- which will appear at the front of the compressed output --- using a bit sequence 0 0000 0000 0000 0000 1. 

Is this true for the RAW (`exomizer raw -b infile`) mode? And that bit stream is the reason why the safety offset is at least 2? 

> We can throw these bits away if we know the uncompressed data size.

We definitely know the uncompressed data size, it will be the original a.out tseg and dseg values. I haven't tried removing those bit signals, as I am trying to use essentially unmodified decompressor source for the first port, until the entire system is better understood. We might then switch to the super-small/fast 8086 asm decompressor for speed after everything is working :)

> The cases with "safety offset" > 2 might be trickier to deal with though. 

My testing has shown various safety offsets greater than 2 so far: 3 and 6. The documentation says that it is possible for it to increase without limit, although rarely. Because of this, I came up with a further revised design.

I have settled on an alternate method that can guarantee to function every time in the kernel, with a couple constraints, which I have coded up and working, outside the kernel. The system uses backwards in-place decompression.  I realized that, given a "large" safety offset, which is configurably set to 16 (1 paragraph), the decompression input/output buffer can be made an extra 16 bytes long, with the output buffer at the high address, and the input buffer and the low address. This guarantees that any compressed file with safety offset <= 16 can be decompressed into an output buffer 16 bytes larger than the original file size. Then, after decompression, the output buffer is memcpy'd 16 bytes lower to its required start address. Its an extra copy operation, but *much* simpler than the alternatives, for the kernel.

The cost for this design is an extra paragraph of memory in each kernel text segment allocated. The data segment will have no cost, as there's always stack above data+bss. So, although the system will work slightly differently when running a compressed binary, 16 bytes extra runtime is a small price to pay. The `elks-compress` program will have to check for safety offset > 16 and/or larger compressed data than original, and reject the conversion. The extra paragraph is easily allocated by incrementing a kernel variable, and the proper placement of the input/output pointers are handled within the decompressor function wrapper.

I have the compression/decompression code up and running, tested and operational outside the kernel. I was in the middle of moving it to the kernel, when I ran into a very strange bug in the kernel, which only happens using the updated toolchain. Four hours later, I still can't completely track the problem down. I'm going to post a separate issue for that.

Thank you!

closed 2021-04-06 by ghaerr