ghaerr/elks #909: Possible continuing toolchain issues?

ghaerr, 2021-04-03:
Hello @tkchia,

Using the most recent source and toolchain, I have run into a *very* strange problem when compiling the kernel and booting ELKS. The problem seemingly manifests only with the new toolchain, and removing both of our recent group of  commits (back to Jan 16) makes no difference: adding the following single line causes a startup failure using the new toolchain, and not with the old:
```
diff --git a/elks/fs/exec.c b/elks/fs/exec.c
index b168ce49..80233e76 100644
--- a/elks/fs/exec.c
+++ b/elks/fs/exec.c
@@ -337,6 +337,7 @@ int sys_execve(char *filename, char *sptr, size_t slen)
 		   retval, (size_t)mh.tseg);
 	    goto error_exec4;
 	}
+printk("compr tseg xx bytesx1\n");
 #ifdef CONFIG_EXEC_MMODEL
 	if ((size_t)esuph.esh_ftseg) {
 	    currentp->t_regs.ds = seg_code->base + bytes_to_paras((size_t)mh.tseg);
```
With the above printk added, things fail. If a single byte is removed, the trailing "1" in the printk, things work.

The failure itself is quite strange: the boot appears to hang after the kernel relocation is applied:
[image]
However, the system will in fact boot and run seemingly normally if two characters are typed on the console!

The map files show no difference in text sections (when removing the single "1" character). I have printk-debugged the problem all the way into 'elks/arch/i86/drivers/char/xt_key.c::kbd_init()':
```
void kbd_init(void)
{
    /* Set off the initial keyboard interrupt handler */

    if (request_irq(KBD_IRQ, keyboard_irq, NULL))
    panic("Unable to get keyboard");

    set_leds();
    kb_read();      /* discard any unread keyboard input*/
}
```
The boot "hangs" just during calling set_leds(), and what appears to be happening is continuous interrupts (calls to) 'keyboard_irq'. I haven't been able to get much deeper as I'm only able to printk-debug and can't run elks-emu. Typing not one, but two characters at the console and the system continues the boot normally.

Does this happen on your system? 

My system recently force-updated QEMU, and its now running v5.2.0. It is possible that's part of the reason, but that still doesn't answer why the problem manifests itself only with the added printk statement! My desktop iMac is running the old toolchain and QEMU v4.2.0, and I can't get the problem to happen there, using the latest source or rolled-back commits.

The map files show just address differences of 4 bytes in the .data section (with and without the "1"), with a single byte difference in .rodata, as to be expected. I can't tell whether this is an old kernel bug (unlikely), a linker issue (likely), or a strange QEMU bug (possible). It is entirely possible the problem has nothing to do with kbd irq, but just happens to show up there. There are no text segment map file differences, so very hard to tell.

If you have any ideas, I'd love to hear them!

Thank you!

ghaerr, 2021-04-03:
Update: Can't get the problem to happen when compiling the kernel in small model (no FARTEXT).
Here's the config file that shows the bug:
[config.bug.zip](https://github.com/jbruchon/elks/files/6253748/config.bug.zip)

tkchia, 2021-04-04:
Hello @ghaerr,

Thanks for the report!  I am able to reproduce the problem.  It seems to be a buffer read overflow bug in the IRQ 1 handler (`keyboard_irq( )`).

ghaerr, 2021-04-04:
Hello @tkchia,

Super work on finding this kernel bug. Please accept my apologies for suggesting the problem being more likely in the linker than the kernel. I won't do that again!! :)

Thank you!

closed 2021-04-04 by ghaerr