ghaerr/elks #916: Serial Console speed

Mellvik, 2021-04-08:
It would be really useful to be able to set the serial console speed in '/bootopts', so as not to be locked to 9600 for the boot messages (say, when getty speed for ttyS0 is 38400).

ghaerr, 2021-04-08:
@Mellvik,

This should probably be easily upgraded - /bootopts now allows a single digit to be specified, which is used as the `init` initial run level. I'm thinking that if the number is instead > 100, it would be interpreted as the serial baud rate. I'll look into a PR for this.

Thank you!

Mellvik, 2021-04-08:
Thanks - sounds like a great solution.
-M

ghaerr, 2021-04-08:
Hello @Mellvik,

This is proving to be more complicated than initially believed: /bootopts can have an option to set a serial console baud rate on boot - but what happens later when/if that same port is also setup in /etc/initab to operate as a getty line? 

The getty line will have an explicit baud rate specified. Should that baud rate be overwritten by /bootopts? If not, then the serial port will change baud rate as soon as getty runs after boot, causing a problem. If so, then is the system supposed to ignore all baud rate changes to that port? The kernel doesn't know what application is opening the port, so an override would essentially force that port to a certain baud rate, and even `stty` couldn't change it. I'm not sure I like that option either.

If we only implement a baud rate in /bootopts that works if there is no "multiuser" mode that involves serial ports, but it seems a bit short-sighted, running any program that opened that line (e.g. getty) would change the baud rate.

Thus, this "fix" is kind of a hack if implemented without thinking through exactly how the serial port should behave.

Thank you!

Mellvik, 2021-04-09:
Thank you @ghaerr.

IMHO - the scenario you describe is the scenario we have today, and the purpose of the change is to eliminate that scenario.

Today: Console speed is set to 9600 during boot, then set som something else (or the same) in inittab.
With change: Console is set to desired speed during boot, and to desired speed in inittab. If they are different, it's because it was configured that way. The chance of making mistakes are the same...

AFAIK - the change is to add an option that was missing, there are no new complications. The defaults should be left as they are.

So the proposal is to add this to the boot process, leave everything else as is...

Thank you for looking into this, it would be a very helpful addition (I guess there are few regular serial console users out there at this time :-) ).

-- Mellvik

ghaerr, 2021-04-09:
Hello @Mellvik,

OK, it seems your setup is running *without* a getty. In that case, there is no option to set the boot serial console baud rate. I'll figure a way to implement that. The problem currently is that ELKS doesn't actually set the baud rate at boot anywhere - it relies on the hardware or BIOS resetting to some baud rate at startup, which seems to be 9600. 

> I guess there are few regular serial console users out there at this time :-) 

I guess there are too few ELKS users out there... we need to get more word out. Did you ever update the ELKS Wikipedia page on v0.4 feature updates and the commits increase, etc? Or should I do that?

Thank you!

Mellvik, 2021-04-09:
Hi @ghaerr,

I am indeed running getty, which is where the problem originates. 
I'm running serial console getty via inittab at 38400 which is fine except the boot messages are garbage, so I'm looking for an easy way out. Which seems to be /bootopts.

I've been looking for a way to change the default port speed, but there is no such thing, at least not on the venerable Compaq.

Wikipedia page: No I did not. It's close to the top of my list, and I figured doing it would force me to update myself on the parts of ELKS that I haven't been involved with.

I suspect your list of pending ELKS fixes is longer than mine, so I'll do it and you fix my mistakes after. Sounds OK?

—M

ghaerr, 2021-04-09:
> I suspect your list of pending ELKS fixes is longer than mine, so I'll do it and you fix my mistakes after. Sounds OK?

The long list is [here](https://github.com/jbruchon/elks/issues/873), take a look at the existing Wikipedia entry, update the stats since they've changed greatly, and add major features from the above list you think are interesting to others, thanks!

Mellvik, 2021-04-10:
@ghaerr,

first cut of Wikipedia update is out. I suggest you take it from here.

—Mellvik

Mellvik, 2021-04-10:
Thanks @ghaerr,

tested using the image provided, works fine - with and without the speed option present. This is good.

Update the wiki?

--Mellvik

ghaerr, 2021-04-10:
Thanks for testing @Mellvik, wiki updated.

Mellvik, 2021-04-12:
@ghaerr,

seems we have to reopen this one.

Doing the keyboard LED testing for @tkchia sent me back to the physical console, I have only been using serial for a while.

Surprisingly, no boot messages on the console. Turns out, boot message are directed to ttyS0 regardless of the /bootopts config. I'm assuming it has to do with the changes related to setting console speed in /bootopts - what do you think?

--mellvik

ghaerr, 2021-04-12:
Hi @Mellvik,

> Surprisingly, no boot messages on the console. Turns out, boot message are directed to ttyS0 regardless of the /bootopts config. 

You're saying that the physical console is getting absolutely nothing, or just the ELKS Setup and login: messages (see attached screenshot):
[image]

> I'm assuming it has to do with the changes related to setting console speed in /bootopts - what do you think?

That is quite strange, since I just tested here on QEMU and boot messages appear on the console, unless there is a "console=ttyS0" line in /bootopts.

Can you make absolutely sure by catting /bootopts after boot and making sure there is no uncommented line, or perhaps by deleting /bootopts completely, syncing and rebooting?

Thank you!

Mellvik, 2021-04-13:
Thanks @ghaerr,

this turns out to be even more interesting.
First - yes, the image booted is pristine from the build process and has everything commented out in /bootopts.
And yes, the console looks exactly like your picture, while the boot messages appear on the serial line @ 9600 bps. And no getty of course.

Here's where it's getting more interesting: Removing (or moving) /bootopts causes boot failure. Physically it appears like a bad sector (repeated retries) "ELKS….*****1!<cr>Press key", but the sectors are fine (after testing on several floppies). After just rewriting the image to the floppy - with /bootopts intact - the machine boots fine. 

Since GitHub filters images, I'll post the screen image separately.

-Mellvik

Mellvik, 2021-04-13:
[image]

ghaerr, 2021-04-13:
Hello @Mellvik,

I would like to subdivide this problem into two parts: A) that of the serial line being used as console at boot, rather than the physical console, and B) removing /bootopts causing boot failure.

Regarding A) the serial console can also be automatically chosen (without /bootopts) by having CONFIG_SERIAL_CONSOLE=y in your .config. Can you check to see if this happens to be the case?

Regarding B) I can duplicate removing /bootopts failing the boot on QEMU (matching your picture), so I will check in to fixing that.

Thank you!

Mellvik, 2021-04-13:
Spot on, @ghaerr ---

> 
> Regarding A) the serial console can also be automatically chosen (without /bootopts) by having CONFIG_SERIAL_CONSOLE=y in your .config. Can you check to see if this happens to be the case?
> 
That's it, never even crossed my mind. that's an embarrassing one - mea culpa.

Some consolation that I found that other issue. Thanks for fixing!

—mellvik

closed 2021-04-13 by Mellvik