ghaerr/elks #920: elks boot issue on IBM XT with XT-CF-lite

riktw, 2021-04-12:
**Description**
I wanted to try elks on my IBM XT as it looks like a very interesting project but I ran into an issue with booting.
It seems that there is a problem reading files, as elks cannot read the bootopts file and cannot run /bin/init. This results in not having a shell. Is the XT-CF-lite card supported to boot from, or is my configuration not yet supported by elks?

**Configuration**
- Version (commit or release):
Latest version from Git, commit 65e652e
- Emulator or HW platform:
IBM PC XT with 256K RAM, 8088 CPU, XT-CF-lite card as hard drive. Image used is the 32MB HD image.

I changed the following settings in the menuconfig:
- Select IBM-PC variant: PC-XT
- Target image: Medium (HD), Filesystem(Minix)

To install, I used the DD command to write the bootable hd.bin file to the compact flash card used in the XT-CF-line card.
I also tested the image in qemu and it works as expected.

**How to reproduce ?**
This issue always happens on boot.

**Raw data**
Boot log: 
[image]

As I found the Ignoring /bootopts rather strange, I added a bit of code to print the content of bootopts into the elks/init/main.c file in the parse_options() function. It seems the bootopts is all 0x00. The same image on qemu does print the bootopts file as expected.

With printing bootopts:
[image]

Mellvik, 2021-04-12:
Rik,

thank you for trying (and testing) ELKS.

The most likely reason for the problem you're experiencing is that the CHS values (cylinders/heads/sectors) value of your CF card are different from the values the image was created with. So ELKS will find data correctly at the beginning of the drive, but nothing else. 

I suggest you create a bootable floppy to begin with, that will give you an operational system (if somewhat limited if the floppies are 360k).

Then you  may try to reconfigure the parameters of your CF card to match those of the provided image (see the wiki), or build your own image with parameters matching your CF card.

We look forward to hearing from you - and to help you get going.

Thank you!

—Mellvik

ghaerr, 2021-04-12:
Hello @riktw,

I think @Mellvik is correct, as your screenshots show CHS as 982,128,63 and the default HD build is 63,16,63. These options are set with menuconfig under the "Target Image" section, when HD is selected.  You can select the FD1440 (1.44M floppy) instead, as @Mellvik is suggesting, or try changing the HD values there. Selecting the 1.44M floppy option will give you most all the ELKS applications in /bin.

The reason the image likely works in QEMU is that the CHS values are pretty much faked-up to match a linear file seek onto the HD image file on the host filesystem. This approach isn't available on real hardware, which must use actual CHS values for the older systems ELKS is targeting.

Thank you!

riktw, 2021-04-12:
Hello and thanks for the very quick response. I lack a modern computer with a 5.25" floppy drive, so I can only make a bootable floppy on my IBM XT. Luckily, I got a compact flash card with DOS 3 on it. 
I added the 360KB images to the floppy, one build from the latest git commit and the ones from the 0.4.0 release.
Using the rawrite tool I wrote these to a 5.25" floppy and tried booting from that.

Sadly, the results where oddly similar. 

Booting using the 0.4.0 release FD360 minix :
[image]
[image]
The bootopts seems to be detected, the CF card as well. But sadly, again no shell.

Booting the fat one is similar:
[image]

I did also try it with the XT-CF-lite disabled. elks did detect the original HDD in the IBM XT but again no shell.
The image I build from the latest commit have the same issue.

Rik

ghaerr, 2021-04-12:
Rik,

I notice on your boot screens that you only have 123k free memory before running /bin/init. Init takes about 15k memory and /bin/sh takes ~65k, although it needs 46,000 contiguous bytes to load, which might not be available, due to known early system memory fragmentation.

How much main memory does your system have?

Perhaps try logging in as "toor" instead of root: that will try to load /bin/sash, which requires 50k ram, 31k contiguous.

So far, we can't tell whether @Mellvik is right, and the filesystem is corrupted (we have seen that before), or whether the "/bin/init: exec failed: /bin/sh" is because of no memory (ENOMEM = 12), or not.

Depending on the result, you could add the following for further debugging. Change line 273 of elksmd/sys_utils/init.c to:
```
fatalmsg("exec failed: %s (%d)\r\n", argv[0], errno);
```
This will give the kernel error number at boot that can be looked up in elks/include/linuxmt/errno.h.

Another idea would be to add the following line in elkscmd/rootfs_template/etc/passwd and run "make" build it into the fd1440.bin image:
```
mem::0:0:System Administrator:/root:/bin/meminfo
```
This will allow you to login as "mem" and get a free memory listing showing the kernel memory layout.

Thank you!

Mellvik, 2021-04-13:
@ghaerr,

actually, I think adding the mem login to /etc/passwd permanently is a good idea. Another tool in the same category as the 'tty' login.

—M

riktw, 2021-04-13:
I have 256K of RAM. I did not see any minimum requirements on the wiki, so I hoped that this was enough.

I did try toor first, sadly no luck. I then added the changes to init.c and passwd. The error code I got was indeed 12, so ENOMEM. Sadly, the mem user did not give me any memory listing.
After a little searching I found out why. meminfo is only included on the FD1440 image and above. I changed this in elks/elkscmd/Applications and luckily adding meminfo did fit on the FD360 image.

This then quickly showed my why I could not start any shell.
[image]

29K free, whoops. Are there any ways to lower the memory usage, or would a bare minimum amount of RAM required be 320KB (and a more usable system perhaps 640K?)

Rik

ghaerr, 2021-04-13:
Hello Rik,

Thanks for your continued testing and descriptive comments, they help a lot.

There are ways we can make more free RAM available within ELKS. We'll start using the memory map you've uploaded. See the large 65535 (64K) segment on the third line of type "BUF"? That memory is reserved for kernel system buffers.

Lets reduce that from 64K to 8K. Edit .config, find the line
```
CONFIG_FS_NR_EXT_BUFFERS=64
```
and change it from 64 to 16 (less than 16 has issues when booting from the MINIX filesystem).

You may also want to comment out the rc.sys script from running, which invokes /bin/sh and a host of other utilities. This can by commenting out (inserting a '#' in the first column) of elkscmd/rootfs_template/etc/inittab:
```
#si::sysinit:/etc/rc.d/rc.sys
```

You'll need to `make clean;  make` then try logging in as mem and toor, and send back another picture of the meminfo layout. 

There are some other things we can do to reduce memory usage, but I'll get into them after seeing this first result, which is going to be tight.

Thank you!

ghaerr, 2021-04-13:
Hello Rik,

I have a simulated 256K system running - if you change CONFIG_FS_NR_EXT_BUFFERS=16 above, as well as comment out the sysinit line in etc/inittab, the system will boot and the standalone shell `sash` will run if you login as 'toor'.

However, that system is so tight on RAM that you can't actually run any external commands, such as `meminfo`. Special sash internal commands will execute, such as ls. 

I will post a fix that allows `sash` to be built with no internal commands, which will release enough space so that many external commands will run. That will make the system usable.

Also, you will want to use `make menuconfig` to eliminate as any features you don't need, like networking, etc to make the kernel smaller. This will then use less RAM for the kernel, allowing more for user applications.

Thank you!

ghaerr, 2021-04-14:
Hello Rik,

See additional instructions in PR #925 for how to build a small `sash` shell, after which, by also included the items above, you should have a usable ELKS system in 256K! More work can be done fine-tuning, but we'll leave that for a bit later :)

Thank you!

toncho11, 2021-04-14:
I remember having this problem when I left out my memory expansion card. 
Maybe for ELKS 0.5 there should be also an ELKS_256_360KB release.

riktw, 2021-04-14:
Thank you all a ton. With the PR and changes suggested. I now have a booting elks on my IBM XT :D 
I can even login as root, but even ls gives me an out of memory. Using Toor, I have a mind boggling 39K of free RAM to play with :) 

[image]

I was now quite curious about the original issue of my CF card not working. So I went back to menuconfig, selected HD image, generated a 32MB HD image and wrote it on the CF card with DD. And what do you know, CHS was not the issue, it was RAM.

[image]

As this boots a LOT faster then a 5.25" floppy and has a lot more space, I think I will keep it on the CF card. Thanks again for the help and quick responses.

closed 2021-04-14 by riktw

ghaerr, 2021-04-14:
Hello Rik,

Well, very nice to see your 256K RAM IBM XT has a newfound use! Although with just 39k free, I'm not sure exactly what useful programs you might run on it :) I think with a bit more work, we could carve out another 30-40k, by shrinking the kernel stack (defaults to 64K) and removing some unneeded configured items in the kernel. Programs also default to 4K heap and 4K stack, most of which could still use fine-tuning.

I notice in your last screenshot (thanks for them) that /bootopts is still being ignored on your HD boot. I'm wondering what the reason for that might be. Can you `cat /bootopts` (or ls -l to check its size) to see what is in it?

Thank you!

ghaerr, 2021-04-14:
Hi @toncho11,

> Maybe the system can also make a check: no 640 KB RAM print "Warning: less than 640 KB memory detected, shell might not load."

PR #925 adds an error number to the "exec failed: ..." message to now indicate the reason for failure. It might make sense to update similar error messages as well, including login's "No shell!".

Thank you!

riktw, 2021-04-14:
Yeah I am planning on ordering a ISA RAM card to increase the RAM to a more comfortable 640K. But for now, it is very nice to see it works.
The bootopts warning is interesting, as ls -l and cat seem to spit out something that makes sense:
[image]

ghaerr, 2021-04-14:
Rik,

The /bootopts failure is strange. It seems that it could be either that the very-early boot code isn't reading it properly, due to its possibly being in a then-inaccessible disk/inode location, or, that your XT BIOS is overwriting a special area used to communicate /bootopts from the boot loader to the kernel, which is currently configured to be at 0x0050:0000.

Here, I'm using the `ls -il` and `hd` commands to show /bootopts inode number, as well as the contents of memory at segment 0x50 offset 0:
[image]
Can you try those on your system?

riktw, 2021-04-15:
I tried that and yeah, 50:0 is completely empty. I am not sure if it could be a quirck from just throwing an HD image on the CF card, as when booting from floppy, bootopts works as expected. 

[image]

Mellvik, 2021-04-15:
Rik, @ghaerr,

Are we ruling out the possibility that the data buffer bootopts is being read into is outside the phys memory range, so whatever ends up at 50:00 is actually coming from empty memory space?

-M

ghaerr, 2021-04-15:
Hello @riktw,

Given that /bootopts is working on floppy boots, that pretty much eliminates the idea of the BIOS overwriting the area at 0x50:0.

>  it could be a quirck from just throwing an HD image on the CF card

I'm guessing that the problem is in the boot loader code that loads this sector from the inode number found in the root directory entry for /bootopts. This boot loader code is extremely tight on space and may not correctly load inodes using higher block numbers. I don't have an easy way of showing the block numbers used for /bootopts, so I'll have to think a bit more about this.

Thank you!

ghaerr, 2021-04-15:
Hello @Mellvik,

> Are we ruling out the possibility that the data buffer bootopts is being read into is outside the phys memory range, so whatever ends up at 50:00 is actually coming from empty memory space?

Yes, because 50:00 is the data buffer; the sector is read directly into this physical memory address.

There may be some confusion as to what the ELKS memory layout is, I created this map of the entire 1M address space, which I'll post up in Documentation:
```
ELKS typical memory layout of 1M physical memory
  
 +--------------------  seg  off  linear size (size decimal)  --------------------+
 | interrupt vectors    0000:0000 00000  0400 (1024)
 | ...
 | ...
 | ...
 | BIOS data area       0040:0000 00400  0100 (256)
 | bootopts             0050:0000 00500  0100 (256)
 | kernel setup data    0060:0000 00600  0200 (512)
 | ...
 | disk track cache     0080:0000 00800  2400 (9216 = SECTOR_SIZE * 18)
 | ...
 | ...
 | kernel code segment  02C0:0000 02C00  F910 (63760)
 | ...
 | ...
 | ...
 | ...
 | ...
 | kernel data segment  1251:0000 xxxxx 10000 (65536=64K)
 | ...
 | ...
 | ...
 | application memory   2251:0000 xxxxx       (remaining memory after kernel data)
 | ...
 | ...
 | top of 640k ram      9FFF:000F 9FFFF
 | EGA ram              A000:0000 A0000 10000 (65536)
 | MDA monochrome ram   B000:0000 B0000  4000 (16384)
 | CGA color ram        B800:0000 B8000  4000 (16384)
 | network ROM          C000:0000 C0000 10000 (65536)
 | ramdisk memory       D000:0000 D0000 20000 (131072=128K)
 | ramdisk memory       E000:0000 E0000
 | BIOS ROM             F000:0000 F0000 10000 (65536)
 +--------------------------------------------------------------------------------+
```

Thank you!

toncho11, 2021-04-16:
@ghaerr

Maybe you could write a small wiki page on all the config options to build a low memory ELKS version.

Mellvik, 2021-04-16:
Thank you, @ghaerr - 
Very valuable indeed. AAMF kernel docs have improved dramatically recently.

Great contributions.

-M