ghaerr/elks #932: Question: Porting guide

cocus, 2021-05-08:
Hi!
I have a board that is not PC-compatible, but I was trying to write a BIOS for it to make it "sort of" PC compatible.
However, my end goal would be to run something interesting, and DOS seems to be a pretty limiting choice for this board (since I don't have any PC peripherals, not even a VGA/EGA/CGA card).
Now, I've stumbled across this project thanks to emu86, and I think this might be an interesting thing to try there.
This board contains a 80C188EB (2x UARTs, no DMA, no MMU, 8088 bus compatible with 80186 instructions, PIC + PIT embedded), 128k of RAM (I'm seriously thinking in adding more to fill it up) and 512k of FLASH. The original peripherals were literally disoldered from the board as I didn't have any use for them (this is an old HDSL/E1 board). I've hardwired an SD card to some GPIO pins of this CPU, and I've already validated that it works (with some custom code, of course!). Of course, everything is being piped thru one of the UARTs peripherals on the CPU chip.

Since I would rather have "something" that just starts without the need of a BIOS, I was thinking in embedding this system directly on the flash.

I know that ELKS depends on bios services (somewhat), but I would like to understand how I could port it to not require any BIOS services, and just "start". I can write a loader, or a custom entry assembly code that jumps into the OS' entry point. Then I should be able to specify that I have 128k of RAM, and there's no BDA, so that area should be free to use.

However, I couldn't find any (with a quick glance) guides or docs on how to port this OS to a new architecture that's not "PC" compatible. I would appreciate any help that you could provide to achieve this :).

Thanks!

ghaerr, 2021-05-08:
Hello Santiago and welcome!

It's a somewhat long story to get you completely where you want to go, but I think it will be easier if we take it in a few steps, with more explanations along the way. The good news is there has been a lot of work done getting ELKS ready to be more easily ported to non-PC hardware. There may be a bit of a learning curve finding where everything is in ELKS, but I can help with that.

I recommend the following:
- Follow the directions on the main page to build the toolchain (i.e. run 'tools/build.sh') and then run "make menuconfig" and make to create a standard ELKS system. This can then be run on QEMU using 'qemu.sh'
- A very small ELKS system can be easily built, which uses a ROM filesystem (or read-only flash). This is done by copying 'emu86-rom.config' to '.config' and remaking the system. That binary can be run on the EMU86 emulator by using the 'emu86.sh' script.

That ROM version of ELKS uses a "headless console" that operates using only two BIOS INT 16h calls for keyboard, and a very few INT 10h calls for "output". As you're quite familiar with EMU86, those are in EMU86 rom-elks.c.

After you have ROM version of ELKS built and running on EMU86, which is about the smallest version we have available, the following will have to be done in order to port to your board:
- Setting ROM and RAM addresse values in elks/include/config.h for your system: CONFIG_ROM_CODE, CONFIG_ROM_SETUP_DATA, etc. See .config for values as well.
- Slight rewrite of 8259 PIC to your version (see elks/include/arch/ports.h for addresses and elks/arch/i86/kernel/irq.c for PIC code).
- Slight rewrite of 8253 PIT to your version (see ports.h and timer.c in above directory)
- I'm sure I'm forgetting other important stuff!

Take a look at this and I'll be glad to help with more questions, or rearranging code to ease your progress.
This should be a really fun project!

Thank you!

mfld-fr, 2021-05-08:
I would be also interested in having ELKS running on the 80C188EB, as it would facilitate my own port to the R8810 (clone of the earlier, but with a 'modernized' PCB).

ghaerr, 2021-05-08:
Hello @cocus,

As a result of @mfld-fr's decision for how to proceed with the ROM BIOS services on EMU86 for booting MS-DOS in https://github.com/mfld-fr/emu86/pull/51#issuecomment-835173693, I now see clearly how we can proceed with getting ELKS ported to both your and @mfld-fr's R8810 target devices, and being able to reliably test during the process.

We will use the mechanism @mfld-fr has envisioned for EMU86, which is proving quite useful for ELKS ROM testing, for porting ELKS to new targets:
- Hardware-specific code (essentially drivers) for ELKS will be split off into seperate files, with the specific chip or target name in the filename (such as pic-8259.c, pit-8253.c, xxx-pc, etc).
- ELKS can then be built for a "target" device/platform which will be a collection of specific drivers or hardware devices.
- The same group of specific drivers will also be written and emulated in EMU86 such as rom-or556.c, io-or556.c, etc).

Thus, new chip or driver support for ELKS will be added as seperate files, and emulated in EMU86 before being tested on real hardware. To start a new target, various working chips/drivers for both ELKS and EMU86 can be copied into new target-named files, and then changed in sync to conform to the new hardware. This will ensure that any ELKS or EMU86 target has the minimal functionality needed for the port, which is important on embedded devices.

I will help split up existing ELKS chip support into seperate files (or of course a PR can be submitted to do the same).

cocus, 2021-05-08:
Hi @ghaerr , I've tried to build it on my local machine, but I wasn't able to do so. I have gcc-ia16 installed system-wide, and I think (not sure) that's interfering with a build step.

Have a look at this:
```
ia16-elf-gcc -fno-inline -melks-libc -mcmodel=small -mno-segment-relocation-stuff -mtune=i8086 -Wall -Os -maout-heap=8192 -maout-stack=2048 -o ash builtins.o cd.o dirent.o error.o eval.o exec.o expand.o input.o jobs.o mail.o main.o memalloc.o miscbltin.o mystring.o nodes.o options.o parser.o redir.o show.o signames.o syntax.o trap.o output.o var.o init.o bltin/echo.o bltin/expr.o bltin/regexp.o bltin/operators.o linenoise_elks.o autocomplete.o 
ia16-elf-gcc: error: unable to locate default linker script ‘elks-small.ld’ in the library search paths
/usr/lib/x86_64-linux-gnu/gcc/ia16-elf/6.3.0/../../../../../ia16-elf/bin/ld.gold: error: cannot open : No such file or directory
/usr/lib/x86_64-linux-gnu/gcc/ia16-elf/6.3.0/../../../../../ia16-elf/bin/ld.gold: error: cannot find crt0.o
cd.o:function docd.isra.0: error: undefined reference to 'chdir'
cd.o:function cdcmd: error: undefined reference to 'abort'
error.o:function exraise: error: undefined reference to 'abort'
error.o:function exraise: error: undefined reference to '_longjmp'
error.o:function onint: error: undefined reference to '_exit'
eval.o:function evalpipe: error: undefined reference to 'pipe'
eval.o:function evalcommand: error: undefined reference to 'pipe'
eval.o:function evalcommand: error: undefined reference to '_setjmp'
eval.o:function evalcommand: error: undefined reference to '_longjmp'
eval.o:function evalcommand: error: undefined reference to '_setjmp'
eval.o:function evalbackcmd: error: undefined reference to 'pipe'
exec.o:function shellexec: error: undefined reference to 'errno!'
exec.o:function shellexec: error: undefined reference to 'errno!'
exec.o:function tryexec: error: undefined reference to 'errno!'
exec.o:function tryexec: error: undefined reference to 'errno!'
exec.o:function find_command: error: undefined reference to 'geteuid'
exec.o:function find_command: error: undefined reference to 'getegid'
expand.o:function expmeta: error: undefined reference to 'opendir'
expand.o:function expmeta: error: undefined reference to 'closedir'
expand.o:function expmeta: error: undefined reference to 'readdir'
jobs.o:function waitproc: error: undefined reference to 'waitpid'
mail.o:function chkmail: error: undefined reference to 'abort'
main.o:function main: error: undefined reference to '_setjmp'
memalloc.o:function stunalloc: error: undefined reference to 'abort'
miscbltin.o:function umaskcmd: error: undefined reference to 'umask'
miscbltin.o:function umaskcmd: error: undefined reference to 'umask'
miscbltin.o:function umaskcmd: error: undefined reference to 'umask'
parser.o:function readtoken1: error: undefined reference to '_setjmp'
parser.o:function readtoken1: error: undefined reference to '_longjmp'
redir.o:function openhere: error: undefined reference to 'pipe'
redir.o:function openhere: error: undefined reference to '_exit'
trap.o:function exitshell: error: undefined reference to '_exit'
output.o:function xioctl: error: undefined reference to 'ioctl'
var.o:function initvar: error: undefined reference to 'getuid'
init.o:function init: error: undefined reference to 'environ!'
bltin/expr.o:function expr_operator: error: undefined reference to 'geteuid'
bltin/expr.o:function expr_operator: error: undefined reference to 'getegid'
linenoise_elks.o:function linenoiseNoTTY: error: undefined reference to 'stdin'
linenoise_elks.o:function linenoiseNoTTY: error: undefined reference to 'stdin!'
linenoise_elks.o:function disableRawMode.constprop.4: error: undefined reference to 'tcsetattr'
linenoise_elks.o:function enableRawMode.constprop.5: error: undefined reference to 'tcgetattr'
linenoise_elks.o:function enableRawMode.constprop.5: error: undefined reference to 'tcsetattr'
linenoise_elks.o:function linenoise: error: undefined reference to 'stdout'
linenoise_elks.o:function linenoise: error: undefined reference to 'stdout!'
linenoise_elks.o:function linenoise: error: undefined reference to 'stdin'
linenoise_elks.o:function linenoise: error: undefined reference to 'stdin!'
autocomplete.o:function completion: error: undefined reference to 'opendir'
autocomplete.o:function completion: error: undefined reference to 'readdir'
autocomplete.o:function completion: error: undefined reference to 'closedir'
autocomplete.o:function completion: error: undefined reference to 'opendir'
autocomplete.o:function completion: error: undefined reference to 'readdir'
autocomplete.o:function completion: error: undefined reference to 'stdout'
autocomplete.o:function completion: error: undefined reference to 'stdout!'
autocomplete.o:function completion: error: undefined reference to 'closedir'
/usr/lib/x86_64-linux-gnu/gcc/ia16-elf/6.3.0/../../../../../ia16-elf/lib/libc.a(lib_a-sbrkr.o):function _sbrk_r: error: undefined reference to '_sbrk'
/usr/lib/x86_64-linux-gnu/gcc/ia16-elf/6.3.0/../../../../../ia16-elf/lib/libc.a(lib_a-signalr.o):function _kill_r: error: undefined reference to '_kill'
/usr/lib/x86_64-linux-gnu/gcc/ia16-elf/6.3.0/../../../../../ia16-elf/lib/libc.a(lib_a-signalr.o):function _getpid_r: error: undefined reference to '_getpid'
/usr/lib/x86_64-linux-gnu/gcc/ia16-elf/6.3.0/../../../../../ia16-elf/lib/libc.a(lib_a-sysisatty.o):function isatty: error: undefined reference to '_isatty'
[... cut ...] nap/bin:/home/cocus/mystuff/80c188/smlr-c-stuff/SmallerC:/home/cocus/mystuff/80c188/smlr-c-stuff/SmallerC
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
LC_NUMERIC=es_AR.UTF-8
OLDPWD=/home/cocus/mystuff/elks/cross
TERM_PROGRAM=vscode
VSCODE_IPC_HOOK_CLI=/run/user/1000/vscode-ipc-5d54cbd7-ede7-42bf-ad81-dd3f17dd47b1.sock
_=/usr/bin/env
```

Now, after I am able to build ELKS and validate it boots on emu86, I would like to dive deep in the PR#934 and start changing stuff around ;) 

Thanks!

ghaerr, 2021-05-09:
Hello @cocus,

Make sure you have sourced the required environment using `. env.sh` from the main ELKS directory, and try running the regular make as just `make` without parallel -j12. (However, it looks like your environment is correct).

It seems having the system-wide `ia16-elf-gcc` installed may be the issue, since the ELKS version uses its own crt0.S and libc, which are in different places, and your link is showing a search in /usr/lib/x86_64-linux-gnu/gcc/ia16-elf/6.3.0/../../../../../ia16-elf/lib/libc.a, which is definitely incorrect. 

The compiler maintainer, @tkchia, may have more to say about whether or not two versions can be simultaneously installed and work.

Thank you!

cocus, 2021-05-09:
@ghaerr I figured out why the build wasn't picking up the toolset provided on cross/. Turns out, the packaged pre-compiled binaries don't have the +x bit set 🤦‍♂️ 
I've removed the gcc-ia16-elf package from my system and tried again, and as you can see, the env is properly set, but the binaries don't have the execute bit set:
```
cocus@u20vm:~/mystuff/elks$ ia16-elf-gcc --version
bash: /home/cocus/mystuff/elks/cross/bin/ia16-elf-gcc: Permission denied
cocus@u20vm:~/mystuff/elks$ stat /home/cocus/mystuff/elks/cross/bin/ia16-elf-gcc
  File: /home/cocus/mystuff/elks/cross/bin/ia16-elf-gcc
  Size: 4682152         Blocks: 9152       IO Block: 4096   regular file
Device: 805h/2053d      Inode: 3424840     Links: 1
Access: (0664/-rw-rw-r--)  Uid: ( 1000/   cocus)   Gid: ( 1000/   cocus)
Access: 2021-04-30 17:00:02.000000000 -0300
Modify: 2021-04-30 17:00:02.000000000 -0300
Change: 2021-05-08 20:48:20.574506832 -0300
 Birth: -

```

tkchia, 2021-05-09:
Hello @cocus, hello @ghaerr,

> The compiler maintainer, @tkchia, may have more to say about whether or not two versions can be simultaneously installed and work.

Well, yes, that is possible, and this is what I do sometimes.  But of course you need to make sure the `$PATH` is correct, depending on what you want to do at each point.

For building ELKS with the specific `gcc-ia16` version that ELKS uses, what I normally do is to set the `$PATH` in its own subshell, like so:
```
$ cd .../elks
$ (. ./env.sh && make all)
```
This way, the path settings in `env.sh` will only apply to the commands inside the `(`...`)`.  Once the ELKS build is done, you will automagically get back to your system-wide `gcc-ia16` version.  This procedure should reduce the chances of confusion.

Hope this helps.

Thank you!

cocus, 2021-05-11:
Thanks!
I think it did it, and I've managed to build it, by just setting the PATH thru sourcing ./env.sh. The issues I had before were just because the cross.zip I gathered from the CI had non-executable binaries (i.e. bit x missing). I did a local build of the toolchain and voilà.
As a side note, is there some "minimal" memory recommendation? My board has 128k and I think I should add additional memory to run this.

mfld-fr, 2021-05-12:
You can start to save RAM by using the ROM version of the kernel, that would  be a 64K BIOS extension @ E0000h, as your target has 512K ROM. See emu-rom.config.

ghaerr, 2021-05-12:
So far, the minimum RAM demonstrated to boot the kernel and run a shell on a PC has been 256k, but this can likely be shaved down with a little more work. @mfld-fr's suggestion of using the ROM version of the kernel is a very good one as 64k RAM is immediately saved. 

There are two version of the ROM configuration that are currently tested: emu86-rom.config uses a ROM filesystem, while  emu86-rom-full.config uses BIOS INT 13h for a disk filesystem, both with kernel in ROM.

cocus, 2021-05-12:
Hi all, I am planning to use the ROM version for sure since I have a lot of flash at my disposal. Right now I am in the middle of modifying the irq and pit code, but I will have to rebase my changes after your latest PR is merged. I've started to add some configs (some of them apply to the PC version as well) on my repo.
In any case, my end goal would be to run this from flash without a pc bios (only a small stub that initializes the hardware and jumps into the elks rom code). After that works (at least on the emu86), I plan to add some additional features like the LEDs and the SD card I've hacked on this board.

mfld-fr, 2021-05-14:
As a side note, I just compared the PCB registers of my R8810 against your 80C188EB ones, and they look to have the same bits.

The only differences are:
1. the register offsets are rearranged in the R8810 to host both the DMA x 2 and the serial port (the R8810 merges the 2 PCB layouts found respectively in the 80C188XL and the 80C188EB),
2. the IRQ line numbering: 8 lines for the 88C188xx, and 12 for the R8810 (+4 for 2 DMA, 1 watchdog and 1 reserved).

It means that you just have to modify the I/O mapping in EMU86 and reuse the existing 'xxx-advantech' PIC, PIT and serial emulated devices; and that I would just have to adjust the register offsets and the IRQ line numbering in your ELKS port :smile:

ghaerr, 2021-05-14:
@mfld-fr : 

> It means that you just have to modify the I/O mapping in EMU86 and reuse the existing 'xxx-advantech' PIC, PIT and serial emulated devices; and that I would just have to adjust the register offsets and the IRQ line numbering in your ELKS port 

Does this mean we should agree on some file naming conventions now for the shared ELKS kernel files using the R8810/80C188 architecture? I am not up on the two devices so I would suggest you and @cocus work out a naming scheme that works for both devices. We might want to do the same in EMU86 as well ?

mfld-fr, 2021-05-16:
@ghaerr : it looks like you forgot to remove the INT 10h from the init / setup code in #941... that is a PC BIOS dependency.

ghaerr, 2021-05-16:
@mfld-fr : 

> it looks like you forgot to remove the INT 10h from the init / setup code in #941... that is a PC BIOS dependency.

Ahh... you mean the one required for `putc`?  Not sure how/if to handle that one, should non-PC version just not output any text at all at boot?

Also saw that non-CONFIG_ROMCODE REL_SYS version still uses extra INT 12h to get memory size a 2nd time. REL_SYS versions could be built for non-IBMPC platforms...

mfld-fr, 2021-05-17:
@cocus : reviewed your [bootstrap first version](https://github.com/cocus/8018x-elks-ldr/commit/7756abeb6d0925e14b260684aea14faa9ee07624). Just spotted two points about the initial CS and stack. It would be nice if you could add the datasheet to that project, for the reader to understand the bits.

mfld-fr, 2021-05-18:
Hello @cocus, could you please open a PR in EMU86, submitting your branch `orckitt-566-emu86` **as is**, so that we can start to work in mainstream on the needed emulation to support ELKS 8018X configuration ?

cocus, 2021-05-18:
Sure, I will do that tomorrow if I have some time. I will have to rebase (from or566) and push my latest changes on top since my branch is outdated and I didn't push the latest changes.

I think I haven't hacked too much the sources, which means that for instance, I am still using elks code (like for the timer and interrupts). At that stage I wasn't concerned about it, but now it needs to be revised.

ghaerr, 2021-05-19:
After more deeply reviewing the ELKS character device directory for setup, BIOS, and BDA dependencies, more reorganization of source files is needed as we move forward with the 8018X porting effort. Almost all of this work won't affect the embedded port work in progress, but to avoid code duplications in the headless console, and show the what is coming, I put together a map of the changes. This will move into porting-guide.txt when finished:
```
Console Devices and dependencies
================================

Old filename        New filename & Setup/BDA Usage
------------        -----------------------------------------------------
                    Direct Console
dircon.c            console-direct.c (6845 CRTC Direct Video RAM Display)
                        BDA:0x4A    VID_COLS
                        25          VID_LINES
                        BDA:0x63    CRTC_PORT (6845 CRTC)
                        BDA:0x4C    VID_PAGE_SIZE
                        BDA:0x49    VID_MODE (7=MDA for video seg, force 1 console)
                        BDA:0x50    VID_CURSOR_X
                        BDA:0x51    VID_CURSOR_Y
                    kbd-scancode.c (uses IRQ 1 handler to read 8042 kbd controller)
bell.c              bell-pc.c (speaker port, 8254 timer chip)
                        SPEAKER_PORT        0x61
                        TIMER2_DATA_PORT    0x42
                        TIMER_CMDS_PORT     0x43

                    BIOS Console
bioscons.c          console-bios.c (BIOS Console Display)
                        SETUP_VID_COLS      setupb(7)
                        SETUP_VID_LINES     setupb(14)
                        BDA:0x49    VID_MODE (7=MDA, force 1 console)
bioscon-asm.S       console-bios-asm.S (all INT 10h and INT 16h BIOS Console routines)
                        INT 16h AH=01h  Poll kbd
                        INT 16h AH=00h  Read kbd
                        INT 10h AH=02h  Set cursor position 
                        INT 10h AH=03h  Get cursor position
                        INT 10h AH=05h  Set active page 
                        INT 10h AH=06h  Scroll active page up
                        INT 10h AH=07h  Scroll acrive page down 
                        INT 10h AH=09h  Write character and attribute
kbd-bios.c          kbd-poll.c (Polling keyboard using kernel timer callback)
bios-asm.S          conio-bios.S (INT 16h polling conpoll)
                        INT 16h AH=01h  Poll kbd
                        INT 16h AH=00h  Read kbd
                    bell-pc.c (uses speaker port, 8254 timer chip)
        
                    Headless Console (BIOS or serial port output)
headless-bios.c     console-headless.c (Headless Console arch-neutral polling kbd)
kbd-bios.c          kbd-poll.c (Polling keyboard using kernel timer callback)
bios-asm.S          conio-bios.S (INT 16h conpoll and conout) 
                        INT 16h AH=01h  Poll kbd
                        INT 16h AH=00h  Read kbd
                        INT 10h AH=0Eh  Write text teletype mode
                    bell.c (^G output to conout)

                    Headless Console (8018X serial output)
headless-8018x.c    console-headless.c (Headless Console arch-neutral polling kbd)
                    kbd-poll.c (Polling keyboard using kernel timer callback)
                    conio-8018x.c (serial conpoll and conout)
                        coninit()   initialize serial port
                        conpoll()   return nonzero character if received
                        conout(c)   output character to serial

```
At this point, I think it would be best to implement new hardware devices in the EMU86 emulator than just jumping in to trying a kernel boot.

The biggest issues for the 8018X port (shown above) is that the bell() code needs pulling out, as it talks directly with the 8254 timer 2, and that there are only three routines needed for the 8018X "console" - `coninit`, `conpoll`, and `conout`. If these routines have already been written in headless-8081x.c, go ahead with a PR and I'll get the moved to the reorganization afterwards.

mfld-fr, 2021-05-19:
Implemented some 8018X stuff in EMU86 (branch `or556`) and tried to run ELKS from PR #949. Got the first line:
```
mfld@home /mnt/data/home/mfld/advantech/emu86 $ ./pcat `cat emu86.pts`
8018X serial headless console
```

But EMU86 complains about an implemented access to I/O:
```
error: I/O write byte Ch to unmapped 3F2h
```

That is coming from:
```
void INITPROC blk_dev_init(void)
{
    outb_p(0xc, (void *) 0x3f2);	//FIXME move somewhere
```

What is the purpose of this line ?

mfld-fr, 2021-05-19:
If commenting that line:
```
mfld@home /mnt/data/home/mfld/advantech/emu86 $ ./pcat `cat emu86.pts`
8018X serial headless console
8018X machine, 256K base RAM.
ELKS kernel 0.4.0 (25440 text, 0 ftext, 3216 data, 35408 bss, 26912 heap)
Kernel text at e062:0000, data 80:0000, top 4000, 190K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
#
```

ghaerr, 2021-05-19:
> outb_p(0xc, (void *) 0x3f2);	//FIXME move somewhere

Geez. Looks like you and your EMU86 have found another vagabond OUTB. I'm not certain what its for, but https://stanislavs.org/helppc/ports.html shows that its likely associated with something required by the 8237 DMA controller.

For the time being, it could be wrapped with CONFIG_ARCH_IBMPC, or if 0, since DMA is not used in ELKS at this time. We need to research whether this is required for DMA hardware at boot, or not.

> mfld@home /mnt/data/home/mfld/advantech/emu86 $ ./pcat `cat emu86.pts`
8018X serial headless console
8018X machine, 256K base RAM.
ELKS kernel 0.4.0 (25440 text, 0 ftext, 3216 data, 35408 bss, 26912 heap)
Kernel text at e062:0000, data 80:0000, top 4000, 190K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
\#

Nice work! Looks like you've implemented the keyboard polling in ELKS as well as the serial I/O and PIC/PIT emulation in EMU86. Please post a PR on 8018X changes to ELKS so that I can use those updates in the console source reorg mentioned above.

ghaerr, 2021-05-19:
Are you running with the unmodified config.h? It is hard-coded for 256K RAM, and you have 512, correct? The available memory will want to be set somehow in the arch-specific setup.S ultimately.

Good to know there's 190K free (before shell) with 256K RAM.

ghaerr, 2021-05-19:
> I'm not certain what its for, but https://stanislavs.org/helppc/ports.html shows that its likely associated with something required by the 8237 DMA controller.

Oops! I was reading the outb arguments in the wrong direction. This is sending 0x0C to port 0x3F2, which is a floppy disk controller commands port. Guessing yet again, I bet this is motor off command for floppy. I'll check more into it.

ghaerr, 2021-05-19:
The IBM PC FDC discussed in https://wiki.osdev.org/Floppy_Disk_Controller under "DOR Bitflag Definitions" seems to show that 0x0C sent to the DOR register (0x03F2 on PC) turns off all drive motors, and enables floppy IRQ and DMA.

I don't really know whether this is needed or not. Suggest at this point moving the OUTB to the only working ELKS FD/HD driver, which is the BIOS controller, bioshd.c::bioshd_init(). Since this is the next routine called, lets move it to the first line in the that routine, just before the call to bioshd_getfdinfo(). If you don't want to mess with sticking that into your next PR, let me know and I'll move it.

Thanks!

cocus, 2021-05-19:
> If commenting that line:
>
> ```
> mfld@home /mnt/data/home/mfld/advantech/emu86 $ ./pcat `cat emu86.pts`
> 8018X serial headless console
> 8018X machine, 256K base RAM.
> ELKS kernel 0.4.0 (25440 text, 0 [...]

This is literally amazing @mfld-fr! Wow!
What have you used to bootstrap ELKS? I would love to try this on my local emu86 branch today!

cocus, 2021-05-19:
> Are you running with the unmodified config.h? It is hard-coded for 256K RAM, and you have 512, correct? The available memory will want to be set somehow in the arch-specific setup.S ultimately.
> 
> Good to know there's 190K free (before shell) with 256K RAM.

It would be nice to see how much it's free after launching the shell. If it's still within the 128K boundaries, I could load this up on my board today!
This would be a good thing to talk about afterwards. ELKS must be explicitly avoiding some memory areas (like BDA, EBDA, extension ROMs like VBIOS), but in our scenario, we don't have any of that. We would like to use the entirety of the available RAM.

ghaerr, 2021-05-19:
> It would be nice to see how much it's free after launching the shell. If it's still within the 128K boundaries, I could load this up on my board today!

Here's what I think is a likely layout of ELKS running CONFIG_ROM on 8018X:
```
ELKS typical ROM memory layout
  
 +--------------------  seg  off  linear size (size decimal)  --------------------+
 | interrupt vectors    0000:0000 00000  0400 (1024)
 | ...
 | ...
 | ...
 | BIOS data area       0040:0000 00400  0100 (256) (UNUSED in ROM config)
 | bootopts             0050:0000 00500  0100 (256) (UNUSED in ROM config)
 | kernel setup data    0060:0000 00600  0200 (512) (CONFIG_ROM_SETUP_DATA=0x0060)
 | ...
 | kernel data segment  0080:0000 00800 10000 (65536=64k, can be reduced)
 | ...
 | ...
 | external sys buffers 1080:0000 01800 10000 (default=64k, CONFIG_FS_NR_EXT_BUFFERS=64)
 | ...
 | ...
 | application memory   1080:0000 xxxxx       (starts at ~66K w/no sys buffers)
 | ...
 | top of 128k ram      2000:0000 20000
 | ...
 | application memory   2080:0000 xxxxx       (starts at ~130K w/no 64K sys buffers)
 | ...
 | ...
 | top of 256k ram      4000:0000 40000
 | ...
 | top of 512k ram      8000:0000 80000
 | start of ROM disk    8000:0000 80000 30000 (typical size 180K)
 | end of ROM disk      B000:0000 B0000
 | ...
 | BIOS ROM             F000:0000 F0000 10000 (65536)
 +--------------------------------------------------------------------------------+
```

You'll need to eliminate the external system buffers by setting CONFIG_FS_EXTERNAL_BUFFER=n or set CONFIG_FS_NR_EXT_BUFFERS=1, after which it looks like you'd start with 62K free on your 128K system. The standalone shell `sash` uses about 51K (run it and `ps` to see), and that means you'd have 11K free. I'd suggest changing these options and running on EMU86 with the memory set to 128K to check. I know ELKS won't run with MINIX or FAT filesystems with external system buffers set to less than 12, but ROMFS doesn't use them.

> ELKS must be explicitly avoiding some memory areas (like BDA, EBDA, extension ROMs like VBIOS), but in our scenario, we don't have any of that. We would like to use the entirety of the available RAM.

From the memory layout above, BDA and OPTSEG only use 512 bytes, but that can reduced. The kernel heap, shown at boot at 26,912 bytes, is much larger than is needed, and that can be reduced as well. Running `meminfo` will show what's using up kernel local heap, as well as external system buffer allocations.

mfld-fr, 2021-05-20:
> What have you used to bootstrap ELKS?

I put a part of your bootstrap code in 'rom-or556.c'.

mfld-fr, 2021-05-21:
I updated to latest ELKS and EMU86. Input looks like to be active, but SASH responses are not what is expected:
```
mfld@home /mnt/data/home/mfld/advantech/emu86 $ ./pcat `cat emu86.pts`
Headless console
8018X machine, 256K base RAM.
ELKS kernel 0.4.0 (25520 text, 0 ftext, 3232 data, 33664 bss, 28640 heap)
Kernel text at e062:0000, data 80:0000, top 4000:0, 190K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# pwd

# cd /

# ls

# cd /bin

# ls -l

# echo Bouh!

#
```

ghaerr, 2021-05-21:
> Input looks like to be active, but SASH responses are not what is expected

I'll look into it and post a fix, give me a sec.

ghaerr, 2021-05-21:
@mfld-fr : Strange. I'm running the latest changes, and `emu86-rom.config` as .config. I'm running an older commit of EMU86 0b42574428b08b21d47fc64f47b655dbe14e8dc0 and everything is working. I'm running CONSOLE=stdio.
```
Gregs-MBP2:elks-gh greg$ ./es
info: load address E0000h
info: load file elks/arch/i86/boot/Image
info: file size=10000h
success: file loaded
info: load address 80000h
info: load file image/romfs.bin
info: file size=2C4D9h
success: file loaded

ELKS Setup Headless console
PC/AT class machine, NEC V20 CPU, 512K base RAM.
ELKS kernel 0.4.0 (30208 text, 0 ftext, 3840 data, 37664 bss, 24032 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 446K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# 
# 
# 
# 
# ls -l
dr-xr-xr-x  1 root     root           76  Jan 01 1970  .
dr-xr-xr-x  1 root     root           76  Jan 01 1970  ..
dr-xr-xr-x  1 root     root          296  Jan 01 1970  bin
-r-xr-xr-x  1 root     root          249  Jan 01 1970  bootopts
dr-xr-xr-x  1 root     root          310  Jan 01 1970  dev
dr-xr-xr-x  1 root     root           95  Jan 01 1970  etc
dr-xr-xr-x  1 root     root            9  Jan 01 1970  home
dr-xr-xr-x  1 root     root           24  Jan 01 1970  lib
dr-xr-xr-x  1 root     root            9  Jan 01 1970  mnt
dr-xr-xr-x  1 root     root           29  Jan 01 1970  root
dr-xr-xr-x  1 root     root            9  Jan 01 1970  tmp
dr-xr-xr-x  1 root     root           15  Jan 01 1970  var
# 
```

Haven't tried 8018X.config, as I haven't updated EMU86 yet.

ghaerr, 2021-05-21:
Is serial T0BUF being overrun by chance? Looks like fast output may be being dropped by EMU86?
```
void conio_putc(byte_t c)
{
    while((inw(0xff00 + 0x66) & 8) == 0);   /* S0STS */
    outb(c, 0xff00 + 0x6A);         /* T0BUF */
}
```

ghaerr, 2021-05-21:
Latest version of EMU86 works fine with latest ELKS using emu86-rom.config, when using CONSOLE=stdio.

Both are running headless console: difference is in conio-8018x.c, which uses serial port rather than BIOS for console I/O.

cocus, 2021-05-22:
> Is serial T0BUF being overrun by chance? Looks like fast output may be being dropped by EMU86?
> 
> ```
> void conio_putc(byte_t c)
> {
>     while((inw(0xff00 + 0x66) & 8) == 0);   /* S0STS */
>     outb(c, 0xff00 + 0x6A);         /* T0BUF */
> }
> ```

I didn't check for overruns on the emu86 code, there's a TODO still in place. I think the timer (and the emulation) on emu86 makes this more evident by just "lagging" a lot when writing to the console. Maybe when we add IRQs we'll have a much more responsive interaction.

ghaerr, 2021-05-22:
> I think the timer (and the emulation) on emu86 makes this more evident by just "lagging" a lot when writing to the console.

What "lagging" exactly are you talking about? Character echo when typing at the ELKS shell prompt?

I haven't looked at the 8018X timer emulation, but the PC/ELKS timer (timer-elks.c) has a "TIMER_MAX" which is the number of 8086 instructions to execute before triggering an interrupt. EMU86 isn't using the actual timer register countdown.

Nonetheless, @mfld-fr is saying that EMU86/ELKS for 8018X is now broken... is it working on your repo? That might help in chasing down the current problem.

cocus, 2021-05-22:
> Nonetheless, @mfld-fr is saying that EMU86/ELKS for 8018X is now broken... is it working on your repo? That might help in chasing down the current problem.

I've updated emu86 to master, but still using my prebuilt images of ELKS, it still works.
Now, checked the latest ELKS master, built, and it still boots.

ghaerr, 2021-05-22:
>  it still boots.

Does it also display properly after doing `ls -l /`? 

@mfld-fr's boots, but seemingly doesn't display output after commands.

cocus, 2021-05-22:
> Does it also display properly after doing `ls -l /`?

It does!
```
info: load address E0000h
info: load file /home/cocus/mystuff/elks/elks/arch/i86/boot/Image
info: file size=10000h
success: file loaded
info: load address 80000h
info: load file /home/cocus/mystuff/elks/image/romfs.bin
info: file size=BF54h
success: file loaded
Headless console
8018X machine, 128K base RAM.
ELKS kernel 0.4.0 (25616 text, 0 ftext, 3232 data, 35424 bss, 26880 heap)
Kernel text at e062:0000, data 80:0000, top 2000:0, 62K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# ls -l /
dr-xr-xr-x  1 root     root           76  Jan 01 1970  .
dr-xr-xr-x  1 root     root           76  Jan 01 1970  ..
dr-xr-xr-x  1 root     root           28  Jan 01 1970  bin
-r-xr-xr-x  1 root     root          249  Jan 01 1970  bootopts
dr-xr-xr-x  1 root     root          310  Jan 01 1970  dev
dr-xr-xr-x  1 root     root           95  Jan 01 1970  etc
dr-xr-xr-x  1 root     root            9  Jan 01 1970  home
dr-xr-xr-x  1 root     root           24  Jan 01 1970  lib
dr-xr-xr-x  1 root     root            9  Jan 01 1970  mnt
dr-xr-xr-x  1 root     root            9  Jan 01 1970  root
dr-xr-xr-x  1 root     root            9  Jan 01 1970  tmp
dr-xr-xr-x  1 root     root           15  Jan 01 1970  var
# 
```

ghaerr, 2021-05-22:
Very strange. Looks like you're running a slightly modified version of ELKS than straight from the repo, since I see 128K RAM, rather than the still-default 256k (SETUP_MEM_KBYTES in config.h).

BTW, with the 128K system you're emulating and the sash shell (/bin/sash) running, I think you've got only 11K free, so you can't actually run hardly any programs from the shell prompt. Most of the commands, such as ls and pwd are actually built into sash. I've got a PR to reduce the 26,880 kernel heap down to less than 1K that will help you get more memory. I'm interested in trying to get a decent working ELKS on 128K, so don't solder more RAM in yet :)

mfld-fr, 2021-05-22:
Got it, @cocus uses `SERIAL=stdio`, so the serial input rate is actually the typing rate, while I am using `SERIAL=pty`, with full input speed by default. So ELKS got only the ending LF for each command, as previous characters were over flooded.

It works when slowing down the serial input with `pcat -d 1000` (1000 ms is too much, that's just for testing). It also works on my side with `SERIAL=stdio`, but because of a known problem that will be fixed later, that option cannot be used for debugging in interactive mode or after ^C.

ghaerr, 2021-05-22:
I have OR556 target running now on EMU86, thanks to @mfld-fr fixes above.

I still think we have a problem with EMU86 8018X timer emulation which uses actual timer countdown instead of instruction count (as does ELKS/PC timer emulation). While OR556 target is operational, it isn't much usable. Typing 'ls' quickly results in the following:
```
Gregs-MBP2:emu86-new greg$ ./es
info: load address E0000h
info: load file ../elks-gh/elks/arch/i86/boot/Image
info: file size=10000h
success: file loaded
info: load address 80000h
info: load file ../elks-gh/image/romfs.bin
info: file size=C259h
success: file loaded
Headless console
8018X machine, 256K base RAM.
ELKS kernel 0.4.0 (25136 text, 0 ftext, 3168 data, 33792 bss, 28576 heap)
Kernel text at e062:0000, data 80:0000, top 4000:0, 190K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
#   
warning: serial port input overflow
s
```
One must type slowly in order to use emulator, which is not realistic. Perhaps the timer countdown should be divided by a constant to better reflect real time taken by EMU86 emulating instructions.

mfld-fr, 2021-05-22:
Can you please measure the IPS on your machine with the `-m` option ?

ghaerr, 2021-05-22:
Here you go:
```
#  info: ips=3093150
info: ips=3118727
info: ips=3067477
info: ips=3131079
info: ips=3171691
info: ips=3154688
info: ips=3165168
info: ips=3155489
info: ips=3130660
info: ips=3132269
lsinfo: ips=2991768
info: ips=3118448

.
..
bin
bootopts
dev
etc
home
lib
mnt
root
tmp
var
# info: ips=2978391
info: ips=3050889
```

toncho11, 2021-05-22:
I do not want to steal this thread, but in the future I would love to see a port to the Sanyo MBC 555

https://www.seasip.info/VintagePC/sanyo.html#ros

It is a "PC" that has no BIOS and the BIOS is loaded from floppy together with DOS 2.11. 
I just wanted to say that is a another BIOS-less 8088 computer.

ghaerr, 2021-05-22:
Hello @toncho11, nice to hear from you!

> I would love to see a port to the Sanyo MBC 555 (I have one)
> I just wanted to say that is a another BIOS-less 8088 computer.

With the changes occurring for the 8018X ports, things are rapidly moving in the right direction to allow a version of ELKS running on bare metal for PCs. It's on my list. There will be issues on how to get it booted, seperate from the ELKS kernel issues, though.

mfld-fr, 2021-05-22:
Thanks. So your EMU86 iterates at ~3M cycles (or instructions) / second. The 8018X timer 2 counter is incremented each cycle. ELKS expects the timer 2 frequency to be 1KHz, so sets the maximum count to 5000. In EMU86, that frequency is 3M / 5K = 600 Hz.

That timer 2 pre-scales the timer 0. ELKS expects the timer 0 frequency to be 100 Hz, so sets the maximum count to 10. In EMU86, that frequency is 600 / 10 = 60 Hz (1.6 ms interval).

I hardly see any emulation problem with the timers with such limited slowdown of 40% compared to the real 8018X target, and I would rather investigate in the polling timing.

ghaerr, 2021-05-22:
I think I'm following you, you're talking only about emulating the 8018X port on ELKS, correct?

The PC ELKS port uses timer2 as the speaker frequency, and uses a fixed 5000 instructions (MAX_TIMER) countdown, ignoring timer 0 register countdown. The ELKS PC emulation works great as far as no lag, although since 8251 UART not being emulated, unsure about whether serial port would be overwritten on input.

mfld-fr, 2021-05-22:
>I think I'm following you, you're talking only about emulating the 8018X port on ELKS, correct?

No, I am dealing with 8018X **timer** emulation.

By the way, my SBC has the same CPU frequency @ 20 MHz, and from mfld-fr/sys86#2, I measured that the real R8810 is executing around 275 instructions / 208 us, so roughly 1.3 instruction / us, and so **130K** instructions each timer 0 interval.

By setting MAX_TIMER to **5K**, you would interrupt ELKS too much for a correct emulation of the real 8018X target. Such correct emulation is critical for @cocus and me to increase our confidence in the current code before burning it into the ROM.

Even for the PC emulation, with a 8086 running @ 4.77 MHz and executing around 0.33 instruction / us, that TIMER_MAX would interrupt ELKS @ 66 Hz on the real target, compared to the intended 100 Hz. The correct value for an accurate timing should rather be 0,33 MHz / 18.2 Hz = 18132.

Moreover, as EMU86 runs 9x faster than a real 8086, you interrupt ELKS @ 600 Hz, so accelerating the keyboard polling. User experience could be better, sure, but the emulation accuracy is not good.

mfld-fr, 2021-05-22:
In other words, the input lag you are observing with EMU86 when typing some commands like `ls` is what you would observe (but 40% faster) on the real target. This is why I am starting to think that the design option of polling the serial port from the console is not the right one for the OR566 or the SNMP-1000.

cocus, 2021-05-22:
> In other words, the input lag you are observing with EMU86 when typing some commands like `ls` is what you would observe (but 40% faster) on the real target. This is why I am starting to think that [...]

I would rather use the serial interrupts instead of the polling, but it's true that polling was quicker to implement :)

cocus, 2021-05-22:
@mfld-fr please review the following table that I've put together to show the main differences between the R8810 and the 80188EB. I've only selected the interesting registers for our needs, but I assume there are more items.

| 80188EB Offset | R8810 Offset | Register                                                                            | Notes                                                                                                                                                                                                                                                                                                                |
|----------------|--------------|-------------------------------------------------------------------------------------|----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| 66h            | 82h          | Serial Status Register                                                              | bit mapping is different. R8810: Missing the different break scenario on 80188EB's DBRK1. Missing CTS bit. OE = OER, TXE = TEMT, FE = FER, TI kinda THRE, RI kinda RDR, RB8/PE kinda PER (RB8 is not impemented on R8810?), DBRK0 = BRK1                                                                                   |
| 6ah            | 84h          | Serial Transmit Buffer Register                                                     | same                                                                                                                                                                                                                                                                                                                 |
| 60h            | 88h          | Baud Rate Compare Register (80188EB) Serial Port Baud Rate Divisor Register (R8810) | 80188EB: If BCLK is not used, then the baudrate is B = Fcpu / (register 60h + 1) or 1/8 of that value if mode 1 thru 4 is used. R8810: Baudrate is B = Microprocessor clock / (16 x register 88h).                                                                                                                   |
| 02h            | 22h          | End-of-Interrupt Register                                                           | same                                                                                                                                                                                                                                                                                                                 |
| 08h            | 28h          | Interrupt Mask Register                                                             | different mapping of serial port interrupt and external interrupts bits: 80188EB: Missing WD and D0, D1 interrupt sources. TMR = TMR, SER = SPI, INT4 = I4, INT3:0 = I3:0. These values are RO on R8810 and RW on the 80188EB. Each bit is a mirror of the interrupt mask source bit on each interrupt control register. |
| 14h            | 44h          | Serial Port Interrupt Control Register                                              | same                                                                                                                                                                                                                                                                                                                 |
| 12h            | 32h          | Timer Interrupt Control Register                                                    | same                                                                                                                                                                                                                                                                                                                 |
| 36h            | 56h          | Timer 0 Control Register                                                            | same                                                                                                                                                                                                                                                                                                                 |
| 46h            | 66h          | Timer 2 Control Register                                                            | same                                                                                                                                                                                                                                                                                                                 |

ghaerr, 2021-05-22:
>  I am starting to think that the design option of polling the serial port from the console is not the right one for the OR566 or the SNMP-1000.

No problems there. The headless console was originally developed as a first-pass at getting ELKS running on EMU86, using only three BIOS calls. Its usefulness is its simplicity for porting. The proper way would be to implement a standard ELKS serial driver that is completely configurable via stty, termios, etc and then have that driver automatically become a console via CONFIG_CONSOLE_SERIAL option. The serial driver would be included when CONFIG_CHAR_DEV_RS is set, and the boot-time settings for the UART will be taken from the default termios values associated with /dev/ttyS0 in ntty.c.

On the Advantech, the serial port could be configured as a normal /dev/ttyS1, perhaps not using serial console, with external access dynamically set to allow login via network or serial, depending upon init run level.

I'll move forward with the small changes required to compartmentalize the PC 8251 normal and fast serial drivers to CONFIG_ARCH_IBMPC so that a new 8018X-compatible, interrupt-driven serial driver can be written and integrated easily.

ghaerr, 2021-05-22:
> the following table that I've put together to show the main differences between the R8810 and the 80188EB

Very interesting, thanks for posting! I'll keep an eye out for these differences as both of you push forward with your ports.

What other items are still not done on your port (other than interrupt-driven serial driver)?

cocus, 2021-05-22:
> What other items are still not done on your port (other than interrupt-driven serial driver)?

Well, I want to validate that a hand-crafted 512k flash image still boots on emu86 (removing the hardcoded loader code), and then I would try to load this on the physical board.

After that, I have the bit-banged sd-card that I would like to add support for.
Maybe I could hack an ethernet card, or something else. My idea is to try to do something interesting (maybe not so useful) with these boards.

ghaerr, 2021-05-23:
>  I have the bit-banged sd-card that I would like to add support for.

That sounds interesting. What hardware interface is that using, serial? Will it require ELKS' FAT filesystem, or does it have its own implementation and just require block (simulated disk sector) I/O?

> Maybe I could hack an ethernet card

Take a look at elks/arch/i86/drivers/net/wd.c, its a network driver written entirely in C for the WD8003. You might use that as a model for another card.

cocus, 2021-05-23:
> That sounds interesting. What hardware interface is that using, serial? Will it require ELKS' FAT filesystem, or does it have its own implementation and just require block (simulated disk sector) I/O?

I'm using some of the GPIOs (which can also be used as programmable chip select pins) in order to emulate an SPI host peripheral, with only one chip select. Some voltage translation (just 1k resistors, hackish as hell!), and voila. Now, on the processor, the sd card exposes a lot of things, but most importantly, the access to data thru data blocks of 512bytes. In my tests, I have used the petitfs (FAT16/32) from ElmChan, and that worked perfectly fine. For this platform I would like to access it somehow, then we can discuss which filesystem to use on it (since that should be pretty trivial to change afterwards). 

> Take a look at elks/arch/i86/drivers/net/wd.c, its a network driver written entirely in C for the WD8003. You might use that as a model for another card.

I have some rtl8019s and rtl8029s. I think I also have one of thos wd8003 cards which I couldn't make it run under a modern Linux on an old pentium 2 machine, but it probably works fine. All of these cards are ISA, so I should rig something up to use them

ghaerr, 2021-05-23:
>  but most importantly, the access to data thru data blocks of 512bytes.

Perfect.

> For this platform I would like to access it somehow

An ELKS block device driver can be written for that. The way they work in ELKS is that 1024-byte block I/O requests from the filesystems (FAT or MINIX, but not ROMFS) are buffered if possible in memory buffers, otherwise sent through a low-level block read/write routine that is not interrupt driven to do the I/O. The I/O is then done synchronously to the sd card, and copied directly into a specified segment:offset buffer. After I/O completion, the ELKS block/buffer/filesystem mechanism handles everything automatically. Either MINIX or FAT filesystems on the sd card can be mounted and read/written to normally.

We can start with the "BIOS" floppy/hard disk driver bioshd.c, which I can strip down to bare essentials for you to then fill in the details for your sd card volume information, and write the actual I/O routine, etc. Then, some changes to the general-purpose low-level block device handler, ll_rw_blk.c will be made, which makes a callback into your sd block driver to actually do the I/O. That will be fun.

On another note, there is an unused SSD block driver ssd.c that could be worth looking at. I have no idea if it works, but it basically does the same thing discussed above but could be modified to read/write flash blocks, if desired. It would be interesting to get that working.

> then we can discuss which filesystem to use on it (since that should be pretty trivial to change afterwards).

Yes, the desired filesystems can be compiled in with configuration options easily.

cocus, 2021-05-23:
> The I/O is then done synchronously to the sd card, and copied directly into a specified segment:offset buffer.

This os good, however, is the block to be read contained in these segments or do they cross boundaries? Having to check for that usually slows things down because I have to add a few additional instructions.

My driver was written in assembly and called by C code, since there's an exceptionally large difference in speed (~10kHz of clock with the C code vs ~250kHz with the handwritten assembly code). The good thing is that I have some code that emulates the sd card on emu86, by monitoring the GPIOs. So this code can be checked suepr easily without having to resort to the actual device.

ghaerr, 2021-05-23:
> is the block to be read contained in these segments or do they cross boundaries?

The segment passed to your I/O routine will not require changing, just copy 1024 bytes from the sd card to/from that segment offset 0 to 1023. If no external buffers are used, that segment will be the kernel DS. However, far better for your routine to be able to copy to a passed segment, but no segment arithmetic/wrap ever needed.

> written in assembly and called by C code, since there's an exceptionally large difference in speed (~10kHz of clock with the C code vs ~250kHz with the handwritten assembly code).

Yes, all the low-level block, filesystem and network driver code was also rewritten as well to use a fast kernel `fmemcpyw` ASM routine that copies words between two segment:offset pairs. This improved ELKS speed considerably.

[11 more replies]

closed 2021-05-24 by cocus