ghaerr/elks #939: Issues compiling ELKS on WSL

agithubuser444, 2021-05-15:
Hello, i am trying to compile ELKS on WSL, but i keep running into a snag.  It keeps doing conflicts with something, i really don't know what's going on since my skills with programming are not good.

Here is a link to the output https://pastebin.com/Nh8RQfpf

ghaerr, 2021-05-15:
Hello @agithubuser444,

Thanks for trying ELKS out. The build is breaking in compiling `nano`, a recently-added editor with what seems to be too many dependencies between system variations. Try commenting the build of nano out by removing line 34 in the application Makefile in elkscmd/Makefile and re-running make from top-level.

Thank you!

agithubuser444, 2021-05-15:
@ghaerr Thank you, it is compiling flawlessly now.

ghaerr, 2021-05-15:
Thank you @agithubuser444! Let us know if you need more help. 

We'll probably leave the `nano` compile in for now, but it is proving to be somewhat non-portable.

closed 2021-05-15 by ghaerr