ghaerr/elks #951: /bootopts enhancements

Mellvik, 2021-05-20:
[Continued from #940 ]
As mentioned in #940, there are practical and technical issues with `/bootopts`. 

- The technical issue is that if the file exceeds 255 bytes, everything is ignored. It would make sense (to me) if whatever is within the first 255 bytes is being acted upon.
- As pointed out by @ghaerr in #944, the error message about the '##' preamble missing or the file exceeding 255 bytes easily scrolls off the screen too fast to notice, even when booting from floppy. A possible (albeit slow) way to 'fix' this is to check and report in rc.sys. 

### Further enhancement
The availability of `netirq=` is sort of a game changer - triggers all kinds of ideas for simplifying development and user config of ELKS. Adding iobase= for network and serial (and other ) interfaces would be great, irq= for serial devices, shmem= for some network devices etc. I don't know how complicated this would be, and in pure self interest I'd suggest the network iobase to be first on the list: I'm diving bak into fixing the 3C509 driver.

On a different tangent, it would be very useful if the Wiki mentioned (under boot options) that if the serial console configuration option is active, the `console` option in /bootopts has no effect.

--Mellvik

mfld-fr, 2021-05-20:
Are you reinventing the Linux device tree 😏 ?

ghaerr, 2021-05-20:
> the error message about the '##' preamble missing or the file exceeding 255 bytes easily scrolls off the screen too fast to notice, even when booting from floppy. 

Much work has been done to keep boot screen from scrolling information such as the NE2K IRQ \# or serial IRQ off the screen at boot. It is important that user be able to verify either compiled-in or bootopts settings with this display. 

Is the error message about the '##' preamble missing or the file exceeding 255 bytes being scrolled off screen? If so, it should be moved to later so that user realizes configuration specification error.

> A possible (albeit slow) way to 'fix' this is to check and report in rc.sys.

The rc.sys script can't check that, and sometimes is not run. The error message could possibly be moved to somewhere it won't be scrolled off screen though.

>  It would make sense (to me) if whatever is within the first 255 bytes is being acted upon.

I think it bad idea to silently ignore some boot options, that is purpose of error message.

> Adding iobase= for network and serial (and other ) interfaces would be great, irq= for serial devices, shmem= for some network devices etc. I don't know how complicated this would be, and in pure self interest I'd suggest the network iobase to be first on the list

All these are easy to do, IF the driver is rewritten to use a variable as port address, etc. Currently, only the serial driver is written that way. I didn't (yet) add com1= etc because didn't know the usefulness of com1 vs com2/com3/com4 etc and adding tons of options that aren't used could be a problem for kernel space, especially for infrequently used options.

However, given the multitudes of NICs having separate port addresses out there, I think it would be a great idea to rewrite NIC driver(s) to support variable port address. First step would be to assign an integer variable such as `int netport = NE2K_PORT` in the driver source. Kernel bootopts routine then sets that variable before driver init routine called.

Mellvik, 2021-05-20:
@ghaerr,
> the error message about the '##' preamble missing or the file exceeding 255 bytes easily scrolls off the screen too fast to notice, even when booting from floppy.
>
> Much work has been done to keep [...]
I fully agree. And it's not something that is fully controllable since it depends on the # of devices (e.g. I have 3 serial ports). Also - and this may only apply to me - I'm remotely accessing the ELKS machine, so unless the console redirection works I have no idea what happened :-).
> Is the error message about the '##' preamble missing or the file exceeding 255 bytes being scrolled off screen? If so, it should be moved to later so that user realizes configuration specification error.
> 
If this is easy to do, it's a great idea!
> A possible (albeit slow) way to 'fix' this is to check and report in rc.sys.
> 
> The rc.sys script can't check that, and sometimes is not run. The error message could possibly be moved to somewhere it won't be scrolled off screen though.
> 
Again, this may not be a good idea since it's going to be slow, but I don't see any problem doing it in rc.sys (script).
> It would make sense (to me) if whatever is within the first 255 bytes is being acted upon.
> 
> I think it bad idea to silently ignore some boot options, that is purpose of error message.
> 
I didn't suggest to silently ignore anything. To me it seems reasonable to report the problem and do what's possible.
> Adding iobase= for network and serial (and other ) interfaces would be great, irq= for serial devices, shmem= for some network devices etc. I don't know how complicated this would be, and in pure [...]
The kernel size consideration is important. The amount of sanity checking to implement is also a question. Still - the flexibility in having some of these parameters available at runtime is extremely attractive - in particular during development.
> However, given the multitudes of NICs having separate port addresses out there, I think it would be a great idea to rewrite NIC driver(s) to support variable port address. First step would be to [...]
Great!! I'll submit a PR for that tomorrow (net drivers) - if I get my toolchain back working.
> 

Thank you!

--Mellvik

ghaerr, 2021-05-20:
> And it's not something that is fully controllable since it depends on the # of devices (e.g. I have 3 serial ports). 

Yes, I see. Its likely the three serial init displays that is causing the bootopts error message to scroll off; I didn't think about that. I'll look into a better time to display the message so it can be seen more clearly.

> I'll submit a PR for that tomorrow (net drivers)

Thanks! Please test on various different port addresses (by compile-time initialization of netport variable) so that we know it works before submitting the PR. IIRC, there's quite a bit of assembly language in the NE2K driver that will have to change. 

The WD8003 driver, completely written in C, looks far easier to change, but we will need testers for it. It only needs all occurrences of WD_PORT to be replaced by `netport` and that should complete it. If you like, I can post a PR for that to show what I'm talking about.

Mellvik, 2021-05-21:
Thank you @ghaerr.

> 20. mai 2021 kl. 19:26 skrev Gregory Haerr ***@***.***>:
> 
> 
> And it's not something that is fully controllable since it depends on the # of devices (e.g. I have 3 serial ports).
> 
> Yes, I see. Its likely the three serial init displays that is causing the bootopts error message to scroll off; I didn't think about that. I'll look into a better time to display the message so it can be seen more clearly.
> 
> I'll submit a PR for that tomorrow (net drivers)
> 
> Thanks! Please test on various different port addresses (by compile-time initialization of netport variable) so that we know it works before submitting the PR. IIRC, there's quite a bit of assembly language in the NE2K driver that will have to change.
> 
Will do - this may not be completed today. There will be an extra (OR) instruction per IOoperation now that the port# is no longer a constant (ne2k), but still very efficient compared to compiled C code.
> The WD8003 driver, completely written in C, looks far easier to change, but we will need testers for it. It only needs all occurrences of WD_PORT to be replaced by netport and that should complete it. If you like, I can post a PR for that to show what I'm talking about.
> 
> 
I'll include that when the ne2k changes have been tested. 

Thank you.

—Mellvik

ghaerr, 2021-05-21:
> There will be an extra (OR) instruction per IOoperation

Actually, an ADD operation. After changing the register array to offsets only (no base + ...):
```
...
io_ne2k_unicast    = 0x01  // page 1 - 6 bytes
...
.extern net_iobase
...
mov $io_ne2k_unicast,%dx  [replace this with the following]
...
mov net_iobase,%dx
add $io_ne2k_unicast,%dx
```

Mellvik, 2021-05-21:
Yes, ADD will do the same as OR in this context. I chose OR out of old habit assuming it would be faster. Turns out it isn't.

I've been testing the changes this afternoon, including the init update. Looks good.

—Mellvik

ghaerr, 2021-06-12:
Hello @Mellvik,

> it would be very useful if the Wiki mentioned (under boot options) that if the serial console configuration option is active, the console option in /bootopts has no effect.

Updated wiki as requested.

If #965 fixes your issue with the /bootopts error message not being seen, please close this issue. We can discuss netbase= or netirq= config issues with your PR when submitted.

Thank you!

Mellvik, 2021-06-13:
Thank you @ghaerr,
This is good.

And for the config issues, I agree.

- Mellvik

closed 2021-06-13 by Mellvik