ghaerr/elks #957: Potential bug in kernel hear heap manager

ghaerr, 2021-05-22:
Opening new issue for tracking from original posting in https://github.com/jbruchon/elks/pull/956#issue-650572091.

Something very strange is going on when kernel near heap gets near full. What appears to be happening when heap is specified to be small (but large enough for kernel operation) is that the last list item in the heap linked list loops back to the starting point, rather than pointing to &_heap_all. Thus, meminfo loops continuously if heap size is less than 704, say 700.

In order to test that it isn't meminfo's problem, HEAP_DEBUG can be turned on in heap.h, and then unmodified kernel heap testing code will loop continuously, which shows heap list corruption. This code will run triggered by meminfo running:
```
// Dump heap
#ifdef HEAP_DEBUG
void heap_iterate (void (* cb) (heap_s *))
{
    list_s * n = _heap_all.next;

    while (n != &_heap_all) {
        heap_s * h = structof (n, heap_s, all);
        (*cb) (h);
        n = h->all.next;
    }
}
#endif /* HEAP_DEBUG */
```
I haven't gotten too far into the heap manager yet, but there seems to be data corruption in the heap manager when manipulating the end list item (which is only seen when a small heap is configured). I suspect this error has not been seen before since ELKS has always had a large enough heap such that the last link item has always been free.

This problem is easily reproducible when compiling a standard kernel, no special options, then changing SETUP_MEMSIZE to a number less than 704. It seems ELKS needs around 624 byte heap minimum.

mfld-fr, 2021-05-24:
I am wondering if the line 122 is correct:
```
	// Try to merge with previous block if free

	list_s * p = h->all.prev;
	if (&_heap_all != p) {
		heap_s * prev = structof (p, heap_s, all);
		if (prev->tag == HEAP_TAG_FREE) {
			list_remove (&(prev->free));  <--- SUSPICIOUS LINE
			heap_merge (prev, h);
			i = _heap_free.prev;
			h = prev;
		} else {
			h->tag = HEAP_TAG_FREE;
		}
	}
```

ghaerr, 2021-05-25:
Hello @mfld-fr : In looking closely at your heap manager, I think I see why we may be having the problem described, it has to do with the list management routines, which are very straightforward.

It seems that when a list is initialized (and subsequently maintained), the "next" list item will always point back to the root. Thus, when the heap manager calls `list_insert_after`, this "circular" mechanism is maintained. However, the heap manager never checks for this wrap as end-of-list; instead, it checks the end of list as being `&_heap_free`:
```
   list_s * n = _heap_free.next;
    while (n != &_heap_free) {
        heap_s * h = structof (n, heap_s, free);
  ...
        n = h->free.next;
    }
```
I haven't yet completely looked at all the heap manager code, but I'm thinking that perhaps there comes a case when the last list item, which seemingly always points back around to the root, isn't equal to the address of _heap_free.

This would explain how `meminfo` loops continuously (before my temp hack), and would also show that the heap manager mechanism is buggy by using &_heap_free as the stopping point for all its internal searching. It may also mean that the heap manager incorrectly allocates/searches in this particular case.

What do you think?

mfld-fr, 2021-05-25:
The end loop condition is correct, but I think there is a problem in the following lines:
```
	// Try to merge with next block if free

	list_s * n = h->all.next;
	if (n->next != &_heap_all) {
```

mfld-fr, 2021-05-25:
Should rather be:
```
	// Try to merge with next block if free

	list_s * n = h->all.next;
	if (n != &_heap_all) {  <--- CHECK NEXT LIST NODE IS NOT ROOT NODE
```

ghaerr, 2021-05-25:
I see: so previous code would merge last block on list incorrectly at times.

I will test.

mfld-fr, 2021-05-25:
By the way there is the same problem in the segment allocator:
```
	// Try to merge with next segment if free

	list_s * n = seg->all.next;
	if (n->next != &_seg_all) {
```

ghaerr, 2021-05-25:
Thanks, I forgot they are extremely similar. Are we pretty sure this is the bug? I will test heap allocator, since we have a test case there with heap size < 704. Sounds like no one has really run ELKS with application space full.

What exactly is the test case for this? Bug occurs when last area of memory in heap or main memory is used, then freed, which incorrect sets last segment size. Then problem seen when last list item is used or searched again?

ghaerr, 2021-05-25:
Ok, confirmed above suggested fix clears up corruption problem. Turns out it was not easy to get this to replicate again, the smallest change and wouldn't reproduce.

Can you confirm that we agree that bug is because last item in list would be improperly checked for merge, thus settings its length incorrectly and merging with non-existent "next" item?

PR forthcoming, but untested as far as segment allocator is concerned.

mfld-fr, 2021-05-25:
See also : https://github.com/mfld-fr/elks/commit/c6e8c267c3256f91202918e0775bbb8484790021

ghaerr, 2021-05-25:
> See also : mfld-fr@c6e8c26

Thanks. Do you have test cases for the additional check for "h->tag == HEAP_TAG_FREE"? 

I don't fully understand that fix yet, and I get nervous making changes to major routines... can you explain that please?

PR #963 (2021-05-26): [kernel] Fix heap and segment allocator merge bug
Fixes #957.

Tested on heap allocator. Untested on segment allocator, but believed correct.

Also fixes compilation bug introduced by previous PR when building BIOS Console.
closed 2021-05-26 by ghaerr