ghaerr/elks #960: 8018X port (phase 2)

mfld-fr, 2021-05-24:
Continuing the porting effort after #932

ghaerr, 2021-05-24:
@mfld-fr : Have you designed a way to handle the different internal device I/O port addresses and possible C code cleanly?

I would prefer that all port addresses and possibly values written have mnemonics and the differences be kept in header files, if possible.

I don't know enough yet about the C issues to know whether separate .c files for (internal) devices may be necessary, but that's probably ok if some are quite different.

ghaerr, 2021-05-24:
As can be seen in #959, which is the first step for serial driver cleanup, architecture-specific files like serial_regs.h are being moved out of include/linuxmt into include/arch. 

I have not yet looked at this for timer and pic, but any PC-specific chip headers will be moved in future PRs as part of the source reorganization to properly handle 8018X and IBM PC separation.

In addition, arch/ports.h is entirely PC-specific, and will get renamed eventually after thinking through some of the reorganization further. It was created a while back to allow simpler reconfiguration of the port and irq addresses of network cards and serial ports, which was very complicated until the dynamic interrupt handler was added in #940. Most of ports.h can go away, but it will need to be renamed as it is entirely PC-centric and we probably don't want to use it for 8081X base addresses, etc. In general, this mechanism was used because it would have been very complicated to implement this within the config.in structure. That may not still be the case.

The developer who writes the 8018X serial port should remove 'ports.h' from serial-template.c (future serial-8018x.c) and it will use a header from the new 8018x header design.

I won't comment too much further at this time, but wanted to give a heads-up as to changes I will be working on while the 8081x second port is occurring.

ghaerr, 2021-05-24:
The beginnings of a design to handle the complexity of the multiple consoles available and their associated possible low-level differences is started first-step in the elks/arch/i86/char/Makefile in PR #959. This tries to keep config.in from being too complicated (yet) while allowing the low-level to be changed out similar to the method employed in EMU86. The serial driver uses the same mechanism, at least for now.

This is a work in progress, which will likely take a few PRs to finalize. The intent is to not make too many changes to the config.in configuration system until it is known and agreed what should be handled there vs header files, and the overall complexity of specifying a new 8018X chip variant. For instance, if all variances were handled in the .config system, then essentially all differences would be hidden in various .config files. This could be good or bad, I am not sure yet.

cocus, 2021-05-25:
So I've tried to flash this on my board's flash, and... I got some interesting results. It didn't work, but it did print some text. It usually halts when executing set_irq(); on device_init() at init.c. The only way I got out of this situation is by "constantly" sending a character to the serial port. Sadly, it doesn't register it, but that might be because the timer is not triggering IRQs.
```
!Headless console                                                             
console_init: done                                                              
in device_init                                                                  
after chr_dev_init                                                              
after blk_dev_init                                                              
!Headless console                                                               
console_init: done                                                              
in device_init                                                                  
after chr_dev_init                                                              
after blk_dev_init                                                              
after set_irq                                                                   
device_init: done                                                               
sock_init: done                                                                 
fs_init: done                                                                   
finalize_options: done                                                          
8018X machine, 128K base RAM.                                                   
ELKS kernel 0.4.0 (25728 text, 0 ftext, 3488 data, 35424 bss, 26624 heap)       
Kernel text at e062:0000, data 80:0000, top 2000:0, 62K free                    
mm_stat: done                                                                   
kernel_init: done                                                               
kfork_proc: done                                                                
wake_up_process: done                                                           
VFS: Mounted root 0x0600 (romfs filesystem).                                    
No init - running /bin/sh                                                       
# 
```
Some notes on this, the "!" at the beginning is my bootloader writing a character before jumping into ELKS.
And I've added a write statement on the timer_proc of the kbd to print a "T" every time it runs, but sadly that doesn't happen on my board (but it does on emu86).
I'll investigate this a little bit further.

ghaerr, 2021-05-25:
My take: the kernel is running through initialization steps first time with interrupts off. As soon as set_irq() first time, a hardware interrupt immediately occurs, (possibly 0:0?) and looks like reboot occurs, with "!Headless console" and same startup sequence occurring again. 2nd time through seems to work, not sure why.

Perhaps on power up, all low-memory interrupt vectors should be initialized to point to an IRET. That would tell us whether a stray interrupt is causing a reboot, perhaps. Or, you could put printk in request_irq to see what IRQs have been setup before set_irq. There should be only IRQ 0, perhaps another interrupt is firing for some reason?

ghaerr, 2021-05-25:
[image]

A quick look at the interrupt vector table shows Timer 0 is mapped to Type 8 (IRQ 0), which is the same as IBM PC, but most others are different. If Timer 1 or 2 is enabled for interrupts, (Type 18 or 19), or serial recv/transmit (Type 20, 21), you'll have problems, since ELKS doesn't set/catch those vectors unless programmed to explicitly.

cocus, 2021-05-25:
I just printed some registers, and indeed, I had the serial tx and rx interrupts unmasked since I didn't mask them, ever. I've masked them now (something that might be added to ELKS so the bootloader doesn't have to do), and I get a consistent console prompt, but NO timer0 interrupts. I've checked all the related registers values (TCUCON, IMASK, T0CON) and nothing weird pops out.
At this stage I'm not sure if I've crashed it, or if it's actually the lack of interrupts.

ghaerr, 2021-05-25:
```
/* TMR and SER interrupts are enabled by default */
unsigned int cache_08 = 0xfa;
```
Yes, comment above shows SER interrupts enabled, should be disabled. Also, reading the manual says that all unused mask bits should be set to 0, not 1. That's not likely the problem, but above, you're only setting 8 of 15 bits, for instance - upper 8 bits to 0, then others 1. That routine also should be rewritten to read the mask, AND out the bit, then write it. That is, lose the cache.

> At this stage I'm not sure if I've crashed it, or if it's actually the lack of interrupts.

Since you don't get the 'T' ever, and prompt is shown, I'd say you've not crashed, but the timer 0 setup may not be correct, as well as possibly the enable_irq routine.

ghaerr, 2021-05-25:
> I've masked them now (something that might be added to ELKS so the bootloader doesn't have to do),

Bootloader should probably initialize interrupt mask register to all disabled. ELKS will work properly if you rewrite routine to read/AND/write interrupt mask register as described above.

ghaerr, 2021-05-25:
```
/* 
 *      Low level interrupt handling for the X86 8018X
 *      platforms 
 */     

void init_irq(void)
{
}
```
The above routine in irq-8018x.c can be used to initialize the interrupt control unit, masks, etc as discussed in section 8 of the manual.

ghaerr, 2021-05-25:
@cocus : I suggest you move your interrupt initialization code into init_irq. I haven't looked at exactly the settings after RESET, but otherwise we can have a look at it to help.

I also noticed some other issues that will be important when interrupt-driven serial I/O is implemented: it seems the 80188EB uses two interrupt vectors (recv interrupt 20h and transmit interrupt 21h), but only one interrupt mask bit (IRQ 2). The recently-added dynamic interrupt handler only allows a one-to-one mapping for mask bits to interrupt vectors. That won't be a problem if the transmit interrupt isn't used (it isn't used in the serial template handler). However, `irq_vector` will have to be updated, as the mapping of interrupt request numbers is not linear to the vectors on the 80188 as it is in the 8086.

mfld-fr, 2021-05-25:
Just a question, IF = 0 after INT 19h. Where is the first STI ?

cocus, 2021-05-25:
> Just a question, IF = 0 after INT 19h. Where is the first STI ?

I think it's the one at `init.c:device_init()` with the `set_irq()` after `blk_dev_init()`

ghaerr, 2021-05-25:
> I think it's the one at init.c:device_init() with the set_irq() after blk_dev_init()

Yes, this is why request_irq and other routines needed the save/restore_flags: the majority of the kernel init takes place with interrupts disabled, just until the partition management, where the kernel basically operates normally.

cocus, 2021-05-25:
I've pulled the latest changes from this repo, and now I can't boot on emu86:
```
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh

panic: No init or sh found
apparent call stack:
Line: Addr    Parameters
~~~~: ~~~~    ~~~~~~~~~~
   0: 009C => 642F 7665 632F 6E6F 6F73 656C 5500
   1: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
   2: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
   3: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
   4: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
   5: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
   6: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
   7: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
   8: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
SYSTEM HALTED - Press CTRL-ALT-DEL to reboot:
```
This didn't happen before I pulled the changes. Do I have to change the address of the romfs or something?

ghaerr, 2021-05-25:
Are you running the 8018x.config? There were some small changes made there. The error above is that neither /bin/init, /bin/sh or /bin/ash are found on ROMFS.

cocus, 2021-05-26:
> Are you running the 8018x.config? There were some small changes made there. The error above is that neither /bin/init, /bin/sh or /bin/ash are found on ROMFS.
> 
> Can you show the entire startup messages?

I didn't copy the defaultconfig into my .config. I've done so, and now it boots again! I can continue investigating what's going on with these interrupts. I've already tried to write into the TCUCON (timer interrupt register) instead of the IMASK (something that should happen for the R8810), and still no dice. I've followed some other projects that do use the interrupts on these 8018x systems and I think I'm doing everything correctly, but seems like something's acting up.

ghaerr, 2021-05-26:
I suggest you put (almost) all your startup code in the irq_init routine, so we can look at it. I don't know the 8018X well (yet), but I'll take a harder look. Obviously there is something still amiss with regards to getting the timers setup and interrupts ticking.

ghaerr, 2021-05-26:
Another idea would be to hand-craft your startup code to also setup the timer2 in the same way as ELKS, and then also setup timer 0 and its vector 8. Then print 'T' from startup code on clock tick using serial port and don't jump to ELKS. This keeps the problem constrained to only setting up chip for IRQ 0 and timer 0/2, along with serial putc routine.

cocus, 2021-05-26:
> I suggest you put (almost) all your startup code in the irq_init routine, so we can look at it. I don't know the 8018X well (yet), but I'll take a harder look. Obviously there is something still amiss with regards to getting the timers setup and interrupts ticking.

For now it looks like this:
```

struct {
    int irq;
    int pcb_reg;
    int default_prio; /* don't set bit 4, that's the Mask bit! */
    int vector;
} const interrupt_control_registers[] = {
    { 0,    0xff00 + 0x12,  7,  0x08 },  /* irq0 = timer (Timer Control) */
};

/*
 *	Low level interrupt handling for the X86 8018X
 *	platforms
 */

void init_irq(void)
{
    /* Mask all interrupt sources */
    outw(0xfd, 0xff00 + 0x08); /* IMASK */
}

void enable_irq(unsigned int irq)
{
    if (irq == 0) {
        /* bit MSK clear */
        outw(interrupt_control_registers[irq].default_prio, interrupt_control_registers[irq].pcb_reg);
    }
}
```

> Another idea would be to hand-craft your startup code to also setup the timer2 in the same way as ELKS, and then also setup timer 0 and its vector 8. Then print 'T' from startup code on clock tick [...]

This would be a good idea, I will try to do something like this

cocus, 2021-05-26:
> This would be a good idea, I will try to do something like this

I ended up doing this, and I can confirm that I don't see any "t"s on the real hardware, but I do see them on emu86!

```
.code16

#define VECT_TIMER0 0x08

// This should land on 0xF000:0xFFF0
.section .reset.text, "ax"
.globl _cpu_reset_entrypoint
_cpu_reset_entrypoint:
    ljmp $0xf000, $_init8018x_and_start

// This should land on 0xF000:0xFF00
.section .text, "ax"
.globl _init8018x_and_start
_init8018x_and_start:
    xor %ax, %ax
    mov %ax, %ds
    mov %ax, %es
    mov %ax, %ss        // DS = ES = SS = 0x0000
    
    mov $0x600, %ax
    mov %ax, %sp        // Set a stack pointer starting at 0x0000:0x0600 (elks doesn't use this addresses)

    // Setup UCS and LCS

    mov $0xffa4, %dx
    mov $0x8000, %ax
    out %ax, %dx

    mov $0xffa6, %dx    // FINISH SETTING UP UCS#
    mov $0x000e, %ax    // set ISTOP for 0FFFFF ending loaction
    out %al, %dx        // REMEMBER, BYTE WRITE WORK OK

    mov $0xffa0, %dx    // SET UP LCS START REGISTER
    mov $0x0000, %ax    // LCS STARTS AT 0H, ZERO WAIT STATES!
    out %al, %dx        // REMEMBER, BYTE WRITES WORK OK

    mov $0xffa2, %dx    // SET UP LCS STOP REGISTER
    mov $0x200A, %ax    // 128kB Installed
    out %al, %dx        // REMEMBER, BYTE WRITES WORK OK
    // Now full FLASH and RAM should be properly mapped

    // Setup internal serial port

    mov $0xff60, %dx    // B0CMP
    mov $0x8033, %ax    // 38400 @ 16MHz
    out %ax, %dx

    mov $0xff64, %dx    // S0CON
    mov $0x0021, %ax    // mode 1 asynchronous 10-bit
    out %ax, %dx

    xor %ax, %ax
    mov %ax, %es
    cli

    // Set Timer2 Maxcount register
    //outw(TIMER2_INTERVAL, 0xff00 + 0x42); // T2CMPA
    mov $0xff42, %dx
    mov $0x0fa0, %ax
    out %ax, %dx

    // Clear the Timer count register before starting it
    //outw(0x0000, 0xff00 + 0x40); // T2CNT
    mov $0xff40, %dx
    mov $0x0000, %ax
    out %ax, %dx

    // Set Timer0 Maxcount register
    //outw(TIMER0_INTERVAL, 0xff00 + 0x32); // T0CMPA
    mov $0xff32, %dx
    mov $0x000a, %ax
    out %ax, %dx

    // Clear the Timer count register before starting it
    //outw(0x0000, 0xff00 + 0x30); // T0CNT
    mov $0xff30, %dx
    mov $0x0000, %ax
    out %ax, %dx

    // Enable Timer2, release inhibit to change EN bit, continuous mode
    //outw(0xc001, 0xff00 + 0x46); // T2CON
    mov $0xff46, %dx
    mov $0xc001, %ax
    out %ax, %dx

    // Enable Timer0, release inhibit to change EN bit, INT enabled, use Timer2 as source, continuous mode
    //outw(0xe009, 0xff00 + 0x36); // T0CON
    mov $0xff36, %dx
    mov $0xe009, %ax
    out %ax, %dx

    // Timer vector

    mov     $4*VECT_TIMER0, %bx
    movw    $int_timer0, %es:(%bx)
    mov     %cs, %es:2(%bx)

    mov $0x00fc, %ax // unmask tmr int
    mov $0xff08, %dx
    out %ax, %dx

    sti

    mov $0xff6a, %dx
    mov $'!', %ax
    out %al, %dx

no_more_text:
    jmp no_more_text

int_timer0:
    // 
    push %ax
    push %dx

    mov $0x8000, %ax // set the NSPEC bit on the
    mov $0xff02, %dx // EOI register so the ICU
    out %ax, %dx // acks the highest priority interrupt

    mov $0xff6a, %dx
    mov $'t', %ax
    out %al, %dx

    mov $0xff66, %dx    // S0STS
serial_wait_isr:
    in %dx, %al
    test 8, %al         // TXE (transmitter empty)
    jz serial_wait_isr  // and if it hasn't finished, keep looping

    pop %dx
    pop %ax
    iret
```
As you can see, the code resembles the init stuff we already have on irq, timer and the serial console.

I might be missing something fundamental, but none of the code I've checked are doing something different. I'm out of ideas

ghaerr, 2021-05-26:
> I can confirm that I don't see any "t"s on the real hardware, but I do see them on emu86!

Well remember that EMU86 is emulating only a very small portion of all the 8018X interrupt control registers, timers, etc. And, EMU86 is only emulating what you/we thought worked, not what actually works! So, I am not surprised. I will take a closer look at your startup code tomorrow, thanks for posting it.

cocus, 2021-05-26:
> Well remember that EMU86 is emulating only a very small portion of all the 8018X interrupt control registers, timers, etc. And, EMU86 is only emulating what you/we thought worked, not what actually [...]

Yep, I understand. Thanks!

mfld-fr, 2021-05-26:
@cocus : good idea to write a kind of unit test juster after CPU reset. After reading it, I have the following suggestions:
1. Reset all the timers **before** programming any of them : write 0x4000 to T0CON, T1CON & T2CON
2. Check T2 is really counting by dumping T2CNT to the serial port after a short loop
3. Change max count to 1000 in T0 (for one second wrap)
3. Check the same for T0, but after a long loop (at least 1 ms ~ 1000 instructions)

After checking the counting is OK:
1. Unmask and set priority to 0 for T0: write 0x0000 to TCUCON
2. Unmask T0 again: write 0x00FC to IMASK

More generally, write 0 in reserved bits.

mfld-fr, 2021-05-26:
See also SYS86 timer initialization : https://github.com/mfld-fr/sys86/blob/master/timer.c

cocus, 2021-09-04:
Hi guys! Glad to talk to you again. I had some rough times lately so I didn't have enough time to dive into this.
I ended up following @mfld-fr's suggestions, and I can validate that T2CNT is actually counting up. But T0CNT is not. I see that T2CNT goes up but never above T2CMPA (which in my example code is 0x0fa0). My latest code (albeit a little bit messy) prints all the PCB registers when a character is received on the serial port. I've taken most of that code from Wichite Sirichote's SBC188 proejct (http://web.archive.org/web/20130823172544/http://www.kmitl.ac.th/~kswichit/C188/boot.asm).

What I can tell from it:
1. Timer 2 works fine. Bit MC on T2CON is set (from the datasheet: `This bit is set when the counter reaches a maximum count. The MC bit must be cleared by writing to the Timer Control register. This is not done automatically. If MC is clear, the counter has not reached a maximum count.`) and T2CNT goes up and restarts before hitting the T2CMPA value.
2. Timer 0's T0CON is written properly with the data 0xe009 and reads back as 0xa009 (bit INH is always read as zero, it's used to not change the T0CON values inadvertantly), but T0CNT never goes up. Bit MC on T0CON is not set. I've double checked that 0xe009 should enable the timer, set it as continuous mode, use timer 2's overflow events as the increment source for this timer, and interrupt source enabled.

As you can see, this is the dump of the PCB registers before even initializing the timers:

```
FF00  0000 0000 0000 0000 00FD 0007 0000 0004
FF10  0010 000F 000F 000F 000F 000F 000F 000F
FF20  FF20 FF22 FF24 FF26 FF28 FF2A FF2C FF2E
FF30  0000 FFFF FFFF 2021 0000 FFFF FFFF 2020
FF40  0000 FFFF FFFF 0000 FF48 FF4A FF4C FF4E
FF50  FFFF FFFF FFFF FFFF FFFF FFC3 FFFF FFFF
FF60  8033 8019 0021 0003 00FF 0020 0000 0000
FF70  0000 8000 0000 000B 00FF 0000 0000 0000
FF80  FFCF FFC3 FFCF FFC3 FFCF FFC3 FFCF FFC3
FF90  FFCF FFC3 FFCF FFC3 200F 2048 FFCF FFC3
FFA0  0000 200A 8000 000E 00FF 00CF 00CF 00CF
FFB0  0000 0000 0000 00FE FF00 FFBA FF02 FFBE
FFC0  FFC0 FFC2 FFC4 FFC6 FFC8 FFCA FFCC FFCE
FFD0  FFD0 FFD2 FFD4 FFD6 FFD8 FFDA FFDC FFDE
FFE0  FFE0 FFE2 FFE4 FFE6 FFE8 FFEA FFEC FFEE
FFF0  FFF0 FFF2 FFF4 FFF6 FFF8 FFFA FFFC FFFE
```

After a short while:
```
FF00  0000 0000 0000 0000 00FC 0007 0000 0004
FF10  0018 0000 000F 000F 000F 000F 000F 000F
FF20  FF20 FF22 FF24 FF26 FF28 FF2A FF2C FF2E
FF30  0000 100A FFFF A009 0000 FFFF FFFF 2020
FF40  0BBC 0FA0 FFFF 8021 FF48 FF4A FF4C FF4E
FF50  FFFF FFFF FFFF FFFF FFFF FFC3 FFFF FFFF
FF60  8033 801F 0021 0003 000D 0020 0000 0000
FF70  0000 8000 0000 000B 00FF 0000 0000 0000
FF80  FFCF FFC3 FFCF FFC3 FFCF FFC3 FFCF FFC3
FF90  FFCF FFC3 FFCF FFC3 200F 2048 FFCF FFC3
FFA0  0000 200A 8000 000E 00FF 00CF 00CF 00CF
FFB0  0000 0000 0000 00FE FF00 FFBA FF02 FFBE
FFC0  FFC0 FFC2 FFC4 FFC6 FFC8 FFCA FFCC FFCE
FFD0  FFD0 FFD2 FFD4 FFD6 FFD8 FFDA FFDC FFDE
FFE0  FFE0 FFE2 FFE4 FFE6 FFE8 FFEA FFEC FFEE
FFF0  FFF0 FFF2 FFF4 FFF6 FFF8 FFFA FFFC FFFE```

Source:
```
.code16

#define VECT_TIMER0 0x08

.equ IMASK, 0xff08
.equ TCUCON, 0xff12

.equ B0CMP, 0xff60
.equ B0CNT, 0xff62
.equ S0CON, 0xff64
.equ S0STS, 0xff66
.equ R0BUF, 0xff68
.equ T0BUF, 0xff6A

.equ LCSST, 0xffA0
.equ LCSSP, 0xffA2
.equ UCSST, 0xffA4
.equ UCSSP, 0xffA6

.equ T0CNT, 0xff30
.equ T0CMPA, 0xff32
[... 291 lines cut ...]
    mov $'t', %ax
    out %ax, %dx

    // check if the serial port has finished sending a byte
    mov $S0STS, %dx    // S0STS
serial_wait_isr:
    in %dx, %ax
    test $8, %ax         // TXE (transmitter empty)
    jz serial_wait_isr      // and if it hasn't finished, keep looping

    mov $0x8000, %ax // set the NSPEC bit on the
    mov $0xff02, %dx // EOI register so the ICU
    out %ax, %dx // acks the highest priority interrupt

    pop %dx
    pop %ax
    iret

message:
    .ascii "Welcome! Built at " BUILD_TIMESTAMP
    .asciz "\r\n"```

Do you have any thoughts on why T0 is not incrementing while T2 does?

Thanks and sorry for the delay!

cocus, 2021-09-04:
As another update to this, I've enabled T1 with the exact same config EXCEPT for the INTs, and it does count up!
Have a look at offsets 0xff38 (T1CNT) and 0xff3e (T1CON, bit MC got set after some seconds):
```
FF00  0000 0000 0000 0000 00FC 0007 0000 0004
FF10  0018 0000 000F 000F 000F 000F 000F 000F
FF20  FF20 FF22 FF24 FF26 FF28 FF2A FF2C FF2E
FF30  0000 100A FFFF A009 002E 100A FFFF 8029
FF40  03D5 0FA0 FFFF 8021 FF48 FF4A FF4C FF4E
FF50  FFFF FFFF FFFF FFFF FFFF FFC3 FFFF FFFF
FF60  8033 801C 0021 0003 000D 0020 0000 0000
FF70  0000 8000 0000 000B 00FF 0000 0000 0000
FF80  FFCF FFC3 FFCF FFC3 FFCF FFC3 FFCF FFC3
FF90  FFCF FFC3 FFCF FFC3 200F 2048 FFCF FFC3
FFA0  0000 200A 8000 000E 00FF 00CF 00CF 00CF
FFB0  0000 0000 0000 00FE FF00 FFBA FF02 FFBE
FFC0  FFC0 FFC2 FFC4 FFC6 FFC8 FFCA FFCC FFCE
FFD0  FFD0 FFD2 FFD4 FFD6 FFD8 FFDA FFDC FFDE
FFE0  FFE0 FFE2 FFE4 FFE6 FFE8 FFEA FFEC FFEE
FFF0  FFF0 FFF2 FFF4 FFF6 FFF8 FFFA FFFC FFFE
```

ghaerr, 2021-09-04:
Hello @cocus,

Good to hear from you again! Thanks for your post. I've just returned from vacation and have some catching up to do, and can then look further into this with you. To refresh my memory, we need a countdown timer so that it can trigger an interrupt, which will be used for the ELKS timer interrupt. We were trying to use T0, which isn't yet working, but you've now got T1 working, correct? Can that be attached to an interrupt to use T1 (for the time being) instead of T0, which would then allow the ELKS keyboard polling to work and thus your initial port operational?

Thank you!

cocus, 2021-09-04:
> Can that be attached to an interrupt to use T1 (for the time being) instead of T0, which would then allow the ELKS keyboard polling to work and thus your initial port operational?

Hi @ghaerr, yes indeed, I've got T1 not only to count using T2 as its prescaler, but it also produces the appropriate interrupts.
I had to tweak my changes on the irq file for the 8018x so I can take the constant of TIMER_INTERRUPT as the T1. In theory by returning the appropriate vector type for TIMER_INTERRUPT irq on the get_irq_vector I should be set, right? It was returning a wrong value before, but now it should return 18 (T1 vector) for TIMER_INTERRUPT. I'll try to test this out tomorrow and I'll let you know.
Thanks!

cocus, 2021-09-07:
@ghaerr (or even @mfld-fr ) do you mind reviewing these changes? I didn't manage to commit them to a branch, but I have a diff. 
```
diff --git elks/arch/i86/drivers/block/init.c elks/arch/i86/drivers/block/init.c
index ee6e4842..716ba399 100644
--- elks/arch/i86/drivers/block/init.c
+++ elks/arch/i86/drivers/block/init.c
@@ -26,6 +26,7 @@
 #include <linuxmt/string.h>
 #include <linuxmt/memory.h>
 
+#include <arch/io.h>
 #include <arch/system.h>
 #include <arch/segment.h>
 
@@ -34,14 +35,24 @@ int boot_rootdev;	/* set by /bootopts options if configured*/
 void INITPROC device_init(void)
 {
     register struct gendisk *p;
-
+    word_t read;
+    printk("in device_init\n");
     chr_dev_init();
+    printk("after chr_dev_init\n");
     blk_dev_init();
+    printk("after blk_dev_init\n");
 
+    /*read = inw(0xff00 + 0x08); printk("IMASK: %x\n", read);
+    read = inw(0xff00 + 0x0e); printk("REQST: %x\n", read);
+    read = inw(0xff00 + 0x10); printk("INTSTS: %x\n", read);
+    read = inw(0xff00 + 0x12); printk("TCUCON: %x\n", read);*/
     set_irq();
+    printk("after set_irq\n");
 
-    for (p = gendisk_head; p; p = p->next)
-	setup_dev(p);
+    for (p = gendisk_head; p; p = p->next) {
+	    setup_dev(p);
+        printk("after setup_dev\n");
+    }
 
 #ifdef CONFIG_BLK_DEV_BIOS
     /*
diff --git elks/arch/i86/drivers/char/kbd-poll.c elks/arch/i86/drivers/char/kbd-poll.c
index f41feffe..173fd9ab 100644
--- elks/arch/i86/drivers/char/kbd-poll.c
+++ elks/arch/i86/drivers/char/kbd-poll.c
@@ -66,7 +66,8 @@ static void kbd_timer(int data)
 	    }
 	}
     }
-    restart_timer();
+	Console_conout(NULL, 'T');
+	restart_timer();
 }
 
 static void restart_timer(void)
diff --git elks/arch/i86/kernel/irq-8018x.c elks/arch/i86/kernel/irq-8018x.c
index 94b2f1da..bc0e863b 100644
--- elks/arch/i86/kernel/irq-8018x.c
+++ elks/arch/i86/kernel/irq-8018x.c
@@ -11,11 +11,18 @@
 #include <linuxmt/sched.h>
 #include <linuxmt/types.h>
 
+#include <arch/ports.h>
 #include <arch/io.h>
 #include <arch/irq.h>
 
-/* TMR and SER interrupts are enabled by default */
-unsigned int cache_08 = 0xfa;
[... 126 lines cut ...]
     sock_init();
 #endif
+	printk("sock_init: done\n");
 
     fs_init();
+	printk("fs_init: done\n");
 
 #ifdef CONFIG_BOOTOPTS
     if (opts)
 	finalize_options();
     else printk("/bootopts IGNORED: header not ## or size > 255\n");
 #endif
+	printk("finalize_options: done\n");
 
     mm_stat(base, end);
+	printk("mm_stat: done\n");
 }
 
 /* this procedure runs in user mode as task 1*/

```

Please note that my main changes are on irq-8018x.c and timer-8018x.c. The ones I'm not a 100% sure about are the ones on the irq file.

Thanks again!

ghaerr, 2021-09-07:
Hi @cocus,

You've got a lot of printk's in there! I will comment separately on the individual file changes, but I have a question: does this work?

Thank you!

cocus, 2021-09-07:
> You've got a lot of printk's in there! I will comment separately on the individual file changes, but I have a question: does this work?

Those printk are for debug purposes :)
No, I couldn't load this for some reason. My elks-loader assembly code is flawed, and I think I screwed it up somehow and now it doesn't work. So I couldn't test the elks changes yet. I didn't even test it on the emulator since I've changed the timer and a lot of changes should be applied (which I don't feel to do as of now).

ghaerr, 2021-09-07:
> Those printk are for debug purposes :)

Using kernel printk is potentially tricky until the (serial) console is initialized by elks/init/main.c::console_init(), but there is special-handling code that should redirect printk to the headless elks/arch/i86/drivers/char/console-headless.c::Console_conout() function. That said, your BIOS/hard boot code initializes the serial port outside of the kernel, you should be OK.

With regards to debug output, I recommend that most of it go into the main init routine elks/init/main.c, rather than in  the block device driver areas like you have it now.

> My elks-loader assembly code is flawed, and I think I screwed it up somehow and now it doesn't work. So I couldn't test the elks changes yet.

I heavily recommend that you start with the current version of ELKS (without your most recent changes), and build that version and debug loading it, before trying to test this interrupt timer code. That is because ELKS should boot and run just fine without the kernel timer, until it comes time to poll the headless (serial) keyboard. So, the boot should work, and everything run up until the shell displays '#', after which keyboard input won't work.

After getting that working, then add this diff.

I'll take a quick look at the non-debug changes in your diff, and comment separately.

ghaerr, 2021-09-07:
Hi @cocus,

Ok, here's what I think: this port is getting kind of messy. I suggest a cleanup, which I think you should do, rather than I, since that will help you understand the code a bit better, and also allow much easier changes between different boards. The cleanup will move all 8081X port and register addresses to a new file: elks/include/arch/8018x.h. This file will use the capitalized register names of the various processor registers used in the timer-8018x.c and irq-8018x.c files, and also \#include the header 8018x.h file.

For starters, create the following file 8018x.h:
```
/* timer registers */
#define T0CMPA (0xff00 + 0x32)
#define T1CMPA (0xff00 + 0x38)
#define T0CNT (0xff00 + 0x30)
#define T1CNT (0xff00 + 0x3a)
... (list all the above defines in sorted order by the last hex 0x30 etc values)

/* interrupt registers */
... etc

/* THIS port's register mappings */
/* this section maps the above register values to constants used in the irq-8018x.c and timer-8018x.c files*/
#define TCMPA   T1CMPA
#define TCNT     T1CNT
... etc
```
Then replace all the hard-coded constants in the timer and irq files with the named mappings from the last lines of this file.

Get rid of the following struct, and just use direct defines from the above header file instead, there is no need for this complexity at this time, since only one timer and interrupt is used by the kernel. We can extend this later after we get the initial port working and an interrupt driven serial port is added.
```
+struct {
+    int irq;
+    int pcb_reg;
+    int default_prio; /* don't set bit 4, that's the Mask bit! */
+    int vector;
+} const interrupt_control_registers[] = {
+    { TIMER_IRQ,    0xff00 + 0x12,  7,  18 },
```
In general, it looks like the only changes you're making are moving from T0 to T1, etc, and this new method will allow handling all of that in a new 8018x.h file instead of hacking irq\*.c and timer\*.c for each port.

Are you following what I am suggesting? If so, lets have you prepare a first PR which just replaces hard coded values for any 8018X port or register in the timer and interrupt files with a define from the 8018x.h file. I think this will make things much easier to understand. We can then enhance the 8081x.h file for multiple processor ports after we get your initial port running. I can commit your PR upon inspection, while we continue to work on getting everything working. This way things will stay quite simple until they need to get complicated, greatly reducing the number of moving parts.

Thank you!

cocus, 2021-09-07:
@ghaerr That makes sense, I was trying to add that struct for later int sources. There's one thing I need to keep, which is the irq_vector() to return 18 if the int is TIMER_INTERRUPT, and 0 otherwise (for now).
I think the other clean up you've mentioned is fine and would make this port a little bit more easy to understand. However, I don't think that it's okay to just change the PCB register address to select if you're using the T0 or T1; that's something the timer-8018x.c should do (if needed). There are some subtle differences between T0 and T1, and of course, this would screw up the IRQs if it's not properly modified.

I'll try to get this booting first, cleaning it up and then checking if these changes do work.

ghaerr, 2021-09-07:
> That makes sense, I was trying to add that struct for later int sources. There's one thing I need to keep, which is the irq_vector() to return 18 if the int is TIMER_INTERRUPT, and 0 otherwise (for now).

That's fine - don't include ports.h and include the new 8018x.h instead, and use TIMER_IRQ instead, set to 18. TIMER_IRQ is defined in ports.h, there's no reason to name every architecture differently when talking about things like what IRQ the timer is using. In this way, the name stays the same, and the include file changes. Eventually, the PC-centric ports.h will be changed or renamed accordingly, rather than trying to combine ports.h and 8018x.h.

> I think the other clean up you've mentioned is fine and would make this port a little bit more easy to understand.

Great, lets do it.

> However, I don't think that it's okay to just change the PCB register address to select if you're using the T0 or T1; that's something the timer-8018x.c should do (if needed). There are some subtle [...]

Thats fine - this is just a suggestion so that things are easier to understand. Try to use defines whenever possible, so that hard addresses aren't coded into the driver routines. I just spent months cleaning all this up from the previous PC-only port, which also used hard addresses all over the place, making porting to new systems harder.

I understand your point about T0 and T1 being different - but it would be nice to use the same names if possible in the drivers when possible, rather than using special numbered names. For instance, in the PC architecture code, it doesn't show which port or interrupt number or counter number is in use, just a general defined name that allows one to change it in ports.h without having to change the driver .c code. Frankly, I would rather have it written in this more simpler way initially, then add complexity later for T1 vs T0 etc once the initial port is working.

> 
> I'll try to get this booting first, cleaning it up and then checking if these changes do work.

Thank you!

ghaerr, 2021-09-17:
Hello @cocus,

I didn't mean to scare you away with my requests for cleaning up the source. I've reviewed your posted changes from 10 days ago and, aside from the many printk debugging statements, it is pretty straightforward and should work, as most of the real changes are just addressing changes for the T0 vs T1 timer. I can make those cleanup changes if that would help, although we might be best off to wait until you have time to test booting an ELKS image on your hardware again with the existing source.

Thank you!

cocus, 2021-09-17:
Hi @ghaerr , no you didn't scare me away, it's just that I can't find that much time to do this, so I was waiting until next time to give you an update. All changes or PoCs that you might have would be really appreciated in the meantime!

Thanks!

cocus, 2021-09-26:
Hi all! With my changes on that branch I've mentioned, I can write with my keyboard via the serial console! I also have a lot of "T"s printed on the serial console because I left that as a debug excersise. So whatever I have added there, it works. Now we can proceed with cleaning it up. Except if you @ghaerr would fancy to do/run some other tests.

Thanks!

ghaerr, 2021-09-26:
Hello @cocus,

Very nice! How's it feel to be the first one to port ELKS to a new processor? :)

> So whatever I have added there, it works. Now we can proceed with cleaning it up. Except if you @ghaerr would fancy to do/run some other tests.

How about let's start with getting your updated drivers submitted as a PR against the current tree, so that we'll have a base version that works. I suggest submitting a PR with most all the debug statements from the rest of the kernel removed, but with your driver files mostly as-is (except for 'T' debugs removed). We can then have a look at it and discuss what cleanups might be done, or whether to commit it first, and cleanup later.

Thanks for working on this!

cocus, 2021-09-26:
@ghaerr It was a really interesting and enjoyable experience, I wasn't that well versed in 8086 code implementation before, and now I think I understand it pretty good. I think it is also good to have found such a great project to try to run on this piece of hardware that probably no one else has.

Regarding the changes and the PR, indeed. I will do so.
I still think we need to tweak the IRQ number define for this particular platform so we don't have to do a LUT afterwards to know which one you're using.

cocus, 2021-09-26:
While PR #971 is on its way, I would like to ask how to include my own app into the rootfs, and how to just rebuild it (without having to resort to use `build.sh`, which takes ages)? I've followed a declined PR for a test app for the /dev/port driver, but I hit some issues when trying to invoke the "install" action of its makefile, so I just made it copy it to another directory and used that in the Applications file. My app gets bundled and works fine when I run `build.sh`.
The idea behind this its just to test the SD card interface from a userland app (since we don't have any restrictions to the IO) so I have a boilerplate working on ELKS of my SD interface driver that should be used later with your SSD block driver.

ghaerr, 2021-09-26:
> I would like to ask how to include my own app into the rootfs

A simple way would be to just add your app to elkscmd/file_utils, and modify file_utils/Makefile to add the name of the app to PRGS= and a few lines to build it, also in that Makefile. 

Then you would also add a line in elkscmd/Applications (look at mkdir for example) and it should automatically build and be installed into the /bin directory of the target image. For the special ROMFS build, the .config you're using only specifies using CONFIG_APP_FILE_UTILS=y and CONFIG_SH_UTILS=y, so your app needs to be in either of those directories, as the ROMFS size is quite limited.

cocus, 2021-09-27:
Ahh, bugger I ran out of memory trying to run my app:
```
8018x ELKS loader built at Sun 26 Sep 2021 02:44:32 PM PDT
Now jumping into ELKS...
Headless console
8018X machine, 128K base RAM.
ELKS kernel 0.4.0 (25728 text, 0 ftext, 3360 data, 33792 bss, 1024 heap)
Kernel text at e062:0000, data 80:0000, top 2000:0, 88K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# sd
sd: Out of memory
# meminfo
  HEAP   TYPE  SIZE    SEG   TYPE    SIZE  CNT
  912e   SEG     16    9d3   CSEG   31040    1
  914a   INT      6
  915c   INT      6
  916e   TTY     80
  91ca   TTY     80
  9226   SEG     16   1167   DSEG   20992    1
  9242   SEG     16   1687   free   20992    0
  925e   SEG     16   1ba7   CSEG    4560    1
  927a   SEG     16   1cc4   DSEG    4432    1
  9296   SEG     16   1dd9   free    8816    0
  92b2   free   624
  Heap/free    1024/  624 Total mem   90832
  Memory usage   89KB total,   60KB used,   29KB free
#
```

Whenever I have a minute, I'll try to upgrade the RAM to 512k.

ghaerr, 2021-09-27:
> Ahh, bugger I ran out of memory trying to run my app:

Yeah, you're tight on RAM. Now you need to use linker options to set the default app stack and heap size (default at 4K each) down to very little. If your program uses no heap, and very small variables on the stack, the following should work (taken from the `ps` Makefile in sys_utils):
```
meminfo: meminfo.o
    $(LD) $(LDFLAGS) -maout-heap=1 -maout-stack=512 -o meminfo meminfo.o $(LDLIBS)
```

cocus, 2021-09-27:
I ended up removing the Elm Chan PetitFS code and leaving my SD bitbang implementation. It turns out it still works, it can read sectors fine. For some weird reason, passing uint32_t as arguments to that function doesn't work properly (or is it because i'm passing a uint8_t *? not sure). Either way, I hardcoded these values so I don't have to pass them, and I've managed to read an entire sector out of my SD card on my test app. I didn't have to tweak the stack, but maybe that helps? Not sure. In any case, I'm done for today. Thanks!

cocus, 2021-09-29:
Good news! My solder job didn't break the board, and now I have 512kB of RAM! I've updated the config.h and my bootloader to enable it and looks like "it works":
```
8018x ELKS loader built at Tue 28 Sep 2021 07:34:20 PM PDT
Now jumping into ELKS...
Headless console
8018X machine, 512K base RAM.
ELKS kernel 0.4.0 (25728 text, 0 ftext, 3360 data, 33792 bss, 1024 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 472K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# meminfo
  HEAP   TYPE  SIZE    SEG   TYPE    SIZE  CNT
  912e   SEG     16    9d3   CSEG   31040    1
  914a   INT      6
  915c   INT      6
  916e   TTY     80
  91ca   TTY     80
  9226   SEG     16   1167   DSEG   20992    1
  9242   SEG     16   1687   free   20992    0
  925e   SEG     16   1ba7   CSEG    4560    1
  927a   SEG     16   1cc4   DSEG    4432    1
  9296   SEG     16   1dd9   free  402032    0
  92b2   free   624
  Heap/free    1024/  624 Total mem  484048
  Memory usage  473KB total,   60KB used,  413KB free
#
```
Much better huh?

cocus, 2021-09-29:
Now, I'm having some issues with my test program. My "block read" function looks like this:
```
DRESULT disk_readp (
	uint8_t *buff,		/* Pointer to the read buffer (NULL:Read BYTEs are forwarded to the stream) */
	uint32_t sector,	/* Sector number (LBA) */
	uint16_t offset,	/* uint8_t offset to read from (0..511) */
	uint16_t count		/* Number of BYTEs to read (ofs + cnt mus be <= 512) */
)
{
	DRESULT res;
	uint8_t d;
	uint16_t bc, tmr;

	if (sd_is_sdv2) sector *= 512;	/* Convert to uint8_t address if needed */

	printf("dr: 0, cnt=%d, buff=%p\n", count, buff);
/// etc
```
I'm calling it like this: `if (disk_readp(NULL, 0, 0, 512))`, but I get the following:
`dr: 0, cnt=249, buff=000000`
Of course, if I pass different values, this changes, but the values this function's receiving are wrong. What could be causing this? I have to tell that the `disk_readp` is on sd_elm.c and the main function is on sd.c, and this is what the Makefile looks like (I've added it to the file_utils makefile):
```
sd: sd_elm.o sd.o sd_asm.o
	$(LD) $(LDFLAGS) -o sd sd_elm.o sd.o sd_asm.o $(LDLIBS)
```
(Note: sd_asm.S contains some highly optimized assembly IO routines to boost the throughtput of the bit-banged SPI interface, it exports 3 functions that are used on disk_readp but AFTER what I've shown here).

Do you have any advice? I want to have this function working properly when passing a buffer, so I can add it to your ssd driver later (for which I'll open another ticket).

Thanks!

ghaerr, 2021-09-29:
> I'm calling it like this: if (disk_readp(NULL, 0, 0, 512)), but when I run it, I get the following:
dr: 0, buff=000000, sector=0, offs=200, count=130

I would say it seems that the `uint32_t` (long) 'sector' parameter is being passed as a `uint16_t` (short).
This could happen if the disk_readp function prototype is not declared properly or seen by the compiler prior to its call. Suggest perhaps forcing the long parameter by calling using `if (disk_readp(NULL, 0L, 0, 512)) ...`

cocus, 2021-09-29:
> I would say it seems that the `uint32_t` (long) 'sector' parameter is being passed as a `uint16_t` (short).
> This could happen if the disk_readp function prototype is not declared properly or seen by [...]

Ah, for some reason I forgot to add that particular header, I thought the compiler was going to be a little bit more pedantic about it. Anyway, now that works and I can read FAT16 filesystems using ElmChan's PFFs library!

> Yes - I would agree that we should discuss SSD driver issues in another issue, and limit discussion in this issue to the 8018X kernel porting issues only, thank you!

I've opened #972 to keep talking about the SD card driver.

For this ticket, I think we should start working on the pure serial console using IRQs and such, since the delay on the polled serial console is really annoying.

Thanks!

cocus, 2021-09-29:
I don't have "date" installed. However, I have one of my test apps that run a busy loop with a 1s delay by calling `sleep(1);` and the output is actually pretty good. I didn't time it, but I checked it against my PC clock and looks good.

What I meant by the "delay" is that I write a letter, and maybe it's lost. I do write pretty fast tho, and considering there's no hardware flow control, it might be possible for the receiver hardware to overrun the buffer.

In any case, let's move forward with the serial IRQ implementation

ghaerr, 2021-09-29:
> What I meant by the "delay" is that I write a letter, and maybe it's lost. I do write pretty fast tho, and considering there's no hardware flow control, it might be possible for the receiver hardware to overrun the buffer.

I would like to debug this situation so that I know that the basics of the ELKS port for 8018X is in fact working properly.

It sounds like you're saying that typing too fast causes a character to be dropped. Is this frequent? Increasing the polling frequency or lowering the baud rate should help, as it sounds like there may only be a single character input buffer on the 8018X chip?

To increase the polling frequency to every 4ms, just change the line below in elks/arch/i86/drivers/char/kbd-poll.c from + 8 to + 4:
```
static void restart_timer(void)
{
    static struct timer_list timer;

    init_timer(&timer);
    timer.tl_expires = jiffies + 8;     /* every 8/100 second*/
    timer.tl_function = kbd_timer;
    add_timer(&timer);
}
```

> In any case, let's move forward with the serial IRQ implementation.

Ok, it's ready to go for you. In elks/arch/i86/drivers/char, rename serial-template.c to serial-8018x.c and rewrite the functions for the 8018X UART, rather than the PC 8250. I have already cleaned up this file to its minimum functionality, so it will be lots easier to start from than the somewhat beastly serial-8250.c version which supports FIFO and other features. I would suggest you get this working for output, before enabling the receive IRQ for one of the two serial ports. Let me know what help you need.

cocus, 2021-10-30:
> Perhaps just set CONFIG_HEADLESS_CONSOLE and comment out all the code in condo-8018x.c so it doesn't interfere with your serial port for now. Of course, also set CONFIG_CONSOLE_SERIAL. We'll figure out a true headless console after you get the serial port working.

Commenting the source of the conio-8018x.c creates a ton of warnings (I assume it's because of:
```
int conio_poll(void)
{
    return 0;
}
```

> That is pretty strange. Another idea might be to temporarily not check the TI (or TXE) bit and just outb the character to the transmit buffer for the time being. Yes, you will probably get all sorts of overruns, but that might allow you to test the RX side until the polling strangeness is figured out.

Tried this, RX is still not working :( I don't get any RX IRQs. My code is not touching the serial IRQ register (SCUCON) at any given time (but the request_irq is, indirectly). Could it be that the CPU crashed? I think going back to the headless stuff to test the IRQs might be a good idea. I don't like these strangeness we're observing, but oh well...

ghaerr, 2021-10-30:
> Commenting the source of the conio-8018x.c creates a ton of warnings 

You're just commenting out the `inw` and `outb` code? Not sure how removing that code could add warnings?

> I think going back to the headless stuff to test the IRQs might be a good idea. 

Well, it would eliminate a lot of variables. I can't think of any more reasons why your code isn't working... :(

You might try just some very basic coding directly in the conio-8018x.c file, and not calling any of the other routines (except possibly request_irq, but only to setup the interrupt vector in low memory, NOT to write the SCUCON or other interrupt registers).

cocus, 2021-10-30:
> You're just commenting out the `inw` and `outb` code? Not sure how removing that code could add warnings?
> 
> >

Because conio_poll ALWAYS returns zero, which should be optimized out by the compiler, I guess? Same for conio_putc() which doesn't do anythin. Maybe I should add `(void)c` into conio_putc().

> You might try just some very basic coding directly in the conio-8018x.c file, and not calling any of the other routines (except possibly request_irq, but only to setup the interrupt vector in low memory, NOT to write the SCUCON or other interrupt registers).

Will do.

cocus, 2021-10-30:
I've done exactly that, and I can confirm I'm receving RX and TX interrupts on the headless conio code. In fact, I can see the timer polling the uart via the LED flashing.

```
/*
 * conio API for 8081X Headless Console
 *
 * This file contains code used for the embedded 8018X family only.
 * 
 * 16 May 21 Santiago Hormazabal
 */

#include <linuxmt/types.h>
#include <linuxmt/config.h>
#include <arch/io.h>
#include <arch/8018x.h>		/* definitions of UART0_IRQ_RX/UART0_IRQ_TX*/
#include <linuxmt/config.h>
#include <linuxmt/sched.h>
#include <linuxmt/debug.h>
#include "conio.h"

static volatile char leds = 0xfe;

void rs_irq_tx(int irq, struct pt_regs *regs)
{
    leds ^= 2;
    outb(leds, 0x2002);
}

/* serial interrupt routine, reads and queues received character */
void rs_irq_rx(int irq, struct pt_regs *regs)
{
    leds ^= 4;
    outb(leds, 0x2002);
}

/* initialize*/
void conio_init(void)
{
    int r; int good_init = 0;
    r = request_irq(UART0_IRQ_RX, rs_irq_rx, INT_GENERIC);
    good_init |= r;
    if (r)
        printk("Can't get serial IRQ for rx (%d) ret=%d\n", UART0_IRQ_RX, r);
    r = request_irq(UART0_IRQ_TX, rs_irq_tx, INT_GENERIC);
    good_init |= r;
    if (r)
        printk("Can't get serial IRQ for tx (%d) ret=%d\n", UART0_IRQ_TX, r);
    
    if (good_init == 0) {
        printk("conio_init: irq rx %d tx %d\n", UART0_IRQ_RX, UART0_IRQ_TX);
    }
}

/*
 * Poll for console input available.
 * Return nonzero character received else 0 if none ready.
 * Called approximately every ~8/100 seconds.
 */
int conio_poll(void)
{
    leds ^= 8;
    outb(leds, 0x2002);

    /* S0STS bit 0x40 RI (receive interrupt) */
    if (inw(0xff00 + 0x66) & 0x40) {
        return inw(0xff00 + 0x68); /* R0BUF */
    }
    return 0;
}

void conio_putc(byte_t c)
{
    while((inw(0xff00 + 0x66) & 8) == 0) {
        leds ^= 0x10;
        outb(leds, 0x2002);
    }	/* S0STS */
    outb(c, 0xff00 + 0x6A);			/* T0BUF */
}
```

Works literally a treat!

ghaerr, 2021-10-30:
>  I can confirm I'm receving RX and TX interrupts on the headless conio code. In fact, I can see the timer polling the uart via the LED flashing.

Pretty neat. I presume you've got 3 LEDs and can see each blink on associated activity.

Funny thing, you're now saying that the receive or transmit UART polling by busy looping does nothing to prohibit the RX or TX interrupt, while earlier there were all sorts of problems? Strange.

Now that things seem to work, I suggest the following:
```
void conio_putc(byte_t c)
{
    while((inw(0xff00 + 0x66) & 8) == 0) {
```
- Change to using TI rather than TXE for TX polling so as to match eventual interrupt handler (see above).
- Instead of actually polling the S0STS RI bit in `conio_poll`, introduce a new global variable which is the character received (read in only by the RX interrupt handler). When that variable is nonzero, return it in conio_poll and set it back to 0. This will determine whether we can get the RX interrupt handler completely working before messing with TX. In the serial driver, this will be replaced by queueing the character to a chqueue (as you've already coded there).
- After getting RX working, do similar with conio_putc, except that global var only keeps status of TI ready. Conio_putc can then poll that (volatile) global variable and output to the UART when ready. Unlike RX handler, we will need to keep the actual transmit output to the UART in normal kernel, NOT the TX interrupt handler. This is because the tty_output and associated routines can't easily be called at interrupt time when in the serial driver.

If all this works, then we are led to believe that the irq-8018x.c code is likely correct, and instead we have something going awry in the setup of the serial driver. So far, I don't know what that is, but will be suggesting commenting out almost everything in the serial driver, and replacing sections with working code from here in a bare skeleton in the beginning to find out where the problem is.

cocus, 2021-11-05:
Hi @ghaerr , so, I didn't change the check on putc() from TXE to TI, and I still get my code working. What did I do? I removed kbd_poll and headless-console out of the build and added a temporary serial-temp.c file with the following contents:
```
#include <arch/io.h>
#include <linuxmt/types.h>
#include <linuxmt/config.h>
#include <linuxmt/errno.h>
#include <linuxmt/kernel.h>
#include <linuxmt/sched.h>
#include <linuxmt/chqueue.h>
#include <linuxmt/ntty.h>
#include "console.h"

#include <linuxmt/types.h>
#include <linuxmt/config.h>
#include <linuxmt/timer.h>
#include <linuxmt/sched.h>
#include "console.h"
#include "conio.h"

char kbd_name[] = "polling";

static volatile char leds = 0xfe;

/* serial interrupt routine, reads and queues received character */
void rs_irq_rx(int irq, struct pt_regs *regs)
{
    leds ^= 0x2;
    outw(leds, 0x2002);

    register struct tty *ttyp = &ttys[0];	/* /dev/tty1*/

    unsigned char c = inb(0xff68);    /* Read received data */

    if (!tty_intcheck(ttyp, c))
        chq_addch(&ttyp->inq, c);
}

/* serial interrupt routine, reads and queues received character */
void rs_irq_tx(int irq, struct pt_regs *regs)
{
    leds ^= 0x4;
    outw(leds, 0x2002);
}

void conio_putc(byte_t c)
{
    while((inw(0xff00 + 0x66) & 8) == 0);	/* S0STS */
    outb(c, 0xff00 + 0x6A);			/* T0BUF */
}

// Called from main.c!
void console_init(void)
{
    request_irq(1, rs_irq_rx, INT_GENERIC);
    request_irq(2, rs_irq_tx, INT_GENERIC);

    printk("Headless consossssssssssssssssssssleggggggggggggg\n");
}

void Console_conout(dev_t dev, char Ch)
{
    if (Ch == '\n')
        conio_putc('\r');
    conio_putc(Ch);
}

void bell(void)
{
    conio_putc(7);	/* send ^G */
}

static int Console_ioctl(struct tty *tty, int cmd, char *arg)
{
    switch (cmd) {
    case TCSETS:
    case TCSETSW:
    case TCSETSF:
        return 0;
    }

    return -EINVAL;
}

static int Console_write(register struct tty *tty)
{
    int cnt = 0;

    while (tty->outq.len > 0) {
        conio_putc((byte_t)tty_outproc(tty));
        cnt++;
    }
    return cnt;
}

static void Console_release(struct tty *tty)
{
    ttystd_release(tty);
}

static int Console_open(register struct tty *tty)
{
    return ttystd_open(tty);
}

struct tty_ops headlesscon_ops = {
    Console_open,
    Console_release,
    Console_write,
    NULL,
    Console_ioctl,
    Console_conout
};
```

This works a treat, I have the RX irqs firing perfectly, and the console works really good! I can type REALLY fast and everything is fine.

> * After getting RX working, do similar with conio_putc, except that global var only keeps status of TI ready. Conio_putc can then poll that (volatile) global variable and output to the UART when [...]

I'm missing this, but I assume your original idea of the tx queue might work better.

I would like to ask you how to translate this code "properly" to a true serial driver, since it seems like what I did before, didn't work correctly. I would also like to understand how to do this without involving that PC-related terminal stuff you've mentioned before without hacking up the Makefile.

Thanks!

ghaerr, 2021-11-05:
Hello @cocus,

Well you've done a great job putting together a very slim serial console driver, nice! Funnily enough, it uses RX interrupts and polled TX output, which was my original suggestion. IMO, that's pretty much good enough, as in almost every case, the writing application blocks anyways until the outq is emptied.

>  I didn't change the check on putc() from TXE to TI

Why not? At least test it, using TI is faster (by one character transmit time) and actually more proper. TXE waits until the transmitter is finished, and doesn't coincide with the TI interrupt when/if you go that route.

>  I assume your original idea of the tx queue might work better.

Unless you really want hardware flow control, I don't see adding TX interrupts to be worth the trouble. Nonetheless, I think we should commit working RX serial first, then add TX later so there's a working point of reference.

> we need to introduce a new file called console-serial-8018x.c that's similar to the console-headless.c BUT its more or less contains the code that I've added on this comment

Good idea, since you've now got something working. If you add that file, and update the Makefile to use this for 8018X regardless of CONFIG_CONSOLE_SERIAL or CONFIG_CONSOLE_DIRECT, CONFIG_CONSOLE_HEADLESS etc, we can commit this for a starting reference.

> I don't think the serial-8018x.c file I have been working on is kinda "okay" to be used with the "console-direct" because of that PC interactions (mostly on console-direct.c) as you've mentioned.

You definitely should not use any console-direct.c code in your build; it writes all over segment 0xB800 and also tries to program the PC CRT controller.

> I would like to ask you how to translate this code "properly" to a true serial driver, since it seems like what I did before, didn't work correctly. 

I really don't know why the serial driver isn't working properly for you. I would suggest backing up serial-8081x.c and starting by removing all 8018x-specific code, then adding in this working code, and keeping the driver extremely lightweight. There must be some problem of bad data, perhaps mis-selection of the ports[] values, or ? If you insert this hardcoded solution procedure-by-procedure, we should be able to track it down.

> I would also like to understand how to do this without involving that PC-related terminal stuff you've mentioned before without hacking up the Makefile.

We may need a CONFIG_CONSOLE_NONE that defines a console-none.c that defines the entry points and does nothing, then use CONFIG_CONSOLE_SERIAL to automatically work with the serial driver, or something close to that.

Thank you!

cocus, 2021-11-07:
> Why not? At least test it, using TI is faster (by one character transmit time) and actually more proper. TXE waits until the transmitter is finished, and doesn't coincide with the TI interrupt when/if you go that route.
> 

So, there's a problem with this. My code now reads the status register on the RX irq. If my putc() busy-loops wait for the TI bit to be set, they get stuck. Now, if I poll for TXE, they don't get stuck. This is because TXE doesn't get clear if you read the status register.

> Good idea, since you've now got something working. If you add that file, and update the Makefile to use this for 8018X regardless of CONFIG_CONSOLE_SERIAL or CONFIG_CONSOLE_DIRECT, CONFIG_CONSOLE_HEADLESS etc, we can commit this for a starting reference.

What I'm currently doing is setting this console as a replacement of the "direct" console, WITHOUT adding ANY "PC related" direct console stuff, by doing the following:
```
ifdef CONFIG_CONSOLE_DIRECT
ifeq ($(CONFIG_ARCH_8018X), y)
	# 8018x internal serial port used as a headless console
	OBJS += console-serial-8018x.o
else
	OBJS += console-direct.o bell-8254.o
endif
endif
```
And exporting the tty_ops as "dircon_ops".

We might want to add a different "console over serial" option that doesn't hardcode these things, I guess, but this is a good start.

I have the 8018x.h changes as well, do you mind if I push both in the upcoming PR? It'll make code more readeable not only on this new .c file, but also on the irq-8018x.c and timer-8018x.c files.

Thanks

[11 more replies]

closed 2021-11-18 by ghaerr