ghaerr/elks #962: Cross build broken by latest EMU86 config for ELKS

mfld-fr, 2021-05-25:
The automatic build of the cross tools is broken by the latest EMU86 configuration for ELKS. That configuration requires SDL2. Here is the patch to fix:
```
diff --git a/.github/workflows/cross.yml b/.github/workflows/cross.yml
index 78ceb8b1..8df8c92b 100644
--- a/.github/workflows/cross.yml
+++ b/.github/workflows/cross.yml
@@ -12,8 +12,11 @@ jobs:
     runs-on: ubuntu-latest
 
     steps:
-      - name: setup
-        run: 'sudo apt install bison flex texinfo'
+      - name: setup-1
+        run: 'sudo apt-get update'
+
+      - name: setup-2
+        run: 'sudo apt-get install libsdl2-dev'
 
       - name: checkout
         uses: actions/checkout@v2
```
Note 1: bison, flex and texinfo are reported to be already installed.
Note 2: there is still missing the `cp config-elks.mk config.mk` in the cross tools Makefile.

ghaerr, 2021-05-25:
> Note 2: there is still missing the cp config-elks.mk config.mk in the cross tools Makefile.

I had that commented out until stable version of EMU86. Shall I set it now along with this patch to the latest version?

mfld-fr, 2021-05-25:
Yes please, as I got the cross build error while trying to update to the latest EMU86.

ghaerr, 2021-05-26:
@mfld-fr : Looks like the build is failing from this PR #964 (at least on my repo), can you debug? Unfortunately I don't have time tonight to look into this, thanks.
```
2021-05-26T02:43:03.1744315Z rm -rf /home/runner/work/elks/elks/cross/build/emu86-f99937262b2566d3d20339e5b200e8483cdec3a2
2021-05-26T02:43:03.1752360Z cd /home/runner/work/elks/elks/cross/build && tar -xzf /home/runner/work/elks/elks/cross/dist/emu86-f99937262b2566d3d20339e5b200e8483cdec3a2.tar.gz
2021-05-26T02:43:03.1942546Z rm -rf /home/runner/work/elks/elks/cross/build/emu86
2021-05-26T02:43:03.1951073Z cd /home/runner/work/elks/elks/cross/build && mv emu86-f99937262b2566d3d20339e5b200e8483cdec3a2 emu86
2021-05-26T02:43:03.1968045Z touch /home/runner/work/elks/elks/cross/build/.emu86.src
2021-05-26T02:43:03.1977151Z cp /home/runner/work/elks/elks/cross/build/emu86/config-elks.mk /home/runner/work/elks/elks/cross/build/emu86/config.mk
2021-05-26T02:43:03.1992272Z make -C /home/runner/work/elks/elks/cross/build/emu86 all
2021-05-26T02:43:03.2007864Z make[1]: Entering directory '/home/runner/work/elks/elks/cross/build/emu86'
2021-05-26T02:43:03.2009026Z cc -g -Wall -Werror -DSDL=1   -c -o list.o list.c
2021-05-26T02:43:03.2192404Z cc -g -Wall -Werror -DSDL=1   -c -o op-common.o op-common.c
2021-05-26T02:43:03.2419105Z cc -g -Wall -Werror -DSDL=1   -c -o op-id-name.o op-id-name.c
2021-05-26T02:43:03.2648458Z cc -g -Wall -Werror -DSDL=1   -c -o op-class.o op-class.c
2021-05-26T02:43:03.3337787Z cc -g -Wall -Werror -DSDL=1   -c -o emu-mem-io.o emu-mem-io.c
2021-05-26T02:43:03.3684092Z cc -g -Wall -Werror -DSDL=1   -c -o emu-proc.o emu-proc.c
2021-05-26T02:43:03.4150745Z cc -g -Wall -Werror -DSDL=1   -c -o op-exec.o op-exec.c
2021-05-26T02:43:03.5321989Z cc -g -Wall -Werror -DSDL=1   -c -o emu-int.o emu-int.c
2021-05-26T02:43:03.5629378Z cc -g -Wall -Werror -DSDL=1   -c -o emu-main.o emu-main.c
2021-05-26T02:43:03.6146938Z cc -g -Wall -Werror -DSDL=1   -c -o op-print-att.o op-print-att.c
2021-05-26T02:43:03.6480630Z cc -g -Wall -Werror -DSDL=1   -c -o emu-con.o emu-con.c
2021-05-26T02:43:03.6709720Z cc -g -Wall -Werror -DSDL=1   -c -o con-sdl.o con-sdl.c
2021-05-26T02:43:03.6806414Z con-sdl.c:13:10: fatal error: SDL2/SDL.h: No such file or directory
2021-05-26T02:43:03.6806990Z    13 | #include <SDL2/SDL.h>
2021-05-26T02:43:03.6807497Z       |          ^~~~~~~~~~~~
2021-05-26T02:43:03.6808029Z compilation terminated.
2021-05-26T02:43:03.6822327Z make[1]: *** [<builtin>: con-sdl.o] Error 1
2021-05-26T02:43:03.6823218Z make[1]: Leaving directory '/home/runner/work/elks/elks/cross/build/emu86'
2021-05-26T02:43:03.6824874Z make: *** [Makefile:183: /home/runner/work/elks/elks/cross/build/.emu86.build] Error 2
2021-05-26T02:43:03.6826215Z make: Leaving directory '/home/runner/work/elks/elks/tools'
2021-05-26T02:43:03.6835650Z ##[error]Process completed with exit code 2.
```

mfld-fr, 2021-05-26:
I see no error in the Actions tab for this PR : https://github.com/jbruchon/elks/actions/runs/877003466

ghaerr, 2021-05-26:
Yes, I mentioned that the error is occurring in my repo from that commit, which is even with the master repo: https://github.com/ghaerr/elks/actions/runs/877005759. I don't know much about how GitHub workflows function. Do you know why this may be, or how to fix it? Thanks.

mfld-fr, 2021-05-27:
The workflow that failed is the 'main' one, that is supposed to reuse the cached output of the 'cross' worflow. As the 'cross' one is far longer (~20 mins), the 'main' missed the cached output. That looks like a transient failure.

You could also improve the 'cross' worflow by adding  an 'on pull_request' clause, with the same filter as the 'on push' clause, following the same model as the 'main' workflow.

ghaerr, 2021-05-27:
I understand now, thanks to your explanation!

ghaerr, 2021-06-13:
Hello @mfld-fr,

> The workflow that failed is the 'main' one, that is supposed to reuse the cached output of the 'cross' worflow. As the 'cross' one is far longer (~20 mins), the 'main' missed the cached output. That looks like a transient failure.

Both ELKS master and my repo are again failing due to the enhanced EMU86 build requiring SDL 2 and a compile failure on include <SDL2/SDL.h>. It seems it isn't a transient failure.

I don't understand workflow automation well, do you have any other ideas on how to fix this? Perhaps we should remove the requirement SDL requirement from the default EMU86 build?

Thank you!

closed 2023-08-03 by ghaerr