ghaerr/elks #968: Is ELKS alive?

marcin-laszewski, 2021-07-30:


Mellvik, 2021-07-30:
Hi Marcin - thanks for asking.

I guess it depends on what you mean by 'alive'. From my perspective ELKS is very much alive, and we're having a (presumably short) hiatus, coping with our own individual challenges in a world that seems to be either burning or flooding - and a lot more.

When summer is over, I guarantee more activity. I have several small and lager projects 'in the works', and I'm sure there are many others.

—Mellvik

ghaerr, 2021-07-30:
Hello @marcin-laszewski,

There's been a lot accomplished this year, compressed executables, dynamic IRQ management with its resulting runtime configurability, much progress on embedding ELKS and lots more. I think everyone's still around, perhaps enjoying some of the outdoors given last year's relative confinement. I'd like to see a v0.5.0 with all these additions, but that'll probably wait until more contributors and volunteers have free time.

As always, please open an issue or comment for anything you'd like to see or having problems with.

Thank you!

toncho11, 2021-08-02:
[ELKS on ESP32]([](https://www.youtube.com/watch?v=Tr2yMjrgP8o&t=0s&ab_channel=FabrizioDiVittorio))

There is an IBM PC emulation with everything running on the ESP32 micro controller.

Mellvik, 2021-08-02:
Interesting @toncho11,

	link please?

Thank you.

—Mellvik

toncho11, 2021-08-02:
The link above should work. But here it again:

https://www.youtube.com/watch?v=Tr2yMjrgP8o&t=0s&ab_channel=FabrizioDiVittorio

closed 2021-10-19 by ghaerr