ghaerr/elks #969: Community space for ELKS

kehak, 2021-08-08:
It occurred to me ELKS, while a great project, currently lacks a place to exchange about it, especially for newcomers.
I see two issues with GitHub in that regard. First, it constitutes a barrier to entry for new users who might not want to register an account, log a ticket, and possibly be ridiculed for asking a beginner question.
Second, GitHub should be focused on bug reports and enhancements, not user support or general talks.

Some platforms come to mind for such a space. Twitter is great for announcements but not for casual conversation. IRC is nice for casual conversation but lacks a backlog and topics, meaning precious information become lost.
So I was thinking of a Reddit channel (/r/elks86), which would fit the bill for such a community.
The Alpine reddit, among others, is a good example of such a small community: https://www.reddit.com/r/AlpineLinux

This could be the place to publish announcements, ask general questions, receive some community based technical support, or even display some setups with a very low barrier to entry. Anything not worthy of creating a ticket here.

Of course, it may not be worth it for a very low volume of exchange. But it may also bring publicity, life, newcomers and maintainers to ELKS, and grow from here.

What do you think about this?

mfld-fr, 2021-08-08:
ELKS has a mailing list : linux-8086@vger.kernel.org. Archives can be found here : http://mail.spinics.net/lists/linux-8086/maillist.html.

kehak, 2021-08-09:
I didn't know about the mailing list, thank you.

I had first envisioned something a little more "modern", but after giving it a second thought, the mailing list suits the projects well. It's webless (that is, technically simple and efficient), and it can be used from within ELKS if required.

closed 2021-08-09 by kehak