ghaerr/elks #972: SSD driver implementation example

cocus, 2021-09-29:
Now that ELKS boots using the polled keyboard mechanism on my i80c188eb board, I would like to add support of SD cards via SPI bitbang on the GPIOs found on this CPU.
I have some code that actually works and exposes the required API for ElmChan's petit filesystem library to read a FAT16 filesystem on the SD card. Some of this code is assembly, while the other is C. This API is pretty basic, init, read sector, write sector (which assumes a sector of 512 bytes).
```
DSTATUS disk_initialize (void);
DRESULT disk_readp (BYTE* buff, DWORD sector, WORD offset, WORD count);
DRESULT disk_writep (const BYTE* buff /* Pointer to the BYTEs to be written (NULL:Initiate/Finalize sector write) */, DWORD sc /* Number of BYTEs to send, Sector number (LBA) or zero */);
```
LBA translation is automatically done by this code based on the type of card being used (older cards used raw sector addressing, while newer use LBA).

How to add a new driver for this platform? How to enable it on the config? And finally, how to use it on the target? (Assuming I can enable FAT fs support on the elks kernel).

Thanks!

ghaerr, 2021-09-30:
> How to add a new driver for this platform? 

I have written a template SSD block device driver, waiting for you to fill in the more gory details of actually reading/writing the low-level SSD device. All of this lives in elks/arch/i86/drivers/block/. There are two portions to the driver: the higher level block device driver ssd.c, and the "test" driver ssd_test.c. You should study both files. 

>  And finally, how to use it on the target? (Assuming I can enable FAT fs support on the elks kernel).

The "test" driver allows you to use the internal "ramdisk" to use as a "disk" that returns data through the SSD block device. It is fully tested and working, and the instructions are in the ssd_test.c file. This will show you how to use it on the target.

You will then rewrite the ssd_test.c read and write block functions to use your special SSD code and everything should then automatically work, except that the real SSD device will be used instead of the ramdisk for data.

> How to enable it on the config? 

Enabling CONFIG_BLK_DEV_SSD should make it all work. I added that option in the `make menuconfig` under kernel block device drivers under "SSD Disk Support".

cocus, 2021-09-30:
Hi @ghaerr ,

> The "test" driver allows you to use the internal "ramdisk" to use as a "disk" that returns data through the SSD block device. It is fully tested and working, and the instructions are in the ssd_test.c file. This will show you how to use it on the target.

You're correct, however, when I've added `diskutils` and SSD support by enabling `CONFIG_BLK_DEV_SSD `, I tried to run the example commands to load the ramdisk, but... I got the following:
```
# ramdisk /dev/ssd make 192
ramdisk: No such device
```
I've set DEBUG to 1 on the ssd_test.c so I can see what happens, but I don't see anything. `/dev/ssd` seems to exist:
```
brw-rw-rw-  1 root     root       1,   0  Jan 01 1970  rd0
brw-rw-rw-  1 root     root       1,   1  Jan 01 1970  rd1
brw-rw-rw-  1 root     root       2,   0  Jan 01 1970  ssd
```
I've tried with other values instead of 192, and I still get the same error.

By the way, when I get the `RDCREATE` ioctl, should I init the SD card? or that should take place on the `ssddev_init()`? I technically should store 1 value on a global variable that stores if the SD card is SDv2 or older, but I only know about that after initializing the card.
Are other ioctls that might show how large device is? How could I do the testing of my SD driver afterwards without using "ramdisk"?

Thanks!

ghaerr, 2021-09-30:
> ramdisk /dev/ssd make 192
ramdisk: No such device
> I've tried with other values instead of 192, and I still get the same error.

Hmmm... strange. I will look into this. It seems that ramdisk.c is failing in the initial open of /dev/ssd, for reasons yet unknown. That's why you're not seeing any debug printk's in ssd_test.c.

Is this running on the 8018X with ROMFS? I've only tested with the non-ROM filesystem.

> By the way, when I get the RDCREATE ioctl, should I init the SD card? or that should take place on the ssddev_init()?

The init should take place in ssddev_init(), since the RDCREATE ioctl's won't ever take place on a non-test (non-ramdisk) system. (That is, most all of ssddev_ioctl() can be removed).

>  I technically should store 1 value on a global variable that stores if the SD card is SDv2 or older, but I only know about that after initializing the card.

Ok, if that is needed for your sub driver.

> Are other ioctls that might show how large device is? 

Not really, the block device subsystem will know how big your drive is by returning the max # 512-byte sectors from ssddev_init(), which is then stored in the ssd.c init function. The mounted filesystem will then be used by other programs to determine the usable filesystem size.

> How could I do the testing of my SD driver afterwards without using "ramdisk"?

You will mount the filesystem directly using something like `mount -t msdos /dev/ssd /mnt`.

cocus, 2021-09-30:
> Is this running on the 8018X with ROMFS? I've only tested with the non-ROM filesystem.
> 

That is correct, it's on the 80c188eb with the ROMFS root filesystem.

> The init should take place in ssddev_init(), since the RDCREATE ioctl's won't ever take place on a non-test (non-ramdisk) system. (That is, most all of ssddev_ioctl() can be removed).
> 

I see, so, if I want to swap the SD card I would have to reboot the system to reload the driver? or is the _init() function called dynamically when mounting it?

> Not really, the block device subsystem will know how big your drive is by returning the max # 512-byte sectors from ssddev_init(), which is then stored in the ssd.c init function. The mounted filesystem will then be used by other programs to determine the usable filesystem size.

Okay, that makes sense. Returning -EFAIL is okay in case of failure?

> You will mount the filesystem directly using something like `mount -t msdos /dev/ssd /mnt`.

Cool!

Thanks

ghaerr, 2021-09-30:
> That is correct, it's on the 80c188eb with the ROMFS root filesystem.

I'm thinking a possible reason for the ramdisk failure is that the ssd_test.c code ignores the 4th argument to ramdisk and instead uses the hard-coded NUM_SECTS value in ssd_test.c. Change that to 64, which will allocate only 64K of memory. I suspect that there may not be enough memory with ROMFS stuck at 512K main RAM.

> if I want to swap the SD card I would have to reboot the system to reload the driver? or is the _init() function called dynamically when mounting it?

It looks like ssd_init() is only called once at kernel init, and then it calls ssddev_init only once at the same time. So we're going to have to change the initialization to occur in the ssd_open routine in ssd.c (which also sets the block device size using inode->i_size). This somewhat breaks my initial design of having separate ssd.c and ssd_test.c sub drivers. You'll likely have to add the ioctl and block read/write functions directly into ssd.c and rename it ssd-something.c instead.

> Returning -EFAIL is okay in case of failure?

You mean in the block read/write functions? No. You just return the # of sectors read, which would be 0 in that case.

ghaerr, 2021-09-30:
> Returning -EFAIL is okay in case of failure?

If you mean in the ssd_init() routine(s), we're going to use ssd_open() instead to do the initialization and determination of physical volume size. So in the case of failure, you'll use -ENXIO like ssd_open does now.

The `register_blkdev` call will remain in init, but the NUM_SECTS determination will move to open.

cocus, 2021-09-30:
> I'm thinking a possible reason for the ramdisk failure is that the ssd_test.c code ignores the 4th argument to ramdisk and instead uses the hard-coded NUM_SECTS value in ssd_test.c. Change that to [...]

I set it to 64, then ran build.sh, now I have the following output:
```
# ramdisk /dev/ssd make 192
ramdisk: 192 KB ramdisk created on /dev/ssd
# mkfs /dev/ssd 192
64 inodes
192 blocks
Firstdatazone=6 (6)
Zonesize=1024
Maxsize=196608

mkfs: write failed in write_block
# sync
# fsck -lvf /dev/ssd
fsck 1.0.1 - 09/01/20
fsck: unable to read super block
#
```
I know it failed, but I wanted to try with fsck to see what's going on. Also, I don't see any debug messages even tho DEBUG is set to 1.
Tried with 64 instead of 192:
```
# ramdisk /dev/ssd make 64
ramdisk: 64 KB ramdisk created on /dev/ssd
# mkfs /dev/ssd 64
No /bin/sh found to execute complex command
# mkfs /dev/ssd 64
21 inodes
64 blocks
Firstdatazone=5 (5)
Zonesize=1024
Maxsize=65536

mkfs: write failed in write_block
#
```

Should I run something different?

ghaerr, 2021-09-30:
With your limited memory situation, we'll need to use 64 instead of 192 for all SSD testing. Also, I highly suggest first doing this on a workstation so as to familiarize yourself with what is needed to get it working, THEN run on the 8018X.

I have also determined that the instructions in ssd_test.c are incorrect: The 192 argument to `ramdisk` is supposed to be in 1K blocks, not sectors, so it should be 96 instead of 192. However, this is ignored in the SSD case. I will fix that documentation. `mkfs` also takes an argument in blocks (1K bytes).

First, ensure that the boot shows the SSD being created (these show the unmodified 192 sectors = 96K system). Your system should set NUM_SECTS to 128 which indicates 64K:
```
Direct console, scan kbd 80x25 emulating ANSI (3 virtual consoles)
ttyS0 at 0x3f8, irq 4 is a 16550A
lp: no ports found
eth: NE2K at 0x300, irq 12, MAC 52:54:00:12:34:56
rd: 64K ramdisk at d000:0000  <--- this may not work on 8018X since ramdisk ram isn't at D000
bioshd: 1 floppy drive and 0 hard drives
ssd: 96K disk  <--- yours should say 64K
```
Not sure why your system had this error (could that be a typo on your part with bad backspace settings?)
```
# mkfs /dev/ssd 64
No /bin/sh found to execute complex command
```
The following sequence worked on my system:
```
# cat zz1
set -x
ramdisk /dev/ssd make 64
mkfs /dev/ssd 64
sync
fsck -lvf /dev/ssd
mount /dev/ssd /mnt
cp /bin/p* /mnt
sync
df /dev/ssd
umount /dev/ssd
fsck -lvf /dev/ssd
# 
```
Output:
```
# ./zz1
+ ramdisk /dev/ssd make 64
ramdisk: 64 KB ramdisk created on /dev/ssd
+ mkfs /dev/ssd 64
21 inodes
64 blocks
Firstdatazone=5 (5)
Zonesize=1024
Maxsize=65536

+ sync
+ fsck -lvf /dev/ssd
fsck 1.0.1 - 09/01/20
Forcing filesystem check on /dev/ssd.

     1 inodes used (5%)
     6 zones used (9%)

     0 regular files
     1 directories
     0 character device files
     0 block device files
     0 links
     0 symbolic links
------
     1 files
+ mount /dev/ssd /mnt
+ cp /bin/partype /bin/passwd /bin/poweroff /bin/printenv /bin/proto /bin/ps /bin/pwd /bin/pwdauth /mnt
+ sync
+ df /dev/ssd
Filesystem    1k-Blocks     free     used    %  FUsed%
/dev/ssd             64       12       52  82%    48%
+ umount /dev/ssd
+ fsck -lvf /dev/ssd
fsck 1.0.1 - 09/01/20
Forcing filesystem check on /dev/ssd.
     2 100755   1 /partype
     3 100755   1 /passwd
     4 100755   1 /poweroff
     5 100755   1 /printenv
     6 100755   1 /proto
     7 100755   1 /ps
     8 100755   1 /pwd
     9 100755   1 /pwdauth

     9 inodes used (45%)
    52 zones used (81%)

     8 regular files
     1 directories
     0 character device files
     0 block device files
     0 links
     0 symbolic links
------
     9 files
```

cocus, 2021-10-01:
@ghaerr Sorry I skipped your instructions using the memory test SSD, but I went directly to the SD code I had working before, and hacked up a test. It initializes the card OK, tho I'm returning a value for the numsects about 20MB but something overlows. 
```
/* initialize SSD device */
int ssddev_init(void)
{
	return disk_initialize() == 0 ? (20480 * 2) : 0;
}
```
And I see the following on the startup log:
```
ssd: 2147471360K disk
```

My SD card is 2GB and it has a 2GB FAT16 partition.

I've tried to mount the partition using `mount -t msdos /dev/ssd /mnt` but the FAT driver fails. Here's the log:
```
FAT: me=0,csz=216,#f=251,floc=-16242,fsz=-18170,rloc=10608,#d=190,dloc=10619,#s=49020,ts=129940992
FAT: Unsupported format
mount failed: Invalid argument
```

Now, since my original code did cope with segments and offsets for the read of an entire sector into a buffer, I just copied that code. I've used pokeb() to accomplish so on the byte read loop, as follows:
```
[....]
	pokeb(buff, seg, d); //*buff = d;
	buff++;
[....]
```
Where both `buff` and `seg` comes from the argument of this read function:
```
[....]
	char *buff,
	seg_t seg,
[....]
```
And I do call this function TWICE on the `ssddev_read_blk`:
```
/* read one 1K block (two sectors) from SSD */
int ssddev_read_blk(sector_t start, char *buf, seg_t seg)
{
	printk("SSD: read blk %x, seg %x\n", start, seg);
	disk_readp(buf, seg, start, 0, 512);
	disk_readp(buf, seg + 0x20, start + 1, 0, 512);
	return 2;	/* # sectors read */
}
```

Do you recommend creating a smaller FAT partition and using a smaller value as the hardcoded number of sectors?

Thank you!

ghaerr, 2021-10-01:
>  I'm returning a value for the numsects about 20MB but something overflows.
> 
> ```
> /* initialize SSD device */
> int ssddev_init(void)
> {
> 	return disk_initialize() == 0 ? (20480 * 2) : 0;
> }
> ```

I suspect the problem here is the ssddev_init is returning an 40960 as a 16-bit 'int' which is negative number, which then is sign-extended into NUM_SECTS, which is a sector_t (long). So try using 10Mb instead so that we don't get integer overflow. ssddev_init() will have to become an unsigned long function, but we will worry about that later.

> My SD card is 2GB and it has a 2GB FAT16 partition.

It might be a good idea to use a much smaller SD card while you get things working, although I would assume that the FAT cluster size is very large in order to get the 2GB with 16-bit FAT indices.

> ```
> FAT: me=0,csz=216,#f=251,floc=-16242,fsz=-18170,rloc=10608,#d=190,dloc=10619,#s=49020,ts=129940992
> FAT: Unsupported format
> mount failed: Invalid argument
> ```

With 'fsz' negative, I can see how this failed. Likely related to volume size being too large for ELKS too handle, or other overflows we'll have to chase down. Suggest using much smaller SD card if possible for initial testing.

> ```
> [....]
> 	pokeb(buff, seg, d); //*buff = d;
> 	buff++;
> [....]
> ```

That looks fine but you'll probably want to use a static buffer and the much faster fmemcpyw function for speed after getting this working.

> ```
> /* read one 1K block (two sectors) from SSD */
> int ssddev_read_blk(sector_t start, char *buf, seg_t seg)
> {
> 	printk("SSD: read blk %x, seg %x\n", start, seg);
> 	disk_readp(buf, seg, start, 0, 512);
> 	disk_readp(buf, seg + 0x20, start + 1, 0, 512);
> 	return 2;	/* # sectors read */
> }
> ```

This also looks fine, but I recommend not incrementing segment by 0x20, but instead just using 512 offset for the second call. Easier to understand and less code. But ultimately doesn't matter, except assumes segment arithmetic in real mode.

> 
> Do you recommend creating a smaller FAT partition and using a smaller value as the hardcoded number of sectors?

Yes!

Thank you!

ghaerr, 2021-10-01:
@cocus,

> Likely related to volume size being too large for ELKS too handle, or other overflows we'll have to chase down

A good way to test would be to use desktop ELKS, as suggested earlier, for initial testing. For this particular issue, that would mean plugging in a 2GB USB flash stick into ELKS (or even with QEMU) and seeing whether ELKS can handle the 2GB drive.

I suspect we may have to chase down some more ELKS problems relating to super large drives. We already have a hard limit on the CHS disk addressing scheme,  but I fixed up the FAT32 subsystem to allow 32-bit sector numbers some time ago. However, not sure what a large FAT16 volume (with a large cluster size) might do, as I never tested that.

However, returning a much smaller volume size using the hack in ssddev_init() should work for testing, provided that you know for sure that all accessible files on the FAT16 volume are within the early portion of the physical volume.

Thank you!

cocus, 2021-10-02:
> It might be a good idea to use a much smaller SD card while you get things working, although I would assume that the FAT cluster size is very large in order to get the 2GB with 16-bit FAT indices.

Sadly, I've ditched all my small SD cards long time ago (< 1GB) so this is the smallest one I have which actually seem to work fine.

> I suspect the problem here is the ssddev_init is returning an 40960 as a 16-bit 'int' which is negative number, which then is sign-extended into NUM_SECTS, which is a sector_t (long). So try using [...]

Right, I've partitioned the SD card with gparted and it only let me create a 16MiB partition for fat16, since that's its minimum. I didn't know that. Anyway, that number is still above the 16bit int signed value and should overflow as well. I'm using 10M as you've suggested for now.

> This also looks fine, but I recommend not incrementing segment by 0x20, but instead just using 512 offset for the second call. Easier to understand and less code. But ultimately doesn't matter, except assumes segment arithmetic in real mode.

So, can I just use `buf + 512` without any issues? What if the actual buffer is 512bytes below 0xffff? Is this even possible?

> That looks fine but you'll probably want to use a static buffer and the much faster fmemcpyw function for speed after getting this working.
> 

Oh, okay. I'll play with this afterwards.

> A good way to test would be to use desktop ELKS, as suggested earlier, for initial testing. For this particular issue, that would mean plugging in a 2GB USB flash stick into ELKS (or even with QEMU) and seeing whether ELKS can handle the 2GB drive.

Right, but I don't mind trying it on my hardware. I'm quite used to it and I think most of these things I'm seeing are hardware related.

> I suspect we may have to chase down some more ELKS problems relating to super large drives. We already have a hard limit on the CHS disk addressing scheme, but I fixed up the FAT32 subsystem to [...]

No worries, I'm able to trick the system into thinking this ssd device has not that many sectors. We can deal with large addressing modes for later.

> However, returning a much smaller volume size using the hack in ssddev_init() should work for testing, provided that you know for sure that all accessible files on the FAT16 volume are within the early portion of the physical volume.

Technically I just have a single file that's 84 bytes long.

In any case, it still doesn't like it:
```
ssd: 10240K disk
8018X machine, 512K base RAM.
ELKS kernel 0.4.0 (38256 text, 0 ftext, 4976 data, 38320 bss, 1024 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 466K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# mount -t msdos /dev/ssd /mnt
SSD: read blk 0, seg 0
dr: 0, buff=4ac6, seg=80, sector=0, offs=0, count=200
sc: cmd=51, arg=0, n=8, res=0
dr: result = 0
dr: 0, buff=4ac6, seg=a0, sector=512, offs=0, count=200
sc: cmd=51, arg=512, n=8, res=0
dr: result = 0
FAT: me=0,csz=216,#f=251,floc=-16242,fsz=-18170,rloc=10608,#d=190,dloc=10619,#s=                                       49020,ts=129940992
FAT: Unsupported format
mount failed: Invalid argument
```

But my test app `sd` is still able to re-initialize the SD card and read the FAT filesystem using the ElmChan PFFs (I know I shouldn't be disturbing the SD card if it's managed by a kernel driver, but I'm just testing).

Thanks!

cocus, 2021-10-02:
I've modified the code to dump whatever is reading, and looks like it's reading correctly the SD card:
```
# mount -t msdos /dev/ssd /mnt
SSD: read blk 0, seg 0
dr: 0, buff=4af6, seg=80, sector=0, offs=0, count=200
sc: cmd=51, arg=0, n=8, res=0
dr: 1, buff=4af6, sector=0, offs=0, count=200
dr: 2, d=fe, cnt=512
dr: 3, d=fe, cnt=512, offset=0, bc=2
buff = 4af6, value read = 0xeb, value stored = 0xeb
buff = 4af7, value read = 0x63, value stored = 0x63
buff = 4af8, value read = 0x90, value stored = 0x90
buff = 4af9, value read = 0x10, value stored = 0x10
buff = 4afa, value read = 0x8e, value stored = 0x8e
buff = 4afb, value read = 0xd0, value stored = 0xd0
buff = 4afc, value read = 0xbc, value stored = 0xbc
buff = 4afd, value read = 0x0, value stored = 0x0
buff = 4afe, value read = 0xb0, value stored = 0xb0
buff = 4aff, value read = 0xb8, value stored = 0xb8
buff = 4b00, value read = 0x0, value stored = 0x0
buff = 4b01, value read = 0x0, value stored = 0x0
buff = 4b02, value read = 0x8e, value stored = 0x8e
buff = 4b03, value read = 0xd8, value stored = 0xd8
buff = 4b04, value read = 0x8e, value stored = 0x8e
buff = 4b05, value read = 0xc0, value stored = 0xc0
buff = 4b06, value read = 0xfb, value stored = 0xfb
buff = 4b07, value read = 0xbe, value stored = 0xbe
buff = 4b08, value read = 0x0, value stored = 0x0
buff = 4b09, value read = 0x7c, value stored = 0x7c
buff = 4b0a, value read = 0xbf, value stored = 0xbf
buff = 4b0b, value read = 0x0, value stored = 0x0
buff = 4b0c, value read = 0x6, value stored = 0x6
buff = 4b0d, value read = 0xb9, value stored = 0xb9
buff = 4b0e, value read = 0x0, value stored = 0x0
buff = 4b0f, value read = 0x2, value stored = 0x2
buff = 4b10, value read = 0xf3, value stored = 0xf3
buff = 4b11, value read = 0xa4, value stored = 0xa4
buff = 4b12, value read = 0xea, value stored = 0xea
buff = 4b13, value read = 0x21, value stored = 0x21
buff = 4b14, value read = 0x6, value stored = 0x6
buff = 4b15, value read = 0x0, value stored = 0x0
buff = 4b16, value read = 0x0, value stored = 0x0
buff = 4b17, value read = 0xbe, value stored = 0xbe
buff = 4b18, value read = 0xbe, value stored = 0xbe
buff = 4b19, value read = 0x7, value stored = 0x7
buff = 4b1a, value read = 0x38, value stored = 0x38
buff = 4b1b, value read = 0x4, value stored = 0x4
buff = 4b1c, value read = 0x75, value stored = 0x75
buff = 4b1d, value read = 0xb, value stored = 0xb
buff = 4b1e, value read = 0x83, value stored = 0x83
buff = 4b1f, value read = 0xc6, value stored = 0xc6
buff = 4b20, value read = 0x10, value stored = 0x10
buff = 4b21, value read = 0x81, value stored = 0x81
buff = 4b22, value read = 0xfe, value stored = 0xfe
buff = 4b23, value read = 0xfe, value stored = 0xfe
buff = 4b24, value read = 0x7, value stored = 0x7
buff = 4b25, value read = 0x75, value stored = 0x75
buff = 4b26, value read = 0xf3, value stored = 0xf3
buff = 4b27, value read = 0xeb, value stored = 0xeb
buff = 4b28, value read = 0x16, value stored = 0x16
buff = 4b29, value read = 0xb4, value stored = 0xb4
buff = 4b2a, value read = 0x2, value stored = 0x2
buff = 4b2b, value read = 0xb0, value stored = 0xb0
buff = 4b2c, value read = 0x1, value stored = 0x1
buff = 4b2d, value read = 0xbb, value stored = 0xbb
buff = 4b2e, value read = 0x0, value stored = 0x0
buff = 4b2f, value read = 0x7c, value stored = 0x7c
buff = 4b30, value read = 0xb2, value stored = 0xb2
buff = 4b31, value read = 0x80, value stored = 0x80
buff = 4b32, value read = 0x8a, value stored = 0x8a
[... 1007 lines cut ...]
00000210  7d 08 00 0f 84 e2 00 80  7c ff 00 74 46 66 8b 1d  |}.......|..tFf..|
00000220  66 8b 4d 04 66 31 c0 b0  7f 39 45 08 7f 03 8b 45  |f.M.f1...9E....E|
00000230  08 29 45 08 66 01 05 66  83 55 04 00 c7 04 10 00  |.)E.f..f.U......|
00000240  89 44 02 66 89 5c 08 66  89 4c 0c c7 44 06 00 70  |.D.f.\.f.L..D..p|
00000250  50 c7 44 04 00 00 b4 42  cd 13 0f 82 af 00 bb 00  |P.D....B........|
00000260  70 eb 66 66 8b 45 04 66  09 c0 0f 85 97 00 66 8b  |p.ff.E.f......f.|
00000270  05 66 31 d2 66 f7 34 88  54 0a 66 31 d2 66 f7 74  |.f1.f.4.T.f1.f.t|
00000280  04 88 54 0b 89 44 0c 3b  44 08 7d 79 8b 04 2a 44  |..T..D.;D.}y..*D|
00000290  0a 39 45 08 7f 03 8b 45  08 29 45 08 66 01 05 66  |.9E....E.)E.f..f|
000002a0  83 55 04 00 8a 54 0d c0  e2 06 8a 4c 0a fe c1 08  |.U...T.....L....|
000002b0  d1 8a 6c 0c 5a 52 8a 74  0b 50 bb 00 70 8e c3 31  |..l.ZR.t.P..p..1|
000002c0  db b4 02 cd 13 72 46 8c  c3 8e 45 0a 58 c1 e0 05  |.....rF...E.X...|
000002d0  01 45 0a 60 1e c1 e0 03  89 c1 31 ff 31 f6 8e db  |.E.`......1.1...|
000002e0  fc f3 a5 1f be 23 81 e8  57 00 61 83 7d 08 00 0f  |.....#..W.a.}...|
000002f0  85 24 ff 83 ef 0c e9 16  ff be 25 81 e8 42 00 5a  |.$........%..B.Z|
00000300  ea 00 82 00 00 be 28 81  e8 36 00 eb 06 be 2d 81  |......(..6....-.|
00000310  e8 2e 00 be 32 81 e8 28  00 eb fe 6c 6f 61 64 69  |....2..(...loadi|
00000320  6e 67 00 2e 00 0d 0a 00  47 65 6f 6d 00 52 65 61  |ng......Geom.Rea|
00000330  64 00 20 45 72 72 6f 72  00 bb 01 00 b4 0e cd 10  |d. Error........|
00000340  46 8a 04 3c 00 75 f2 c3  00 00 00 00 00 00 00 00  |F..<.u..........|
00000350  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000003f0  00 00 00 00 02 00 00 00  00 00 00 00 65 00 20 08  |............e. .|
00000400
```

ghaerr, 2021-10-02:
Your desktop dump shows the problem! We're using a partitioned volume but ELKS isn't using its partitioning identification code that determines the actual volume start sector. The GRUB string in the first sector dump shows this. That means, for now, you need to add a "volume start sector offset" to the passed low-level sector to get the actual start of the FAT volume. A quick look shows that the FAT volume starts in sector 2, so add 1.

That should do it!

ghaerr, 2021-10-02:
> So, can I just use buf + 512 without any issues? What if the actual buffer is 512bytes below 0xffff? Is this even possible?

Yes. The way ELKS buffer (and most other large allocation) memory is managed is to allocate a segment, which is large enough to use any offset within the allocated memory. Thus, for system buffers, which are 1K bytes, you're passed the same segment in the block driver, and any offset 0-1023 will work and should be used.

cocus, 2021-10-02:
> Your desktop dump shows the problem! We're using a partitioned volume but ELKS isn't using its partitioning identification code that determines the actual volume start sector. The GRUB string in the first sector dump shows this. That means, for now, you need to add a "volume start sector offset" to the passed low-level sector to get the actual start of the FAT volume. A quick look shows that the FAT volume starts in sector 2, so add 1.
> 
> [EDIT: Oops - it looks like there may be an MBR sector and a spare, so try adding 2 also if 1 doesn't work.]
> 
> That should do it!

I've tried adding 1 or 2 to the sector number, but no dice. Here's the output when adding just 1:
```
# mount -t msdos /dev/ssd /mnt
SSD: read blk 0, buff 4a76, seg 80
dr: 0, buff=4a76, seg=80, sector=512, offs=0, count=200
sc: cmd=51, arg=512, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
dr: 0, buff=4a76, seg=a0, sector=1024, offs=0, count=200
sc: cmd=51, arg=1024, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
FAT: me=e2,csz=139,#f=125,floc=-31955,fsz=-32768,rloc=813,#d=8,dloc=813,#s=33807,ts=72190822
FAT: Unsupported format
mount failed: Invalid argument
#
```

And this is when adding 2:
```
# mount -t msdos /dev/ssd /mnt
SSD: read blk 0, buff 4ac6, seg 80
dr: 0, buff=4ac6, seg=80, sector=1024, offs=0, count=200
sc: cmd=51, arg=1024, n=8, res=0
dr: 1, buff=4ac6, sector=1024, offs=0, count=200
dr: 2, d=fe, cnt=512
dr: 3, d=fe, cnt=512, offset=0, bc=2
dr: result = 0
dr: 0, buff=4ac6, seg=a0, sector=1536, offs=0, count=200
sc: cmd=51, arg=1536, n=8, res=0
dr: 1, buff=4ac6, sector=1536, offs=0, count=200
dr: 2, d=fe, cnt=512
dr: 3, d=fe, cnt=512, offset=0, bc=2
dr: result = 0
FAT: me=ff,csz=116,#f=181,floc=0,fsz=-1,rloc=-181,#d=86,dloc=-176,#s=65024,ts=-1070466560
FAT: Unsupported format
mount failed: Invalid argument
#
```

Please note that the "sector=xxxxx" on the dumps refer to the actual low-level sector address sent to the SD card. SDv2 expects LBA addressing (hence the *512 of that value).

> Yes. The way ELKS buffer (and most other large allocation) memory is managed is to allocate a segment, which is large enough to use any offset within the allocated memory. Thus, for system buffers, which are 1K bytes, you're passed the same segment in the block driver, and any offset 0-1023 will work and should be used. The idea is that segment arithmetic should be avoided and only offsets used, as this would greatly easy the transition into any protected mode enhancement, as the allocation routines only return a single segment descriptor value that is valid for all offset up to the size allocated.

I see. I will move to using + 512 once this works.

> No, not with the way ELKS memory management works.

Got it!

cocus, 2021-10-02:
I did format the entire disk with FAT16 (even tho I can't set the size of the filesystem), and remove the "offset", this time it worked!
```
ssd: 10240K disk
8018X machine, 512K base RAM.
ELKS kernel 0.4.0 (38224 text, 0 ftext, 4896 data, 38320 bss, 1024 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 466K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# mount -t msdos /dev/ssd /mnt
SSD: read blk 0, buff 4a76, seg 80
dr: 0, buff=4a76, seg=80, sector=0, offs=0, count=200
sc: cmd=51, arg=0, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
dr: 0, buff=4a76, seg=a0, sector=512, offs=0, count=200
sc: cmd=51, arg=512, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
FAT: me=f8,csz=64,#f=2,floc=64,fsz=256,rloc=576,#d=1024,dloc=640,#s=3862528,ts=3862528
FAT: 1931M, fat16 format
SSD: read blk 240, buff 4e76, seg 80
dr: 0, buff=4e76, seg=80, sector=294912, offs=0, count=200
sc: cmd=51, arg=294912, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
dr: 0, buff=4e76, seg=a0, sector=295424, offs=0, count=200
sc: cmd=51, arg=295424, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
# cd /mnt
# ls
.
..
SSD: read blk 40, buff 5276, seg 80
dr: 0, buff=5276, seg=80, sector=32768, offs=0, count=200
sc: cmd=51, arg=32768, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
dr: 0, buff=5276, seg=a0, sector=33280, offs=0, count=200
sc: cmd=51, arg=33280, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
System Volume
message.txt
# more message.txt
SSD: read blk 340, buff 5676, seg 80
dr: 0, buff=5676, seg=80, sector=425984, offs=0, count=200
sc: cmd=51, arg=425984, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
dr: 0, buff=5676, seg=a0, sector=426496, offs=0, count=200
sc: cmd=51, arg=426496, n=8, res=0
dr: 2, d=fe, cnt=512
dr: result = 0
This is a test message written into a 2GB SD card and probably read by an i80c188eb!
#
```

ghaerr, 2021-10-02:
> I did format the entire disk with FAT16 (even tho I can't set the size of the filesystem), and removed the "offset", this time it worked!

Congratulations! You've just implemented the first new ELKS block device driver in quite a while!! :)

> I've tried adding 1 or 2 to the sector number, but no dice

Notice the output from your desktop `hd -n 1024 /dev/sdb`: only the "boot" partition of 0x400 bytes is dumped, but nothing after that. This means that the FAT partition isn't being read, and we don't actually know it's start offset. For some reason reading /dev/sdb isn't the raw device and the entire device isn't being read. Once you can get that information, using a start sector offset will work. That offset could be as high as 32 or more on some devices, especially those that allow FAT32.

Go ahead and do a couple of the minor cleanups previously discussed (offset+512 and unsigned long return on ssddev_init), and perhaps we can figure out how to make this work using the previously-formatted partitioned volume. For now, that could use a static compiled offset, or the initialization routine could try to determine whether there's a boot partition. Unfortunately, getting the ELKS standard partition code working on this is going to be a bit of work, as we'd also need /dev/ssd1, /dev/ssd2, etc., which might not be worth the effort.

Once ssddev_init is defined as unsigned long, it would be nice to test with the previous 20M size, just to check our basic volume size handling, which I think will work.

Nice work!!

cocus, 2021-10-04:
> Congratulations! You've just implemented the first new ELKS block device driver in quite a while!! :)
> 

Good progress!

> Notice the output from your desktop `hd -n 1024 /dev/sdb`: only the "boot" partition of 0x400 bytes is dumped, but nothing after that. This means that the FAT partition isn't being read, and we [...]

I've defined ssd_init to use sector_t (which is unsigned long anyway) and I hardcoded the size of the SSD block to 1.9GB (using the exact byte value reported from my OS), and it still works!
```
ssd: 1977286656K disk using sector_t
8018X machine, 512K base RAM.
ELKS kernel 0.4.0 (38608 text, 0 ftext, 4912 data, 38320 bss, 1024 heap)
Kernel text at e062:0000, data 80:0000, top 8000:0, 466K free
VFS: Mounted root 0x0600 (romfs filesystem).
No init - running /bin/sh
# mount -t msdos /dev/ssd /mnt
SSD: read blk 0, buff 4a86, seg 80
sc: cmd=51, arg=0, n=8, res=0
dr: 0, buff=4a86, seg=80, sector=0, result=0
sc: cmd=51, arg=512, n=8, res=0
dr: 0, buff=4a86, seg=a0, sector=512, result=0
FAT: me=f8,csz=64,#f=2,floc=64,fsz=256,rloc=576,#d=1024,dloc=640,#s=3862528,ts=3862528
FAT: 1931M, fat16 format
SSD: read blk 240, buff 4e86, seg 80
sc: cmd=51, arg=294912, n=8, res=0
dr: 0, buff=4e86, seg=80, sector=294912, result=0
sc: cmd=51, arg=295424, n=8, res=0
dr: 0, buff=4e86, seg=a0, sector=295424, result=0
#
```
Now, the only issue I'm encountering is that I can't write anything. I don't see any calls to the `ssddev_write_blk()` function, even if I added some debug (using printk), have a look:
```
# touch /mnt/a
SSD: read blk 242, buff 5286, seg 80
sc: cmd=51, arg=295936, n=8, res=0
dr: 0, buff=5286, seg=80, sector=295936, result=0
sc: cmd=51, arg=296448, n=8, res=0
dr: 0, buff=5286, seg=a0, sector=296448, result=0
SSD: read blk 244, buff 5686, seg 80
sc: cmd=51, arg=296960, n=8, res=0
dr: 0, buff=5686, seg=80, sector=296960, result=0
sc: cmd=51, arg=297472, n=8, res=0
dr: 0, buff=5686, seg=a0, sector=297472, result=0
SSD: read blk 246, buff 5a86, seg 80
sc: cmd=51, arg=297984, n=8, res=0
dr: 0, buff=5a86, seg=80, sector=297984, result=0
sc: cmd=51, arg=298496, n=8, res=0
dr: 0, buff=5a86, seg=a0, sector=298496, result=0
SSD: read blk 248, buff 5e86, seg 80
sc: cmd=51, arg=299008, n=8, res=0
dr: 0, buff=5e86, seg=80, sector=299008, result=0
sc: cmd=51, arg=299520, n=8, res=0
dr: 0, buff=5e86, seg=a0, sector=299520, result=0
SSD: read blk 24a, buff 6286, seg 80
sc: cmd=51, arg=300032, n=8, res=0
dr: 0, buff=6286, seg=80, sector=300032, result=0
sc: cmd=51, arg=300544, n=8, res=0
dr: 0, buff=6286, seg=a0, sector=300544, result=0
SSD: read blk 24c, buff 6686, seg 80
sc: cmd=51, arg=301056, n=8, res=0
dr: 0, buff=6686, seg=80, sector=301056, result=0
sc: cmd=51, arg=301568, n=8, res=0
dr: 0, buff=6686, seg=a0, sector=301568, result=0
SSD: read blk 24e, buff 6a86, seg 80
sc: cmd=51, arg=302080, n=8, res=0
dr: 0, buff=6a86, seg=80, sector=302080, result=0
sc: cmd=51, arg=302592, n=8, res=0
dr: 0, buff=6a86, seg=a0, sector=302592, result=0
SSD: read blk 250, buff 6e86, seg 80
sc: cmd=51, arg=303104, n=8, res=0
[... 109 lines cut ...]
sc: cmd=51, arg=325632, n=8, res=0
dr: 0, buff=6686, seg=80, sector=325632, result=0
sc: cmd=51, arg=326144, n=8, res=0
dr: 0, buff=6686, seg=a0, sector=326144, result=0
SSD: read blk 27e, buff 6a86, seg 80
sc: cmd=51, arg=326656, n=8, res=0
dr: 0, buff=6a86, seg=80, sector=326656, result=0
sc: cmd=51, arg=327168, n=8, res=0
dr: 0, buff=6a86, seg=a0, sector=327168, result=0
SSD: read blk 240, buff 6e86, seg 80
sc: cmd=51, arg=294912, n=8, res=0
dr: 0, buff=6e86, seg=80, sector=294912, result=0
sc: cmd=51, arg=295424, n=8, res=0
dr: 0, buff=6e86, seg=a0, sector=295424, result=0
#
```
I have a suspicion that the fat16 driver might be the culprit here as you have mentioned before. This happened before, when I used the 10MB limit as well.

For now I think it's fine to keep the filesystem in the raw device, no problems for now.

Now, two questions, when you mentioned to use a static buffer to read into it and then use the fmemcpyw function, how can I achieve this? The sample ssd_test creates a buffer (which is a segment_s), but it always use some fmem*** function to access it. I don't know how I should access the buffer to read into it. In the other case where I should use a local char buffer, then how can I interact to it using fmemcpyw?

And the other question, how can I create an init script that automatically mounts the SD card on boot and then runs the shell? I find really annoying to have to type the mount command everytime.

Thanks!

ghaerr, 2021-10-04:
> I've defined ssd_init to use sector_t (which is unsigned long anyway) and I hardcoded the size of the SSD block to 1.9GB (using the exact byte value reported from my OS), and it still works!

Great! Using sector_t is better than unsigned long, thanks.

> I have a suspicion that the fat16 driver might be the culprit here as you have mentioned before. This happened before, when I used the 10MB limit as well.

Try running `sync`. Everything is happening in kernel buffers until that happens.

> For now I think it's fine to keep the filesystem in the raw device, no problems for now.

Ok - but it would be nice to have a sector_t 'start_offset' (set to 0 for you for now) that could be set to the beginning of a partition so that partitioned devices could work as well for future block device drivers. I'm still wondering why your `hd` desktop command didn't read the full device...  ?

> when you mentioned to use a static buffer to read into it and then use the fmemcpyw function, how can I achieve this? The sample ssd_test creates a buffer (which is a segment_s), but it always use [...]

Does your bit-bang driver ever read more than one byte at a time? That is, can you pass a seg:offset buffer address to it, or does it only read a single character from a sector every time? I haven't seen the source to be able to best help you on this. Ultimately, it would be best if your low-level driver either buffered the sector, or there was a very fast way to call it multiple times with little overhead to get each SSD byte.

> how can I create an init script that automatically mounts the SD card on boot and then runs the shell? I find really annoying to have to type the mount command everytime.

The elkscmd/rootfs_template/etc/rc.d/rc.sys file is waiting and ready to go for you: just add your own mount command in the obvious section and it'll get mounted for you on boot :)

Thank you!

ghaerr, 2021-10-04:
> I hardcoded the size of the SSD block to 1.9GB (using the exact byte value reported from my OS), and it still works!

We're going to need a way to figure out the physical volume size if possible for two reasons: obviously an accurate displayed value would be nice, but also we're going to have to reject devices > 4GB (sizeof unsigned long). ELKS just can't handle larger volumes as it only keeps 32 bits for file and volume sizes throughout the system. However, by using the start_offset described above, we could still support partitioned volumes up to 4GB but on way larger devices, provided your bit-banger reader can address it somehow.

ghaerr, 2021-10-04:
> Now, since my original code did cope with segments and offsets for the read of an entire sector into a buffer, I just copied that code. I've used pokeb() to accomplish so on the byte read loop, as follows:
```
[....]
	pokeb(buff, seg, d); //*buff = d;
	buff++;
[....]
```
```
disk_readp(buf, seg, start, 0, 512);
```
You don't need a separate `fmemcpy` if your low-level driver just reads a full sector directly into the arguments passed to `disk_readp`. We can use a `char __far *` pointer to directly write the characters if your driver is completely in C. If in ASM, then the low level should use the passed segment for all writing directly.

cocus, 2021-10-04:
> Try running `sync`. Everything is happening in kernel buffers until that happens.
> 

You're right!
```
# sync
SSD: WRITE WRITE WRITE blk 240, seg 0
ssddisk:I/O error
dev 200, sector 576
```
Now I need to continue working on the write sector code.

> Ok - but it would be nice to have a sector_t 'start_offset' (set to 0 for you for now) that could be set to the beginning of a partition so that partitioned devices could work as well for future [...]

It didn't read the full device because I capped it to 1024 bytes with `-n 1024` (or 2 sectors). I agree it would be nice to have that value for general purpose usage.

> Does your bit-bang driver ever read more than one byte at a time? That is, can you pass a seg:offset buffer address to it, or does it only read a single character from a sector every time? I haven't [...]

My bit-bang code reads a single byte at a time, that is because the assembly code only returns a byte at a time. I can buffer a single sector locally (or maybe two) if needed, but I'm not sure how to do that, either using a local char buffer, or using the segment_s allocating it once the driver starts.

> We're going to need a way to figure out the physical volume size if possible for two reasons: obviously an accurate displayed value would be nice, but also we're going to have to reject devices > [...]

I have a code that reads the size of the card, but the value I'm getting is a little bit shy of what my computer reports (my code calculates a little bit more space thant my pc). There might be an issue on my code, but it's a start. I can return that size on the ssd_init function.

> You don't need a separate `fmemcpy` if your low-level driver just reads a full sector directly into the arguments passed to `disk_readp`. We can use a `char __far *` pointer to directly write the [...]

Okay, but the driver's read function receives a char* and a seg_t argument. I know I should be able to combine both into a char __far*, but is this the way to do it?
My code does read 1 byte at a time, and right now I'm using pokeb to write the byte into the seg:buffer passed.

Thanks!

ghaerr, 2021-10-04:
> My bit-bang code reads a single byte at a time, that is because the assembly code only returns a byte at a time

Just pass the segment and offset to the bit-bang code in ASM; put the segment into ES and then use a single byte move to write the character directly into the passed segment and offset. Since the segment will stay the same, you can then just increment the offset 512 times directly in your ASM routine.

> I can buffer a single sector locally (or maybe two) if needed

No need - essentially we want to have disk_readp be the entry point to your ASM routine, and be passed three parameters: sector #, segment and offset. It will read 512 bytes into that buffer.

> Okay, but the driver's read function receives a char* and a seg_t argument. I know I should be able to combine both into a char __far*, but is this the way to do it?
> My code does read 1 byte at a time, and right now I'm using pokeb to write the byte into the seg:buffer passed.

If you want get rid of `pokeb`, you can use something like this until you rewrite your low level driver:
```
unsigned char __far *buffer = _MK_FP(0, segment); /* init far pointer to segment:0 */
...
   buffer[offset++] = c; /* write byte c to segment:offset (in loop) */
```

> the value I'm getting is a little bit shy of what my computer reports (my code calculates a little bit more space thant my pc

Could be it's multiplying by 1000 rather than 1024...

ghaerr, 2021-10-04:
> An even faster far pointer loop would be:
```
unsigned char __far *buffer = _MK_FP(offset, segment); /* init far pointer to segment:offset */
...
   *buffer++ = c; /* write byte c to segment:offset (in loop) and increment offset */
```

cocus, 2021-10-04:
> unsigned char __far *buffer = _MK_FP(segment, offset); /* init far pointer to segment:offset */
> ...
>    *buffer++ = c; /* write byte c to segment:offset (in loop) and increment offset */

I had to cast my char* buff into unsigned long for the _MF_FP(), not sure if this is ideal, but that's the type of argument I get from the `ssddev_read_blk` function, as follows:
```
unsigned char __far *buffer = _MK_FP(seg, (unsigned long)buff); /* init far pointer to segment:offset */
[...]

// then used as follows on the loop
*buffer++ = rcvr_mmc();
```

> The elkscmd/rootfs_template/etc/rc.d/rc.sys file is waiting and ready to go for you: just add your own mount command in the obvious section and it'll get mounted for you on boot :)

I've tried this, and the file is on the romfs, and the contents are correct, but I still don't get the mount command to run. Should I enable something on the configuration?

ghaerr, 2021-10-04:
> I had to cast my char* buff into unsigned long for the _MF_FP()

Ok, 'but' is a 'char *' so it needs to be cast to a non-pointer type., Just casting to (unsigned) should be fine and potentially better code.

> I've tried this, and the file is on the romfs, and the contents are correct, but I still don't get the mount command to run.

Ooops - you're not running /bin/init, so the rc.sys file doesn't get run! Running that requires the larger bash shell also, and you've only got /bin/sash (as /bin/sh) on ROMFS. This can be sorted out, but there isn't enough space in the ROM filesystem to fit in addressable memory at the moment. You might be able to use the 'alias' command in /root/.sashrc however. (Read up on that in the sash source or play with it on the desktop first).

cocus, 2021-10-06:
In any case, I have my changes almost ready to go. Please have a look at them before I even open a PR, just to have a green light.

https://github.com/jbruchon/elks/compare/master...cocus:cocus/80188eb-work?expand=1

Thanks!

ghaerr, 2021-10-06:
Hello @cocus,

> In any case, I have my changes almost ready to go. Please have a look at them before I even open a PR, just to have a green light.

I was traveling yesterday. I just took a quick look at your proposed PR, you have been busy! Looks nice and very clean, but I do have a few comments, before you submit, which I will get to a bit later today.

> is there a "microsecond sleep" function available on the ELKS for the block drivers? I was resorting to "nop"s but that's not elegant, nor accurate.

You can schedule a kernel timer as described in https://github.com/jbruchon/elks/issues/960#issuecomment-930628010, but I'm not sure the extra size and complexity is worth the trouble. I'll take a look at your .S file for the NOPs and comment further.

Thank you!

cocus, 2021-10-06:
> I was traveling yesterday. I just took a quick look at your proposed PR, you have been busy! Looks nice and very clean, but I do have a few comments, before you submit, which I will get to a bit later today.

No worries. We can discuss these comments whenever you're free.

> You can schedule a kernel timer as described in [#960 (comment)](https://github.com/jbruchon/elks/issues/960#issuecomment-930628010), but I'm not sure the extra size and complexity is worth the trouble. I'll take a look at your .S file for the NOPs and comment further.

No, I don't think it would be a good idea. And as far as I remember, the timer resolution is 1ms, so a microsecond delay wouldn't be achievable. I'll let the "nop"'s in place. The delay function is on the sd card code, not on the low level (since the low level shouldn't add any delay at all).

Thanks!

ghaerr, 2021-10-07:
Hello Santiago (@cocus),

I have reviewed your SSD SD card driver. Overall, very impressive work and quite well done, thank you!!

Almost all of the C code can probably stay just as you wrote it; my concerns are mostly over file naming conventions and ELKS integration issues, which are pretty small. Here are my thoughts:

- Rename the following files (note, we're trying to move to using hyphen rather than underscore in driver filename options, see character device drivers):
-- ssd_generic_sd.c -> ssd-sd.c
-- ll_8018x_ssd_ssdcard.S -> spi-8018x.S
-- ll_ssd_ssdcard.h -> spi->8018x.h
-- ssd_test.c -> ssd-test.c
- Move ssd_ioctl() function to ssd.c
- Remove CONFIG_BLK_DEV_SSD and CONFIG_BLK_DEV_SSD_SD from Makefile and config.in; instead of four options, just have single 'choice' in config.in choosing between CONFIG_BLK_DEV_SSD_TEST, CONFIG_BLK_DEV_SSD_SD8018X and no options. Keeping things very simple, we only need a single choice selection for SSD Disk Support: 'None' (no options defined), 'Test Driver', and '8018X SD Card' or something similar.

You've done a fantastic job on this. With the above changes in a PR, I will comment on some minor issues for clarity, but we should be able to commit the code as-is.

Thank you!!

cocus, 2021-10-07:
> * Rename the following files (note, we're trying to move to using hyphen rather than underscore in driver filename options, see character device drivers):
>   -- ssd_generic_sd.c -> ssd-sd.c
>   -- ll_8018x_ssd_ssdcard.S -> spi-8018x.S
>   -- ll_ssd_ssdcard.h -> spi->8018x.h
>   -- ssd_test.c -> ssd-test.c

Done!

> * Move ssd_ioctl() function to ssd.c

I'm unsure about this one.

> * Remove CONFIG_BLK_DEV_SSD and CONFIG_BLK_DEV_SSD_SD from Makefile and config.in; instead of four options, just have single 'choice' in config.in choosing between CONFIG_BLK_DEV_SSD_TEST, [...]

Done!

I'll open a PR right now. However, I'll be out until next week's friday, so if there are some changes for the PR, you might push to my PR branch to amend the changes, or wait until friday.

Thanks!

ghaerr, 2021-10-07:
> Move ssd_ioctl() function to ssd.c
>I'm unsure about this one.

My point is that `ioctl` will always be unimplented in the SSD driver, so why not just move the ssd_ioctl entry point into the top-level ssd.c file, that's all.

ghaerr, 2022-01-12:
Hello @cocus,

I'm not sure if you've had any time keep up with some of ELKS recent changes, but #1098 and #1099 introduced the ability for block device sector sizes smaller or larger than 512 bytes. I thought you might be interested that this also applies to the SSD driver, for devices that might not support 512 sector size, FYI.

BTW, I haven't had any chance to actually test your 8018X SSD driver with these changes, but I did test with ssd-test.c, so I'm fairly certain everything will continue to work fine.

Thank you!

cocus, 2022-01-12:
I didn't have enough time to check those, but I can tell you that as long as the 512bytes per sector still work, then the SD driver should still work. I don't recall if there was any means to query the driber for this sector size, were any way to do this?
In any case, most of the old SD cards won't support a sector size other than 512 (and my code doesn't support other size either way).

cocus, 2022-01-16:
UPDATE: I re-built elks with these changes, and the SD driver still works fine!