ghaerr/elks #991: Error Building with CGAF

SonnoOcea, 2021-11-07:
**Description**
- What is expected ?
- What is observed ?

**Configuration**
- Version (commit or release):
- Emulator or HW platform:

**How to reproduce ?**
- Does the problem always happen ?
- List the steps to reproduce the problem.

**Raw data**
- Screenshot(s):
- Log file:

**Additional information**
Add any other context information about the problem here.

ghaerr, 2021-11-08:
Hello @SonnoOcea,

Thanks for the bug report.

The problem is in the file 'elkscmd/cgatext/makefile'. Change the LIBS= line in that file to the following and cgatext should compile:
```
LIBS   = $(ELKS_LIB86)/lib86.a
```
I'll get this fixed in an upcoming PR.

Thank you!

SonnoOcea, 2021-11-08:
Thanks a lot ghaerr!
The edit did indeed work and I had a succesful build,
I'll check later if it works or not on the hardware itself and I'll let you know and close the issue.

SonnoOcea, 2021-11-09:
I tested it on a pentium (overkill, but the v20 xt clone I want to use it on needs maintenance) and **it does work correctly** indeed, thanks again for the help.

For future reference the images built can be used with a gotek by just renaming the file the bin to img (both formats are raw images of floppy disks).

Have a nice day!

[image]
[image]

closed 2021-11-09 by SonnoOcea