ghaerr/elks #996: Network socket close bug

Mellvik, 2021-11-10:
When an application closes a network socket just after having written to it, the socket remains open 'under the hood' for up to 10 seconds. 

This is evidenced by new attempts to open a connection using the same local port number, which gives a `EADDRINUSE` error until some timeout causes a real close.

The test case is the (upcoming) ELKS `ftpd` server and the `MGET` command, which sends a series of files back-to-back back to the client. When programmed to reuse the same port and wait until the `EADDRINUSE` error goes away, it takes between 5 and 10 seconds for that to happen.

If the socket is being read from instead of written to by ELKS, as is the case with the `FTP` `MPUT` command, there is no problem. The socket is being fully closed immediately and can be reopened back-to-back using the same local port number.

The new `ftpd` which turns out to be very helpful provoking the issue, will be posted shortly.

--Mellvik

ghaerr, 2021-11-10:
> This is evidenced by new attempts to open a connection using the same local port number, which gives a EADDRINUSE error until some timeout causes a real close.

I assume you're testing using an ftp client from outside ELKS to your new FTPD server on ELKS.

Is it the external ftp client that is getting the EADDRINUSE error? Is that on the data connection (not command channel)?

Mellvik, 2021-11-10:
Additional comments:
- The timeout before a socket again becomes available is always exactly 10 seconds. The variation mentioned before came from an experiment using 2 different port numbers every other time.
- From the TCP trace, the connection is always properly closed. A FIN which is immediately ACKed.

ghaerr, 2021-11-10:
> From the TCP trace, the connection is always properly closed. A FIN which is immediately ACKed.

There needs to be two FINs, one sent from each side, each ACKed, in order to fully shut the TCP connection down without going into TIME WAIT state. Are you seeing the second ACK? Not sure yet which direction we're talking about.

ghaerr, 2021-11-10:
> When programmed to reuse the same port and wait until the EADDRINUSE error goes away

Can you remind me exactly what is supposed to happen on an FTP transfer (from ftpd to ftp)? Does the client specify a port number on its side that the daemon is supposed to connect to? Is that a reserved port, or is there communication as to a new port number every time? I am trying to understand the binding (and reuse) of the port numbers and don't have the FTP protocol in my head, to debug this problem.

Mellvik, 2021-11-10:
Hi @ghaerr - the problem appears ONLY when ELKS is the sender.
Here's the code that waits for the socket to become available:
```
        while ((bind(fd, (struct sockaddr*) &tempaddr, sizeof(tempaddr))) < 0) {
                if (sockwait++ > 10 || errno != EADDRINUSE) {
                        printf("Bind: Could not connect on port %d\n", server_port-1);
                        perror("bind error");
                        //tempaddr.sin_port   = htons(some_other_port);
                        return -1;
                }
                if (debug) printf(".");
                sleep(1);
                //server_port = 876; /* TESTING - 'random' (!!) #  below 1024 */
                //tempaddr.sin_port   = htons(server_port - sockwait);
        }
        if (debug) printf("\n");
```

BTW - the `if (sockwait++` statement never kicks in.

--M

Mellvik, 2021-11-10:
@ghaerr, the closing of the socket is fine protocol wise. Elks sends a FIN, the client reponds with a FIN that ACKs the first.

```
16:35:09.939738 IP (tos 0x0, ttl 64, id 1699, offset 0, flags [none], proto TCP (6), length 310)
    10.0.2.15.ftp-data > 10.0.2.2.46825: Flags [P.], cksum 0x90bd (correct), seq 11777:12047, ack 1, win 512, length 270
16:35:09.939839 IP (tos 0x8, ttl 64, id 21719, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.46825 > 10.0.2.15.ftp-data: Flags [.], cksum 0x182b (incorrect -> 0x8b16), ack 12047, win 53760, length 0
16:35:09.952273 IP (tos 0x0, ttl 64, id 1700, offset 0, flags [none], proto TCP (6), length 64)
    10.0.2.15.ftp > 10.0.2.2.53536: Flags [P.], cksum 0x686c (correct), seq 860:884, ack 261, win 512, length 24: FTP, length: 24
	226 Transfer Complete.
16:35:09.952342 IP (tos 0x10, ttl 64, id 28995, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.53536 > 10.0.2.15.ftp: Flags [.], cksum 0x182b (incorrect -> 0xc739), ack 884, win 29200, length 0
16:35:09.963545 IP (tos 0x0, ttl 64, id 1701, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp-data > 10.0.2.2.46825: Flags [F.], cksum 0x5b16 (correct), seq 12047, ack 1, win 512, length 0
16:35:09.964447 IP (tos 0x8, ttl 64, id 21720, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.46825 > 10.0.2.15.ftp-data: Flags [F.], cksum 0x182b (incorrect -> 0x8b14), seq 1, ack 12048, win 53760, length 0
```

ghaerr, 2021-11-10:
I can't see from the code fragment what address ftpd is trying to bind to. That is, I am trying to understand exactly which TCP connection (source and dest port numbers) is used, which direction it was started (passive or active open), and how it was closed. I assume the first transfer works, and the bind fails on the second transfer? That would kick in the sockwait++ code, unless the problem is occurring in the active (connect) or passive open (accept).

ghaerr, 2021-11-10:
> the closing of the socket is fine protocol wise. Elks sends a FIN, the client reponds with a FIN that ACKs the first.

Got it, thanks for the trace. So the connection is ftpd's data port (22) to ftp client's port 46825. I can see both FINs, so that's definitely a bug. The bug being that the next ftpd data port connection from 22 to, anything fails since port 22 is still in use on ELKS, correct?

Mellvik, 2021-11-10:
Some FTP stuff:
- PASSIVE is not implemented (yet)
- The command channel is always open (server port 21)
- The data channel is opened and closed per transfer, initiated (since we're in active mode) by a PORT command from the client, indicating the IP address and port number to connect to.
- The server opens the data connection from port 20, although any port will work. I tried to use a new port for every connection and ran out of `ktcp` socket memory.

Mellvik, 2021-11-10:
@ghaerr, 
yes, your observation above is correct (except the port #).

And again, it's always exactly 10 seconds.

M

ghaerr, 2021-11-10:
> yes, your observation above is correct

I see. So FTPD is getting a EADDRINUSE error on the bind statement, even after the double FIN close?

It is possible that the TCP time wait state still needs to happen in the double FIN close case; I think this was the reason that the SO_REUSEADDR was put in - so that servers can bind to the same LOCAL port number even though there is a time wait going on. 

Does the remote ftp client give a new port number to connect to each time? That is, if we were to implement SO_REUSEADDR, would that fix the issue, since the TCP connection would be do a new port number remotely each time?

ghaerr, 2021-11-10:
> I was thinking the same, but then, why does it work the transferring the opposite way?

I'm thinking it's because that direction, the FTP client is binding using SO_REUSEADDR, since in that direction it is using the same port number 20, and the ELKS FTPD is issuing a different port.

> Yes, there is a different port every time. 

Good, that means the fix above should work. If it does, I'll add a SO_REUSEADDR option that can be used to not check for local port number and the bind will work.

ghaerr, 2021-11-10:
Can you show me the code just above the bind statement in your server, that sets the tempaddr fields?

Mellvik, 2021-11-10:
> 10. nov. 2021 kl. 18:03 skrev Gregory Haerr ***@***.***>:
> 
> 
> I was thinking the same, but then, why does it work the transferring the opposite way?
> 
> I'm thinking it's because that direction, the FTP client is binding using SO_REUSEADDR, since in that direction it is using the same port number 20, and the ELKS FTPD is issuing a different port.
> 
> 
No, the procedure is exactly the same for both ways.

—M

Mellvik, 2021-11-10:
I posted the PR now.

—M

closed 2021-11-13 by ghaerr