ghaerr/elks #1012: Data loss using telnet

ghaerr, 2021-11-16:
Opening seperate issue for @Mellvik from discussion in https://github.com/jbruchon/elks/pull/1005#issuecomment-970133803:

If you telnet to elks from the mac (or whatever client), then telnet back to the same machine (or a different one, which becomes somewhat more complicated with the QEMU config), not even a 'ls -l' will come out as it should.

So it seems we're getting closer to the telnet problem thanks to the recent networking updates  - which still may well be in the pty driver … The trace below shows the test. Notice the 'helge$' at the end of the last line - which is part of the shell prompt after ls completed. The second output is what the 'ls' should have looked like.

```
www9:~ helge$ telnet localhost 2323
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
login: root
# telnet 10.0.2.2
Connecting to 10.0.2.2 (10.0.2.2:23)
Connected

Darwin/BSD (www9.local) (ttys006)

login: helge
Password:
Last login: Tue Nov 16 09:59:58 on ttys000
www9:~ helge$ ls -l
total 17743832
drwx------    4 helge  staff         128 Nov 18  2015 Applications
drwxr-xr-x    4 helge  staff         128 Dec 23  2011 Beth Hart
drwxr-xr-x    4 helge  staff         128 Jul 30 22:19 Chris Botti
drwxr-xr-x    8 helge  staff         256 Oct  7 11:53 Consensys
drwx------@ 277 helge  staff        8864 Oct 22 14:20 Desktop
drwx------@  91 helge  staff        2912 Nov 16 09:40 Documents
-rw-------    1 helge  staff    25000448 Oct 17  2019 DosTest [6.22].ova
drwx------+   6 helge  staff         192 Jul 31 12:45 Downloads
drwx------@  71 helge  staff        2272 Apr  8  2021 Library
drwx------+  14 helge  staff         448 Jan 11  2020 Movies
drwx------+   9 helge  staff         288 Oct  7 11:53 Music
drwx------+   3 helge  staff          96 Nov 16  2014 Pictures
drwxr-xr-x+   5 helge  staff         160 Nov 16  2014 Publhelge$

——————————————————————————————————————————————

www9:~ helge$ ls -l
total 17743832
drwx------    4 helge  staff         128 Nov 18  2015 Applications
drwxr-xr-x    4 helge  staff         128 Dec 23  2011 Beth Hart
drwxr-xr-x    4 helge  staff         128 Jul 30 22:19 Chris Botti
drwxr-xr-x    8 helge  staff         256 Oct  7 11:53 Consensys
drwx------@ 277 helge  staff        8864 Oct 22 14:20 Desktop
drwx------@  91 helge  staff        2912 Nov 16 09:40 Documents
-rw-------    1 helge  staff    25000448 Oct 17  2019 DosTest [6.22].ova
drwx------+   6 helge  staff         192 Jul 31 12:45 Downloads
drwx------@  71 helge  staff        2272 Apr  8  2021 Library
drwx------+  14 helge  staff         448 Jan 11  2020 Movies
drwx------+   9 helge  staff         288 Oct  7 11:53 Music
drwx------+   3 helge  staff          96 Nov 16  2014 Pictures
drwxr-xr-x+   5 helge  staff         160 Nov 16  2014 Public
-rw-r--r--@   1 helge  staff       50953 May  7  2020 Skattemelding2019tore.pdf
-rw-r--r--@   1 helge  staff       92452 May  7  2020 Skatteoppgjørkari2019.pdf
drwxr-xr-x    5 helge  staff         160 Oct 17  2019 VirtualBox VMs
-rw-r--r--    1 helge  staff  8060338437 Jul 20  2019 helge-hdsound.tjz
-rw-r--r--    1 helge  wheel      174188 Nov 20  2014 itunes.bad
-rw-r--r--    1 helge  staff       24945 Aug  5  2020 ne2k-mac.S
-rw-r--r--    1 helge  staff   910174999 Sep  3  2019 raspi-radio-full.tjz
drwxr-xr-x    9 helge  staff         288 Nov 10 14:42 src
drwxr-xr-x    6 helge  staff         192 Nov 11 10:20 tmp
-rw-r--r--    1 helge  staff    21411840 Feb  6  2016 venix-full.img
-rw-r--r--    1 helge  staff    21411840 Feb 18  2016 venix-hd.img
drwx------   13 helge  staff         416 Mar  5  2016 venix21
-rwxr-xr-x    1 helge  staff    15974400 May 16  2017 venix86hd.chd
-rwxr-xr-x    1 helge  staff    11161600 Feb  5  2016 venixhd.img
www9:~ helge$
```

ghaerr, 2021-11-16:
Geez @Mellvik, just when I thought we'd crushed all the TCP bugs you've gone and shown we still have another!

I'm wondering whether this problem occurs only when the 2nd session is running from within another telnet, or whether it would occur with the 2nd telnet from ELKS to remote is started from another console or terminal. That would help determine if the problem is related to mixing packets within ktcp, or perhaps a PTY related issue.

I'll try to duplicate on my side, and squash this (last?) known networking issue.

Mellvik, 2021-11-16:
Well, sorry to disappoint you @ghaerr — I may have yet another one.

But first - I've only been able to reproduce this telnet problem when running two telnets. I'll try other variants again - we do have a different scenario now, so that makes sense.
And - since the problem is repeatable on QEMU, it eases debugging a lot.

Even if the scenario provoking the problem is indeed an odd one, I guess the problem may manifest it self in the most unexpected ways in other contexts. So it's worth chasing.

Now, while debugging the ftpd PASSIVE mode, it ran into a new problem when transferring files back to back - in active mode. It seems ktcp under certain circumstances gets confused because the file operations are too slow, the client is too impatient, banging in with retransmits and ktcp crashes. I'll open a new issue with details.

Oh well….

— Mellvik

PR #1021 (2021-11-22): [kernel] Fix dropping data in PTY
Fixes problem associated with overrunning output buffer on TTY writes, notably associated with PTY driver.

Fixes #1012, problem duplicated on QEMU.
closed 2021-11-22 by ghaerr