ghaerr/elks #1013: [ktcp] Another tcp retransmit issue

Mellvik, 2021-11-16:
Like you said @ghaerr, 
the sweat from the previous one has just dried out and we have another one (or two) on our hands.

The problem I just discovered is that `ktcp` some times fail to remove acked packets from the queue,  which down the line creates havoc. This one is somewhat hard to reproduce because it is timing related, but essentially - here's the summary from the packet trace:
We have a series of ftp retreive operations (the ELKS ftp server sends multiple files back to the client) going (MGET).  The files are small and ELKS buffering makes transfer of several files back to back fast even on a floppy based system. Then ELKS needs to get more data, which takes some time, but not necessarily more than a second or so.

In summary - everything works fine until a retransmit timer kicks off. Then ELKS starts retransmitting packets that have been acked, which eventually sends us into an endless loop.

So here we go starting with the end of a file transfer - packets are D for data channel, C for command channel, E for server, K for client  origin. The problem starts at C10 - the first ELKS retransmit:
- C1E: ELKS detects end of file, sends 'Transfer Complete' to the client via the CMD channel.
- D2E: ELKS sends FIN on the data channel
- D3K: Client reponds with a FIN on the data channel, ACKs the FIN from ELKS
- C4K: Client ACKs C1 from ELKS.
- C5K: The client sends a new **PORT** command on the data channel, preparing for transfer of the next file
- D6K: The client notices that its FIN packet (data channel) has not been ACKed, resends D2 
- C7K: Another 0.3 sec and the client finds the PORT command un-ACKed, resends C5
- D8K: Another 0.4 secs, the FIN on the data channel has still not been ACKed, clients resends D2 again
- C9K: The PORT command on the CMD channel i still not ACKed, so the clients resends C5 again
- C10E: It has been 1.1 secs since P1, ELKS is back - on the CMD channel - with resend of C1, which was ACKed by C4!
- C11K: The client ACKs C10 again anyway
- C12E: ELKS now seems thoroughly confused, resends C-10 (minus 10), the CMD channel packet that started the now completed file transfer: 'Opening Binary Connection ....'. It's not as bad as it seems. This is outgoing packet -1 on the CMD channel, the one before C1. What's bad is that this packet was ACKed by the client in packet C-4.
- C13K: Is a resend of C11, again ACKing C10 (which is a duplicate of C1)
- D14K: is a repeat of D8, yet another ACK of D2
- C15K: Repeats the PORT command (C5, C9) again
- D16E: Elks resends D2, the FIN on the data channel
- D17K: Client ACKs D16, this time without including the FIN flag
- C18E: Elks repeats C1, 'Transfer Complete' - just like C10
- C19K: Client ACKs C18 
- D20E: Elks resends D-2 (minus 2), which was previously ACKed by D-1K.
- D21K: Clients ACKs D20 again
- D22E: Elks resends an even older packet, D-4 (the oldest packet in the retransmit buffer?), which was previously ACked by D-3.
- D23K: Client repeats the ACK in D21, which is ahead of D-4 anyway.
- C24E: ELKS resends C12 one more time
- C25K: The client acknowledges not C24 (really C12), the newer C18 (this ACK is a duplicate of C19K).
- D26K: The client then repeats the FIN packet D2.
- C27E: ... While ELKS repeats C1
At this point we have a loop - ELKS keeps resending old and ACKed packets, the client keeps ACKing them.
The packet trace has been included  for reference. The first packet in the trace is P-4 from the description. The console output has also been included, unfortunately no ^P.

```
14:17:39.077446 IP (tos 0x0, ttl 64, id 1752, offset 0, flags [none], proto TCP (6), length 516)
    10.0.2.15.ftp-data > 10.0.2.2.45505: Flags [P.], cksum 0x3e66 (correct), seq 1025:1501, ack 1, win 4380, length 476
14:17:39.077559 IP (tos 0x8, ttl 64, id 4882, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.45505 > 10.0.2.15.ftp-data: Flags [.], cksum 0x182b (incorrect -> 0xd1e9), ack 1501, win 32160, length 0
14:17:39.092041 IP (tos 0x0, ttl 64, id 1753, offset 0, flags [none], proto TCP (6), length 540)
    10.0.2.15.ftp-data > 10.0.2.2.45505: Flags [P.], cksum 0x3c72 (correct), seq 1501:2001, ack 1, win 4380, length 500
14:17:39.092101 IP (tos 0x8, ttl 64, id 4883, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.45505 > 10.0.2.15.ftp-data: Flags [.], cksum 0x182b (incorrect -> 0xcbc5), ack 2001, win 33232, length 0
14:17:39.099223 IP (tos 0x0, ttl 64, id 1754, offset 0, flags [none], proto TCP (6), length 64)
    10.0.2.15.ftp > 10.0.2.2.53676: Flags [P.], cksum 0x999e (correct), seq 12962:12986, ack 4501, win 4380, length 24: FTP, length: 24
	226 Transfer Complete.
14:17:39.106168 IP (tos 0x0, ttl 64, id 1755, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp-data > 10.0.2.2.45505: Flags [F.], cksum 0x3c79 (correct), seq 2001, ack 1, win 4380, length 0
14:17:39.106357 IP (tos 0x8, ttl 64, id 4884, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.45505 > 10.0.2.15.ftp-data: Flags [F.], cksum 0x182b (incorrect -> 0xcbc3), seq 1, ack 2002, win 33232, length 0
14:17:39.106492 IP (tos 0x10, ttl 64, id 30185, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.53676 > 10.0.2.15.ftp: Flags [.], cksum 0x182b (incorrect -> 0x0788), ack 12986, win 29200, length 0
14:17:39.107050 IP (tos 0x10, ttl 64, id 30186, offset 0, flags [DF], proto TCP (6), length 61)
    10.0.2.2.53676 > 10.0.2.15.ftp: Flags [P.], cksum 0x1840 (incorrect -> 0xd87b), seq 4501:4522, ack 12986, win 29200, length 21: FTP, length: 21
	PORT 10,0,2,2,208,3
14:17:39.306846 IP (tos 0x8, ttl 64, id 4885, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.45505 > 10.0.2.15.ftp-data: Flags [F.], cksum 0x182b (incorrect -> 0xcbc3), seq 1, ack 2002, win 33232, length 0
14:17:39.326821 IP (tos 0x10, ttl 64, id 30187, offset 0, flags [DF], proto TCP (6), length 61)
    10.0.2.2.53676 > 10.0.2.15.ftp: Flags [P.], cksum 0x1840 (incorrect -> 0xd87b), seq 4501:4522, ack 12986, win 29200, length 21: FTP, length: 21
	PORT 10,0,2,2,208,3
14:17:39.716829 IP (tos 0x8, ttl 64, id 4886, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.45505 > 10.0.2.15.ftp-data: Flags [F.], cksum 0x182b (incorrect -> 0xcbc3), seq 1, ack 2002, win 33232, length 0
14:17:39.756825 IP (tos 0x10, ttl 64, id 30188, offset 0, flags [DF], proto TCP (6), length 61)
    10.0.2.2.53676 > 10.0.2.15.ftp: Flags [P.], cksum 0x1840 (incorrect -> 0xd87b), seq 4501:4522, ack 12986, win 29200, length 21: FTP, length: 21
	PORT 10,0,2,2,208,3
[... 126 lines cut ...]
tcp retrans: seq 12962 size 24 rcvwnd 29200 unack 82 rto 32 rtt 1 (RETRY 3 cnt 5
 mem 1158)
tcp retrans: seq 12904 size 58 rcvwnd 29200 unack 82 rto 32 rtt 1 (RETRY 3 cnt 5
 mem 1158)
tcp retrans: seq 12962 size 24 rcvwnd 29200 unack 82 rto 64 rtt 1 (RETRY 4 cnt 5
 mem 1158)
tcp retrans: seq 12904 size 58 rcvwnd 29200 unack 82 rto 64 rtt 1 (RETRY 4 cnt 5
 mem 1158)
tcp retrans: seq 2001 size 0 rcvwnd 31088 unack 977 rto 64 rtt 14 (RETRY 2 cnt 5
 mem 1158)
tcp retrans: seq 1501 size 500 rcvwnd 31088 unack 977 rto 64 rtt 14 (RETRY 2 cnt
 5 mem 1158)
tcp retrans: seq 1025 size 476 rcvwnd 31088 unack 977 rto 64 rtt 14 (RETRY 2 cnt
 5 mem 1158)
tcp retrans: seq 12962 size 24 rcvwnd 29200 unack 82 rto 64 rtt 1 (RETRY 5 cnt 5
 mem 1158)
tcp retrans: seq 12904 size 58 rcvwnd 29200 unack 82 rto 64 rtt 1 (RETRY 5 cnt 5
 mem 1158)
tcp retrans: seq 12962 size 24 rcvwnd 29200 unack 82 rto 64 rtt 1 (RETRY 6 cnt 2
 mem 122)
tcp retrans: max retries exceeded seq 12962 unack 82 time 64
tcp retrans: seq 12904 size 58 rcvwnd 29200 unack 82 rto 64 rtt 1 (RETRY 6 cnt 1
 mem 78)
tcp retrans: max retries exceeded seq 12904 unack 82 time 64
```

ghaerr, 2021-11-16:
Good gosh, this seems complicated! I have a few questions to ask, before trying to dive into it:

- How is your testing or setup different from when you said that everything was working perfectly?
- What is the actual set of commands used to reproduce, and can it be done on QEMU?
- Is this using any new code in the ftpd server or ftp client? If so, what is changed?

I am wondering why we lost any packets in the first place. We need to try to find that out, so we can then understand whether this was caused by a packet being dropped on one side or the other, which then caused a retry.

Do you think that the reason for the packet retransmit is not that packets were dropped, but because ktcp timed out on its retrans timer? If so, we can try increasing ktcp TCP_RETRANS_MAXWAIT, which is currently set to 4 seconds.

I notice there is some discussion on a FIN that didn't get ACK'd - is that from a previous file transfer? If so, perhaps this problem is a result of the SO_REUSEADDR. (That should only be a problem if both local and remote port numbers are identical for each transfer).

Finally, recompile ktcp with DEBUG_RETRANS set to 1, and then ^P will show exactly when the retrans packets are being discarded by ktcp. Ktcp only discards them when the ACK received is greater than what was previously sent. That's why this bug is pretty strange. Note that ktcp subtracts the initial sequence number, unlike your packet tracer, so that the "ack" and "unack" values are smaller and readable. It makes them harder to match with your trace though.

Mellvik, 2021-11-16:
@ghaerr,
I believe this must be a minor thing with big(ger) side effects.
> Good gosh, this seems complicated! I have a few questions to ask, before trying to dive into it:
> 
> How is your testing or setup different from when you said that everything was working perfectly?
It's not. I ran perfect transfers before and after.
> What is the actual set of commands used to reproduce, and can it be done on QEMU?
I seriously doubt this can be recreated, it's very timing-sensitive.
> Is this using any new code in the ftpd server or ftp client? If so, what is changed?
No changes, we're using active connection mode.
> I am wondering why we lost any packets in the first place. We need to try to find that out, so we can then understand whether this was caused by a packet being dropped on one side or the other, [...]
I think this is close to target.  I'm only seeing the wire, not what elks actually get. Still I don't think anything was lost, but that a delay caused by floppy access disturbed the flow and revealed a state sensitivity we haven't  seen before. I need to look closer at what it takes to trigger the situation tomorrow. I'm pretty sure I can do it via other activities on the machine while the transfer is going on.
My theory (which admittedly has holes) is that the incoming packets are piling up while elks is doing floppy i/o. Then when the retransmit timer kicks in, ktcp starts retransmitting immediately instead of processing the pending acks first. Increasing MAXWAIT may help, but we're nowhere close to 4 seconds. The hole in the theory is that the oldest data packets - from before the assumed floppy activity, got immediately acked. Why didn't those acks make it into the stack?
> I notice there is some discussion on a FIN that didn't get ACK'd - is that from a previous file transfer? If so, perhaps this problem is a result of the SO_REUSEADDR. (That should only be a problem if both local and remote port numbers are identical for each transfer).
> 
No, this is not an issue.
> Finally, recompile ktcp with DEBUG_RETRANS set to 1, and then ^P will show exactly when the retrans packets are being discarded by ktcp. Ktcp only discards them when the ACK received is greater than [...]
Ok, that will be the next step. Stay tuned.

- Mellvik

ghaerr, 2021-11-17:
Hello @Mellvik,

> Still I don't think anything was lost, but that a delay caused by floppy access disturbed the flow and revealed a state sensitivity we haven't  seen before. 

Yes, I think that is likely what is happening. And it's not a big deal, but upsets the whole apple cart and things spiral out of control...

> My theory (which admittedly has holes) is that the incoming packets are piling up while elks is doing floppy i/o. Then when the retransmit timer kicks in, ktcp starts retransmitting immediately instead of processing the pending acks first.

Yes, a definite possibility:

I have (finally) read through your detailed report. Thank you for all that work, I can see that line C10E, where ELKS retransmits a packet, after for sure having received an ACK on it, is definitely in error. Why would that be? I am guessing the likely place to look is in ktcp's main loop, where I can see a case where it may be possible that after disk I/O, caused by a received packet, ktcp may still be in the middle of its main loop for an extended period, and then return, only to start processing the retransmit queue, rather than say, processing another packet. Another simultaneous possibility is that the timer countdown is not accurate, for reasons unknown.

> The hole in the theory is that the oldest data packets - from before the assumed floppy activity, got immediately acked. Why didn't those acks make it into the stack?

I'm not quite following you here. Are you talking about oldest received data packets sent by the client to ELKS server, and ELKS then ACKed them? I am not seeing that. If you're talking about the oldest data packets sent by ELKS got ACKed by the client, then that could happen - the ACKs sent by the client may not even been read by ktcp from the ethernet driver yet.

Here is the ktcp main loop:
```
while (1) {
        if (tcp_timeruse > 0 || tcpcb_need_push > 0 ||
            cbs_in_time_wait > 0 || cbs_in_user_timeout > 0) {. <--- Section 1

            /* don't wait long if data needs pushing to tcpdev */
            if (tcpcb_need_push) {
                timeint.tv_sec  = 0;
                timeint.tv_usec = 1000; /* 1msec */
            } else {
                timeint.tv_sec  = 1;
                timeint.tv_usec = 0;
            }
            tv = &timeint;
        } else {
            tv = NULL;          /* no timeout if no timers active or push needed */
        }

        FD_ZERO(&fdset);   <--- Section 2
        FD_SET(intfd, &fdset);
        FD_SET(tcpdevfd, &fdset);
        count = select(intfd > tcpdevfd ? intfd + 1 : tcpdevfd + 1, &fdset, NULL, NULL, tv);
        if (count < 0) {
                if (errno == EINTR)
                        continue;
                printf("ktcp: select failed errno %d\n", errno);
                return;
        }

        Now = timer_get_time(); <--- Section 3

        /* expire timeouts*/
        if (cbs_in_time_wait > 0 || cbs_in_user_timeout > 0) {. <--- Section 4
            debug_tcp("tcp: time_wait %d user_timeout %d\n",
                cbs_in_time_wait, cbs_in_user_timeout);
            tcpcb_expire_timeouts();
        }

        /* always push data*/
        //if (tcpcb_need_push > 0)
            tcpcb_push_data();  <--- Section 5

        /* process received packets*/
        if (FD_ISSET(intfd, &fdset)) { <--- Section 6
                if (linkprotocol == LINK_ETHER)
                        eth_process();
                else slip_process();
        }

        /* process application socket actions*/
        if (FD_ISSET(tcpdevfd, &fdset)) <--- Section 7
                tcpdev_process();

[... 1 lines cut ...]
        if (tcp_timeruse > 0) <--- Section 8
                tcp_retrans();
}
```
A brief explanation:
- 1 - The first section sets either a 1ms, 1s or no timeout for the select, based on what is going on.
- 2 - Select. This returns with the file descriptors that have I/O ready: /etc/tcpdev (for kernel read/write to application), /etc/eth (packet received)
- 3 - The current time is taken.
- 4 - The socket time_wait and close_wait timers are checked for expiration, and control blocks possibly freed.
- 5 - Any received TCP data that needs to be sent to an application is pushed by writing /etc/tcpdev.
- 6 - Read a SINGLE ethernet packet and process it.
- 7 - Read a SINGLE /etc/tcpdev request. (Application read/wite, socket change, bind, connect, etc).
- 8 - Compare the time taken in (3) with EVERY retransmit packet, and retransmit as many as might be necessary. (This would normally be just one, since the time would likely be different on each).

So, it is easy to see that a retransmit packet could be sent while several ethernet packets have not even been read from the device driver yet, especially if the ethernet device driver queues packets and the main loop is slowed down for a long period in one of the earlier sections.

There's a couple ways we might solve/test this, one would be to keep running the main loop while any packets are ready to be processed, and not process any retrans packets until all received packets are read. The other would be to loop inside eth_process, until read returns -EAGAIN from the driver. I think the former would be better, as that would allow received data to be pushed to the application in between reading packets, then when no more are available, check the retrans list.

[... cut ...] e that does this.

> Increasing MAXWAIT may help, but we're nowhere close to 4 seconds. 

You may want to recompile ktcp with TCP_RETRANS_MAXWAIT set to (10<<4), which is 10 seconds, after testing with the new main loop, just to see what happens. When you run with DEBUG_RETRANS and ^P, we should be able to determine if we also have a timer countdown problem as well. If your packet trace is showing retransmit activity in way less than 4 seconds, then there seems to be a timer problem.

Thank you!

Mellvik, 2021-11-17:
Thank you much @ghaerr - 
I'm on this now - needed to take the day 'off' to deliver pending projects.

Feedback later today or tomorrow.

—M

Mellvik, 2021-11-17:
@ghaerr,
Thank you for a very thorough (and educating) analysis and discussion. That really made the hour studying packet traces worthwhile. And I think your diagnosis and cure will most likely not only solve the problem, but also improve the overall network response.
I have a couple of comments and question that will have to wait till tomorrow.

I only managed a little bit of testing this afternoon, confirming the expected: That this is a VERY hard (time sensitive) problem to repeat. Which makes it only harder. OTOH all the banging on ftp etc really pushes the implementation (ktcp) - and the retransmit 'curcuitry' in particular, which is good.

To be continued.

-M

Mellvik, 2021-11-18:
Thanks again @ghaerr - 
while I'm trying to find a scenario that makes the problem reliably repeatable - some clarifications and questions.

> Yes, a definite possibility:
>
> I have (finally) read through your detailed report. Thank you for all that work, I can see that line C10E, where ELKS retransmits a packet, after for sure having [...]

It is interesting - maybe worrisome - that the time between the first retransmission from ELKS, Packet 10, and the original packet, P1, is 1 second (on the command channel).
Further - the second retransmission from ELKS (P12) is of an older packet (seq12962), the one before P1 (seq12986) on the command channel. I don't have that part of the trace anymore, but the timing between them is between 1.5 and 2 seconds.
This leaves 2 questions: Why the short timeout (compared to 4 seconds) and why the newer packet first?

> I'm not quite following you here. Are you talking about oldest received data packets sent by the client to ELKS server, and ELKS then ACKed them? I am not seeing that. If you're talking about the [...]

I should have been more specific on this one. What I'm referring to is P22 in the trace, which resends ELKS data packet seq#1501  (Packet # minus 4), which was ACKed by P-3, then by P-1 (ACK#2001) and then by P3, P6, P8, P14, P17.

I guess this reinforces the impression that ktcp is retransmitting without reading pending incoming packets.

> Here is the ktcp main loop:
>
> ```
> while (1) {
>         if (tcp_timeruse > 0 || tcpcb_need_push > 0 ||
>             cbs_in_time_wait > 0 || cbs_in_user_timeout > 0) {. <--- Section 1
>
>             /* [...]

This is very educating - I'm following you and agreeing, that's what it looks like.

> There's a couple ways we might solve/test this, one would be to keep running the main loop while any packets are ready to be processed, and not process any retrans packets until all received packets [...]

I agree, the former must be better/safer. Then again - if ktcp always prioritizes incoming over outgoing, what happens when there are many connections going, say 2 FTPs and a telnet, wouldn't that adversely affect the others? I guess ktcp doesn't handle URG packets at this time? Can we distinguish between connections in the main loop?

> To do this, we'll just keep a simple variable as to whether any ethernet or socket actions have been processed, and run the main loop again until there are none, before running the retrans list. [...]

OK, I'll run the proposed changes even before I've found a reliable scenario to repeat the problem. Given the logic of your discovery, these fixes are required regardless, we just need a setting to test them.

--Mellvik

ghaerr, 2021-11-18:
>  if ktcp always prioritizes incoming over outgoing, what happens when there are many connections going, say 2 FTPs and a telnet, wouldn't that adversely affect the others? 

That can all get pretty complicated. This PR doesn't directly change any ordering of reading from the select file descriptors. The intent was to skip *only* the retransmit queue check until all file descriptors (packets and sockets) were handled.

As mentioned above, for reasons I don't understand, this change required increasing the ktcp heap space for fast back-to-back transfers, and should not have, it would seem. However, it does very slightly change the timeout when data is required to be pushed to the application from 1ms to 0, I notice now. That could possibly change the timing. I will look further into that.

> I guess ktcp doesn't handle URG packets at this time? Can we distinguish between connections in the main loop?

No and no.

>  I'll run the proposed changes even before I've found a reliable scenario to repeat the problem. Given the logic of your discovery, these fixes are required regardless, we just need a setting to test them.

The best approach would be to use the old code and try to get a duplication, even if it takes 10 boots/retries, for instance. Then, run the new code and if it doesn't happen, we can likely assume that this fixes that problem.

I'm almost certain that you've identified a problem where retransmits are occurring after ACKs have been received. The only way this could happen is if the packets never got read, as the retransmit queue is expired (i.e., retransmits removed earlier than the last ACK received) every ktcp main loop cycle.

There could also be a timer/timing problem, as we're seeing some unexplained behavior with times much less than 4 seconds (I still recommend changing the retransmit timeout back to 10 seconds so that the values are quite a bit larger).

To summarize:
- Test with old code and get to repeat somehow
- Decrease retrans timeout to 1-2 seconds to get easier repeat, if needed
- Test with new code (even with lower retrans timeout, if needed) to see if problem solved

I will look into trying to understand why a larger heap is required, by changing the select timeout code as described above.

Finally, if we can't get this to repeat, it is probably worth committing this fix, since we're almost certain there is a problem, and the old behavior can be easily reverted by not setting the two "loopagain = 1" statements.

Mellvik, 2021-11-18:
Thank you @ghaerr - 

> To summarize:
> Test with old code and get to repeat somehow
> Decrease retrans timeout to 1-2 seconds to get easier repeat, if needed
> Test with new code (even with lower retrans timeout, if needed) to [...]

Agreed - I'll continue this tomorrow, based on the message you sendt while I was writing this!

—Mellvik

ghaerr, 2021-11-18:
> Agreed - I'll continue this tomorrow, based on the message you sendt while I was writing this!

I'll look into the FTP PASV issues you've posted, but we should keep all that completely separate, including any changes to ftpd, until we solve this issue, if possible. The PR 1014 fix for this is now pared down to minimal modifications and repeatable main loop behavior (until a retransmit occurs) so it will hopefully be simple to confirm, once you find a way to reliably repeat the original problem. Thank you!

closed 2021-11-21 by ghaerr

Mellvik, 2021-11-21:
Thank you, @ghaerr -
I've made serious efforts to recreate this problem w/o success. Given that the problem has been thoroughly investigated, understood, and presumably fixed, it makes sense to close it. 

Should it pop up again, which I seriously doubt- we'll take it from there.
From my point of view- another one bites the dust!

-M

PR #1014 (2021-11-21): [net] Process all received packets and sockets before retransmits
Attempts to fix #1013.

@Mellvik,

This is pretty straightforward, except a complication where we have to expire the retransmit list every cycle or ktcp runs out of memory from packet processing. The main loop won't process actual retransmits required until all socket and packets waiting to be processed have completed.

I tested this and it worked well, except ktcp started running out of memory when running the fast file transfer test I use with back-to-back ftpget/ftpput's. I'm not sure why this is, but increased ktcp's heap to [... cut]