ghaerr/elks #1016: [net][ftpd] Challenges with FTP passive mode

Mellvik, 2021-11-18:
While working on **passive mode** for `ftpd` - which is taking a lot longer than anticipated, I've come across a number of issues that need to be discussed.

First of all - some background.
FIle transfers with `ftp` come in two flavours. FTP operates via two channels or connections: The command channel which is open for the duration of the session and passes commands and status responses back and forth, usually between an interactive client and a server. 
When the client wats to send or receive data, a data channel is opened for that particular transfer, normally a file or a directory listing. In the old days, when the Internet was 'open', the client would tell the server
'I have opened a socket on address/port a,b,c,d,x,y' where a,b,c,d is the IP address in 4 decimal numbers and 'x,y' is the port number, x*256+y. All in ASCII, easy to see, send, decode etc. This is know as **active mode**: The server connects back to the client for a transfer. When the transfer is complete (or has an error), the data channel is closed. If we're transferring 100 or 10.000 files, a new connection is opened per file. Simple, if not very efficient. 

Active mode works fine in a LAN environment, but not on the Internet: In todays NAT-oriented internet, the client is normally hidden behind a NAT router, possibly a firewall, and connecting to it from the outside is - or should be - impossible. Thus **passive mode**, which is just like active mode, but the opposite. The sever provides the client with a point to connect to for the data transfer:
'I've opened a socket on address/port a,b,c,d,x,y, ready for you to use' and the client initiates the connection. No problems with NAT or firewalls - if the port number is acceptable, which it normally is, but that's a different story.

When opening sockets on Unix/Linux systems, probably others too, a program/server normally leaves to the OS to add local addresses, some times also port numbers for the connection. This is convenient for a number of reasons, relieving the application from keeping track of interfaces and their addresses. On the other hand, in some cases the program needs to find out what address (and port #) the OS assigned to a connection. Such as with  our Passive Mode, where we're supposed to tell the client in clear text where to connect - address and port.

On most  systems, this is easy - such as calling `getsockname` on the socket.  But ELKS does not have this function. How do we implement PASV (technical name for Passive Mode)? Either we implement `getsockname`, which I discussed with @ghaerr and turns out to be a major effort in both code and time, use hard-coding (not really acceptable) or use the `/etc/hosts` file and the `in_gethostbyname` library routine added by @ghaerr last year.

We decided on the latter - for now. The primary reason a `getsockname` implementation would be expensive is that ELKS has  a userspace TCP implementation. `ktcp` runs as a process and sending specific datasets back and forth between the kernel and the process requires an API - and code. More code, more RAM, ...

Thus `/etc/hosts` becomes an active part of ELKS networking, and we may just (suggestion) start using it with `ktcp` start-up: Instead of specifying IP addresses in the startup files, use names. Which will make life easier when we have several ELKS systems on the same network, which will happen real soon now.

As alluded to above, there have been other surprises when working on passive mode as well. In order for these issues not to become buried under a long rant like this, I'll put them in separate posts.

-M

Mellvik, 2021-11-18:
I'm running into a number of (probably related) snags when running FTP in passive mode. 

I seem to be able to reuse the socket (at least the port number), but the resources don't get unallocated.
So - if I repeat the same FTP (DIR in this case) command 4 times (PASV mode), the 4th will fail because `ktcp` runs out of memory - as evidenced by the following console output - first the netstat after the 3rd command, then the debug printout and finally the C-code:

```
 No        State    RTT lport        raddress  rport
-----------------------------------------------------
  1  ESTABLISHED 1000ms  1025         0.0.0.0      2
  2       LISTEN 1000ms 49821         0.0.0.0      0
  3       LISTEN 1000ms 49821         0.0.0.0      0
  4       LISTEN 1000ms 49821         0.0.0.0      0
  5  ESTABLISHED  562ms    21        10.0.2.2  53882
  6       LISTEN 1000ms    21         0.0.0.0      0
  7   CLOSE_WAIT 1000ms 49821        10.0.2.2  49374 (-1074.3 secs)
  8       LISTEN 1000ms    80         0.0.0.0      0
  9       LISTEN 1000ms    23         0.0.0.0      0
```

```
New Client Detected...
Child running.
cmd: 'SYST' 29
cmd: 'PASV' 9
tcp: port 49821 reused, can't free previous socket in state 7   <--- from the previous instance of ftpd
Accepted connection from 6b6c6500 on port 29547, fd 6 <-- this is data returned from the accept() call, usually but not always 0
cmd: 'LIST' 26
LIST/NDIR: '/bin/ls -l' (6)
121
cmd: 'PASV' 9
tcp: port 49821 reused, can't free previous socket in state 1
Accepted connection from 0 on port 0, fd 7
cmd: 'LIST' 26
LIST/NDIR: '/bin/ls -l' (7)
121
cmd: 'PASV' 9                         <---- third run
tcp: port 49821 reused, can't free previous socket in state 1
Accepted connection from 0 on port 0, fd 8
cmd: 'LIST' 26
LIST/NDIR: '/bin/ls -l' (8)
121
cmd: 'PASV' 9
tcp: port 49821 reused, can't free previous socket in state 1
sys_brk(11) fail: brk e6cc over by 2878 bytes
sys_brk(11) fail: brk dcce over by 320 bytes
sys_brk(11) fail: brk dcd0 over by 322 bytes
sys_brk(11) fail: brk dcd2 over by 324 bytes
sys_brk(11) fail: brk db92 over by 4 bytes
sys_brk(11) fail: brk ee50 over by 4802 bytes
sys_brk(11) fail: brk e452 over by 2244 bytes
sys_brk(11) fail: brk df52 over by 964 bytes
...
```

Here's the code implementing the PASV command:
```
/* Passive mode: Server listens for incoming data connection */
int do_pasv(int controlfd, int *datafd) {
        int fd, i = 1, port = PASV_PORT;
        char *p, *a;
        char str[50], *pasv_err = "425 Can't open passive connection";
        struct sockaddr_in pasv, ss;

        if ((fd = socket(AF_INET, SOCK_STREAM, 0)) < 0) {
                write(controlfd, pasv_err, strlen(pasv_err));
                if (debug) printf("PASV: Socket open failed.\n");
                return -1;
        }

        if (setsockopt(fd, SOL_SOCKET, SO_REUSEADDR, &i, sizeof(i)) < 0)
                perror("SO_REUSEADDR");
        bzero(&pasv, sizeof(pasv));
        pasv.sin_family = AF_INET;
        pasv.sin_addr.s_addr = myaddr,
        //pasv.sin_addr.s_addr = htonl(INADDR_ANY);
        pasv.sin_port = htons(port);
        i = 0;
        while (bind(fd, (struct sockaddr *)&pasv, sizeof(pasv)) < 0) {
                printf("PASV: host %lX port %d\n", myaddr, port);
                if (i++ > 10 || errno != EADDRINUSE) {
                        if (debug) printf("Bind: Could not connect on port %d\n", port);
                        perror("bind error");
                        write(controlfd, pasv_err, strlen(pasv_err));
                        return -1;
                }
                pasv.sin_port   = htons(++port);
        }
#if 0
        int len = sizeof(pasv);
        if (getsockname(fd, (struct sockaddr *) &pasv, &len) < 0) {
                printf("Error in getsockname, should not happen\n");
                retrurn -1;
        }
#endif
        if (listen(fd, 1) < 0) {
                write(controlfd, pasv_err, strlen(pasv_err));
                if (debug) printf("PASV: Listen failed.\n");
                close(fd);
                return -1;
        }
        a = (char *) &pasv.sin_addr;
        p = (char *) &pasv.sin_port;
        sprintf(str, "227 Entering Passive Mode (%d,%d,%d,%d,%d,%d)\r\n", UC(a[0]),
                UC(a[1]), UC(a[2]), UC(a[3]), UC(p[0]), UC(p[1]));
        write(controlfd, str, strlen(str));
        //if (debug) printf("%s", str);
        int sslen = sizeof(ss);
        if ((i = accept(fd, (struct sockaddr *)&ss, (unsigned int *)&sslen)) < 0 ) {
                perror("accept error");
                return -1;
        }
        //close(fd);
        printf("Accepted connection from %lx on port %u, fd %d\n", ss.sin_addr.s_addr, ss.sin_port, i);
        *datafd = i;

        return 0;

}
```

Any suggestions, @ghaerr?

I'm getting other anomalies too, but this is the one most easily repeatable.

--Mellvik

ghaerr, 2021-11-18:
The first basic problem, showed by your `netstat` listing, is that the out of memory is caused by excessive LISTEN states. After `accept` in the above code, you absolutely must close the listen socket: `listen(fd)`; first because that listen socket won't be reused and second because it's control block is using up memory. Also, remember that we implemented SOL_RCVBUF? That should be used to allocate only 128 bytes for the listen TCP control block. See any of the telnetd, httpd, or ftpd servers, they all use it.

The second problem is that the SO_REUSEADDR is failing:
```
tcp: port 49821 reused, can't free previous socket in state 7 
```
State 7 (look in tcp.h) is TS_CLOSE_WAIT, so the address reuse is failing, since the previous socket is still in the close wait state, it's still waiting on an application close from the other side (see TCP state diagram). Without looking too much deeper into your code, for now, lets just try a few msecs sleep using `usleep(unsigned long usecs)` to allow the other end to close the file. You can use the code you've got to increment the local port number, but that's quite memory intensive in ktcp and may run out of memory, depending on how long it takes to have the previous socket get out of CLOSE_WAIT state. (In the netstat listing, it looks as though it were never closed).

For this problem:
```
tcp: port 49821 reused, can't free previous socket in state 1
```
State 1 is TS_LISTEN. We don't want to reuse a port number for an active server already listening on that port!! This is another reason why the listen socket must be closed. The fact that ktcp let you reuse it means now there's two ports with the same port number on the ktcp side, which would be a disaster if the remote end port number matched. If it doesn't, then we're just using up memory unnecessarily since the socket is open. That's why I left that error message in.

Finally, in the printf that displays the address information returned from `accept`, remember I told you there's a bug in accept and that data is never returned, its garbage:
```
Accepted connection from 6b6c6500 on port 29547, fd 6 <-- this is data returned from the accept() call, usually but not always 0
```
Do we need the remote port number? No, correct? So lets leave this out of the equation for now, as I mentioned that fixing the accept but has the same problems as adding getsockname/getpeername - lots of extra code. I'd like to try to solve this without going that route if possible.

So, the biggest two issues seem to be lack of closing the listen socket, and the inability to reuse the local port. When we find the reason why the previous port was not closed, or find that we can wait just a bit to allow that to occur on the client side (see TCP diagram for CLOSE_WAIT), then the port reuse will work and less memory will be used.

ghaerr, 2021-11-18:
@Mellvik another idea:

Debug the new PASV mode using an MGET that returns only a single file initially, so that you debug getting a single file transfer working initially, then check netstat to be sure everything is closed down properly. This will also alleviate the need for having to rebind a port number and those problems because only one transfer will be attempted at a time. 

After you get that working well, then add an MGET with just two files, and go from there...

ghaerr, 2021-11-18:
```
        bzero(&pasv, sizeof(pasv));
        pasv.sin_family = AF_INET;
        pasv.sin_addr.s_addr = myaddr,
        //pasv.sin_addr.s_addr = htonl(INADDR_ANY);
        pasv.sin_port = htons(port);
```
Don't use bzero, don't assume INADDR_ANY is 0. Remove bzero and uncomment line above.

```
pasv.sin_port   = htons(++port);
```
htons is a macro, don't auto increment inside it. Use two statements.

```
a = (char *) &pasv.sin_addr;
        p = (char *) &pasv.sin_port;
        sprintf(str, "227 Entering Passive Mode (%d,%d,%d,%d,%d,%d)\r\n", UC(a[0]),
                UC(a[1]), UC(a[2]), UC(a[3]), UC(p[0]), UC(p[1]));
```
Don't use comma-separate notation for addresses, very confusing, use standard dot notation for addresses and unsigned int for port. Don't assume network/host byte order, use standard macros or functions.
Use in_ntoa to convert network byte ordered IP address to ascii, and use ntohs to convert port and %u format.

Mellvik, 2021-11-18:
> @Mellvik another idea:
>
> Debug the new PASV mode using an MGET that returns only a single file initially, so that you debug getting a single file transfer working initially, then check netstat to be [...]
@ghaerr,
Referring to the previous message, yes - we'll keep the ongoing networking issues separate.

Multi transfers: Absolutely. The test case is repeated DIRs until stable (a DIR is a small GET). MGET and MPUT will be cork-poppers!!

Thanks for the advise on the code. The multiple sockets having the same port explains some of the instability I have experienced (commands working, then not working). I'm confident we'll have great improvemets tomorrow.

Thank you.

-Mellvik

Mellvik, 2021-11-18:
> a = (char *) &pasv.sin_addr;
>         p = (char *) &pasv.sin_port;
>         sprintf(str, "227 Entering Passive Mode (%d,%d,%d,%d,%d,%d)\r\n", UC(a[0]),
>                 UC(a[1]), UC(a[2]), UC(a[3]), [...]
@ghaerr, thanks again for the code comments.

FWIW, the comma separators are mandatory, this is the FTP syntax for the PORT and PASV commands. IIRC I lifted this piece from the 4.3BSD ftpd, including the macro, so it should be reasonably safe.

Getting there.

-Mellvik

ghaerr, 2021-11-18:
> FWIW, the comma separators are mandatory, this is the FTP syntax for the PORT and PASV commands. IIRC I lifted this piece from the 4.3BSD ftpd, including the macro, so it should be reasonably safe.

My mistake: I thought this code was being used to display an address to the user. I see now it's an internal way of sending information for PASV mode.

> The multiple sockets having the same port explains some of the instability I have experienced (commands working, then not working).

Does the other end have the same port number? If so, this is definitely a big problem. 

I'm thinking that in the case displayed "tcp: port %u reused, can't free previous socket in state %d", that perhaps the request should be flatly denied by the ktcp bind code, and changed to "tcp: port %u NOT reused, previous socket in state %d". 

The SO_REUSEADDR should only be allowed when it can be done safely, and even if your code does not use the same remote port number, this suggested change would force one to increment a port number, rather than reuse one that may not work.

Mellvik, 2021-11-18:
> The multiple sockets having the same port explains some of the instability I have experienced (commands working, then not working).
> 
> Does the other end have the same port number? If so, this is definitely a big problem.
> 
It does. But armed with your code comments and other input, that can be fixed.
> I'm thinking that in the case displayed "tcp: port %u reused, can't free previous socket in state %d", that perhaps the request should be flatly denied by the ktcp bind code, and changed to "tcp: [...]
Agreed -  this would be very helpful.

Thank you!

-Mellvik

ghaerr, 2021-11-18:
>  in some cases the program needs to find out what address (and port #) the OS assigned to a connection. Such as with our Passive Mode, where we're supposed to tell the client in clear text where to connect - address and port.

Ok, I'm finally understanding the PASV mode: the server binds to a local port number, then sends that port number back to the client to connect to. Because we're trying to reuse that port, you're setting SO_REUSEADDR, which should work well in the case the previous socket was fully closed and in the TIME_WAIT state, correct? But instead, ktcp is sometimes seeing that the previous socket is in CLOSE_WAIT, and it's allowing port reuse (incorrectly), and I then suggested delaying, correct?

I'm thinking that the real problem is that ktcp should not be allow ftpd to reuse the port number in the latter circumstance, and that ftpd should increment the port number and try again, which you're doing. In that case, it will eventually find a port that's not used, and the previously used ports will go to TIME_WAIT on their own, and be deallocated. (Staying in CLOSE_WAIT would be an error, if you find that).

Thus, add the following code to ktcp to refuse the port reuse, and try this with your modified server code as previously discussed. This is in ktcp/tcpdev.c, function tcpdev_bind:
```
        if (n2) {                       /* port already bound */
            if (!db->reuse_addr) {      /* no SO_REUSEADDR on socket */
                debug_tcp("tcp: port %u already bound, rejecting (use SO_REUSEADDR?)\n", port);
reject:   <--- add this line
                tcpcb_remove(n);
                retval_to_sock(db->sock, -EADDRINUSE);
                return;
            }

            /* remove TCB control block on SO_REUSEADDR to save heap space */
            if (n2->tcpcb.state == TS_TIME_WAIT) {
                LEAVE_TIME_WAIT(&n2->tcpcb);    /* entered via FIN_WAIT_2 state on FIN rcvd*/
                tcpcb_remove(n2);
                debug_tcp("tcp: port %u reused, freeing previous socket in time_wait\n", port);
            } else { <--- add this line and change all below to following
                printf("tcp: port %u NOT reused,previous socket in state %d\n",
                        port, n2->tcpcb.state);
               goto reject;
            }
```
With this change, no delay should be needed, and some extra local ports may be used until the previous time out. You can also change the debug_tcp to printf to see when the port is reused for testing.

Use the following to display the attempted port that failed:
```
printf("PASV: host %s port %d\n", in_ntoa(myaddr), port);
```
Note that currently (in the bind code) ktcp's bind completely ignores "myaddr", you're better off leaving that set to htonl(INADDR_ANY). I bring these points up so that we don't get too far away from known-working code.

ghaerr, 2021-11-18:
> Does the other end have the same port number? If so, this is definitely a big problem.

> It does. But armed with your code comments and other input, that can be fixed.

With the above change to tcpdev_bind to reject the port reuse, I don't suggest you write even more code to change the remote port number, as there's no real need to, and we'd be starting to write even more code that might get confusing or misused.

Thank you!

Mellvik, 2021-11-19:
@ghaerr,
Hundreds of tests later - some incomprehensible :-) - I'm still at loss as to where/what the problem is. Must analyze and summarize tomorrow.

- M

Mellvik, 2021-11-20:
Update @ghaerr - 

while still chasing an instability in passive mode (some times working, some times not), a short notice on sockets and timeouts:

When doing (manually) repeated transfers (from ELKS) in passive mode, the same socket is reused any number of times – until I get an error situation which somehow fails to close the connection. The (sub)process exits, but the data connection remains lingering in CLOSE_WAIT - forever. No timeout.

The timeout value is as far as I can tell being decrementing till it wraps around.

I don't think this is what was expected from the minor change to tcpdev_bind…  Comments?

 No        State    RTT lport        raddress  rport
-----------------------------------------------------
  1  ESTABLISHED 1000ms  1033         0.0.0.0      2
  2  ESTABLISHED  875ms    21        10.0.2.2  42738
  3   CLOSE_WAIT 1000ms 49824        10.0.2.2  37192 (-335.7 secs)
  4       LISTEN 1000ms    21         0.0.0.0      0
  5   CLOSE_WAIT 1000ms 49823        10.0.2.2  44548 (-516.2 secs)
  6   CLOSE_WAIT 1000ms 49822        10.0.2.2  38386 (115.6 secs)
  7   CLOSE_WAIT 1000ms 49821        10.0.2.2  58934 (-79.0 secs)
  8       LISTEN 1000ms    80         0.0.0.0      0
  9       LISTEN 1000ms    23         0.0.0.0      0

At this point ktcp starts running out of memory.

—Mellvik

ghaerr, 2021-11-20:
> When doing (manually) repeated transfers (from ELKS) in passive mode, the same socket is reused any number of times – until I get an error situation which somehow fails to close the connection. The (sub)process exits, but the data connection remains lingering in CLOSE_WAIT - forever. No timeout.

Looking at the TCP state diagram, CLOSE_WAIT means that one side has closed the connection. The TCP stack is waiting for the other side to close, and won't do anything until that happens, period, no timeout. In other words, the TCP full duplex channel is in half duplex mode. I'm still a bit confused as to which side is opening/closing what with your code, so I can't comment further other than to say you need to think clearly on exactly which side is doing the opening, then closing, and when.

If this turns into more trouble, I suggest removing the SO_REUSEADDR request, and as suggested earlier, just getting a single transfer to work well, before proceeding. That will eliminate another variable. Yes, ktcp will be short of memory, but you can verify whether the basic PASV connection works every time with no SO_REUSEADDR: just keep a static variable (for now), and increment the port number in a low range where you know it won't have been used.

> The timeout value is as far as I can tell being decrementing till it wraps around.

I've been thinking of fixing that display, it is negative since the current time is way past the time it went into the CLOSE_WAIT state. But as I mentioned before, the stack will never deallocate it, it's not on a timeout list. It is waiting for the other side to close, in which case the state will change to LAST_ACK and go into a timer state.

> I don't think this is what was expected from the minor change to tcpdev_bind… 

Not sure. I suggest eliminating SO_REUSEADDR as suggested above.

We have too many variables and are losing track of too many changes at once. I suggest we halt work on ftpd until the previously-reported #1013 is tested using old kernel, then solved with #1014. That can then be committed, as that problem is supposed to solve a known problem of packets not being read, which we both are almost sure is a big problem.

After that, I suggest a commit of ktcp rejecting any SO_REUSEADDR for a socket not in time_wait state, as we've pretty much realized that is also in error. If you like, I could add that to the #1014 fix and push it. I would rather not commit #1014 without your specific testing, but feel much better about it now that I realized my initial error in it (decreased timeout requiring more heap space), so I will commit it, along with the SO_REUSEADDR rejection, if you've got too much going on.

Should we take the latter approach, then you could restart your ftpd work with fixes we pretty much know are needed, and ultimately close #1013 after your ftpd testing unless you see it pop up again.

```
 No        State    RTT lport        raddress  rport
-----------------------------------------------------
  1  ESTABLISHED 1000ms  1033         0.0.0.0      2
  2  ESTABLISHED  875ms    21        10.0.2.2  42738
  3   CLOSE_WAIT 1000ms 49824        10.0.2.2  37192 (-335.7 secs)
  4       LISTEN 1000ms    21         0.0.0.0      0
  5   CLOSE_WAIT 1000ms 49823        10.0.2.2  44548 (-516.2 secs)
  6   CLOSE_WAIT 1000ms 49822        10.0.2.2  38386 (115.6 secs)
  7   CLOSE_WAIT 1000ms 49821        10.0.2.2  58934 (-79.0 secs)
  8       LISTEN 1000ms    80         0.0.0.0      0
  9       LISTEN 1000ms    23         0.0.0.0      0
```
> At this point ktcp starts running out of memory.

No doubt - with this many open TCP control blocks, it is very close to running out and there's nothing we can do.

Regarding the many CLOSE_WAIT's above: if the processes that "own" them are terminated, then this is definitely a bug. However, I am guessing they are all part of the ftpd process which is still running. In that case, either ftpd has not closed them, for some reason (which is your bug), or the remote process has not closed them (which is unlikely). You should look into this to find which. I am guessing the former.

Let me know which way you'd like to proceed.

Thank you!

ghaerr, 2021-11-20:
@Mellvik, another thought:

Perhaps put all the current ftpd changes aside, and only test with the checked-in version of ftpd, using the old kernel and/or the new kernel, to determine whether #1013 is a problem, and/or whether #1014 fixes it, or you can't reproduce, in which case we check in #1014 with the additional SO_REUSEADDR I'm suggesting added (before your testing).

We then commit that #1014 and you then start anew with your ftpd changes, chasing down the CLOSE_WAIT problem first, then solve the problem of port-reusability after that by adding back SO_REUSEADDR and its associated code in ftpd.

There are so many variables it is very hard for me to track then all! :)

Mellvik, 2021-11-20:
Additional info on this one @ghaerr - 

I've been banging my head against the wall since thursday - there is no reliable pattern to the problem, which is that some times, I get a broken pipe error when writing to a socket just after the accept has returned. It is the situation that trigger this I've been searching for, and I'm coming to the conclusion that we have a memory corruption problem, probably in ktcp. Here are some observations:
When a certain FTP command (fetching data) works, it works any number of times. Doing some simple commands in between and the next time I use the same command it fails.
Some times I get really odd results, such as DISK IO errors about a sector # that is far off the floppy.
Other times, when restarting ktcp after a cleanup of lingering sockets, ktcp no longer works until a reboot.
In one case I even got a kernel panic in brelse.

I'll keep chasing - if you have any suspicions let me know.

—M

ghaerr, 2021-11-20:
> I've been banging my head against the wall since thursday - there is no reliable pattern to the problem, which is that some times, I get a broken pipe error when writing to a socket just after the accept has returned. 

The EPIPE errno will occur whenever a closed socket is written to.

> When a certain FTP command (fetching data) works, it works any number of times. Doing some simple commands in between and the next time I use the same command it fails.

It is possible this is related to your SO_REUSEADDR, or ktcp's allowance of it. I suggest we remove all SO_REUSEADDR (except for the initial ftpd listen) completely for now in ftpd.

> It is the situation that trigger this I've been searching for, and I'm coming to the conclusion that we have a memory corruption problem, probably in ktcp. Here are some observations:

> Some times I get really odd results, such as DISK IO errors about a sector # that is far off the floppy.
> Other times, when restarting ktcp after a cleanup of lingering sockets, ktcp no longer works until a reboot.
> In one case I even got a kernel panic in brelse.

Yes, these definitely sound like possible corruption problems. I cannot recommend testing using a restarted ktcp, the system is likely very screwed up if ktcp had to be restarted, both from ktcp issues and the tcpdev kernel driver. So lets not do that - that might be adding to the problems.

After reading this, I think that, given the unknown number of problems still in ktcp and the kernel, as well as the hack-arounds you've had to add to ftpd to get PASV working, along with the basic fact that it might be coded incorrectly still, the whole thing is too complicated to reliably debug. I am going to suggest the following:
- I'll add the SO_REUSEADDR reject code to the outstanding #1014 
- You remove all SO_REUSEADDR code period, and only use a static incremented port number. You'll get a hard failure in this case if the port is in use.
- You decide if you want to test #1013 or not in the way previously described, and we then commit #1014
- You eliminate all hack code in ftpd, and simply PASV mode as much as possible. We then debug that after #1014 is committed, and start with obvious problems, like fixing the CLOSE_WAIT problem, before going into faster back-to-back issues. Ftpd needs to work with a single PASV get, time after time, with all sockets closed and timed out, before proceeding further.

Mellvik, 2021-11-20:
Thank you for a thorough walkthru @ghaerr,
Some clarifications:
> When doing (manually) repeated transfers (from ELKS) in passive mode, the same socket is reused any number of times – until I get an error situation which somehow fails to close the connection. The [...]
All testing is about ftp passive mode and the ftp data channel, so elks is accepting incoming connections.
> If this turns into more trouble, I suggest removing the SO_REUSEADDR request, and as suggested earlier, just getting a single transfer to work well, before proceeding.
> 
Per you recommendation earlier in this thread, REUSEADDR was removed when the changes in ktcp were made.
> That will eliminate another variable. Yes, ktcp will be short of memory, but you can verify whether the basic PASV connection works every time with no SO_REUSEADDR: just keep a static variable (for now), and increment the port number in a low range where you know it won't have been used.
> 
Yes, that's what we have.
> The timeout value is as far as I can tell being decrementing till it wraps around.
>
> I've been thinking of fixing that display, it is negative since the current time is way past the time it went into [...]
Ok, I will look closer at that. It doesn't make sense to me that there is no timeout. 
> I don't think this is what was expected from the minor change to tcpdev_bind…
>
> Not sure. I suggest eliminating SO_REUSEADDR as suggested above.
>
> We have too many variables and are losing track of [...]
The thing is, the new ftpd is the only test case, it has PASV mode. Previous versions don't.
> After that, I suggest a commit of ktcp rejecting any SO_REUSEADDR for a socket not in time_wait state, as we've pretty much realized that is also in error. If you like, I could add that to the #1014 fix and push it.
> 
Yes, that's good!
> I would rather not commit #1014 without your specific testing, but feel much better about it now that I realized my initial error in it (decreased timeout requiring more heap space), so I will commit it, along with the SO_REUSEADDR rejection, if you've got too much going on.
> 
I don't think it's a problem introducing this into the equation. The focus of my testing is not the timeouts or socket reuse, but they are part of the equation.
> Regarding the many CLOSE_WAIT's above: if the processes that "own" them are terminated, then this is definitely a bug. However, I am guessing they are all part of the ftpd process which is still running.
> 
No, the owning process is gone, the other end is gone, so the only way to get rid of them is to kill ktcp.

Let me take a look at the other responses before we conclude.

Thank you very much!

-Mellvik

ghaerr, 2021-11-20:
Ok, @Mellvik, thanks for the clarifications. I will push a fix the the SO_REUSEADDR into #1014, even though you've removed that code from ftpd for now (good).

It also seems that the ftpd is fairly clean, as far as your testing goes, good.

I just looked at the ktcp CLOSE_WAIT code, and there is something that could be an issue, but still doesn't explain fully what you're seeing. See below.

>  Regarding the many CLOSE_WAIT's above: if the processes that "own" them are terminated, then this is definitely a bug. 
> However, I am guessing they are all part of the ftpd process which is still running.

> No, the owning process is gone, the other end is gone

Ok, that means you are NOT running ftpd, that ftpd has terminated? Please clarify.

This is very important: from my discussion above, I realize now that it must be ftpd that did not close the file descriptor (for whatever reason). This is the reason that we have the CLOSE_WAIT state: the other side closed it first and sent a FIN to ktcp, and ktcp is waiting for the application ftpd close. It will not timeout, ever, but will enter LAST_ACK (timeout) on a close or program exit. The LAST_ACK timeout, along with the FIN_WAIT_1/2 and CLOSING timeouts, are still 10 seconds.

However - I see from looking at the ktcp code, that when a process is killed or dies, the same "close" code is called that would/should have been from the application: in this case, a FIN is sent, and the state is changed to LAST_ACK. That means that ktcp should still go into a LAST_ACK wait state after ftpd exit. But under no circumstances should ktcp keep a CLOSE_WAIT control block for an application that has exited. If you can duplicate this behavior, we have a new bug. If the ftpd daemon is still running, it is not a bug, but rather just the fact that a close never happened. I hope this clarifies the issue.

> so the only way to get rid of them is to kill ktcp.

Got it - lets not do that, as it introduces more possible problems if there is in fact a data corruption issue.

Mellvik, 2021-11-20:
Thank you @ghaerr -
> Yes, these definitely sound like possible corruption problems. I cannot recommend testing using a restarted ktcp, the system is likely very screwed up if ktcp had to be restarted, both from ktcp issues and the tcpdev kernel driver. So lets not do that - that might be adding to the problems.
> 
Good point. Avoiding reboots to save time (30 seconds or so) may just do the opposite.
> After reading this, I think that, given the unknown number of problems still in ktcp and the kernel, as well as the hack-arounds you've had to add to ftpd to get PASV working,
> 
Except for the port increment & retry in bind, I don't think we have any hacks left.
> along with the basic fact that it might be coded incorrectly still,
> 
That may well be the case. OTOH it is encouraging that I can do hundreds of file transfers in passive mode.
> the whole thing is too complicated to reliably debug. I am going to suggest the following:
>
> I'll add the SO_REUSEADDR reject code to the outstanding #1014
> You remove all SO_REUSEADDR code period, [...]
Yes, this is what I've been using since thursday. Made the environ stable.
> You decide if you want to test #1013 or not in the way previously described, and we then commit #1014
> You eliminate all hack code in ftpd, and simply PASV mode as much as possible. We then debug [...]
I'm not following you here. I think we need to realign bases in order to decide on a reasonable path. The lingering sockets are annoyances, but not blocking my 'quest' for the occasionally (almost randomly) failing accept(). That now looks more and more like memory corruption, but then again...

We wouldn't do this if it was simple, right?

-M

ghaerr, 2021-11-20:
> I'm not following you here. I think we need to realign bases in order to decide on a reasonable path. The lingering sockets are annoyances, but not blocking my 'quest' for the occasionally (almost randomly) failing accept(). That now looks more and more like memory corruption, but then again...

I'm just trying to keep track of all the potential TCP bugs you're throwing at me: 

We've agreed on the first issue, I'll add the port reject code to my outstanding PR, which shouldn't affect your testing but fixes a previous bug (we think). 

Secondly, we have a CLOSE_WAIT problem which I have identified as a clear problem of not having a file descriptor closed, but could be another ktcp bug. I'd like to have you check that closely. 

Thirdly, you're reminding me (again) of a possible accept problem, which I had forgotten about. I'd like to get the previous bugs identified first, but in any case the accept issue may be wire-related with a SYN or ACK dropped, we will probably have to debug that on the line. I would prefer to do that once we fix/identify the previous two, that's all.

ghaerr, 2021-11-20:
Also, regarding accept: remember that accept does NOT return anything valid in the last two parameters passed to it; they are completely ignored (for now). So pass NULL for both in your code, for the time being.

Mellvik, 2021-11-20:
Ok  - this is good, @ghaerr.
I needed those clarifications, I've been totally consumed by the accept() problem these past few days. So, really, we have 4 issues open:
The retransmit/ack issue from early this week
The close_wait issue
The accept issue
The REUSEADDR issue
Your recent discovery about the REUSEADDR problem makes me think it may be the source (or contributing to) the accept problem. Or maybe not since I'm not using this option in passive mode? Also, Remember the change introduced above, :
This is in ktcp/tcpdev.c, function tcpdev_bind:

        if (n2) {                       /* port already bound */
            if (!db->reuse_addr) {      /* no SO_REUSEADDR on socket */
                debug_tcp("tcp: port %u already bound, rejecting (use SO_REUSEADDR?)\n", port);
reject:   <--- add this line
                tcpcb_remove(n);
                retval_to_sock(db->sock, -EADDRINUSE);
                return;
            }

            /* remove TCB control block on SO_REUSEADDR to save heap space */
            if (n2->tcpcb.state == TS_TIME_WAIT) {
                LEAVE_TIME_WAIT(&n2->tcpcb);    /* entered via FIN_WAIT_2 state on FIN rcvd*/
                tcpcb_remove(n2);
                debug_tcp("tcp: port %u reused, freeing previous socket in time_wait\n", port);
            } else { <--- add this line and change all below to following
                printf("tcp: port %u NOT reused,previous socket in state %d\n",
                        port, n2->tcpcb.state);
               goto reject;
            }

I've been using this since then (thursday). I will pull your latest pr tomorrow. It seems the ftpd is the perfect test case for this issue with its heavy (ab)use of sockets. In active mode (in which REUSEADDR is being used), file transfers have been 100% stable btw. Hundreds of files back-to-back. That's encouraging.

-M

ghaerr, 2021-11-20:
> So, really, we have 4 issues open

Yes, and here's the status of what to do about them after I commit #1014 shortly:
- The retransmit/ack issue from early this week

This is your #1013, should be fixed. If it doesn't repeat, please close that issue.

- The close_wait issue

Need your help in finding the missing `close` for this. Should never see close_wait after ftpd exit, but certainly may see if ftpd is buggy in closing a file.

- The accept issue

I am aware that accept does NOT return any connection information in its last two arguments. Please only pass NULL and do not attempt dereferencing anything for the time being. This can be fixed later. 

I am not aware of another accept issue, is it that certain accepts fail? What exactly does that mean?

Frankly, we are mixing together far too many issues with other issues. They are hard to track and will become hard for other users to navigate, should anyone wish to. It may be a good idea to have seperate issues, and only comment in the pertinent issue comment section, especially when bugs or issues break out in swarms.

- The REUSEADDR issue

Yes - do not use it in any PASV mode for the time being. It works only in certain circumstances and is too buggy for new code. It will be fixed later.

> Your recent discovery about the REUSEADDR problem makes me think it may be the source (or contributing to) the accept problem. Or maybe not since I'm not using this option in passive mode? 

If you're not using SO_REUSEADDR in passive mode, then that's not an issue. Lets completely leave it if'defd out of any code being tested for now.

> Also, Remember the change introduced above, This is in ktcp/tcpdev.c, function tcpdev_bind

I (re)made this change in #1014 which I'm about to commit; you'll have to remove that from your repo, or just use the ftpd changes only from the master branch tomorrow morning.

> It seems the ftpd is the perfect test case for this issue with its heavy (ab)use of sockets.

No, we don't want to use ftpd for testing SO_REUSEADDR at this time, it may cause a TCP control block to be removed and replaced with another just like it, right in the middle of a connection, causing all sorts of strange problems. We will revisit a correct SO_REUSEADDR later. Good news is I've finally recognized it could be buggy, so don't use it.

> In active mode (in which REUSEADDR is being used), file transfers have been 100% stable btw. Hundreds of files back-to-back.

Yes, that's the case where it seems to be working, but we need to get passive mode to work without it and for single files and no close_wait remainders first. Too many variables otherwise and too much going on.

Mellvik, 2021-11-22:
Thank you @ghaerr,
> The close_wait issue
> Need your help in finding the missing close for this. Should never see close_wait after ftpd exit, but certainly may see if ftpd is buggy in closing a file.
> 
Yes, it's all about sockets lingering after process exit. I'm on this one.
> The accept issue
> I am aware that accept does NOT return any connection information in its last two arguments. Please only pass NULL and do not attempt dereferencing anything for the time being. This can be fixed later.
> 
That (the parameter thing) was a comment, not an issue. The testing is done with NULL as the 2nd and 3rd parameters to accept(). The issue is accept() returning ok, while the next read/write fails, randomly and with random consequences - such as panic in ktcp. This is what I referred to as a potential memory corruption problem  on saturday.
Given all the fixes you've posted since I started writing this :-) - it will take another round of tests so see if anything has changed.
> The REUSEADDR issue
> Yes - do not use it in any PASV mode for the time being. It works only in certain circumstances and is too buggy for new code. It will be fixed later.
>
> Your recent discovery [...]
Good, thanks. I'm always working off of the latest commit except in cases when something else is recommended.
> It seems the ftpd is the perfect test case for this issue with its heavy (ab)use of sockets.
>
> No, we don't want to use ftpd for testing SO_REUSEADDR at this time, it may cause a TCP control block to [...]
Ok, I guess I got lost on this one - so I need to know what to test. I've been avoiding SO_REUSEADDR for both modes in ftpd since late last week, and - with your recent fixes, port reuse has been working fine. Thousands of back-to-back file transfers in active mode and the port 20 socket gets reused constantly - just an occasional 'tcp: port 20 reused, freeing previous socket in time_wait' message (approx. once per 6th file).

I'll be awaiting your advice on this one, for now staying away from the SO_REUSEADDR option.

--Mellvik
>

ghaerr, 2021-11-22:
Hello @Mellvik,

Sounds like you're all over it. Definitely interested in hearing about what you learn with CLOSE_WAIT.

The last round of networking enhancement PRs probably won't be changing the strange results you're seeing.

I now understand you're saying that accept itself isn't the problem, but that the socket it returns is causing extreme failures sometimes. Note that #1019 fixed the ability to pass non-NULL parameters to accept, for your future use. This won't change the bug you're seeing though.

> I'll be awaiting your advice on this one, for now staying away from the SO_REUSEADDR option.

SO_REUSEADDR still has potential issues, lets not use it (except for the first server port bind when started), until the CLOSE_WAIT and accept bugs are identified. The SO_REUSEADDR is working as intended for your back-to-back transfers, but I've identified that there are other situations that could cause problems. Fix coming.

Thank you!

Mellvik, 2021-11-22:
Thank you @ghaerr,
> Sounds like you're all over it. Definitely interested in hearing about what you learn with CLOSE_WAIT.
> 
Yes - I do have a definitive status on this one: If a process exits without closing a socket and for some reason, data is left in the 'pipeline', that socket is left in close_wait forever.

This may happen in case of bugs or other errors, hardware or OS-related conditions, the point being that a process exit should be able to clean up the situation.

> I now understand you're saying that accept itself isn't the problem, but that the socket it returns is causing extreme failures sometimes. Note that #1019 [...]
I've tested this today (getsockname() and accept() w/parameters), works like a charm. Thank you - this is indeed an important improvement!! And you found a way to do it w/o high cost.
> I'll be awaiting your advice on this one, for now staying away from the SO_REUSEADDR option.
>
> SO_REUSEADDR still has potential issues, lets not use it (except for the first server port bind when [...]
No problem, things are actually working quite well without it. I'll post a report on that separately.

I'm also going to reopen the ACK/timing issue, I think I have a new case in which this beast kicks in. :-)

—Mellvik

ghaerr, 2021-11-22:
> Yes - I do have a definitive status on this one: If a process exits without closing a socket and for some reason, data is left in the 'pipeline', that socket is left in close_wait forever.

Can you post some code that allows me to duplicate this? I am trying to understand if the socket is already in close_wait when the process exits or not.

I've looked hard at ktcp and can't find the error yet. If the remote side sends a FIN, the socket will go into close_wait. If the process is exited or socket closed, we're supposed to go into LAST_ACK state, which should timeout. If the remote side hasn't yet sent a FIN, then a process exit will cause the state to go to FIN_WAIT_1, then timeout. Obviously we're not seeing either of these latter state changes, so I don't get it it.

> This may happen in case of bugs or other errors, hardware or OS-related conditions, the point being that a process exit should be able to clean up the situation.

Yes - but the "cleanup" is limited: as I mentioned earlier, process exit is exactly the same as a socket close when the process is running. That is, no special cleanup is done - the socket should move to either LAST_ACK or FIN_WAIT. So I need some code to duplicate the problem.

> I'm also going to reopen the ACK/timing issue, I think I have a new case in which this beast kicks in. :-)

Geez. This means the mainloop PR change did NOT fix the issue? So you've found another case for this, whereas over the weekend you could not, or is this a new case?

Thank you!

Mellvik, 2021-11-22:
@ghaerr,
update on the CLOSE_WAIT issue:

While the previous description is correct, it turns out we don't need a process exit to get the socket into a forever CLOSE_WAIT mode.

What it takes is a FIN from the other end, which ELKS ACKs but doesn't FIN (the application closes properly).

I cannot post code to repeat this yet - because it requires the ftpd accept error (which isn't really accept() related, but it needs a name) to get into this mode. 

I need to dig more into this in order to create a repeatable test case.

—Mellvik

> 22. nov. 2021 kl. 17:31 skrev Helge Skrivervik ***@***.***>:
> 
> Thank you @ghaerr,
>> Sounds like you're all over it. Definitely interested in hearing about what you learn with CLOSE_WAIT.
>> 
> Yes - I do have a definitive status on this one: If a process exits without closing a socket and for some reason, data is left in the 'pipeline', that socket is left in close_wait forever.
> 
> This may happen in case of bugs or other errors, hardware or OS-related conditions, the point being that a process exit should be able to clean up the situation.
> 
>> I now understand you're saying that accept itself isn't the problem, but that the socket it returns is causing extreme failures sometimes. Note that #1019 <https://github.com/jbruchon/elks/pull/1019> fixed the ability to pass non-NULL parameters to accept, for your future use. This won't change the bug you're seeing though.
>> 
> I've tested this today (getsockname() and accept() w/parameters), works like a charm. Thank you - this is indeed an important improvement!! And you found a way to do it w/o high cost.
>> I'll be awaiting your advice on this one, for now staying away from the SO_REUSEADDR option.
>> 
>> SO_REUSEADDR still has potential issues, lets not use it (except for the first server port bind when started), until the CLOSE_WAIT and accept bugs are identified. The SO_REUSEADDR is working as intended for your back-to-back transfers, but I've identified that there are other situations that could cause problems. Fix coming.
>> 
> No problem, things are actually working quite well without it. I'll post a report on that separately.
> 
> I'm also going to reopen the ACK/timing issue, I think I have a new case in which this beast kicks in. :-)
> 
> 
> —Mellvik
>

ghaerr, 2021-11-22:
> While the previous description is correct, it turns out we don't need a process exit to get the socket into a forever CLOSE_WAIT mode.

Ok, good to know.

> What it takes is a FIN from the other end, which ELKS ACKs but doesn't FIN (the application closes properly).

You're saying that the ELKS receives a FIN from the other end, which is ACKed, and the ELKS application is closing the socket, but a FIN isn't sent out, and somehow the state ends up in CLOSE_WAIT?

I am looking at code and see that there is some trickiness to dealing with unread data, which might come after the state is going into LAST_ACK or particularly FIN_WAIT_1 (from the ELKS application close). Ktcp could be forcing the socket to CLOSE_WAIT somehow when processing that final data packet.

> I cannot post code to repeat this yet - because it requires the ftpd accept error (which isn't really accept() related, but it needs a name) to get into this mode.

It is possible this is a corruption bug due to the accept error, not sure.

I will post a PR with printf statements in ktcp showing the state transitions in these cases. That should help us see when and where ktcp is changing state. That, along with a printf with your close code, should help solve this.

Mellvik, 2021-11-22:
> 
> I'm also going to reopen the ACK/timing issue, I think I have a new case in which this beast kicks in. :-)
> 
> Geez. This means the mainloop PR change did NOT fix the issue? So you've found another case for this, whereas over the weekend you could not, or is this a new case?
> 
> 
This is a new case! and it's slightly different. 

In short what happens is that
- connection is opened (accept()), 
- the client sends 3 full size 1460 packets, 
- ELKS ACKs them one by one, shrinking the window along the way, 
- to 1 in the 3rd ACK. 
Now, this one may be interesting (I need to look more at this later). The window size advertized by the 2nd ACK is 1460 exactly, which is also the size of the next packet payload. 
So the next window should arithmetically be 0, but cannot be zero so it's one by definition. If this is accommodated for in the code, no problem, but what happens next is that the client sends one byte, ELKS ACKs one byte, and this continues for a long time - single byte packets -
until ktcpd finally says 'tcp: dropping packet, no buffer space: 1 > 0' 9 times and hangs. 

I've seen this behaviour a number of times over the past few days, and I don't know for sure that this is at all related to the ACK/timing issue. My full analysis will compare a transfer that works with one as described. What sent me down the ACK/timing road was the fact that in between the packets on the data channel, there is a packet on the command channel that may rock the boat.

This is a preliminary description - i'm working on a more detailed rundown for you to look at.

Thank you.

—Mellvik

closed 2021-11-30 by ghaerr