ghaerr/elks #1022: [net] Receive window size timing issues

ghaerr, 2021-11-22:
Opening new issue for @Mellvik, from https://github.com/jbruchon/elks/issues/1016#issuecomment-975756845.

In short what happens is that
- connection is opened (accept()),
- the client sends 3 full size 1460 packets,
- ELKS ACKs them one by one, shrinking the window along the way,
- to 1 in the 3rd ACK.
Now, this one may be interesting (I need to look more at this later). The window size advertized by the 2nd ACK is 1460 exactly, which is also the size of the next packet payload.
So the next window should arithmetically be 0, but cannot be zero so it's one by definition. If this is accommodated for in the code, no problem, but what happens next is that the client sends one byte, ELKS ACKs one byte, and this continues for a long time - single byte packets -
until ktcpd finally says 'tcp: dropping packet, no buffer space: 1 > 0' 9 times and hangs.

I've seen this behaviour a number of times over the past few days, and I don't know for sure that this is at all related to the ACK/timing issue. My full analysis will compare a transfer that works with one as described. What sent me down the ACK/timing road was the fact that in between the packets on the data channel, there is a packet on the command channel that may rock the boat.

This is a preliminary description - i'm working on a more detailed rundown for you to look at.

Mellvik, 2021-11-22:
Packet trace for the scenario above:
```
10:54:02.732527 IP (tos 0x10, ttl 64, id 33839, offset 0, flags [DF], proto TCP (6), length 51)
    10.0.2.2.42822 > 10.0.2.15.ftp: Flags [P.], cksum 0x1836 (incorrect -> 0x0ac2), seq 92:103, ack 475, win 29200, length 11: FTP, length: 11
	STOR ftpd
10:54:02.759715 IP (tos 0x0, ttl 64, id 9154, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp > 10.0.2.2.42822: Flags [.], cksum 0x1154 (correct), ack 103, win 4369, length 0
10:54:02.780656 IP (tos 0x0, ttl 64, id 9155, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp > 10.0.2.2.42822: Flags [.], cksum 0x1149 (correct), ack 103, win 4380, length 0
10:54:03.065811 IP (tos 0x0, ttl 64, id 9156, offset 0, flags [none], proto TCP (6), length 88)
    10.0.2.15.ftp > 10.0.2.2.42822: Flags [P.], cksum 0x8e47 (correct), seq 475:523, ack 103, win 4380, length 48: FTP, length: 48
	150 Opening BINARY data connection for 'ftpd'.
10:54:03.066268 IP (tos 0x8, ttl 64, id 18596, offset 0, flags [DF], proto TCP (6), length 1500)
    10.0.2.2.38532 > 10.0.2.15.49822: Flags [.], cksum 0x1ddf (incorrect -> 0x29a0), seq 1:1461, ack 1, win 29200, length 1460
10:54:03.066383 IP (tos 0x8, ttl 64, id 18597, offset 0, flags [DF], proto TCP (6), length 1500)
    10.0.2.2.38532 > 10.0.2.15.49822: Flags [P.], cksum 0x1ddf (incorrect -> 0x799f), seq 1461:2921, ack 1, win 29200, length 1460
10:54:03.066459 IP (tos 0x8, ttl 64, id 18598, offset 0, flags [DF], proto TCP (6), length 1500)
    10.0.2.2.38532 > 10.0.2.15.49822: Flags [.], cksum 0x1ddf (incorrect -> 0xbff9), seq 2921:4381, ack 1, win 29200, length 1460
10:54:03.101310 IP (tos 0x0, ttl 64, id 9157, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.49822 > 10.0.2.2.38532: Flags [.], cksum 0xbd1f (correct), ack 1461, win 2920, length 0
10:54:03.112546 IP (tos 0x10, ttl 64, id 33840, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.42822 > 10.0.2.15.ftp: Flags [.], cksum 0x182b (incorrect -> 0xb024), ack 523, win 29200, length 0
10:54:03.136403 IP (tos 0x0, ttl 64, id 9158, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.49822 > 10.0.2.2.38532: Flags [.], cksum 0xbd1f (correct), ack 2921, win 1460, length 0
10:54:03.170666 IP (tos 0x0, ttl 64, id 9159, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.49822 > 10.0.2.2.38532: Flags [.], cksum 0xbd1e (correct), ack 4381, win 1, length 0
10:54:03.402565 IP (tos 0x8, ttl 64, id 18599, offset 0, flags [DF], proto TCP (6), length 41)
    10.0.2.2.38532 > 10.0.2.15.49822: Flags [P.], cksum 0x182c (incorrect -> 0xc005), seq 4381:4382, ack 1, win 29200, length 1
10:54:03.407398 IP (tos 0x0, ttl 64, id 9160, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.49822 > 10.0.2.2.38532: Flags [.], cksum 0xbd1d (correct), ack 4382, win 1, length 0
10:54:03.642569 IP (tos 0x8, ttl 64, id 18600, offset 0, flags [DF], proto TCP (6), length 41)
    10.0.2.2.38532 > 10.0.2.15.49822: Flags [P.], cksum 0x182c (incorrect -> 0xfc04), seq 4382:4383, ack 1, win 29200, length 1
10:54:03.647075 IP (tos 0x0, ttl 64, id 9161, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.49822 > 10.0.2.2.38532: Flags [.], cksum 0xbd1c (correct), ack 4383, win 1, length 0
10:54:03.882565 IP (tos 0x8, ttl 64, id 18601, offset 0, flags [DF], proto TCP (6), length 41)
    10.0.2.2.38532 > 10.0.2.15.49822: Flags [P.], cksum 0x182c (incorrect -> 0x4704), seq 4383:4384, ack 1, win 29200, length 1
10:54:03.887093 IP (tos 0x0, ttl 64, id 9162, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.49822 > 10.0.2.2.38532: Flags [.], cksum 0xbd1b (correct), ack 4384, win 1, length 0
10:54:04.122577 IP (tos 0x8, ttl 64, id 18602, offset 0, flags [DF], proto TCP (6), length 41)

```

ghaerr, 2021-11-22:
Very interesting analysis on the receive window, @Mellvik.

It seems we've been too precise in estimating the size of our TCP buffer: CB_NORMAL_BUFSIZ (in ktcp/tcp.h) is set to 4892, which is exactly 3\*1460 + 512. And ktcp subtracts 512 when advertising the receive size, so the fast remote client sends 1460 packets, exactly as we allowed for. After that, we have 512 bytes left but the ktcp sending routine still subtracts 512 so we get 0, which is changed to 1.

Then, we end up with a 1-byte sized advertisement, and a sequence of single-byte transfers are performed. It's likely a big that ktcp hangs after dropping a packet, but lets hold that problem until later.

This whole problem is described in the RFC's as SWS: Silly window syndrome. The RFCs state an implemention MUST handle it, and ktcp does not. The solution is best described as not allowing the receive buffer to be immediately changed quickly when opening up buffer space, or these ping-pong type things happen. We may also have to slow down the packet sending earlier, or at least advertise some of the remaining 512 bytes available, rather than getting subtracting 512 again and advertising 1.

In our particular case, we could try increasing the 4892 default buffer to larger, or perhaps subtract less when it's less than 512, but the SWS would likely persist, just at different remaining buffer sizes.

A nice solution could be to advertise 0, but then we have to implement the ZWP zero window probe, which means we could have to set a timer as to when the window is non-zero, which is a whole other problem.

At this point, I think studying SWS and its solution would be a good idea. For a temp workaround, you may want to change 4892 to 5020, which would allow a 128-byte packet remainder, instead of 0, just to see what happens.

> I've seen this behaviour a number of times over the past few days, and I don't know for sure that this is at all related to the ACK/timing issue.

I don't think this is a ACK/timing bug, but rather very likely the SWS/small advertised window problem just discussed. That's good news, because we're not seeing lost data, but rather just bad client/server interaction. (With the exception of the ultimate hang, which can wait and is a yet another issue).

Mellvik, 2021-11-22:
How interesting, @ghaerr!

> It seems we've been too precise in estimating the size of our TCP buffer: CB_NORMAL_BUFSIZ (in ktcp/tcp.h) is set to 4892, which is exactly 3*1460 + 512. And ktcp subtracts 512 when advertising the [...]
The speed difference between the nodes gives little leeway for slowdown. The incoming is like a firehose ... wham, and the initially advertized window is filled.
> or at least advertise some of the remaining 512 bytes available, rather than getting subtracting 512 again and advertising 1.
> 
Yes, that sounds like the way to go. A curiosity - and I haven't counted - but I'd be surprised if the number of single byte packets didn't equal 512, or very close.
> In our particular case, we could try increasing the 4892 default buffer to larger, or perhaps subtract less when it's less than 512, but the SWS would likely persist, just at different remaining [...]
Thanks, that's what I needed.
> I've seen this behaviour a number of times over the past few days, and I don't know for sure that this is at all related to the ACK/timing issue.
>
> I don't think this is a ACK/timing bug, but rather [...]
Yes, I've been pondering that one. It seems when the single byte packets starts, there is no more data passer to the waiting process and ktcp eventually stops. Then again, one thing at a time... 

-M

ghaerr, 2021-11-22:
Reading these two articles describes a bit more about the problem of SWS. Our problem is receive-side SWS:
http://www.tcpipguide.com/free/t_TCPSillyWindowSyndromeandChangesTotheSlidingWindow.htm
https://en.wikipedia.org/wiki/Silly_window_syndrome

We may have to implement advertising a zero window, as there is no other way to completely stop the sending system. I may have been wrong in saying we need a special timer, as ktcp is guaranteed to send an ACK on previous data not yet processed.

Both articles indicate definite problems when a single byte buffer is advertised. In our case, that's not even the case, as we advertise one when there's 0, indicating to the sender to keep sending. I'm pretty sure your observation that it takes about 512 of those before ktcp shuts down is correct.

I'm thinking of using David Clark's solution referenced in the second article: once zero is advertised, the receiver should wait until either 1) an MSS sized packet can be received, or 2) half the buffer is empty. I'm not yet sure whether we need to subtract the additional "PUSH_THRESHOLD" from the advertised size, I am thinking not, once we get this working.

> Then again, one thing at a time...

Agreed, we can get the other issues resolved first, then setup a test case to solve this problem. I can't replicate it on QEMU alone.

Mellvik, 2021-11-22:
Fast work again, @ghaerr -
I agree on both accounts. We'll test the other alternative first, if for nothing else, out of curiosity.
Then turn to David Clark's solution.

Thank you.

-M

ghaerr, 2021-11-22:
@Mellvik, I've implemented advertising zero window with Clark's solution and it seems to work well. 

I'll post a PR that should solve this issue, and will commit the close_wait PR for your testing tomorrow.

closed 2021-11-25 by ghaerr