ghaerr/elks #1032: Boot fails with 'ELKS.3!'

ConiKost, 2021-11-26:
**Description**
I am trying to test ELKS on my shiny new NuXT. When booting the floppy, I am only getting ```ELKS.3!```

**Configuration**
- Version (commit or release): 0.4.0
- Emulator or HW platform: NuXT + HxC Floppy Emulator

**How to reproduce ?**
It happens with very boot.

**Additional information**
Since HxC Floppy Emulator needs to convert .img files into it's own format .hxc, it's also possible, that the converted image is becoming faulty, as booting the downloaded .img file in 86Box works. But I don't have any idea, what's the message suppose to say.

ghaerr, 2021-11-26:
Hello @ConiKost,

Thanks for trying ELKS. The '3' error means "Can't find system" for FAT, and "*" means read retry for MINIX.

I am guessing here, but it seems that either 1) the HxC floppy emulator may not be working properly (electrically) with the NuXT, or perhaps the .hxc format isn't placing the sectors in the required order. Generally what happens is the ELKS boot loader (see bootblocks/boot_sect.S and boot_err.h for details), reads the floppy root directory sector looking for /linux (for FAT) and issues '3' if it can't find it (or possibly can't read the sector, I'd have to check), and the MINIX boot sector, which is 512 bytes larger, checks for read errors and displays '*', while having more code to figure out the location of /linux from the MINIX floppy.

Have you tried booting any other OS using the HxC?

Thank you!

ConiKost, 2021-11-26:
Hi! Thank you for the explanation.

> Have you tried booting any other OS using the HxC?

Yes, booting MSDOS floppy images for example works just fine via HxC Floppy emulator.

ghaerr, 2021-11-26:
> Yes, booting MSDOS floppy images for example works just fine via HxC Floppy emulator.

Ok, that likely means its an ELKS incompatibility issue. What's the CHS (cylinders, head, tracks) settings you're using? Are you emulating a 1440K floppy with 18 sectors per track, 2 heads, 80 tracks? Or something else?

ghaerr, 2021-11-26:
Another thought would be to boot MSDOS, then unplug the USB image from the HxC, insert one with an ELKS FAT image on it, and see whether MSDOS will read it.

ConiKost, 2021-11-26:
> What's the CHS (cylinders, head, tracks) settings you're using?

I am not sure, if I can say, as I don't have any options to set that? I only can the in the bios the type of floppy drive.

> Are you emulating a 1440K floppy with 18 sectors per track, 2 heads, 80 tracks? Or something else?

Yes, it's set to 3.5 Inch 1.44 MB Floppy drive setting. Should be the 'default' one.

> Another thought would be to boot MSDOS, then unplug the USB image from the HxC, insert one with an ELKS FAT image on it, and see whether MSDOS will read it.

Good idea. This failed. While I can access the mounted fat floppy disk, it only contains garbage. So this looks like HxC is converting the image to a defective state. Using a MSDOS floppy image shows regular files and directories.

ConiKost, 2021-11-26:
Okay, I figured out :-) Seems something fauly in direct conversion. If I convert to default .img and afterwards again to .hfe, it works. ELKS is booting. So this is definitely nothing related to ELKS.

Thank you!

closed 2021-11-26 by ConiKost