ghaerr/elks #1033: Image Build error

Mellvik, 2021-11-27:
When selecting `Build extra binary images` in `menuconfig`
the build fails when creating the minix 720k image:

```
install /Users/helge/src/mellvik/elks/elks/arch/i86/boot/Image /Users/helge/src/mellvik/elks/target/linux
rm -f /Users/helge/src/mellvik/elks/image/fd720-minix.bin
mfs  /Users/helge/src/mellvik/elks/image/fd720-minix.bin genfs -1 -n14 -i192 -s720 /Users/helge/src/mellvik/elks/target
No free slots in bitmap found
make[2]: *** [minixfs] Error 255
make[1]: *** [fd720-minix] Error 2
```

closed 2021-11-28 by ghaerr

Mellvik, 2021-11-29:
@ghaerr -

I'm still getting errors on this one, although a different error (and still with the `Build extra binary images` menuconfig option:
```
git log --abbrev-commit | head -1 | sed 's/commit/ELKS built from commit/' > /Users/helge/src/mellvik/elks/target/etc/motd
install /Users/helge/src/mellvik/elks/elks/arch/i86/boot/Image /Users/helge/src/mellvik/elks/target/linux
rm -f /Users/helge/src/mellvik/elks/image/fd360-fat.bin
dd if=/dev/zero of=/Users/helge/src/mellvik/elks/image/fd360-fat.bin bs=1024 count=360
360+0 records in
360+0 records out
368640 bytes transferred in 0.002157 secs (170906182 bytes/sec)
mformat -i /Users/helge/src/mellvik/elks/image/fd360-fat.bin -f 360 -M 512 -d 2 -L 9 -r 14 -k -N 0 -c 1
make[2]: mformat: No such file or directory
make[2]: *** [fatfs] Error 1
make[1]: *** [fd360-fat] Error 2
make: *** [all] Error 2
```

-M

ghaerr, 2021-11-29:
> make[2]: mformat: No such file or directory

Yes, you have to have the "mtools" distribution to get the `mformat` utility the build uses to create the FAT images. This is available as mformat v4.0.36 from the web, depending on your Linux distribution. Earlier versions should work as well.

Mellvik, 2021-11-30:
Thank you @ghaerr.
`brew install mtools` is my new friend.

However, being the hard-to-satisfy customer, I'm still not happy.

While the 1200KB 5.25in floppy format is arguably the second most common for todays vintage systems - after the 1.4M 3.5in format, it is missing from the `Build extra binary images` collection.

Admittedly - I need both 1.4M and 1.2M images every day. Thus I'm requesting that the 1.2M format is invited to the party.

-M

PR #1034 (2021-11-28): [images] Fix build for 360k and 720k disk images
Fixes #1033.

Removes /bin/sash to make space for all the applications for 360k and 720k disks. Because of the enhancements made to networking, the 720k distribution has run out of space. 

For now, removing /bin/sash (~36k bytes) is far easier than juggling/removing other applications to make space, since /bin/sh can do almost everything /bin/sash can. I think it's probably time for "selectable" compressed applications, which would allow most applications to be compressed on the 360k and 720k distributions, instead of removing functionality lik [... cut]