ghaerr/elks #1042: Network startup in multi machine configurations

Mellvik, 2021-12-07:
Now that startup issues (#1041 ) are on the  table:

Currently, the `/bin/net` command needs to be user edited in order to add ELKS machines to a network. While this (still) it not something that happens very often, I suggest it be changed to something more manageable (and yes, I'm banging my head against this one every day).

- Change /bin/net to use /etc/hosts (not contributing to boot speed, but then again ...)
- Add lines for a 2nd and 3rd system in `/etc/hosts` (e.g. elks1, elks2 ...), so that activating a new machine is a matter of moving the `elks` name to a new line in `/etc/hosts`.
- An alternative is to put either a name or an explicit address in `/bootopts` - but `bootopts` is about to get overloaded, isn't it?

```
127.0.0.1      localhost
10.0.2.15       elks1 elks 
10.0.2.16       elks2
10.0.2.17       elks3
10.0.2.2        gateway
255.255.255.0   netmask
```

ghaerr, 2021-12-07:
Hello @Mellvik,

All good ideas. I've got an idea that should solve this problem nicely: implement a HOSTNAME= environment variable. With a few mods, which I'll post in a PR, the following scenarios would all work, with say HOSTNAME=elks2 or HOSTNAME=10.0.2.16:
- Modify ktcp to use the HOSTNAME environment variable (as well as GATEWAY and NETMASK), as hostname, gateway and NETMASK if found in the environment.
- HOSTNAME= in /bootopts. Any capitalized environment variable name would be passed through to user applications by /bin/login.
- HOSTNAME= in /etc/profile. Now that this file is sourced by /etc/rc.d/rc.sys, /etc/profile could be used instead of /bootopts for setting the system hostname/IP address and gateway.
- With HOSTNAME= not specified, the IP associated with 'elks' would be used in /etc/hosts.
- The /bootopts or /etc/profile methods could also be used to set the local timezone using TZ=, which allows for easier customization using only /bootopts.

Overall, this mechanism would allow for system-wide settings to be specified either in /bootopts or /etc/profile, allowing one a single file change to setup important aspects of system networking and timezone.

In the future, other important system-wide settings could be implemented merely by the capitalized environment variable mechanism used in /bootopts. Any string containing '=' in /bootopts is already created as an environment variable string; for instance 'net=eth' is passed through to rc.sys for network physical link layer selection. The difference will be that all variable names that are capitalized will also be passed through to the shell and login environment.

Mellvik, 2021-12-07:
Thank you, @ghaerr -
Yes, this is great. Universal and simple.

Much appreciated.

--M

closed 2021-12-08 by ghaerr