ghaerr/elks #1043: [kernel] [ehancement] Confusing error messages when running out of file descriptors

Mellvik, 2021-12-07:
When ELKS runs out of file descriptors (in my case a missing `close()` in the code), the error messages leave little clue as to what is the problem:
```
VFS: iput: trying to free free inode                                            
VFS: device 0000, inode 329, mode=0140000     
```
Also, if attempting to open a socket in this situation:
```                                  
socket error: Unknown error 22      
```

If this is an easy fix, it may be worthwhile.

--M

ghaerr, 2021-12-07:
@Mellvik,

I'll look into the VFS error. The reason the "socket error" message shows Unknown error is that `perror` on ELKS tries to open /lib/liberror.txt to get the error message strings, but can't because there are no more file descriptors. It then prints a generic message.

Thank you!

Mellvik, 2021-12-07:
Thank you @ghaerr,
No rush on this one - it's primary for the record.

I don't have an organized bug-list like yours....

-M

ghaerr, 2021-12-08:
@Mellvik,

> When ELKS runs out of file descriptors (in my case a missing close() in the code)

I wrote a program that opens 25 files, after which `open` returns -1. Can't get this problem to duplicate. Can you forward code showing what you did?

Thank you!

Mellvik, 2021-12-08:
Thanks again for looking into this, @ghaerr.

Interesting - and again this will be easier when I've posted the ftp program.

I don't know if it will be useful, but here's the routine in which I got the problem. Which eventually was solved by the close(fd) at the end. It's also worth noting that the system has many processes running - 3 gettys, all daemons. And possibly a FAT filesystem mounted.

-M

int do_get(int controlfd, char *file1, char *file2, int mode) {
        char iobuf[IOBUFLEN+1];
        int fd, n, datafd;
        int maxfdp1, data_finished = FALSE, control_finished = FALSE;
        fd_set rdset;
        char expect;

        bzero(iobuf, sizeof(iobuf));
        if (*file2 == '\0') file2 = file1;

        if (mode == PASV)
                datafd = do_passive(controlfd);
        else
                datafd = do_active(controlfd);
        if (datafd < 0) return -1;

        sprintf(iobuf, "RETR %s\r\n", file1);

        if (debug) printf("%s", iobuf);
        if ((fd = open(file2, O_WRONLY|O_TRUNC|O_CREAT, 0664)) < 0) {
                printf("GET: Cannot open local file '%s'\n", file2);
                return -1;
        }

        write(controlfd, iobuf, strlen(iobuf));
        if (mode == PORT) {     /* active mode, wait for connection */
                if ((n = dataconn(datafd)) < 0) {
                        printf("Active mode connect failed.\n");
                        return -1;
                }
                datafd = n;
        }

        FD_ZERO(&rdset);
        FD_SET(controlfd, &rdset);
        FD_SET(datafd, &rdset);

        maxfdp1 = MAX(controlfd, datafd) + 1;
        printf("Remote: %s, local: %s\n", file1, file2);

        expect = '1'; /* expect '150 Opening connection' etc. */
        while (1) {
                if (control_finished == FALSE) FD_SET(controlfd, &rdset);
                if (data_finished == FALSE) FD_SET(datafd, &rdset);
                select(maxfdp1, &rdset, NULL, NULL, NULL);

                //bzero(iobuf, (int)sizeof(iobuf));
                if (FD_ISSET(controlfd, &rdset)) {
                        get_reply(controlfd, iobuf, IOBUFLEN);
                        //printf("iobuf len %d, diff %d\n", strlen(iobuf), (strlen(strchr(iobuf, '\n')+1)));
                        if (*iobuf != expect) {
                                printf("Error: %s" , &iobuf[4]);
                                break;
                        }
                        if (debug) printf("%s", iobuf);
                        /* The second part of the if below is for the odd condition
                         * that the 'transfer started' and the 'transfer complete'
                         * are read in the same operation. This happens all the time
                         * when the client is slow and the files are small.
                         */
                        if (expect == '2' || (strlen(strchr(iobuf, '\n')+1) > 0)) {
                                control_finished = TRUE;
                                FD_CLR(controlfd, &rdset);
                        } else
                                expect = '2';   /* 226 Transfer complete */
                }

                if (FD_ISSET(datafd, &rdset)) {
                        while ((n = read(datafd, iobuf, IOBUFLEN)) > 0) {
                                if (write(fd, iobuf, n) != n) {
                                        perror("File write error");
                                        break;
                                        // MAY have to reset data connection
                                }
                        }
                        data_finished = TRUE;
                        FD_CLR(datafd, &rdset);
                }
                if ((control_finished == TRUE) && (data_finished == TRUE))
                        break;

        }
        close(datafd);
        close(fd);
        return 1;
}

ghaerr, 2021-12-08:
> if ((fd = open(file2, O_WRONLY|O_TRUNC|O_CREAT, 0664)) < 0) {

I tried the above open 25 times and still no duplication.

When you get the ftp client finished (and posted), please test this problem by commenting out the close(fd). If it duplicates, I will take it from there by doing the same.

Thank you!

closed 2022-01-05 by Mellvik