ghaerr/elks #1046: [net] ktcp timeout/RST panic

Mellvik, 2021-12-08:
A somewhat hard to recreate problem that I've seen a few times of the past couple of week. Here's one scenario:
- The (new) `ftp` client is left idle - the command channel is open.
- Most servers don't like this, they timeout - and send a message to that effect: 
```
11:34:05.983731 IP (tos 0x10, ttl 64, id 11345, offset 0, flags [DF], proto TCP (6), length 96)
    10.0.2.2.ftp > 10.0.2.15.1620: Flags [P.], cksum 0x1863 (incorrect -> 0x5a02), seq 10987:11043, ack 3150, win 29200, length 56: FTP, length: 56
	421 Timeout (900 seconds): closing control connection.
11:34:05.986030 IP (tos 0x0, ttl 64, id 19983, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1620 > 10.0.2.2.ftp: Flags [.], cksum 0xc9f0 (correct), ack 11043, win 4324, length 0
11:34:06.009972 IP (tos 0x10, ttl 64, id 11346, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.ftp > 10.0.2.15.1620: Flags [F.], cksum 0x182b (incorrect -> 0x68c3), seq 11043, ack 3150, win 29200, length 0
11:34:06.012159 IP (tos 0x0, ttl 64, id 19984, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1620 > 10.0.2.2.ftp: Flags [.], cksum 0xc9ef (correct), ack 11044, win 4324, length 0
```
It would be convenient if `ktcp` acknowledged the FIN from the server and closed the connection properly even if the application is still running and the file descriptor open.
What happens when the client is 'reactivated', i.e. a command is entered, is as follows:
```
ftp> pwd                                                                             
tcp: RST from 10.0.2.2:21->1620                                                 
tcpdev_read: returning -EPIPE to socket read state 10                                                                                                     
ftp> bye                                                                        
ktcp: panic in read                                                             
```
The final exchange on the wire is:
```
12:56:07.950236 IP (tos 0x0, ttl 64, id 19985, offset 0, flags [none], proto TCP (6), length 45)
    10.0.2.15.1620 > 10.0.2.2.ftp: Flags [P.], cksum 0x2b7e (correct), seq 3150:3155, ack 11044, win 4324, length 5: FTP, length: 5
	PWD
12:56:07.950456 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.ftp > 10.0.2.15.1620: Flags [R], cksum 0x274e (correct), seq 1807416926, win 0, length 0
```
It seems the ktcp panic is triggered by a close() call.
The EPIPE return from tcpdev_read is probably OK and can (should) be caught by the application, it is the panic that is troublesome.

-m

ghaerr, 2021-12-08:
> It would be convenient if ktcp acknowledged the FIN from the server and closed the connection properly even if the application is still running and the file descriptor open.

From the trace, it shows that ktcp does ACK the FIN, which is correct, and should place the socket into CLOSE_WAIT state. CLOSE_WAIT allows one-way communication, and waits for the client to close the socket. I'm not following you on "closed the connection properly": the socket is now "half-closed", which is a normal TCP state. TCP won't notify the application of a half-closed socket and the file descriptor on the client side is still open. The CLOSE_WAIT status could be checked by running netstat on another terminal after timeout from the server.

It would seem that the ftp client should close its end of the socket when receiving the "421 Timeout (900 seconds): closing control connection." message. This would probably fix the whole problem (although the "panic read" is definitely a bug).

> ftp> pwd                                                                             
tcp: RST from 10.0.2.2:21->1620

The received RST is correct, since the socket is closed on the server side, and the ftp client wrote to a fully-closed socket on the server.

However, the RST will cause ktcp to fully close the socket on ELKS side. I suspect that ktcp is not notifying the kernel in the case that the socket is half-closed going to full-closed, and that's the reason for the panic.

>  tcpdev_read: returning -EPIPE to socket read state 10        

State 10 is TIME_WAIT, so -EPIPE is correct. It seems that the RST did cause ktcp to close it's half of the socket, and its now in TIME_WAIT state.
                                                                                             
> ftp> bye                                                                        
ktcp: panic in read                     

Agreed this is a bug. Perhaps ktcp is not notifying the kernel that the socket was closed. I'll look into this.

> It seems the ktcp panic is triggered by a close() call.

Would that be the close by your ftp client? I don't have source, so hard to follow what it's doing. Does your client close after getting the -EPIPE on a read (or select)?

> The EPIPE return from tcpdev_read is probably OK and can (should) be caught by the application, it is the panic that is troublesome.

Agreed.

Mellvik, 2021-12-08:
Thank  you @ghaerr,

that was fast!
> ftp> bye
> ktcp: panic in read
>
> Agreed this is a bug. Perhaps ktcp is not notifying the kernel that the socket was closed. I'll look into this.
>
> It seems the ktcp panic is triggered by a close() [...]
Yes, a 'bye' command will close the command channel, regardless of state, and exit. However, before that it will try to send a QUIT command on the command channel which is now gone. That write may also be the trigger of the panic. Source is coming - it's in the 'oh sh…, there's another one' state.
> Does your client close after getting the -EPIPE on a read (or select)?
> 
That depends. If it's a data channel error, it will emit an error, close the file descriptor and go back to the ftp-prompt. If it's the command channel (which we're talking about in this case), checking return codes is more lax (at this point).

—M

ghaerr, 2021-12-08:
Hello @Mellvik,

I'm looking at the ktcp code that handles RST processing in tcp.c:
```
    if (h->flags & TF_RST) {
        /* TODO: Check seqnum for security */
        printf("tcp: RST from %s:%u->%u\n",
            in_ntoa(cb->remaddr), ntohs(h->sport), ntohs(h->dport));
        rmv_all_retrans_cb(cb);

        if (cb->state == TS_CLOSE_WAIT) {
            cbs_in_user_timeout--;
            ENTER_TIME_WAIT(cb);
            tcpdev_sock_state(cb, SS_UNCONNECTED);      /* no wakeup?*/ <--- change to SS_DISCONNECTING
        } else {
            tcpdev_sock_state(cb, SS_DISCONNECTING);    /* wakes up process*/
            tcpcb_remove_cb(cb);        /* deallocate*/
        }
        return;
    }
```
This seems exactly as I described above: the socket state is CLOSE_WAIT, as the remote side closed their end. Then, the RST comes in, ktcp sets the socket to TIME_WAIT, but sets the socket state to SS_UNCONNECTED, which does NOT wakeup the client socket. The comment suggests there might want to be a wakeup. 

If you change it to SS_DISCONNECTING, that will wakeup the client when this occurs, (which will be from select I think), and that should fix the problem. If you could try this and let me know, that would be great.

The "panic in read" occurs when a read is attempted on a closed socket, the above fix may not fix that. That will depend on exactly where the ftp client is (in select vs in read) waiting when the disconnect occurs. If it is in read, then ktcp should return -EPIPE. If in select and is woken up, your code may attempt a subsequent read which would produce the panic. We may have to add code checking for SS_DISCONNECTED in `sock_read`. If this is too hard to follow, I can produce a PR that hopefully does the right thing. It would be nice if we could reproduce this on QEMU.

Thank you!

Mellvik, 2021-12-08:
Thanks again, @ghaerr - 

very thorough indeed.

Actually, the application is waiting ro user input, not for something from the net, so from your explanation the case is definitely a 'write, then read on a closed socket panic'.

If it still makes sense to do the proposed tests, let me know!

-M

ghaerr, 2021-12-08:
>  so from your explanation the case is definitely a 'write, then read on a closed socket panic'.

In that case, checking the return value from write should give you -EPIPE, and you can close the socket, etc and skip the socket read which is causing the panic.

> If it still makes sense to do the proposed tests, let me know!

Definitely. You should add the suggested SS_DISCONNECTING fix as described, which will more properly handle the close in the kernel when a RST is received. It also looks like this should fix any subsequent read by returning 0 (not -EPIPE, but still enough for an application to know that the socket has been closed on the remote end).

After you get everything working, you can submit the tcp.c change along with your ftp client PR.

Mellvik, 2021-12-08:
Thank you @ghaerr,

will do that.

Much appreciated.

-M

ghaerr, 2021-12-08:
Here's a patch that fixes both the remote close notification as well as properly handling subsequent reads if the socket is closed. This second fix does the proper thing is the write return code wasn't checked.

Add this to your version, and when you get everything working, submit it along with the ftp client.
```
diff --git a/elks/net/socket.c b/elks/net/socket.c
index ca8101e5..2ee41ddb 100644
--- a/elks/net/socket.c
+++ b/elks/net/socket.c
@@ -148,6 +148,9 @@ static size_t sock_read(struct inode *inode, struct file *file,
     if (sock->flags & SF_ACCEPTCON)
        return -EINVAL;
 
+    if (sock->state != SS_CONNECTED)
+       return -EPIPE;
+
     if (size == 0)
        return 0;
 
diff --git a/elkscmd/ktcp/tcp.c b/elkscmd/ktcp/tcp.c
index eb0560f6..7e159d44 100644
--- a/elkscmd/ktcp/tcp.c
+++ b/elkscmd/ktcp/tcp.c
@@ -274,7 +274,7 @@ if (!cb->sock) { debug_accept("tcp established: NULL SOCKET\n"); }
        if (cb->state == TS_CLOSE_WAIT) {
            cbs_in_user_timeout--;
            ENTER_TIME_WAIT(cb);
-           tcpdev_sock_state(cb, SS_UNCONNECTED);      /* no wakeup?*/
+           tcpdev_sock_state(cb, SS_DISCONNECTING);    /* wakes up process*/
        } else {
            tcpdev_sock_state(cb, SS_DISCONNECTING);    /* wakes up process*/
            tcpcb_remove_cb(cb);        /* deallocate*/
```

Mellvik, 2021-12-10:
@ghaerr,
While these patches are still part of my testing, it may be helpful for further thinking that the `socket.c` patch introduces a problem at the end of file transfers, as evidenced but the following:

```
# ftpget 10.0.2.2 ftp yyyy xxxxxxxxx > ftp                                     
ftpget: File write error                                                        
```
The actual file transfers fine (error free), so it seems the final `read` is giving an unexpected response.
Removing the change in `socket.c`  gets us back to normal.

I'll be back with details on the timeout and too many open files issues.

--M

ghaerr, 2021-12-10:
> The actual file transfers fine (error free), so it seems the final read is giving an unexpected response.
Removing the change in socket.c gets us back to normal.

I suppose that's what I get for submitting an untested change. Go ahead and remove the socket.c change permanently. I think what's happening is that the final FIN is received, which is marking the socket unconnected, and setting the SF_CLOSING flag. Ktcp later uses that flag to return 0 for a later read. It appears the application read hasn't been issued prior to the FIN received so we can't use socket connected status as I coded.

This shows the complexity of ELKS kernel/ktcp sync issues, which in this case may result in allowing possibly improperly coded applications (for instance, forgotten close) to produce problems which we can't easily fix. But we will fix properly coded programs to work correctly! :)

Mellvik, 2021-12-11:
@ghaerr,
> I suppose that's what I get for submitting an untested change.
> 
Your remote debugging track record is rather remarkable. No one expects every attempt to be successful.
> Go ahead and remove the socket.c change permanently. I think what's happening is that the final FIN is received, which is marking the socket unconnected, and setting the SF_CLOSING flag. Ktcp later [...]
Indeed. An opportunity for learning for all. Would you take a quick look at urlget/ftpget to see if there is something that should be done differently?

Thank you!
-M

ghaerr, 2021-12-11:
> Would you take a quick look at urlget/ftpget to see if there is something that should be done differently?

The file transfers are effected with a tight "read socket, write file" (or vice versa) routine, which is pretty basic. I can't see any changes that could help.

I think the original issue you are having can be fixed with just the SS_DISCONNECTING change. I added the socket.c check as an afterthought while looking at the source. I propose we make as few changes as possible to our working TCP stack; I think the single SS_DISCONNECTED change should fix our issue here. Going too much deeper into this will result in more troubles, I believe, since ktcp doesn't differentiate between full and half duplex sockets, really.

So - continue testing with the latter change only and we'll see whether this resolves the "panic read" issue.

Mellvik, 2021-12-12:
Thank you @ghaerr,
Not unusual for the season, ELKS is getting a lot of competition for time and attention.
> Would you take a quick look at urlget/ftpget to see if there is something that should be done differently?
>
> The file transfers are effected with a tight "read socket, write file" (or vice versa) [...]
Looks good so far, but admittedly a couple of other issues popped up - possibly requiring their own thread, but let me start here anyway.
1) This is a new variant of the 'telnet requires extra newline' problem you fixed a while back. When telnetting into ELKS, then ftp out from ELKS, and doing a DIR command, there is no output until an extra newline is entered. This happens ONLY if the DIR listing requires more than one packet. Smaller listings (<1460) work fine. When directly connected to ELKS (serial or console) there is no problem. Telnet in, then out from ELKS works fine. I'm going to submit a packet trace and ^P trace which hopefully can shed some more light on what's going on,

2) This is really a question about select(). Consider the following case: The ftp client is requesting a file from the server. Connection (passive or active) is properly set up, the transfer request sent, and incidentally, the file has zero bytes, so the server has nothing to send, and (correctly) closes the connection and sends a transfer complete message on the command channel. If the ftp client was reading from the data channel socket directly, it would get a -1 and we would be ok, but select() seems to be stuck wating for data to become available. Did I miss something? BTW the 'transfer complete' on the command channel cannot be used as a flag to catch the situation because it Usually arrives before ELKS has had time to consume all the qued up data.
Again, this is a question, not a bug, It may be a bad idea to use select in this case.

--Mellvik

ghaerr, 2021-12-12:
Hi @Mellvik,

I haven't had time today to look into these, but should have time tomorrow.

> there is no output until an extra newline is entered. 

Just a carriage return, or any character? Or are you talking about a literal newline, meaning ctrl-J?
This sounds very much like the original problem, which turned out not to be PTY but instead related to some packets getting dropped, then the server timing out and and extra ACK sent, which started things working again. If any character will start it working, (as did before) then it's the client sending another packet that caused the resync.

If this is newline only (carriage return or ^J), then this could be a different issue. You might want to turn on ^P and do a packet trace, just to see what's happening on the line when the timeout occurs.

If this is happening ONLY when a ftp is also running, we likely have a networking problem, not sure. Are you saying that you're using your new ftp client from within an inbound telnet connection to cause this problem? If so, yes, we probably need a packet trace, and we need to be sure the code that displays dropped packets is turned on (should be with ^P).

I'll look at your more complex select issue shortly, thanks!

Mellvik, 2021-12-12:
@ghaerr,
Some times an attempt to be unambiguous turns out to be the exact opposite. I chose 'newline' hoping that it would be better than 'enter' or 'return'. Now I'm all smiles knowing that the attempt was futile from the start.
The totally unambiguous expression then, turns out to be 'hitting the enter button', so that's what I'm referring to. And please don't worry about this till I get the traces submitted. We're all tight on time these days.

-M

Mellvik, 2021-12-13:
Having looked at the trace, it's clear that this is a new issue. So in order to keep things separate I'm creating one.

Also - the `ktcp panic` seems to be handled well by the change proposed by @ghaerr in https://github.com/jbruchon/elks/issues/1046#issuecomment-988990338 
above. I'll keep this issue open  for a little longer  though - for proper verification.

 
--M

Mellvik, 2021-12-13:
Output after the server has timed out. Looks good.

```
ftp> quit                                                                       
tcp: RST from 10.0.2.2:21->1033                                                 
tcpdev_read: returning -EPIPE to socket read state 10                           
Closing: quit                                                                   
# 
```

-M

ghaerr, 2021-12-13:
@Mellvik,

Glad this is fixed. You can include the kernel fix in your ftp client PR, or separately, thanks!