ghaerr/elks #1048: [net] ftp/telnet not working well together.

Mellvik, 2021-12-13:
The preamble to this issue can be found in #1046 .

**Summary:**
When telnetting into ELKS, then FTPing out of ELKS and doing a DIR command (long directory listing), the output from the listing hangs until a character is entered in `telnet`. This only happens when the listing is large, requiring more than one, probably many ( 3-4 ) packets. Small listings work as expected.

Looking closely at the packet trace, this is what seems to happen:
- When the `FTP DIR` (`LIST`) command is sent to the server, it immediately responds with as many packets as ELKS can eat.
- This happens too fast for the `telnetd` process to take away anything from the input buffer, so the receiving window on the `ftp` data channel shrinks and eventually ends up at zero.
- When the data receive window reaches 0, everything stops. The server keeps retransmitting its last ack packet and everything is at a standstill until something is entered into `telnet`, which apparently wakes the `telnetd` process so it starts eating off of the input buffer and regular operation resumes.

So - viewed from the outside, it looks like the FTP data receive window reaching zero prevents a wakeup of the process waiting to read the data.

@ghaerr - I suspect this may ring a bell already, let me know if there are supplementary tests the may be useful. I'm not sure this can be recreated using QEMU given that timing is a factor. I might repeat the test using ELKS as the `ftp server`, which would probably work just fine - but the other system was unavailable at the point of testing.

Trace below.

-M

```
10:50:24.019227 IP (tos 0x0, ttl 64, id 711, offset 0, flags [none], proto TCP (6), length 46)
    10.0.2.15.1027 > 10.0.2.2.ftp: Flags [P.], cksum 0x1523 (correct), seq 46:52, ack 534, win 4380, length 6: FTP, length: 6
	LIST
10:50:24.021080 IP (tos 0x10, ttl 64, id 24803, offset 0, flags [DF], proto TCP (6), length 95)
    10.0.2.2.ftp > 10.0.2.15.1027: Flags [P.], cksum 0x1862 (incorrect -> 0xb8c7), seq 534:589, ack 52, win 29200, length 55: FTP, length: 55
	150 Opening ASCII mode data connection for '/bin/ls'.
10:50:24.022817 IP (tos 0x0, ttl 64, id 712, offset 0, flags [none], proto TCP (6), length 87)
    10.0.2.15.telnet > 10.0.2.2.36264: Flags [P.], cksum 0x7aaf (correct), seq 25321:25368, ack 190, win 4380, length 47
10:50:24.022987 IP (tos 0x10, ttl 64, id 53069, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36264 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0xd8bb), ack 25368, win 57888, length 0
10:50:24.026174 IP (tos 0x0, ttl 64, id 713, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1027 > 10.0.2.2.ftp: Flags [.], cksum 0xc1d2 (correct), ack 589, win 4325, length 0
10:50:24.029548 IP (tos 0x0, ttl 64, id 714, offset 0, flags [none], proto TCP (6), length 52)
    10.0.2.15.telnet > 10.0.2.2.36264: Flags [P.], cksum 0x848a (correct), seq 25368:25380, ack 190, win 4380, length 12
10:50:24.029706 IP (tos 0x10, ttl 64, id 53070, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36264 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0xd8af), ack 25380, win 57888, length 0
10:50:24.033253 IP (tos 0x0, ttl 64, id 715, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1027 > 10.0.2.2.ftp: Flags [.], cksum 0xc19b (correct), ack 589, win 4380, length 0
10:50:24.042283 IP (tos 0x0, ttl 64, id 716, offset 0, flags [none], proto TCP (6), length 96)
    10.0.2.15.telnet > 10.0.2.2.36264: Flags [P.], cksum 0xfe92 (correct), seq 25380:25436, ack 190, win 4380, length 56
10:50:24.042450 IP (tos 0x10, ttl 64, id 53071, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36264 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0xd877), ack 25436, win 57888, length 0
10:50:24.046957 IP (tos 0x0, ttl 64, id 717, offset 0, flags [none], proto TCP (6), length 42)
    10.0.2.15.telnet > 10.0.2.2.36264: Flags [P.], cksum 0x9c68 (correct), seq 25436:25438, ack 190, win 4380, length 2
10:50:24.047075 IP (tos 0x10, ttl 64, id 53072, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36264 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0xd875), ack 25438, win 57888, length 0
10:50:24.051663 IP (tos 0x8, ttl 64, id 59661, offset 0, flags [DF], proto TCP (6), length 1500)
    10.0.2.2.46803 > 10.0.2.15.1028: Flags [.], cksum 0x1ddf (incorrect -> 0xbbaa), seq 1:1461, ack 1, win 29200, length 1460
10:50:24.051810 IP (tos 0x8, ttl 64, id 59662, offset 0, flags [DF], proto TCP (6), length 1500)
    10.0.2.2.46803 > 10.0.2.15.1028: Flags [P.], cksum 0x1ddf (incorrect -> 0xb21d), seq 1461:2921, ack 1, win 29200, length 1460
10:50:24.051978 IP (tos 0x8, ttl 64, id 59663, offset 0, flags [DF], proto TCP (6), length 1216)
    10.0.2.2.46803 > 10.0.2.15.1028: Flags [P.], cksum 0x1cc3 (incorrect -> 0x2301), seq 2921:4097, ack 1, win 29200, length 1176
10:50:24.056281 IP (tos 0x10, ttl 64, id 24804, offset 0, flags [DF], proto TCP (6), length 64)
    10.0.2.2.ftp > 10.0.2.15.1027: Flags [P.], cksum 0x1843 (incorrect -> 0x9191), seq 589:613, ack 52, win 29200, length 24: FTP, length: 24
	226 Transfer complete.
10:50:24.064931 IP (tos 0x0, ttl 64, id 718, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1028 > 10.0.2.2.46803: Flags [.], cksum 0x03ec (correct), ack 1461, win 2920, length 0
10:50:24.077113 IP (tos 0x0, ttl 64, id 719, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1028 > 10.0.2.2.46803: Flags [.], cksum 0x03ec (correct), ack 2921, win 1460, length 0
10:50:24.087015 IP (tos 0x0, ttl 64, id 720, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1028 > 10.0.2.2.46803: Flags [.], cksum 0x03ec (correct), ack 4097, win 284, length 0
10:50:24.094323 IP (tos 0x0, ttl 64, id 721, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1028 > 10.0.2.2.46803: Flags [.], cksum 0xfe17 (correct), ack 4097, win 1776, length 0
10:50:24.094433 IP (tos 0x8, ttl 64, id 59664, offset 0, flags [DF], proto TCP (6), length 1500)
    10.0.2.2.46803 > 10.0.2.15.1028: Flags [.], cksum 0x1ddf (incorrect -> 0x393d), seq 4097:5557, ack 1, win 29200, length 1460
10:50:24.111084 IP (tos 0x0, ttl 64, id 722, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1027 > 10.0.2.2.ftp: Flags [.], cksum 0xc19b (correct), ack 613, win 4356, length 0
10:50:24.123063 IP (tos 0x0, ttl 64, id 723, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1028 > 10.0.2.2.46803: Flags [.], cksum 0xfe17 (correct), ack 5557, win 316, length 0
10:50:24.352536 IP (tos 0x8, ttl 64, id 59665, offset 0, flags [DF], proto TCP (6), length 356)
    10.0.2.2.46803 > 10.0.2.15.1028: Flags [P.], cksum 0x1967 (incorrect -> 0x3e0d), seq 5557:5873, ack 1, win 29200, length 316
[... 12 lines cut ...]
10:50:38.512544 IP (tos 0x8, ttl 64, id 59671, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.46803 > 10.0.2.15.1028: Flags [.], cksum 0x182b (incorrect -> 0x8c08), ack 1, win 29200, length 0
10:50:43.008848 IP (tos 0x10, ttl 64, id 53073, offset 0, flags [DF], proto TCP (6), length 42)
    10.0.2.2.36264 > 10.0.2.15.telnet: Flags [P.], cksum 0x182d (incorrect -> 0xcb2b), seq 190:192, ack 25438, win 57888, length 2
10:50:43.011217 IP (tos 0x0, ttl 64, id 725, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36264: Flags [.], cksum 0xa97a (correct), ack 192, win 4378, length 0
10:50:43.034268 IP (tos 0x0, ttl 64, id 726, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36264: Flags [.], cksum 0xa978 (correct), ack 192, win 4380, length 0
10:50:43.075529 IP (tos 0x0, ttl 64, id 727, offset 0, flags [none], proto TCP (6), length 552)
    10.0.2.15.telnet > 10.0.2.2.36264: Flags [P.], cksum 0x1bd8 (correct), seq 25438:25950, ack 192, win 4380, length 512
10:50:43.075623 IP (tos 0x10, ttl 64, id 53074, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36264 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0xd243), ack 25950, win 58960, length 0
10:50:43.078411 IP (tos 0x0, ttl 64, id 728, offset 0, flags [none], proto TCP (6), length 50)
    10.0.2.15.telnet > 10.0.2.2.36264: Flags [P.], cksum 0x9a20 (correct), seq 25950:25960, ack 192, win 4380, length 10
10:50:43.078500 IP (tos 0x10, ttl 64, id 53075, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36264 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0xd239), ack 25960, win 58960, length 0
[... cut ...]
    10.0.2.2.36264 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0xcc09), ack 26472, win 60032, length 0
10:50:43.123162 IP (tos 0x0, ttl 64, id 730, offset 0, flags [none], proto TCP (6), length 50)
    10.0.2.15.telnet > 10.0.2.2.36264: Flags [P.], cksum 0x6555 (correct), seq 26472:26482, ack 192, win 4380, length 10
10:50:43.123249 IP (tos 0x10, ttl 64, id 53077, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36264 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0xcbff), ack 26482, win 60032, length 0
```

ghaerr, 2021-12-13:
@Mellvik,

Thanks for the very complete bug description and trace. I am traveling, and probably can't get too deep in the trace yet, but it sounds like your diagnosis is likely very close to the problem. When you say "This happens too fast for the `telnetd` process to take away anything from the input buffer", I assume you actually mean `ftpd`, right?

I am guessing that, also without running ftp outbound inside telnet, when receiving a long dir command the ktcp receive window very quickly goes to 0 as well. That would be interesting to test. When ktcp is at receive 0, either the remote side needs to "ping" to ask about an updated window size, or ktcp needs to advertise a larger window, in order things to start up again. IIRC, ktcp sends this "extra" ACK (opening the window), only when the application (ftp client in this case) actually reads the data from ktcp. 

It is possible that for some reason, ktcp and/or telnet is hung up in a read and ktcp can't or isn't processing the ftp client read. What we're dealing with is the complexities of ktcp being an ELKS application, and it has to be scheduled to run (as a normal application) to process TCP requests. There could be a deadlock situation, I'm not sure. It seems you're saying that when telnet completes a read, then everything starts up again. 

Is the ftp client also doing I/O (i.e. printf to stdout), during the transfer? That might change things if it were/weren't.

Further testing by setting DEBUG_WINDOW in ktcp config.h and running ^P would help see if this is the case, as that would show when the application read ACK (re-opening the window) occurs. This would be seen as "tcp: extra ACK...".

Mellvik, 2021-12-13:
Thank you @ghaerr,
No rush, enjoy your travelling.

> When you say "This happens too fast for the telnetd process to take away anything from the input buffer", I assume you actually mean ftpd, right?
> 
No, I'm actually talking about telnetd (elks ftpd is not involved). The data arriving to the ftp client which is running in a telnet(d)process, don't get read until telnetd wakes up (my take).
> I am guessing that, also without running ftp outbound inside telnet, when receiving a long dir command the ktcp receive window very quickly goes to 0 as well. That would be interesting to test.
> 
I've tested this briefly and (surprisingly) not been able to replicate the problem.
> When ktcp is at receive 0, either the remote side needs to "ping" to ask about an updated window size, or ktcp needs to advertise a larger window, in order things to start up again. IIRC, ktcp sends [...]
Yes, this is spot on. For some reason telnetd is not scheduled to run,blocked - or missing a tick, and we have a deadlock. And it seems there is an interplay between ftp and telnetd that triggers this situation. Running other tests involving a telnet client and long listings does not create a situation that completely fills the socket data buffer with max size packets in 'no time'. It may be possible to create a similar situation using tcpget (with nc on the server side)from a telnet session...
> Is the ftp client also doing I/O (i.e. printf to stdout), during the transfer? That might change things if it were/weren't.
> 
No, not naturally so (the data is a ls -l listing which runs continuously), but that may be an interesting experiment.
> Further testing by setting DEBUG_WINDOW in ktcp config.h and running ^P would help see if this is the case, as that would show when the application read ACK (re-opening the window) occurs. This would be seen as "tcp: extra ACK...".
> 
Will take look at that, thanks.

-M

Mellvik, 2021-12-22:
@ghaerr,
I just found a new example that triggers this problem, and is easily reproducible in QEMU:
`telnet` into elks and do a `vi /etc/inittab` (or one other file). It hangs after the enter on the command line - until any character is typed on the console.

So the problem seems unrelated to `ftp` and full packets and zero windows.
I've looked at the packet trace, but not found any hint as to what's going on.

Packet trace included below, starting at the [enter] which sends the `vi` command line.
The open line in the trace is where something is typed, and the output continues.

```
11:29:49.064115 IP (tos 0x10, ttl 64, id 12405, offset 0, flags [DF], proto TCP (6), length 41)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [P.], cksum 0x182c (incorrect -> 0xfc4c), seq 113:114, ack 259975, win 65535, length 1
11:29:49.065831 IP (tos 0x0, ttl 64, id 1814, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [.], cksum 0xf839 (correct), ack 114, win 4379, length 0
11:29:49.085391 IP (tos 0x0, ttl 64, id 1815, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [.], cksum 0xf838 (correct), ack 114, win 4380, length 0
11:29:55.004982 IP (tos 0x0, ttl 64, id 1816, offset 0, flags [none], proto TCP (6), length 43)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [P.], cksum 0xe120 (correct), seq 259975:259978, ack 114, win 4380, length 3
11:29:55.005137 IP (tos 0x10, ttl 64, id 12406, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0x0952), ack 259978, win 65535, length 0
11:29:55.090361 IP (tos 0x0, ttl 64, id 1817, offset 0, flags [none], proto TCP (6), length 44)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [P.], cksum 0xa660 (correct), seq 259978:259982, ack 114, win 4380, length 4
11:29:55.090489 IP (tos 0x10, ttl 64, id 12407, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0x094e), ack 259982, win 65535, length 0
11:29:55.098757 IP (tos 0x10, ttl 64, id 12408, offset 0, flags [DF], proto TCP (6), length 47)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [P.], cksum 0x1832 (incorrect -> 0x2e7d), seq 114:121, ack 259982, win 65535, length 7
11:29:55.103497 IP (tos 0x0, ttl 64, id 1818, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [.], cksum 0xf831 (correct), ack 121, win 4373, length 0
11:29:55.132261 IP (tos 0x0, ttl 64, id 1819, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [.], cksum 0xf82a (correct), ack 121, win 4380, length 0
11:29:55.155922 IP (tos 0x0, ttl 64, id 1820, offset 0, flags [none], proto TCP (6), length 52)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [P.], cksum 0xdc68 (correct), seq 259982:259994, ack 121, win 4380, length 12
11:29:55.168233 IP (tos 0x0, ttl 64, id 1821, offset 0, flags [none], proto TCP (6), length 44)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [P.], cksum 0xa649 (correct), seq 259994:259998, ack 121, win 4380, length 4
11:29:55.168397 IP (tos 0x10, ttl 64, id 12409, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0x0937), ack 259998, win 65535, length 0
11:29:55.180563 IP (tos 0x10, ttl 64, id 12410, offset 0, flags [DF], proto TCP (6), length 48)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [P.], cksum 0x1833 (incorrect -> 0x500c), seq 121:129, ack 259998, win 65535, length 8
11:29:55.185384 IP (tos 0x0, ttl 64, id 1822, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [.], cksum 0xf81a (correct), ack 129, win 4372, length 0
11:29:55.212240 IP (tos 0x0, ttl 64, id 1823, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [.], cksum 0xf812 (correct), ack 129, win 4380, length 0
11:29:55.245053 IP (tos 0x0, ttl 64, id 1824, offset 0, flags [none], proto TCP (6), length 47)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [P.], cksum 0x2742 (correct), seq 259998:260005, ack 129, win 4380, length 7
11:29:55.292505 IP (tos 0x10, ttl 64, id 12411, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0x0928), ack 260005, win 65535, length 0

11:30:18.639494 IP (tos 0x10, ttl 64, id 12412, offset 0, flags [DF], proto TCP (6), length 41)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [P.], cksum 0x182c (incorrect -> 0xfc1e), seq 129:130, ack 260005, win 65535, length 1
11:30:18.641214 IP (tos 0x0, ttl 64, id 1825, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [.], cksum 0xf80b (correct), ack 130, win 4379, length 0
11:30:18.656725 IP (tos 0x0, ttl 64, id 1826, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [.], cksum 0xf80a (correct), ack 130, win 4380, length 0
11:30:18.692138 IP (tos 0x0, ttl 64, id 1827, offset 0, flags [none], proto TCP (6), length 552)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [P.], cksum 0xc50b (correct), seq 260005:260517, ack 130, win 4380, length 512
11:30:18.692239 IP (tos 0x10, ttl 64, id 12413, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0x0727), ack 260517, win 65535, length 0
11:30:18.709633 IP (tos 0x0, ttl 64, id 1828, offset 0, flags [none], proto TCP (6), length 349)
    10.0.2.15.telnet > 10.0.2.2.36286: Flags [P.], cksum 0x3d4f (correct), seq 260517:260826, ack 130, win 4380, length 309
11:30:18.709721 IP (tos 0x10, ttl 64, id 12414, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.36286 > 10.0.2.15.telnet: Flags [.], cksum 0x182b (incorrect -> 0x05f2), ack 260826, win 65535, length 0
1
```

ghaerr, 2021-12-23:
Hello @Mellvik,

> telnet into elks and do a vi /etc/inittab (or one other file). It hangs after the enter on the command line - until any character is typed on the console.

Got it to repeat. This is a very strange problem. More investigation reveals that `vi /etc/inittab` fails but `vi` alone or with a very small file succeeds. Increasing the pty driver queue size changes the results to less frequent. It seems to be related to the amount of immediate output `telnetd` is trying to pump through at one time. Turning on DEBUG_TCP and DEBUG_TCPDATA with ^P caused the problem to go away. Definitely timing related.

Finally debugged it to the point of seeing that the problem is that telnetd is hanging in its own `select` forever (or until more I/O, such as a keypress input received from remote telnet). There must be a race condition of some kind, or a possible deadlock, not sure.

However, seeing that select was hanging indefinitely, I added a 50ms timer (quite slow for CPU) which allows telnetd's select to not hang forever - and the problem went away. This issue could be related to an application having to wait for a pty driver queue to drain, with ktcp and that application hanging, or perhaps a race condition with something like that and telnetd's select call, but bottom line is telnetd itself is hanging, and a slow timer temporarily solves this issue without introducing more errors and very little CPU overhead.

I've pushed PR #1073 for this workaround for the time being, can you please test with the original ftp/telnet combination to see whether this fixes that issue as well?

Note: with extensive testing, even with the workaround, I am still seeing an occasional, but separate bug: that of the telnet character echo being "one behind" when typing into the shell. It's not really one behind, but actually still requiring yet another character to be entered to display the "last" character. I can't duplicate this totally yet, but we should open a new issue for that if you/we see it.

Thank you!

Mellvik, 2021-12-23:
wow @ghaerr, that's another elusive chase. You're definitely closing in on the culprit.

I can confirm that the telnet/ftp combination now works as expected. SO however temporary, the fix does the trick. Feel free to close the issue if you think that's appropriate.

As to the 'one char behind' problem, yes, I've seen that occasionally but never found any pattern to its occurrence. Now that it's on the table, I'll keep a open eye for it, not just log out and log back in.

Thank you.

—M

ghaerr, 2021-12-23:
> I can confirm that the telnet/ftp combination now works as expected. SO however temporary, the fix does the trick.

Thanks for your testing @Mellvik, we're getting there!

Mellvik, 2022-01-07:
New evidence in this half-open case, @ghaerr. This time including the zero file length issue. 

The scenario is as follows: external login to (physical hw) ELKS via `telnet`. Then `ftp` back to some external server. The ftp server host may be different from the telnet host. While still in passive mode, `get` a zero length file from the external (ftp)server. Transfers fine.
Change to active mode, repeat. The transfer completes, but the data connection hangs after the FIN from the server which closes the data connection. The FIN is ack'ed by ELKS, then full stop, and the `ftp` client hangs. In any other situation this works fine. Also, if the file has one byte or more, it works fine. Obviously another timing related issue.

Observations:
- Only observable on physical, not on QMU (active mode outgoing is not supported on QEMU) Unfortunate.
- Not repeatable when doing loopback on ELKS - neither on HW nor QEMU.
- The hang is interruptible, after an interrupt everything is normal, no lingering issues.
- Possibly even more interesting: If left hanging, the (external) server will eventually timeout, and send a message on the command channel, which will release the hang, and ELKS continues as if nothing had happened. IOW the activity on the data channel awakens the dialogue in the same way any character in the `telnet` window did in a previous setting with similar symptoms. BTW - typing anything into the `telnet` window has no effect in this case. 
- The fact that this only happens in active mode, is probably important - the data connection is outbound from ELKS, so we're again in `accept()` somehow.
- Notably: While the `telnet` host may be several hops away from the ELKS machine, the `ftp`-server must be on the same network. Any extra packet delay will cause the problem to go away.
- We're stuck in `select()` (with a 1ms timeout which doesn't kick in).

@ghaerr - can you make any sense of this?

Regular case:
```
ftp> passi
parse_cmd: <passi>
Passive mode off.
ftp> get 1byte
parse_cmd: <get 1byte>
parse_cmd: 1byte
---> TYPE I
200 Type set to I.
PORT: Listening on 10.0.2.15 port 1449
---> PORT 10,0,2,15,5,169
200 PORT command successful.
---> RETR 1byte
150 Opening BINARY mode data connection for '1byte' (1 bytes).
Accepted connection from 10.0.2.2 @ 20
remote: 1byte, local: 1byte
226 Transfer complete.
ftp>                                   <---- back to the command prompt immediately
```
Hanging case:
```
ftp> get xdr                       <------------- zero length file
parse_cmd: <get xdr>
parse_cmd: xdr
---> TYPE I
200 Type set to I.
PORT: Listening on 10.0.2.15 port 1447
---> PORT 10,0,2,15,5,167
200 PORT command successful.
---> RETR xdr
150 Opening BINARY mode data connection for 'xdr' (0 bytes).
Accepted connection from 10.0.2.2 @ 20
remote: xdr, local: xdr
226 Transfer complete.
<--------------- hangs here
```
Packet trace excerpt:
Server attempts to close
```
12:17:06.062532 IP (tos 0x8, ttl 64, id 25068, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.ftp-data > 10.0.2.15.1441: Flags [F.], cksum 0x182b (incorrect -> 0x794c), seq 1, ack 1, win 29200, length 0
12:17:06.064817 IP (tos 0x0, ttl 64, id 10607, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1441 > 10.0.2.2.ftp-data: Flags [.], cksum 0xda40 (correct), ack 2, win 4380, length 0
--------------- hang ---------------
-----------the server times out (900 seconds) -------
12:32:05.857979 IP (tos 0x10, ttl 64, id 36784, offset 0, flags [DF], proto TCP (6), length 96)
    10.0.2.2.ftp > 10.0.2.15.1440: Flags [P.], cksum 0x1863 (incorrect -> 0x306f), seq 651:707, ack 84, win 29200, length 56: FTP, length: 56
	421 Timeout (900 seconds): closing control connection.
12:32:05.862726 IP (tos 0x0, ttl 64, id 10608, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1440 > 10.0.2.2.ftp: Flags [.], cksum 0xa05d (correct), ack 707, win 4324, length 0
12:32:05.885301 IP (tos 0x0, ttl 64, id 10609, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1440 > 10.0.2.2.ftp: Flags [.], cksum 0xa025 (correct), ack 707, win 4380, length 0
12:32:05.885675 IP (tos 0x10, ttl 64, id 36785, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.ftp > 10.0.2.15.1440: Flags [F.], cksum 0x182b (incorrect -> 0x3f30), seq 707, ack 84, win 29200, length 0
12:32:05.895157 IP (tos 0x0, ttl 64, id 10610, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1440 > 10.0.2.2.ftp: Flags [.], cksum 0xa024 (correct), ack 708, win 4380, length 0
12:32:05.897189 IP (tos 0x0, ttl 64, id 10611, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1441 > 10.0.2.2.ftp-data: Flags [F.], cksum 0xda3f (correct), seq 1, ack 2, win 4380, length 0   <--- ELKS finally sends the FIN on the data channel
12:32:05.897436 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.ftp-data > 10.0.2.15.1441: Flags [R], cksum 0xf56b (correct), seq 906389161, win 0, length 0  <---- Server just resets, has closed the data channel long ago
```

ghaerr, 2022-01-07:
Hello @Mellvik,

I'm reopening this issue to track it. Thanks for the excellent analysis, I don't get have any real ideas on where the problem lies, and this won't be easy to debug, since QEMU emulation isn't available. A couple questions:

> The scenario is as follows: external login to (physical hw) ELKS via telnet. Then ftp back to some external server. 

I assume that you can't get it to repeat just ftp'ing out to external server, without telnet involved, correct?

This problem may be related to the problem in #1073, where telnetd would not return from select until an additional event occurred, then everything normal. I was never able to figure out why that occurred, other than a likely kernel race condition between multiple applications and ktcp. The workaround was to install a timeout in telnetd's select().

> We're stuck in select() (with a 1ms timeout which doesn't kick in).

Are you talking about ELKS ftp select? Or ftpd? Does that 1ms timeout otherwise normally occur?

> The fact that this only happens in active mode, is probably important - the data connection is outbound from ELKS, so we're again in accept() somehow.

What do you mean "we're again in accept"? I realize we've had a recent accept bug fix, are you thinking that either ftp or ftpd is in accept when this occurs? Meaning that we could have received a SYN, zero length file transfer, and FIN, all before the waiting application accept was called? We could turn on DEBUG_ACCEPT and DEBUG_TCP to show accept and zero-length data transfers with FIN packets, which were used to debug accept and FIN occurring before a final application read, respectively.

Are you testing this with your updated ftp client? If so, lets go ahead and submit a PR for those changes, so we can get them in to v0.5.0, as this bug may take a while to figure out, because of its 

Thank you!

Mellvik, 2022-01-07:
Thank you @ghaerr,

> 
> The scenario is as follows: external login to (physical hw) ELKS via telnet. Then ftp back to some external server.
> 
> I assume that you can't get it to repeat just ftp'ing out to external server, without telnet involved, correct?
> 
Yes.
> This problem may be related to the problem in #1073 <https://github.com/jbruchon/elks/pull/1073>, where telnetd would not return from select until an additional event occurred, then everything [...]
Agree. And it may be useful to underscore that in this case, the trigger seems to be traffic on the FTP command channel. When stuck, keypushes in telnet get transmitted and ack'ed.
> We're stuck in select() (with a 1ms timeout which doesn't kick in).
> 
> Are you talking about ELKS ftp select? Or ftpd? Does that 1ms timeout otherwise normally occur?
> 
This timeout was introduced specifically to handle the 0 size files, it does occur some times with such files, not always
> The fact that this only happens in active mode, is probably important - the data connection is outbound from ELKS, so we're again in accept() somehow.
>
> What do you mean "we're again in accept"? I [...]
Sorry, I screwed up on this one. I mean the connection is inbound for ELKS, thus the accept() involvement. And no, the client is long past the accept() stage, so my reference to accept() lacks foundation. Pls ignore.

The ftp (and ftpd) PRs are coming. the first as we speak. 

—Mellvik

ghaerr, 2022-01-07:
> Are you talking about ELKS ftp select? Or ftpd? Does that 1ms timeout otherwise normally occur?
> This timeout was introduced specifically to handle the 0 size files

Why exactly do you need a timeout in ftp client select? 

Regardless, this could be useful to possibly solve this issue. The idea being just like telnetd - a small timeout which just loops back to select again in the case of (buggy) non-reported I/O. My question above is whether you can code up ftp client so that it has a select timeout, and use it in the same way that this caused the "push" to keep telnet/telnetd sessions working in #1073.

If ftpd is involved (I think not in this issue, correct?) then we might think of using the same workaround in ftpd.

I'm a little worried about why a timeout is required to process zero-length files, it is possible that could be causing desync issues?

Mellvik, 2022-01-07:
> Are you talking about ELKS ftp select? Or ftpd? Does that 1ms timeout otherwise normally occur?
> This timeout was introduced specifically to handle the 0 size files
> 
> Why exactly do you need a timeout in ftp client select?
> 
Like I said, it was introduced to handle the 0 size file case. I kept the prinf in the loop in order to track it, and it's interesting. Anyway, I'll dig up the explanation from the previous thread tomorrow. Timing has very interesting consequences some times. And frustrating. I sort of feel I've been fighting milliseconds all week :-)
> Regardless, this could be useful to possibly solve this issue. The idea being just like telnetd - a small timeout which just loops back to select again in the case of (buggy) non-reported I/O. My [...]
Interesting - I'll look at that in the morning.
> If ftpd is involved (I think not in this issue, correct?) then we might think of using the same workaround in ftpd.
> 
Right, ftpd is out of the loop on this one.
> I'm a little worried about why a timeout is required to process zero-length files, it is possible that could be causing desync issues?
> 
Like I said, we've been down that road. The timeout is there because the expected input in some cases (0 length files in particular) arrive too early for the logic to work. So the timeout is a flag to go back and look, 'you probably got what you wanted already'. Sort of. Off the top of mye head. I'll look at the code again in the morning.

Actually, this is why i wanted to do the ftp thing from scratch. Not having a completely stable platforme, we were bound to encounter issues where it would be hard know whether it is app level or system level. Knowing the app level to the bone eases the process.

In the end, the ambition with ftp -actually ftpd - is to bring it to a level where we can use it as a virtual (net)file system. We're close. The upcoming enhancements are mostly qemu related, but with support for metadata commands (plus a little), ftpd can deliver that. I we can do this, make it stable and keep it small, it's a real feat.

--M

Mellvik, 2022-01-08:
Apropos `ftp` as a net file system: This works fairly well already. MacOS Finder connects to ELKS and navigates the fil system, but unfortunately only downwards from `ftpd`s home directory and read only.

FileZilla (and presumably other clients) are more flexible. Look at the attached screenshot from FileZilla, ELKS connection on the right. Can navigate visually all over and delete/create/transfer files back and forth. Similar apps for Linux and Windows.

-M
[image]

ghaerr, 2022-01-08:
That's pretty cool @Mellvik!

Can you use it for a while without ELKS running out of memory?

> but unfortunately only downwards from ftpds home directory and read only.

I wonder if ftpd chdir'd to /, or if pathnames didn't start with /, would help.

Mellvik, 2022-01-08:
> That's pretty cool @Mellvik!
> 
> Can you use it for a while without ELKS running out of memory?
> 
> but unfortunately only downwards from ftpds home directory and read only.
> 
> I wonder if ftpd chdir'd to /, or if pathnames didn't start with /, would help.
> 
> 
> 
@ghaerr,
The comment was related to macos Finder only. Finder limits the legroom for security reasons. Our ftp server imposes no limitations at all - totally insecure. As demonstrated by the FileZilla clip. It's in /etc - while the connections starts in /root.

I tried copying files in and out, works fine ...

ftpd/ktcp handles the load reasonably well. FIlezilla (and Finder) will traverse any directory tree immediately and at high speed. So there are quite a few sys_brk messages, but nothing fatal during my initial testing. Actually it was fun. It seems both clients decode the ls -l output to determine which files are directories etc. I had expected them to require the ftp metadata commands.

To be continued :-)

-M

Mellvik, 2022-01-12:
@ghaerr,
some more investigation into the zero file size /hang issue:

What seems to happen is that the FIN  (again because the file is zero length) arrives before select() has been called, so the read in the select loop never kicks in. Pretty much a  reprise of the FIN before accept() problem fixed in #1052 ...

Now that it has been identified, the problem is an easy fix in `ftp` and possibly not worth spending more time on, but at least it has been recorded. 
And verified to have nothing to do with this (`telnet/ftp`) issue/thread. The reason `telnet` got involved in the first place was that it was the only setting in which the error manifested itself.

A PR for `ftp` is coming.

Feel free to (re-)close this issue.

-M

Mellvik, 2022-01-12:
well, I accidentally closed it myself, so we're fine :-)
M

closed 2022-01-12 by Mellvik