ghaerr/elks #1051: [net] Possible accept() bug - or feature

Mellvik, 2021-12-15:
Another timing related problem, this time with `accept()` - and while it's unclear whether it really is a bug or a feature, it is a problem.

Summary:
Here's what happens: `accept()` opens a requested connection, 2-way SYN, all good. The next incoming packet is a FIN (the sender had no data to send anyway, and closes down). This happens so fast (all within less than a tenth of a second) that by the time `open_fd` inside `accept` gets data back from `ktcp`(my analysis), the connection is marked as closed. In the application (`ftp` client in this case) `accept` appears to have failed (EINVAL), while it actually was successful. The file was just too small. 

Comments
This happens only with zero length files. Even one byte - meaning one packet transferred before the close - is enough to get this to work right. Also, for now, the problem has been handled in the `ftp` client by detecting EINVAL and acting accordingly. However, there are other conditions that may cause EINVAL to be returned from `accept`, so it's not a reliable solution.
It may be a (another) hard one to fix given the nature of having `ktcp` outside the kernel. If that's the case, a unique error return code would be OK.

BTW - it may not make much sense to transfer zero length files, but it's still a file. And when transferring hundreds or thousands of files from/to a single directory, there are going to be all variants - directories, zero length files, dead symbolic links, special files, whatever. Any standard `ftp`-server will either try to send them or send error codes that must be handled properly.

The packet trace looks like this:
```
16:47:52.094307 IP (tos 0x0, ttl 64, id 9736, offset 0, flags [none], proto TCP (6), length 62)
    10.0.2.15.1476 > 10.0.2.2.ftp: Flags [P.], cksum 0x6ccc (correct), seq 112:134, ack 742, win 4380, length 22: FTP, length: 22
	PORT 10,0,2,15,5,199
16:47:52.095114 IP (tos 0x10, ttl 64, id 12350, offset 0, flags [DF], proto TCP (6), length 70)
    10.0.2.2.ftp > 10.0.2.15.1476: Flags [P.], cksum 0x1849 (incorrect -> 0x740a), seq 742:772, ack 134, win 29200, length 30: FTP, length: 30
	200 PORT command successful.
16:47:52.104039 IP (tos 0x0, ttl 64, id 9737, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1476 > 10.0.2.2.ftp: Flags [.], cksum 0x920c (correct), ack 772, win 4350, length 0
16:47:52.121224 IP (tos 0x0, ttl 64, id 9738, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1476 > 10.0.2.2.ftp: Flags [.], cksum 0x91ee (correct), ack 772, win 4380, length 0
16:47:52.132145 IP (tos 0x0, ttl 64, id 9739, offset 0, flags [none], proto TCP (6), length 50)
    10.0.2.15.1476 > 10.0.2.2.ftp: Flags [P.], cksum 0x4448 (correct), seq 134:144, ack 772, win 4380, length 10: FTP, length: 10
	RETR yyy
16:47:52.132966 IP (tos 0x8, ttl 64, id 65461, offset 0, flags [DF], proto TCP (6), length 60)
    10.0.2.2.ftp-data > 10.0.2.15.1479: Flags [S], cksum 0x183f (incorrect -> 0x5938), seq 965492279, win 29200, options [mss 1460,sackOK,TS val 588497909 ecr 0,nop,wscale 7], length 0
16:47:52.138936 IP (tos 0x0, ttl 64, id 9740, offset 0, flags [none], proto TCP (6), length 44)
    10.0.2.15.1479 > 10.0.2.2.ftp-data: Flags [S.], cksum 0x33cc (correct), seq 462266865, ack 965492280, win 4380, options [mss 1460], length 0
16:47:52.139073 IP (tos 0x8, ttl 64, id 65462, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.ftp-data > 10.0.2.15.1479: Flags [.], cksum 0x182b (incorrect -> 0xea94), ack 1, win 29200, length 0
16:47:52.139240 IP (tos 0x10, ttl 64, id 12351, offset 0, flags [DF], proto TCP (6), length 102)
    10.0.2.2.ftp > 10.0.2.15.1476: Flags [P.], cksum 0x1869 (incorrect -> 0x61be), seq 772:834, ack 144, win 29200, length 62: FTP, length: 62
	150 Opening BINARY mode data connection for 'yyy' (0 bytes).
16:47:52.139623 IP (tos 0x8, ttl 64, id 65463, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.ftp-data > 10.0.2.15.1479: Flags [F.], cksum 0x182b (incorrect -> 0xea93), seq 1, ack 1, win 29200, length 0
16:47:52.142370 IP (tos 0x0, ttl 64, id 9741, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1476 > 10.0.2.2.ftp: Flags [.], cksum 0x91e4 (correct), ack 834, win 4318, length 0
16:47:52.142585 IP (tos 0x10, ttl 64, id 12352, offset 0, flags [DF], proto TCP (6), length 64)
    10.0.2.2.ftp > 10.0.2.15.1476: Flags [P.], cksum 0x1843 (incorrect -> 0x619c), seq 834:858, ack 144, win 29200, length 24: FTP, length: 24
	226 Transfer complete.
16:47:52.144641 IP (tos 0x0, ttl 64, id 9742, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1479 > 10.0.2.2.ftp-data: Flags [.], cksum 0x4b88 (correct), ack 2, win 4380, length 0
16:47:52.146673 IP (tos 0x0, ttl 64, id 9743, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1476 > 10.0.2.2.ftp: Flags [.], cksum 0x91e4 (correct), ack 858, win 4294, length 0
16:47:52.148125 IP (tos 0x0, ttl 64, id 9744, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1476 > 10.0.2.2.ftp: Flags [.], cksum 0x918e (correct), ack 858, win 4380, length 0
```

--Mellvik

ghaerr, 2021-12-15:
Hello @Mellvik,

I would say you are getting quite good at both describing and diagnosing networking problems!! And this is definitely a bug.

Completely agree that `accept` has to work with FIN received before application even executes the `accept` call (fast zero-length files).

I'm looking at this now, trying to see how this could happen with a fast FIN, but not when there's even 1 byte of data followed by FIN.

Have you checked to see whether it is the first or second return of -EINVAL below that is causing the failure:
```
int sys_accept(int fd, struct sockaddr *upeer_sockaddr, int *upeer_addrlen)
{
    struct file *file;
    register struct socket *sock;
    register struct socket *newsock;
    int i;

    if (!(sock = sockfd_lookup(fd, &file)))
        return -ENOTSOCK;
        
    if (sock->state != SS_UNCONNECTED)
        return -EINVAL; <---- #1 (perhaps comment these two lines out to see if problem persists)
    
    if (!(sock->flags & SF_ACCEPTCON))
        return -EINVAL;
        
    if (!(newsock = sock_alloc())) {
        debug_net("NET(%d) sys_accept: no more sockets\n", current->pid);
        return -ENOSR;          /* Was EAGAIN, but we are out of system resources! */
    }   
        
    //newsock->type = sock->type;
    newsock->ops = sock->ops;
    if ((i = sock->ops->dup(newsock, sock)) < 0) {
        sock_release(newsock);
        return i;
    }       
        
    i = newsock->ops->accept(sock, newsock, file->f_flags);
    if (i < 0) {
        sock_release(newsock);
        return i;
    }

    if ((fd = get_fd(SOCK_INODE(newsock))) < 0) {
        sock_release(newsock);
        return -EINVAL; <----- #2
    }
```

ghaerr, 2021-12-15:
Tracing a bit further, here is ktcp's code for processing FIN:
```
    if (h->flags & TF_FIN) {
        cb->rcv_nxt++;
        debug_close("tcp[%p] packet in established, fin: 1, data: %d, setting state to CLOSE_WAIT\n", cb->sock, datasize);

        cb->state = TS_CLOSE_WAIT;
        cb->time_wait_exp = Now;        /* used for debug output only*/
        debug_tcp("tcp: got FIN with data %d buffer %d\n", datasize, cb->buf_used);
        if (cb->bytes_to_push <= 0)
            tcpdev_sock_state(cb, SS_DISCONNECTING);
    }
```
If you change the debug_tcp/debug_close above to printf, we should see the "tcp: got FIN with data" message, which I'd like to have you verify. Since the "bytes to push" (data already received) is 0, then the socket state will be sent to SS_DISCONNECTING, which I am thinking might be interfering with the application waiting in `accept`. That could mean that case #3 above is causing the problem (the return from newsock->ops->accept).

Possible error scenario summary: the application is waiting in inet_accept, and instead of getting the accept return value sent by ktcp, it is getting the SS_DISCONNECTING state change instead from the FIN. To confirm, we turn on the printf above, as well as perhaps place a printk in the #3 case above.

Here is the code in ipv4/af_inet.c handling accept:
```
static int inet_accept(register struct socket *sock, struct socket *newsock, int flags)
{   
    register struct tdb_accept *cmd;
    int ret;
    
    debug_net("INET(%d) accept wait sock %x newsock %x\n", current->pid, sock, newsock);
    cmd = (struct tdb_accept *)get_tdout_buf();
    cmd->cmd = TDC_ACCEPT;
    cmd->sock = sock;
    cmd->newsock = newsock; 
    cmd->nonblock = flags & O_NONBLOCK;
    
    tcpdev_inetwrite(cmd, sizeof(struct tdb_accept));
    
    /* Sleep until tcpdev has news */
    while (bufin_sem == 0) {
        interruptible_sleep_on(sock->wait); <--- ftp client waits for accept return from ktcp here
    }

    // <--- if the FIN sends a change socket state instead of accept return, the following fields are all incorrect!!!
    // ktcp sends different types of structures back to the kernel depending on the type. We are expecting
   // an tdb_accept_ret packet here, NOT A STATE CHANGE. We may have to figure a way to accept either
   // back and act accordingly.
    debug_net("INET(%d) accepted sock %x newsock %x\n", current->pid, sock, newsock);
    newsock->remaddr = ((struct tdb_accept_ret *)tdin_buf)->addr_ip;
    newsock->remport = ((struct tdb_accept_ret *)tdin_buf)->addr_port;
    ret = ((struct tdb_accept_ret *)tdin_buf)->ret_value;
    tcpdev_clear_data_avail();
    if (ret >= 0) {
        newsock->state = SS_CONNECTED;
        ret = 0;
    } // <---- might want to add the following statement: <---------------
   else printk("inet_accept: got error return %d from ktcp\n", ret);
    return ret;
}
```
However, the 'tcpdev_sock_setstate(... SS_DISCONNECTED)` ends up in the kernel here:
```
int inet_process_tcpdev(register char *buf, int len)
{   
    register struct socket *sock;
    
    sock = ((struct tdb_return_data *)buf)->sock;
    debug_net("INET(%d) process_tcpdev sock %x type %d wait %x\n", 
        current->pid, sock, ((struct tdb_return_data *)buf)->type, sock->wait);
    
    switch (((struct tdb_return_data *)buf)->type) {
    case TDT_CHG_STATE:
        sock->state = (unsigned char) ((struct tdb_return_data *)buf)->ret_value;
        tcpdev_clear_data_avail();
        debug_net("INET(%d) chg_state sock %x %d\n", current->pid, sock, sock->state);
        if (sock->state == SS_DISCONNECTING) {
            sock->flags |= SF_CLOSING;
            wake_up(sock->wait);
        }
        break;
```
I'm not sure yet how the -EINVAL gets returned to the application yet, but I think we're getting close.

ghaerr, 2021-12-16:
Hello @Mellvik,

Sorry about the long responses... after having some time to think about this, I think the problem is that the remote side, immediately after the double-SYN connection, immediately sends data (actually no data, just the FIN) before the ELKS application has even been scheduled to execute the `accept` system call. This is almost the same as your diagnosis, except the problem isn't in open_fd.

When ktcp receives the FIN, it changes the socket state to SS_DISCONNECTING. There is a race condition, but it's a bit different than I thought and described above. In order to fully understand, it's necessary to see exactly how `accept` and `listen` work inside the kernel, as well as in ktcp:

What happens is that when `listen` is executed, ktcp clones a separate TCP CB (control block) from the existing listen connection. This new CB is essentially the "accept" TCP CB, except it hasn't been assigned a socket yet. That's what `accept` does in the kernel: it clones a socket and passes that socket to ktcp.

When ktcp completes the double SYN, the already-cloned TCP CB is free to accept data. The problem is, the new socket has not been assigned yet, since `accept` has not been executed, to pass it to ktcp. So, I'm almost sure, what is happening is that when the FIN is received, ktcp changes the socket state to SS_DISCONNECTING, but the BUG is that this is the original socket, not the new socket (because the TCP CB's are literally cloned, and the new and old socket are the same).

Thus, the reason you're getting the -EINVAL is that in the sys_accept routine, it checks the status of the socket to be UNCONNECTED and returns -EINVAL if not:
```
int sys_accept(int fd, struct sockaddr *upeer_sockaddr, int *upeer_addrlen)
{   
    struct file *file;
    register struct socket *sock;
    register struct socket *newsock;
    int i;
    
    if (!(sock = sockfd_lookup(fd, &file)))
        return -ENOTSOCK; 
    
    if (sock->state != SS_UNCONNECTED)
        return -EINVAL; <--- this is the -EINVAL you are getting
 ```
Now, we could just remove those two statements, but the problem is that when ktcp got a FIN, it changed the state on the *wrong* socket. So, the original listen fd socket is messed up.

I think the only easy answer is for ktcp to not change the socket (or TCP) state, when receiving a FIN on an unaccepted TCP CB. Without a lot of redoing, the only way we can do that would be do just drop the packet. This should give time for ELKS to schedule the ftp client into executing `accept`, which will work correctly once the new socket has been passed.

So, I think the following (temporary) one-line fix should make things work. In ktcp/tcp.c function tcp_established, add the following in the FIN processing section:
```
    if (h->flags & TF_FIN) {
        if (cb->unaccepted) return;  // <---- add this line
        cb->rcv_nxt++;
        debug_close("tcp[%p] packet in established, fin: 1, data: %d, setting state to CLOSE_WAIT\n", cb->sock, datasize);

        cb->state = TS_CLOSE_WAIT;
        cb->time_wait_exp = Now;        /* used for debug output only*/
        debug_tcp("tcp: got FIN with data %d buffer %d\n", datasize, cb->buf_used);
        if (cb->bytes_to_push <= 0)
            tcpdev_sock_state(cb, SS_DISCONNECTING);
    }
```
If this one-line fix works, I will submit a PR with a permanent fix. That fix will require dealing with the possibility of the FIN bit set AND data in the packet. We can't just drop the packet after having stored the data into the TCP CB in the earlier part of this function.

Mellvik, 2021-12-16:
@ghaerr,

Interestingly, you're debugging faster by reading code than my coding and testing. Admittedly, I got a good night's sleep in between, but still. And your analysis is spot on, Thank you!

I had an almost completed report with the tests from last night, but a fast forward with the suggested line in tcp.c did the trick - as expected.

So I'm reverting the other changes, including the hack in ftp to handle the error return, and continue the testing of the ftp client. Which, as it turns out, is as much a test of ktcp etc. No doubt, we have more fun coming. But I'll (for once) keep this issue free from other bugs and problems.

Looking forward to your PR.

—M

PR #1052 (2021-12-16): [networking] Fix FIN before accept, RST processing
Fixes #1051, and also #1046.

As described in #1051, when a FIN packet is received before the `accept` system call has been executed, ktcp cannot be allowed to change state. So, we drop the packet to allow more time for the accepting application to be scheduled in. This only seems to occur on fast incoming zero-length file transfers.

When a RST is received, described in #1046, wake up any waiting process so a possible read error can be returned, instead of just deallocating the TCP CB and causing a later read panic in ktcp.
closed 2021-12-16 by ghaerr