ghaerr/elks #1065: Porting yet another editor

toncho11, 2021-12-21:
Hi,

I have found this Linux editor: https://github.com/gphalkes/tilde
I am a fan of DOS editors.
Maybe it can be ported to ELKS?

ghaerr, 2021-12-21:
Hello @toncho11,

A quick look at Tilde's homepage shows that it requires the following packages to be installed. Each of these would have to be ported to ELKS:
flex
gettext
libacl1-dev
libattr1-dev
libgpm-dev
libncurses5-dev
libpcre2-dev
libtool-bin
libunistring-dev
libxcb1-dev and/or libx11-dev
pkg-config
LLnextgen (available here)

It's large, written in C++ and I'd say there's an extremely small chance it could fit in 64K. I would suggest starting with something much smaller, or you'll end up spending tons of time sorting out ./configure-based build environments, which I learned don't work well with ELKS, when trying to port the "small" text editor `nano`.  (It finally ran but could only edit a file a couple kilobytes in size).

toncho11, 2021-12-21:
Not surprising. Only chance is when ELKS runs on 286/386 for example and it is compiled to use more than 64 kbs of memory.

closed 2021-12-21 by toncho11