ghaerr/elks #1085: Add support for NE2000 in 8-bit mode

ConiKost, 2022-01-04:
**Description**
I would like to use an ethernet NE2000 compatible adapter on my NuXT system (V20, 8088).

Normally, NE2000 cards are used in 16-bit mode. Since old 8-bit ISA cards usually don't have any RJ45 connector, there are a couple of modern rebuilds of 8-bit isa network cards, using the Realtek RTL8019AS chipset, a NE2000 compatible one. 

I am using this one: https://github.com/skiselev/isa8_eth.
There are more rebuilds like that, but all share the common, to be based on RTL8019AS and use 8-bit ISA.

Apparently, such configuration is not supported by the NE2000 driver in ELKS, as it reports, that it did't found any NE2000 adapter. I would like to suggest to add 8-bit mode for NE2000, if that's any way possible? As far as I understand, NE2000 in 8-bit mode is not the same as a real NE1000.

In DOS, the NE2000 paket driver (http://crynwr.com/) needs to be patched, to make it work in 8-bit mode. AFAICS, only a few lines have been patched out. If done, it just works fine.

```
--- NE2000.ASM.ORG      1993-06-15 11:39:00.000000000 +0200
+++ NE2000.ASM.MOD      2022-01-04 15:10:22.352991827 +0100
@@ -166,8 +166,8 @@
 block_output:
        assume  ds:nothing
        push    ax              ; save buffer address
-       inc     cx              ; make even
-       and     cx,0fffeh
+;      inc     cx              ; make even
+;      and     cx,0fffeh
        loadport
        setport EN_CCMD
        pause_
@@ -228,7 +228,7 @@
 usage_msg      db      "usage: NE2000 [options] <packet_int_no> <int_level> <io_addr>",CR,LF,'$'

        public  copyright_msg
-copyright_msg  db      "Packet driver for NE2000, version "
+copyright_msg  db      "Packet driver for NE2000 8-bit mod, version "
                db      '0'+(majver / 10),'0'+(majver mod 10),".",'0'+version,".",'0'+dp8390_version,CR,LF,'$'

 int_no_name    db      "Interrupt number ",'$'
@@ -260,7 +260,7 @@
 ;dma address 0. Put it in a buffer called board_data.
        assume  ds:code

-       or      endcfg,ENDCFG_WTS
+;      or      endcfg,ENDCFG_WTS

        loadport
        mov     al,endcfg
@@ -278,7 +278,7 @@
        out     dx,al
        setport EN0_RCNTLO      ; remote byte count 0
        pause_
-       mov     al,20h          ; count is actually doubled.
+       mov     al,20h        ; count is actually doubled.
        out     dx,al
        setport EN0_RCNTHI
        pause_
@@ -302,6 +302,7 @@
        pause_
 sp_read_loop:
        in      al,dx           ; get a byte
+       in      al,dx           ; get a byte
        stosb                   ; save it
        loop    sp_read_loop
```

**Configuration**
- Version (commit or release): Release 0.4.0 
- Emulator or HW platform: NuXT, 640KB Memory, NEC V20 8088 compatible.
- Network card: https://github.com/skiselev/isa8_eth, configured for IRQ2 and I/O 0x260.

**How to reproduce ?**
- Does the problem always happen?: Yes, NE2000 is not being detected.
- List the steps to reproduce the problem.: Boot ELKS from Floppy.

Mellvik, 2022-01-04:
Hi Conrad,
welcome to ELKS - I'm sure we'll get your card working with some debugging and changes.

Your assumption is correct - it's easy to mod the current driver to do single byte transfers instead of word transfers.
And some of the changes in the packet driver apply, although this will look different in the ELKS driver, because we're using the stos/lods loop instructions instead of in/out for data I/O.

Just a few lines of code to change actually, probably (I'd have to look) only in the dma_read & dma_write routines.

However, this may not be the reason your card isn't detected. Have you configured the IO Address and IRQ in ELKS' /bootopts file?

The driver should be able to recognize the card even without working I/O - since the probing is register poking only - ig the I/O address is correct.

The next step, when your card is responding, is to make the changes in the dma_* routines. When the card comes up in ELKS with it's own MAC address (printed during the boot process), you know that you're almost there. The next level is interrupts….

Keep us posted and we'll try to guide you through the process.

—Mellvik

ghaerr, 2022-01-04:
> Ignoring /bootopts: header not ## or size > 255

I don't recommend using /bootopts, as it unfortunately doesn't work when booting from FAT/MSDOS disks. Is that the case for you?

> Are there any nighty builds? As the change you mentioned has been added a few days ago, it's, of course, not included in 0.4.0 release :-)

We don't have nightly builds, but I can compile a system image for you. However, since you have development to do, I highly recommend running tools/build.sh and taking the time for the gcc compiler toolchain to build. (Yes, it does take forever).

Do you just need a fd1440.bin file that has NE2K and byte access enabled?

Also, frankly, v0.4.0 is a very long time ago from the perspective of developers here at ELKS. There have been tons of changes to the NE2K driver as well as the networking subsystem.

ConiKost, 2022-01-04:
> I don't recommend using /bootopts, as it unfortunately doesn't work when booting from FAT/MSDOS disks. Is that the case for you?

Yes, this is the case, I was using FAT floppy image.

> We don't have nightly builds, but I can compile a system image for you. However, since you have development to do, I highly recommend running tools/build.sh and taking the time for the gcc compiler toolchain to build. (Yes, it does take forever).

Sounds good, I would like to try anyway to get a toolchain and build for ELKS. Will report back, if there are any problems. Is Ubtuntu 20.04 LTS ok?

> Do you just need a fd1440.bin file that has NE2K and byte access enabled?

Yes, also netirq=2 and netport=0x260 (I/O is 0x260 here?) in my case? So I could quickly look, if its at least recognized. Thank you.

ghaerr, 2022-01-04:
Conrad,

Here's a fd1440.bin configured for 8-bit NE2K. You can use /bootopts to edit the configuration away from I/O address 0x300 and IRQ 12 by booting, editing that file using vi, sync, then rebooting.
[fd1440.bin.zip](https://github.com/jbruchon/elks/files/7809136/fd1440.bin.zip)

ConiKost, 2022-01-04:
> Here's a fd1440.bin configured for 8-bit NE2K. You can use /bootopts to edit the configuration away from I/O address 0x300 and IRQ 12 by booting, editing that file using vi, sync, then rebooting.

Well, that worked! NE2K is being detected and shows correct mac address.

Mellvik, 2022-01-04:
Now, that's really encouraging, Conrad - congratulations.

I just read up on the controller card, a very interesting project indeed. Did you get a preassembled card, or did you assemble it yourself? I'd be interested in getting one.

Since the card per the spec cannot operate if there is anything else occupying the IO addresses from 0x300 - 0x31F, it would make sense to use that address range for the card, don't you think?

—mellvik

ConiKost, 2022-01-04:
Everything looks good. ```net start``` worked fine after setting ip and gateway. Webserver seems reachable from another PC. So it looks like, current master supports everything with NE2K in 8bit.

ConiKost, 2022-01-04:
> Since the card per the spec cannot operate if there is anything else occupying the IO addresses from 0x300 - 0x31F, it would make sense to use that address range for the card, don't you think?

I have to check the XTIDE, AFAIK it runs on 0x300. But I rembember, that this could be changed. But I will check that later :-)

ConiKost, 2022-01-04:
Good news, I was able to compile current git master. Closing here.

ConiKost, 2022-01-04:
> I just read up on the controller card, a very interesting project indeed. Did you get a preassembled card, or did you assemble it yourself? I'd be interested in getting one.

Sorry, forgot to answer. I got from the athor a preassembled card.

closed 2022-01-04 by ConiKost