ghaerr/elks #1086: Many 'Track read  retry' with floppy emulator

ConiKost, 2022-01-04:
**Description**
When I am using my HxC Floppy emulator, I am getting frequently ```Track read retry``` messages. Those messages occur total random during access. But I have to say, I never have seen any read errors. Read and Write from emulated floppy just works fine, despite seing the 'Track read retry' messages. I currently don't have any real floppy drive here, to test that.

HxC is configured for standard PC 1.44MB emulation. I don't see in DOS or Windows any printed retry messages.
Could be this some timing issue?

**Configuration**
- Version (commit or release): 0.4.0 and git master
- Emulator or HW platform: NuXT, 640kb memory, 

**How to reproduce ?**
- Boot ELKS from HxC floppy emulator

**Raw data**
```VFS: Mounted root 0x0380 (minix filesystem)
bioshd(0): track read retry #1 CHS 0/1/3 count 16
bioshd(0): track read retry #1 CHS 0/0/11 count 8
bioshd(0): track read retry #1 CHS 0/0/17 count 2
Running /etc/rc.d/rc.sys script
bioshd(0): track read retry #1 CHS 0/0/13 count 6
ELKS built from commit 132f2bfe
<Date>
bioshd(0): track read retry #1 CHS 0/1/1 count 18

ELKS 0.4.0

login:
```

**Additional information**
The message `track read retry` also happens during usage, for example ```sync``` command.

ghaerr, 2022-01-04:
Hello @ConiKost,

By default, ELKS tries to read whole tracks at a time from the floppy drive in order to increase throughput. It seems that the HxC floppy emulator is sometimes returning an error on track reads. It is possible the HxC does not support this feature. 

This can be turned off in the block device section of make menuconfig by setting "BIOS drive track caching" to OFF (CONFIG_TRACK_CACHE). Lets see whether that makes a difference. If so, you shouldn't see much throughput decrease with your HxC since it's just flash memory anyway.

Thank you!

ConiKost, 2022-01-05:
I've disabled that option now. While the message is gone, I a, now getting different ones.

Example: ```bioshd(0) cmd 0 retry #1 CHS 0/1/3 count 2```

ghaerr, 2022-01-05:
> While the message is gone, I am now getting different ones.

I was afraid that might be the case - it seems that the BIOS is returning an error on some sector reads (and previously, on some track reads).

At this point, I have no idea what the reason is, unless there might be a spurious interrupt or other activity causing the BIOS to fail.

Did the kernel I sent you issue any of these, or is this only happening with your recently-built kernel? I'm trying to think of what else might be causing this problem. Perhaps you can post your .config file so I can compare that with the one I built the for kernel I sent you earlier.

ConiKost, 2022-01-05:
> I was afraid that might be the case - it seems that the BIOS is returning an error on some sector reads (and previously, on some track reads).

> At this point, I have no idea what the reason is, unless there might be a spurious interrupt or other activity causing the BIOS to fail.

At least, I can say, the BIOS is Open Source and on Github: https://github.com/skiselev/8088_bios
Maybe you have there an idea?

> Did the kernel I sent you issue any of these, or is this only happening with your recently-built kernel?

It does not matter. ELKS 0.4.0 stable, your kernel and mine kernel do cause the messages.

ghaerr, 2022-01-05:
> At least, I can say, the BIOS is Open Source and on Github: https://github.com/skiselev/8088_bios

Oh, I see; possibly you're not running an exact "PC Compatible". I'll look at that bios. What exact hardware are you running, is it a desktop, or is the whole thing homebuild? 

You mentioned previously that this BIOS and HxC does run well with MSDOS, correct?

ConiKost, 2022-01-05:
> Oh, I see; possibly you're not running an exact "PC Compatible". I'll look at that bios.

That Bios is specially made for 8088 compatible systems. I would expect, that it's "PC Compatible".

> What exact hardware are you running, is it a desktop, or is the whole thing homebuild?

It's a NuXT: https://monotech.fwscart.com/NuXT_MicroATX_Turbo_XT_955MHz_832K_RAM_XTCF_SVGA_Floppy_Serial/p6083514_19777986.aspx

While it's a modern ATX build, it uses old original ICs. Like NEC V20 or Faraday FE2010A (Used in 80s for XT clones)
Full specs: https://raw.githubusercontent.com/monotech/NuXTv2/master/Monotech%20NuXT%202.0%20User%20Manual.pdf

> You mentioned previously that this BIOS and HxC does run well with MSDOS, correct?

Correct. No issuses. ELKS itself runs also fine. FTP, WWW works for me. But except those retry messages.

ghaerr, 2022-01-05:
Thanks for the info and links, I'll check them out.

> No issuses. ELKS itself runs also fine. FTP, WWW works for me. But except those retry messages.

Interesting. So it seems that there may be an incompatibility between ELKS and the 8088 BIOS, even though MSDOS runs fine on it, as the current ELKS code runs well on strictly IBM PC-compatible BIOSes.

More specifically, it seems perhaps just the code dealing with the Carry bit return from INT 13h may be a possibility, as ELKS also prints a longer message when returned AX != 0 (something like "bioshd: error out AX=xx in AX=xx") which is not being displayed, correct? That seems to show that the AX return is proper.

I'll look further into the ELKS BIOS INT 13h call function and take a quick look at your 8088 bios source.

ConiKost, 2022-01-05:
> More specifically, it seems perhaps just the code dealing with the Carry bit return from INT 13h may be a possibility, as ELKS also prints a longer message when returned AX != 0 (something like [...]

Correct, I have _not_ seen any message with ```error```.

> I'll look further into the ELKS BIOS INT 13h call function and take a quick look at your 8088 bios source.

Great. If you need me to patch for test something, just say. As I have a working build system, I can quickly test.

ghaerr, 2022-01-05:
> Example: bioshd(0) cmd 0 retry #1 CHS 0/1/3 count 2

Is the count always "2" on errors reported? If not, can you give me some more examples of the CHS values reported on retries?

ConiKost, 2022-01-05:
> Is the count always "2" on errors reported? If not, can you give me some more examples of the CHS values reported on retries?

Yes.

All messages during boot without login in order:
```
bioshd(0) cmd 0 retry #1 CHS 0/1/3 count 2
bioshd(0) cmd 0 retry #1 CHS 0/0/11 count 2
bioshd(0) cmd 0 retry #1 CHS 0/0/17 count 2
bioshd(0) cmd 0 retry #1 CHS 0/0/13 count 2
bioshd(0) cmd 0 retry #1 CHS 0/1/1 count 2
```

After Login:
```bioshd(0) cmd 0 retry #1 CHS 0/0/9 count 2```

ghaerr, 2022-01-05:
It appears that your BIOS does not support using the 1E vector to support an updated DDPT (disk drive parameter table). I will look into this further.

Try the following one-line patch to elks/arch/i86/drivers/bioshd.c:
```
/* do bios I/O, return # sectors read/written */
static int do_bios_readwrite(struct drive_infot *drivep, sector_t start, char *buf,
    ramdesc_t seg, int cmd, unsigned int count)
{   
    int drive, errs;
    unsigned int cylinder, head, sector, this_pass;
    unsigned short in_ax, out_ax;
    
    drive = drivep - drive_info;
    drive = hd_drive_map[drive];
    get_chst(drivep, start, &cylinder, &head, &sector, &this_pass);
    
    /* limit I/O to requested sector count*/
    if (this_pass > count) this_pass = count;
    this_pass = 1;  // <---- INSERT THIS LINE
    if (cmd == READ) debug_bios("bioshd(%d): read lba %ld count %d\n",
                drive, start, this_pass);
```

ConiKost, 2022-01-05:
The message is still there, but count has changed.

```
bioshd(0) cmd 0 retry #1 CHS 0/1/3 count 16
bioshd(0) cmd 0 retry #1 CHS 0/0/11 count 8
bioshd(0) cmd 0 retry #1 CHS 0/0/17 count 2
bioshd(0) cmd 0 retry #1 CHS 0/0/13 count 6
bioshd(0) cmd 0 retry #1 CHS 0/1/1 count 18
```

ghaerr, 2022-01-05:
Hmmm, that is very strange. If the "this_pass = 1" was placed as shown above, it should be impossible to display "count 16" in the printk below, which is:
```
printk("bioshd(%d): cmd %d retry #%d CHS %d/%d/%d count %d\n",
                drive, cmd, MAX_ERRS - errs + 1, cylinder, head, sector, this_pass);
```
You might recheck, as well as change the `debug_bios` to `printk` following. Or perhaps printk("this_pass %d\n", this_pass) BEFORE setting it to 1.

ghaerr, 2022-01-05:
It appears the first line below is commented out?
```
 /* limit I/O to requested sector count*/
    if (this_pass > count) this_pass = count;
    this_pass = 1;  // <---- INSERT THIS LINE
```

ConiKost, 2022-01-05:
Hi, I am sorry, but I am an idiot :-) It was late yesterday and I have overseen this.

First: My change in source code looks ok.

```
diff --git a/elks/arch/i86/drivers/block/bioshd.c b/elks/arch/i86/drivers/block/bioshd.c
index a6133d28..4418df01 100644
--- a/elks/arch/i86/drivers/block/bioshd.c
+++ b/elks/arch/i86/drivers/block/bioshd.c
@@ -730,6 +730,7 @@ static int do_bios_readwrite(struct drive_infot *drivep, sector_t start, char *b
 
        /* limit I/O to requested sector count*/
        if (this_pass > count) this_pass = count;
+       this_pass = 1;  // <---- INSERT THIS LINE
        if (cmd == READ) debug_bios("bioshd(%d): read lba %ld count %d\n",
                                drive, start, this_pass);
```

Second: The count has changed, was correct. BUT: It was not related to the source code change. I had forgotten to disable ```CONFIG_TRACK_CACHE```, as I had to create the config new and forotten to disable it. And as my fault, I wrote the wrong messages. Sorry :-(

Third: I disabled the option now and re-created the floppy image with kernel. The messages are still there, but count is shown as 1, which fits the souce code change.

Current messages:
```
bioshd(0) cmd 0 retry #1 CHS 0/1/3 count 1
bioshd(0) cmd 0 retry #1 CHS 0/0/11 count 1
bioshd(0) cmd 0 retry #1 CHS 0/0/17 count 1
bioshd(0) cmd 0 retry #1 CHS 0/0/13 count 1
bioshd(0) cmd 0 retry #1 CHS 0/1/1 count 1
```

ghaerr, 2022-01-05:
Hello Conrad,

>  I had forgotten to disable CONFIG_TRACK_CACHE

Thanks for letting me know, I couldn't quite figure out how that result could have happened!

> Current messages

Well, this may be hard to debug. What we're now looking at is code that has worked on all PC's except yours. With the one-liner change in, ELKS is now doing single-sector I/O, and it's failing on the first of two sector reads (each), with the retry and then next sector read working each time. Very strange! Are you completely sure you have HxC setup as a 1.44Mb floppy?

> It appears that your BIOS does not support using the 1E vector to support an updated DDPT

I looked at your BIOS source and it appears it does support the 1E DDPT vector. However, that doesn't matter since that only affects multi-sector reads, and we're now doing single sector reads.

I am now wondering whether there is some hardware reason why these particular sector numbers are not working, or whether this is something to do with the BIOS, or possibly ELKS. Can you use a different thumb drive in the HxC to help rule out hardware issues? Is that USB drive rated for the speed the HxC requires?

Looking at the ELKS driver, which works on seemingly all other PCs, the only difference between the first sector read failure and the retry is the driver performs a "disk reset". Lets try commenting that out (leave the "this_pass = 1" in):
```
    /* limit I/O to requested sector count*/
    if (this_pass > count) this_pass = count;
    this_pass = 1;
   // possibly change "debug_bios" below to printk to see other reads which might not be failing
    if (cmd == READ) debug_bios("bioshd(%d): read lba %ld count %d\n",
                drive, start, this_pass);
    
    errs = MAX_ERRS;    /* BIOS disk reads should be retried at least three times */
    do {
        BD_AX = (cmd == WRITE ? BIOSHD_WRITE : BIOSHD_READ) | this_pass;
        BD_CX = ((cylinder << 8) | ((cylinder >> 2) & 0xc0) | sector);
        BD_DX = (head << 8) | drive;
#ifdef CONFIG_FS_XMS_BUFFER
        if (seg >> 16) {
            BD_ES = DMASEG;     /* if xms buffer use DMASEG*/
            BD_BX = 0; 
            if (cmd == WRITE)   /* copy xms buffer down before write*/
                xms_fmemcpyw(0, DMASEG, buf, seg, this_pass*512/2);
            set_cache_invalid();
        } else
#endif  
        {   
            BD_ES = (seg_t)seg;
            BD_BX = (unsigned) buf;
        }
        debug_bios("bioshd(%d): cmd %d CHS %d/%d/%d count %d\n",
            drive, cmd, cylinder, head, sector, this_pass);
        in_ax = BD_AX;
        out_ax = 0;
        
        set_ddpt(drivep->sectors);
        if (call_bios(&bdt)) {
            printk("bioshd(%d): cmd %d retry #%d CHS %d/%d/%d count %d\n",
                drive, cmd, MAX_ERRS - errs + 1, cylinder, head, sector, this_pass);
            out_ax = BD_AX;
            //reset_bioshd(drive); <--- COMMENT out this line
            printk("retry AX=%x\n", out_ax); // <-- add this line to display the BIOS error in AH/AL
        }
    } while (out_ax && --errs); /* On error, retry up to MAX_ERRS times */
``` 
You might also change the debug_bios to printk to see more information, although that might cause stuff to scroll off the screen.

If the system continues to fail sector reads with the disk reset commented out, I would think this means something to do with the BIOS is in error.

Can you also tell us what the boot screen says about how many heads/sectors etc /dev/fd0 has?

ConiKost, 2022-01-05:
> Are you completely sure you have HxC setup as a 1.44Mb floppy?

Yes, 100% for sure.
See config: https://imgur.com/nr1NvOn 
It's currently hard coded to 1.44MB.
Also tried 'Auto' without any luck.

> Can you use a different thumb drive in the HxC to help rule out hardware issues? Is that USB drive rated for the speed the HxC requires?

As I have the SD card version, I tried a different sd card now, without any luck. And yes, sd card should be fast enough.

> Lets try commenting that out (leave the "this_pass = 1" in):

This went wrong. Massive read errors and boot failed. See screenshots, too much to write.
-> https://imgur.com/a/VzOL0SC

> Can you also tell us what the boot screen says about how many heads/sectors etc /dev/fd0 has?

```
fd: found valid ELKS disk parameters on /dev/fd0 boot sector
fd: /dev/fd0 probably has 18 sectors, 2 heads, and 80 cylinders.
```
```

ghaerr, 2022-01-05:
> This went wrong. Massive read errors and boot failed. 

Interesting! So without a disk reset in between read errors everything fails. I will continue reading your BIOS source.

> See screenshots, too much to write.
> https://imgur.com/a/VzOL0SC

I can't get anything from that link, it just plays an unrelated video of a drummer.

It might be interesting to add the removed "reset_bioshd(drive)" BEFORE any disk I/O, thus resetting the disk system before each sector:
```
/* do bios I/O, return # sectors read/written */
static int do_bios_readwrite(struct drive_infot *drivep, sector_t start, char *buf,
    ramdesc_t seg, int cmd, unsigned int count)
{
    int drive, errs;
    unsigned int cylinder, head, sector, this_pass;
    unsigned short in_ax, out_ax;

    drive = drivep - drive_info;
    drive = hd_drive_map[drive];
    get_chst(drivep, start, &cylinder, &head, &sector, &this_pass);

    /* limit I/O to requested sector count*/
    if (this_pass > count) this_pass = count;
    this_pass = 1;
    if (cmd == READ) debug_bios("bioshd(%d): read lba %ld count %d\n",
                drive, start, this_pass);

    errs = MAX_ERRS;    /* BIOS disk reads should be retried at least three times */
    do {
        reset_bioshd(drive); // <--- ADD THIS LINE
        BD_AX = (cmd == WRITE ? BIOSHD_WRITE : BIOSHD_READ) | this_pass;
        BD_CX = ((cylinder << 8) | ((cylinder >> 2) & 0xc0) | sector);
        BD_DX = (head << 8) | drive;
#ifdef CONFIG_FS_XMS_BUFFER
        if (seg >> 16) {
            BD_ES = DMASEG;     /* if xms buffer use DMASEG*/
            BD_BX = 0;
            if (cmd == WRITE)   /* copy xms buffer down before write*/
                xms_fmemcpyw(0, DMASEG, buf, seg, this_pass*512/2);
            set_cache_invalid();
        } else
#endif
        {
            BD_ES = (seg_t)seg;
            BD_BX = (unsigned) buf;
        }
```
Can you read any of the "retry AX=" results from the screen? I'm still interested in seeing what the failed return code is.

ConiKost, 2022-01-05:
> I can't get anything from that link, it just plays an unrelated video of a drummer. [EDIT: the screenshots flash on the screen then are covered by gray... perhaps post the screenshots here? Can you read any of the "retry AX=" results from the screen? I'm still interested in seeing what the failed return code is.

Can you access those?
https://i.imgur.com/9NLePh8.jpeg
https://i.imgur.com/EY33tEH.jpeg
https://i.imgur.com/GCFVu8A.jpeg
https://i.imgur.com/Bjxk5fL.jpeg
https://i.imgur.com/u0KQQCS.jpeg
https://i.imgur.com/HdPka0I.jpeg

ghaerr, 2022-01-06:
> Can you access those?

Yes - thanks. I can see the error return = 0x0400 which means "sector not found".

I have another idea, perhaps interrupts need to be enabled for this BIOS. Try the following, but take out the "extra" call to reset_bioshd(drive) from the previous discussion.
```
int INITPROC bioshd_init(void)
{
    register struct gendisk *ptr;
    int count;

#ifndef CONFIG_SMALL_KERNEL
    printk("bioshd: Driver Copyright (C) 1994 Yggdrasil Computing, Inc.\n"
           "Extended and modified for Linux 8086 by Alan Cox.\n");
#endif

    /* FIXME perhaps remove for speed on floppy boot*/
    outb_p(0x0C, FDC_DOR);      /* FD motors off, enable IRQ and DMA*/

#ifdef CONFIG_BLK_DEV_BFD
    _fd_count = bioshd_getfdinfo();
#if 1 // <--- CHANGE from 0 to 1
    enable_irq(6);              /* Floppy */
#endif
#endif
#ifdef CONFIG_BLK_DEV_BHD
    _hd_count = bioshd_gethdinfo();
#if 1 // <---- CHANGE from 0 to 1
    if (sys_caps & CAP_PC_AT) { /* PC/AT or greater */
        enable_irq(HD_IRQ);     /* AT ST506 */
        enable_irq(15);         /* AHA1542 */
    }
    else {
        enable_irq(5);          /* XT ST506 */
    }
#endif
    bioshd_gendisk.nr_real = _hd_count;
#endif /* CONFIG_BLK_DEV_BHD */
```
That is - enable two sections of code from if 0 to if 1.

ConiKost, 2022-01-06:
> It might be interesting to add the removed "reset_bioshd(drive)" BEFORE any disk I/O, thus resetting the disk system before each sector:

This seems end in a loop. After ```VFS: Mouted root 0x0380 (minix filesystem)``` is being accessed, I can see, that track 35 is forever. It never stops and try over and over. No further output is being printed.

```
--- a/elks/arch/i86/drivers/block/bioshd.c
+++ b/elks/arch/i86/drivers/block/bioshd.c
@@ -730,11 +730,13 @@ static int do_bios_readwrite(struct drive_infot *drivep, sector_t start, char *b
 
        /* limit I/O to requested sector count*/
        if (this_pass > count) this_pass = count;
+       this_pass = 1;  // <---- INSERT THIS LINE
        if (cmd == READ) debug_bios("bioshd(%d): read lba %ld count %d\n",
                                drive, start, this_pass);
 
        errs = MAX_ERRS;        /* BIOS disk reads should be retried at least three times */
        do {
+               reset_bioshd(drive); // <--- ADD THIS LINE
                BD_AX = (cmd == WRITE ? BIOSHD_WRITE : BIOSHD_READ) | this_pass;
                BD_CX = ((cylinder << 8) | ((cylinder >> 2) & 0xc0) | sector);
                BD_DX = (head << 8) | drive;
@@ -761,7 +763,8 @@ static int do_bios_readwrite(struct drive_infot *drivep, sector_t start, char *b
                        printk("bioshd(%d): cmd %d retry #%d CHS %d/%d/%d count %d\n",
                            drive, cmd, MAX_ERRS - errs + 1, cylinder, head, sector, this_pass);
                    out_ax = BD_AX;
-                   reset_bioshd(drive);
+                   //reset_bioshd(drive); <--- COMMENT out this line
+                   printk("retry AX=%x\n", out_ax); // <-- add this line to display the BIOS error in AH/AL
                }
        } while (out_ax && --errs);     /* On error, retry up to MAX_ERRS times */
```

> I have another idea, perhaps interrupts need to be enabled for this BIOS. Try the following, but take out the "extra" call to reset_bioshd(drive) from the previous discussion.

System boots, but no change, still the same retry messages. ```retry AX=400``` in every case.

```
diff --git a/elks/arch/i86/drivers/block/bioshd.c b/elks/arch/i86/drivers/block/bioshd.c
index a6133d28..75995a31 100644
--- a/elks/arch/i86/drivers/block/bioshd.c
+++ b/elks/arch/i86/drivers/block/bioshd.c
@@ -584,13 +584,13 @@ int INITPROC bioshd_init(void)
 
 #ifdef CONFIG_BLK_DEV_BFD
     _fd_count = bioshd_getfdinfo();
-#if 0
+#if 1 // <--- CHANGE from 0 to 1
     enable_irq(6);             /* Floppy */
 #endif
 #endif
 #ifdef CONFIG_BLK_DEV_BHD
     _hd_count = bioshd_gethdinfo();
-#if 0
+#if 1 // <---- CHANGE from 0 to 1
     if (sys_caps & CAP_PC_AT) {        /* PC/AT or greater */
        enable_irq(HD_IRQ);     /* AT ST506 */
        enable_irq(15);         /* AHA1542 */
@@ -730,6 +730,7 @@ static int do_bios_readwrite(struct drive_infot *drivep, sector_t start, char *b
 
        /* limit I/O to requested sector count*/
        if (this_pass > count) this_pass = count;
+       this_pass = 1;  // <---- INSERT THIS LINE
        if (cmd == READ) debug_bios("bioshd(%d): read lba %ld count %d\n",
                                drive, start, this_pass);
 
@@ -761,7 +762,8 @@ static int do_bios_readwrite(struct drive_infot *drivep, sector_t start, char *b
                        printk("bioshd(%d): cmd %d retry #%d CHS %d/%d/%d count %d\n",
                            drive, cmd, MAX_ERRS - errs + 1, cylinder, head, sector, this_pass);
                    out_ax = BD_AX;
-                   reset_bioshd(drive);
+                   reset_bioshd(drive); //<--- COMMENT out this line
+                   printk("retry AX=%x\n", out_ax); // <-- add this line to display the BIOS error in AH/AL
```

ghaerr, 2022-01-06:
Thanks Conrad, you are fast!

This is helping. I'm reading the 8088 BIOS source, it seems it does require the FDC controller interrupt to be enabled. I thought this was 6, but perhaps it's something else. Stay tuned!

ghaerr, 2022-01-06:
Conrad, please try this mod. You can leave the other changes as they were in the last example you set above (where the system booted, but no change). Change this in file elks/arch/i86/kernel/irqtab.S:
```
//
//      IRQ 0 (timer) has to go on to the bios for some systems
//
        decw    _bios_call_cnt_l // Will call bios int?
        jne     do_eoi
        movw    $1,_bios_call_cnt_l    // <---- change from 5 to 1
        pushf
        lcall   *_stashed_irq0_l
        jmp     was_trap        // EOI already sent by bios int
```

ConiKost, 2022-01-06:
I am sorry, no change, still retry messages and AX=400.

```
iff --git a/elks/arch/i86/drivers/block/bioshd.c b/elks/arch/i86/drivers/block/bioshd.c
index a6133d28..75995a31 100644
--- a/elks/arch/i86/drivers/block/bioshd.c
+++ b/elks/arch/i86/drivers/block/bioshd.c
@@ -584,13 +584,13 @@ int INITPROC bioshd_init(void)
 
 #ifdef CONFIG_BLK_DEV_BFD
     _fd_count = bioshd_getfdinfo();
-#if 0
+#if 1 // <--- CHANGE from 0 to 1
     enable_irq(6);             /* Floppy */
 #endif
 #endif
 #ifdef CONFIG_BLK_DEV_BHD
     _hd_count = bioshd_gethdinfo();
-#if 0
+#if 1 // <---- CHANGE from 0 to 1
     if (sys_caps & CAP_PC_AT) {        /* PC/AT or greater */
        enable_irq(HD_IRQ);     /* AT ST506 */
        enable_irq(15);         /* AHA1542 */
@@ -730,6 +730,7 @@ static int do_bios_readwrite(struct drive_infot *drivep, sector_t start, char *b
 
        /* limit I/O to requested sector count*/
        if (this_pass > count) this_pass = count;
+       this_pass = 1;  // <---- INSERT THIS LINE
        if (cmd == READ) debug_bios("bioshd(%d): read lba %ld count %d\n",
                                drive, start, this_pass);
 
@@ -761,7 +762,8 @@ static int do_bios_readwrite(struct drive_infot *drivep, sector_t start, char *b
                        printk("bioshd(%d): cmd %d retry #%d CHS %d/%d/%d count %d\n",
                            drive, cmd, MAX_ERRS - errs + 1, cylinder, head, sector, this_pass);
                    out_ax = BD_AX;
-                   reset_bioshd(drive);
+                   reset_bioshd(drive); //<--- COMMENT out this line
+                   printk("retry AX=%x\n", out_ax); // <-- add this line to display the BIOS error in AH/AL
                }
        } while (out_ax && --errs);     /* On error, retry up to MAX_ERRS times */
 
diff --git a/elks/arch/i86/kernel/irqtab.S b/elks/arch/i86/kernel/irqtab.S
index 96a936f6..15a074ac 100644
--- a/elks/arch/i86/kernel/irqtab.S
+++ b/elks/arch/i86/kernel/irqtab.S
@@ -304,7 +304,7 @@ updct:
 //
        decw    _bios_call_cnt_l // Will call bios int?
        jne     do_eoi
-       movw    $5,_bios_call_cnt_l
+       movw    $1,_bios_call_cnt_l    // <---- change from 5 to 1
        pushf
        lcall   *_stashed_irq0_l
        jmp     was_trap        // EOI already sent by bios int
~
```

ghaerr, 2022-01-06:
> I am sorry, no change, still retry messages and AX=400.

Thanks Conrad! Unfortunately I am running out of ideas...

If you look at your boot screen, I notice several stars (*) instead of dots (.). This also indicates sector retry. So the issue isn't really in the kernel, it is also happening in the boot code, which does its own retries. 

It seems there is some incompatibility between the HxC and your BIOS or machine. It is possible that MSDOS doesn't show retries, and that is why you're thinking it is also OK, when it may not be. 

I'm not sure what to do next (other than remove the retry messages). You might try commenting out the retry printk, and verifying that the kernel and application are working fine. You'll still see the stars in the boot loader, which indicates something is wrong in the low-level BIOS read sector and HxC.

[image]

ConiKost, 2022-01-06:
Hi!

> If you look at your boot screen, I notice several stars (*) instead of dots (.). This also indicates sector retry. So the issue isn't really in the kernel, it is also happening in the boot code, which does its own retries.

That's a good observation. Is at this stage BIOS directly used for floppy access?

> It is possible that MSDOS doesn't show retries, and that is why you're thinking it is also OK, when it may not be.
Yes, I am afraid, that this is quite possible :-(

> It seems there is some incompatibility between the HxC and your BIOS or machine.
I will have to borrow a real floppy, to check, it that's a BIOS or HxC issue. This will take some time. I will report back once tested.

I really appreciate your help. Thank you!

ghaerr, 2022-01-07:
> Is at this stage BIOS directly used for floppy access?

Yes. After having exhausted my initial ideas of what could be changed in the kernel and saw the boot asterisks, I realized that this isn't a kernel issue at all, the boot code calls the BIOS read sector and fails a few times, with the retries working, just like in the kernel. I'm pretty sure that you have the same problem with MSDOS, as it doesn't report retries, only hard failures.

closed 2022-02-08 by ghaerr

ConiKost, 2022-02-19:
Just a quick update: Using a different floppy emulator (Gotek) results in the same problem. So it's at least not releated to HxC directly. I will try a real floppy as next.

ConiKost, 2022-02-20:
@ghaerr I did now tested with a real floppy drive (1.44MB). While I don't get any errors during boot, I am getting some (but different) retry errors after boot. I wonder, if this could be releated to https://github.com/jbruchon/elks/issues/1119?

Using official 0.5.0 image.

```
ELKS built from commit 55f32f28
Sun Feb 20 15:23:44 2022
bioshd(0): track read retry #1 CHS 0/1/15 count 4

ELKS 0.5.0

login: root
bioshd(0): track read retry #1 CHS 0/0/9 count 10
# dir
bioshd(0): track read retry #1 CHS 0/0/7 count 12
dir: not found
# ls /
bioshd(0): track read retry #1 CHS 0/0/9 count 10
System Volume   bin            bootopts      dev
etc                       home        lib                linux
mnt                     root           tmp             var
```

ghaerr, 2022-02-20:
Hello @ConiKost,

Thanks for your update and continued testing!

>  I wonder, if this could be releated to #1119?

No, I don't think so - #1119 is about a problem that occurs only when switching from I/O from one floppy to another. This was tested on (just one other) real hardware system and found that it seems to be a QEMU BIOS problem.

> I am getting some (but different) retry errors after boot.

Currently, we are not getting any reports from sector retries, on either floppy emulators or real hardware, other than your hardware. So I don't know what this problem is. It seems it could be a hardware problem on your system; MSDOS doesn't report sector retries, but ELKS does. The fact you're getting so many, while no one else does, seems to point that direction. However, that isn't proven either. I wonder if there are any MSDOS-based floppy test programs that might help pinpoint whether this could be the issue?

Thank you!

ConiKost, 2022-02-20:
Hi @ghaerr,
thank you for the reply.

> No, I don't think so - https://github.com/jbruchon/elks/issues/1119 is about a problem that occurs only when switching from I/O from one floppy to another. This was tested on (just one other) real hardware system and found that it seems to be a QEMU BIOS problem.

Good to know, thank you.

> Currently, we are not getting any reports from sector retries, on either floppy emulators or real hardware, other than your hardware. So I don't know what this problem is. It seems it could be a [...]

I am just unsure, are the tests with the real floppy drive the same issue? It looks like? But yes, I also tend to says, that this seems be a specific NuXT problem, not a floppy drive problem itself.

> I wonder if there are any MSDOS-based floppy test programs that might help pinpoint whether this could be the issue?

Hm, maybe I can reproduce that with tools like 'VGACopy'. I will try some tools and see, what happens in DOS.

ConiKost, 2022-05-15:
@ghaerr Just a small update. After speaking with NuXT creator, we could nail it down, that this is a bios software issue. Using an external floppy bios resolves this issue. So the issue must be within the 8088 bios. I opened an issue for Sergey to investigate this.

ghaerr, 2022-05-15:
Hello @ConiKost,

Thanks for letting us know. I have looked at your open issue over there, it will be nice to get this fixed for your hardware.

Thank you!

ConiKost, 2023-01-26:
@ghaerr The bios author sergej is currently investigating the bug, but his written floppy image seems to panic. Could you have a look here: https://github.com/skiselev/8088_bios/issues/17#issuecomment-1404448650 Thanks!

ConiKost, 2023-02-01:
Just for reference: This is a bug in the 8088 bios. Current development version contains a fix, which at least works for me on NuXT.

ghaerr, 2023-02-01:
That's good to hear @ConiKost! I see from your comments over there that ELKS is no longer producing any I/O errors, so we can consider this matter completed. Thank you!