ghaerr/elks #1087: Compilation of CONFIG_NANO fails

ConiKost, 2022-01-04:
**Description**
Compiling Git master with ```CONFIG_NANO``` fails.

```
make[3]: Entering directory '/home/elks/Downloads/elks/elks/net'
ia16-elf-gcc -ffreestanding -fno-inline -melks -mcmodel=small -msegment-relocation-stuff -mtune=i8086  -I/home/elks/Downloads/elks/include -I/home/elks/Downloads/elks/elks/include -I/home/elks/Downloads/elks/libc/include -DUTS_RELEASE=\"0.4.0\" -D__KERNEL__ -Wall -Os -Wno-strict-aliasing -fno-delete-null-pointer-checks -Wnull-dereference -c -o socket.o socket.c
ia16-elf-gcc -ffreestanding -fno-inline -melks -mcmodel=small -msegment-relocation-stuff -mtune=i8086  -I/home/elks/Downloads/elks/include -I/home/elks/Downloads/elks/elks/include -I/home/elks/Downloads/elks/libc/include -DUTS_RELEASE=\"0.4.0\" -D__KERNEL__ -Wall -Os -Wno-strict-aliasing -fno-delete-null-pointer-checks -Wnull-dereference -c -o protocols.o protocols.c
ia16-elf-gcc -ffreestanding -fno-inline -melks -mcmodel=small -msegment-relocation-stuff -mtune=i8086  -I/home/elks/Downloads/elks/include -I/home/elks/Downloads/elks/elks/include -I/home/elks/Downloads/elks/libc/include -DUTS_RELEASE=\"0.4.0\" -D__KERNEL__ -Wall -Os -Wno-strict-aliasing -fno-delete-null-pointer-checks -Wnull-dereference -c -o ipv4/af_inet.o ipv4/af_inet.c
ia16-elf-gcc -ffreestanding -fno-inline -melks -mcmodel=small -msegment-relocation-stuff -mtune=i8086  -I/home/elks/Downloads/elks/include -I/home/elks/Downloads/elks/elks/include -I/home/elks/Downloads/elks/libc/include -DUTS_RELEASE=\"0.4.0\" -D__KERNEL__ -Wall -Os -Wno-strict-aliasing -fno-delete-null-pointer-checks -Wnull-dereference -c -o nano/af_nano.o nano/af_nano.c
nano/af_nano.c: In function 'nano_bind':
nano/af_nano.c:152:20: warning: unused variable 'old_ds' [-Wunused-variable]
     unsigned short old_ds;
                    ^~~~~~
nano/af_nano.c:149:14: warning: unused variable 'i' [-Wunused-variable]
     int ano, i;
              ^
nano/af_nano.c:149:9: warning: unused variable 'ano' [-Wunused-variable]
     int ano, i;
         ^~~
nano/af_nano.c: In function 'nano_connect':
nano/af_nano.c:197:22: error: 'sockaddr_na' undeclared (first use in this function)
     if (get_user(&(((sockaddr_na *)uservaddr)->sun_family)) != AF_NANO) {
                      ^~~~~~~~~~~
nano/af_nano.c:197:22: note: each undeclared identifier is reported only once for each function it appears in
nano/af_nano.c:197:35: error: expected expression before ')' token
     if (get_user(&(((sockaddr_na *)uservaddr)->sun_family)) != AF_NANO) {
                                   ^
nano/af_nano.c:209:14: warning: implicit declaration of function 'sock_awaitconn' [-Wimplicit-function-declaration]
     if ((i = sock_awaitconn(sock, serv_upd->npd_socket, flags)) < 0)
              ^~~~~~~~~~~~~~
nano/af_nano.c: In function 'nano_getname':
nano/af_nano.c:292:30: warning: passing argument 1 of 'move_addr_to_user' from incompatible pointer type [-Wincompatible-pointer-types]
     return move_addr_to_user(&upd->npd_sockaddr_na, upd->npd_sockaddr_len,
                              ^
In file included from nano/af_nano.c:12:0:
/home/elks/Downloads/elks/elks/include/linuxmt/socket.h:58:12: note: expected 'char *' but argument is of type 'struct sockaddr_na *'
 extern int move_addr_to_user(char *,size_t,char *,int *);
            ^~~~~~~~~~~~~~~~~
nano/af_nano.c:293:9: warning: passing argument 3 of 'move_addr_to_user' from incompatible pointer type [-Wincompatible-pointer-types]
         usockaddr, usockaddr_len)
         ^~~~~~~~~
In file included from nano/af_nano.c:12:0:
/home/elks/Downloads/elks/elks/include/linuxmt/socket.h:58:12: note: expected 'char *' but argument is of type 'struct sockaddr *'
 extern int move_addr_to_user(char *,size_t,char *,int *);
            ^~~~~~~~~~~~~~~~~
nano/af_nano.c:294:1: error: expected ';' before '}' token
 }
 ^
nano/af_nano.c: At top level:
nano/af_nano.c:460:24: error: unknown type name 'select_table'
          int sel_type, select_table * wait)
                        ^~~~~~~~~~~~
nano/af_nano.c:558:5: error: 'nano_select' undeclared here (not in a function)
     nano_select,
     ^~~~~~~~~~~
make[3]: *** [../Makefile-rules:244: nano/af_nano.o] Error 1
make[3]: Leaving directory '/home/elks/Downloads/elks/elks/net'
make[2]: *** [Makefile:123: net/net.a] Error 2
make[2]: Leaving directory '/home/elks/Downloads/elks/elks'
make[1]: *** [Makefile:73: Image] Error 2
make[1]: Leaving directory '/home/elks/Downloads/elks/elks'
make: *** [Makefile:13: all] Error 2
Build script has terminated with error 5
```

ghaerr, 2022-01-05:
Hi Conrad,

Thanks for posting this. It looks like nobody has used the "nano" sockets support in ELKS for quite some time. The good news is that I don't know of any programs that actually use this feature. It was originally written for client/server Nano-X/Microwindows, but those programs are now compiled without the need for this option.

I'll look into making this compile; is there a specific need you have for this, or is this just the result of having turned on all of ELKS features?

Thank you!

ConiKost, 2022-01-05:
I was indeed just curious and wanted to see, what it's good for, as help is empty :-) Thank you.
Btw, CONFIG_DEV_META also fails compilation. (No clue for what that is good)

ghaerr, 2022-01-05:
>  CONFIG_DEV_META also fails compilation. (No clue for what that is good)

CONFIG_DEV_META seems to be some kind of implementation of user-mode drivers, of which there are none for ELKS. I think I'll move it into "experimental" status, since I doubt much will come of it. I will try to get CONFIG_NANO to compile, separately.

ConiKost, 2022-01-05:
> CONFIG_DEV_META seems to be some kind of implementation of user-mode drivers, of which there are none for ELKS. I think I'll move it into "experimental" status, since I doubt much will come of it.

Just for reference, this is the compile error with 

```
make[3]: Entering directory '/home/elks/Downloads/elks/elks/arch/i86/drivers/char'
ia16-elf-gcc -ffreestanding -fno-inline -melks -mcmodel=small -msegment-relocation-stuff -mtune=i8086  -I/home/elks/Downloads/elks/include -I/home/elks/Downloads/elks/elks/include -I/home/elks/Downloads/elks/libc/include -DUTS_RELEASE=\"0.4.0\" -D__KERNEL__ -Wall -Os -Wno-strict-aliasing -fno-delete-null-pointer-checks -Wnull-dereference -c -o init.o init.c
ia16-elf-gcc -ffreestanding -fno-inline -melks -mcmodel=small -msegment-relocation-stuff -mtune=i8086  -I/home/elks/Downloads/elks/include -I/home/elks/Downloads/elks/elks/include -I/home/elks/Downloads/elks/libc/include -DUTS_RELEASE=\"0.4.0\" -D__KERNEL__ -Wall -Os -Wno-strict-aliasing -fno-delete-null-pointer-checks -Wnull-dereference -c -o mem.o mem.c
ia16-elf-gcc -ffreestanding -fno-inline -melks -mcmodel=small -msegment-relocation-stuff -mtune=i8086  -I/home/elks/Downloads/elks/include -I/home/elks/Downloads/elks/elks/include -I/home/elks/Downloads/elks/libc/include -DUTS_RELEASE=\"0.4.0\" -D__KERNEL__ -Wall -Os -Wno-strict-aliasing -fno-delete-null-pointer-checks -Wnull-dereference -c -o ntty.o ntty.c
ia16-elf-gcc -ffreestanding -fno-inline -melks -mcmodel=small -msegment-relocation-stuff -mtune=i8086  -I/home/elks/Downloads/elks/include -I/home/elks/Downloads/elks/elks/include -I/home/elks/Downloads/elks/libc/include -DUTS_RELEASE=\"0.4.0\" -D__KERNEL__ -Wall -Os -Wno-strict-aliasing -fno-delete-null-pointer-checks -Wnull-dereference -c -o meta.o meta.c
meta.c: In function 'new_request':
meta.c:43:9: warning: unused variable 'i' [-Wunused-variable]
     int i;
         ^
meta.c: At top level:
meta.c:71:13: error: conflicting types for 'do_meta_request'
 static void do_meta_request(kdev_t device)
             ^~~~~~~~~~~~~~~
meta.c:72:1: note: an argument type that has a default promotion can't match an empty parameter name list declaration
 {
 ^
In file included from meta.c:24:0:
../block/blk.h:135:24: note: previous declaration of 'do_meta_request' was here
 #define DEVICE_REQUEST do_meta_request
                        ^
../block/blk.h:145:14: note: in expansion of macro 'DEVICE_REQUEST'
 static void (DEVICE_REQUEST) ();
              ^~~~~~~~~~~~~~
meta.c: In function 'do_meta_request':
meta.c:81:2: error: too many arguments to function 'end_request'
  end_request(0, req->rq_dev);
  ^~~~~~~~~~~
In file included from meta.c:24:0:
../block/blk.h:147:13: note: declared here
 static void end_request(int uptodate)
             ^~~~~~~~~~~
meta.c:120:6: error: too many arguments to function 'end_request'
      end_request(0, req->rq_dev);
      ^~~~~~~~~~~
In file included from meta.c:24:0:
../block/blk.h:147:13: note: declared here
 static void end_request(int uptodate)
             ^~~~~~~~~~~
meta.c:135:2: error: too many arguments to function 'end_request'
  end_request(1, req->rq_dev);
  ^~~~~~~~~~~
In file included from meta.c:24:0:
../block/blk.h:147:13: note: declared here
 static void end_request(int uptodate)
             ^~~~~~~~~~~
meta.c:77:11: warning: variable 'buff' set but not used [-Wunused-but-set-variable]
     char *buff;
           ^~~~
meta.c: At top level:
meta.c:163:5: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
     ubd_ioctl,   /* ioctl */
     ^~~~~~~~~
meta.c:163:5: note: (near initialization for 'ubd_fops.ioctl')
meta.c:165:5: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
     ubd_release,  /* release */
     ^~~~~~~~~~~
meta.c:165:5: note: (near initialization for 'ubd_fops.release')
meta.c:215:5: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
     ucd_read,   /* read */
     ^~~~~~~~
meta.c:215:5: note: (near initialization for 'ucd_fops.read')
meta.c:216:5: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
     ucd_write,   /* write */
     ^~~~~~~~~
meta.c:216:5: note: (near initialization for 'ucd_fops.write')
meta.c:218:5: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
     ucd_select,   /* select */
     ^~~~~~~~~~
meta.c:218:5: note: (near initialization for 'ucd_fops.select')
meta.c:220:5: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
     ucd_open,   /* open */
     ^~~~~~~~
meta.c:220:5: note: (near initialization for 'ucd_fops.open')
meta.c: In function 'meta_ioctl':
meta.c:286:44: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
      blk_dev[driver->udd_major].request_fn = do_meta_request;
                                            ^
meta.c:289:19: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
  driver->udd_task = current;
                   ^
meta.c:300:26: warning: comparison of distinct pointer types lacks a cast
     if (driver->udd_task != current) {
                          ^~
meta.c: At top level:
meta.c:366:5: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
     meta_release,  /* release */
     ^~~~~~~~~~~~
meta.c:366:5: note: (near initialization for 'meta_chr_fops.release')
make[3]: *** [../../../../Makefile-rules:244: meta.o] Error 1
make[3]: Leaving directory '/home/elks/Downloads/elks/elks/arch/i86/drivers/char'
make[2]: *** [Makefile:197: drivers/char/chr_drv.a] Error 2
make[2]: Leaving directory '/home/elks/Downloads/elks/elks/arch/i86'
make[1]: *** [Makefile:75: Image] Error 2
make[1]: Leaving directory '/home/elks/Downloads/elks/elks'
make: *** [Makefile:13: all] Error 2
Build script has terminated with error 5
```

PR #1091 (2022-01-07): [kernel] Fix compilation errors in nano sockets and meta drivers
Fixes #1087.

Fixes compilation errors in CONFIG_NANO (nano sockets) and CONFIG_DEV_META (user device drivers) options. It looks like these two options had not been compiled since the move from the `bcc` compiler.

Untested, as no known programs use nano sockets. Possible use for older MIcrowindows/Nano-X programs compiled client/server (which is not the case today for ELKS).

CONFIG_DEV_META compiles but needs work, as /dev/meta is not created, MAX_BLKDEV is too low, and there are no user device drivers. This option should proba [... cut]
closed 2022-01-07 by ghaerr