ghaerr/elks #1088: Disk size reported wrong suffix

ConiKost, 2022-01-05:
**Description**
It looks like, that the suffix for reported disks is somehow static.
My master disk has always ```Mb``` as suffix, while slave always show ```Gb```
The size seems be correct. Only suffix is wrong.

**Configuration**
- Version (commit or release): Git master
- Emulator or HW platform: NuXT

**How to reproduce ?**
- Does the problem always happen ? Yes
- List the steps to reproduce the problem: Install two disks.

**Raw data**
A) Master: 8GB CF, Slave: 128MB CF
```/dev/hda: 972 cylinders, 255 heads, 63 sectors = 7624.6 Mb``` (This is a 8GB CF)
```/dev/hdb: 993 cylinders, 8 heads, 32 sectors = 124.1 Gb``` (This is a 128MB CF)

B) Master: 128MB CF, Slave: 8GB CF
```/dev/hdb: 993 cylinders, 8 heads, 32 sectors = 124.1 Mb``` (This is a 128MB CF)
```/dev/hda: 972 cylinders, 255 heads, 63 sectors = 7624.6 Gb``` (This is a 8GB CF)

ghaerr, 2022-01-07:
@ConiKost,

What program are you running, or exactly where/when is this size information displayed?

ghaerr, 2022-01-07:
> What program are you running, or exactly where/when is this size information displayed?

Oops, never mind, found it. You're having unset CONFIG_SMALL_KERNEL is causing some code I hadn't noticed before. Fix coming.

PR #1092 (2022-01-07): [kernel] Fix reported disk size at boot
Fixes #1088.

May add this display into standard boot messages later, not just when CONFIG_SMALL_KERNEL not defined.

Also fixes 720k floppy image build, ran out of space (again) due to network program size increases.
closed 2022-01-07 by ghaerr