ghaerr/elks #1094: System reboots - Toshiba T1200

mzryz, 2022-01-07:
Hi there,

I am trying ELKS via the HD Images written to CF card. I have tested both Minix, FAT, both standard & MBR varients. I am using the XTide Bios CF drive called 3inONEder. Toshiba T1200 is an 8086 laptop with an 8087 npu available.
[Link to 3inONEder](https://conventionalmemories.com/wiki_cm/3inONEder_for_Toshiba_portables)

Problem, on boot system says:
```
Booting C>>C
ELKS...................................4!
Press key
```
Then when pressing any key the system will just reboot, go back to ram check and then start the process again. Am I doing something wrong? Thanks.

ghaerr, 2022-01-07:
Hello @mzryz,

Thank you for trying ELKS. The '4' in the boot message means that the kernel header wasn't found from the sectors loaded on disk/CF. This usually means that the CHS (cylinder/head/sector) might be incorrect for the order the sectors are on disk/CF. For the standard build, the floppy is in 1.44Mb format, which expects 18 sectors, 2 heads and 80 cylinders. You might check whether the 3inOneder is setup for that format. Otherwise, you'll have to "cd images; make images" to build a different set of boot images for various disk sizes. (Or configure the target boot disk format in "make menuconfig".

> Then when pressing any key the system will just reboot, go back to ram check and then start the process again. Am I doing something wrong? 

We ran out of space in the boot sector, so "Press key" really means "Press any key to reboot". So that's normal behavior when the system isn't found, which is what's happening.

Thank you!

mzryz, 2022-01-07:
Thanks for the reply! I guess this isn't an issue anymore so I'll close & try floppy image / play with making my own image.

closed 2022-01-07 by mzryz

ghaerr, 2022-01-07:
>  I'll close & try floppy image / play with making my own image.

Ok, let us know if you need more help. I initially thought you were talking about a floppy boot but now that I see the Toshiba CF drive is a hard disk emulation, I'm almost certain the ELKS image CHS isn't matching the CF hard disk CHS settings. By default, ELKS hard disk images use CHS of 63 cylinders, 16 heads and 63 sectors, but can be changed with make menuconfig.