ghaerr/elks #1107: stty not working with ash

Mellvik, 2022-01-15:
The previously reported problem with setting terminal speed via `stty` turns out to apply to all `stty` settings – quietly ignored.

It turns out the problem is in `/bin/ash`, presumably `linenoise` related. `stty` works fine in sash.

@ghaerr, you're far better suited to track down this one than I am.

Thank you. 

-M

ghaerr, 2022-01-15:
> It turns out the problem is in /bin/ash, presumably linenoise related. stty works fine in sash.

Thanks for the report Helge. Linenoise reads the tty settings at startup, then sets raw mode for the command line, then sets cooked mode using the original settings before running the command, then raw mode again at the next prompt.

Linenoise will have to be modified to read the current settings before setting cooked mode. I am sure we can get the baud rate working, not sure about the other tty settings. I'll take a pass at it, and then you can test it, since I don't have access to real hardware for a bit.

Thank you!

Mellvik, 2022-01-15:
> It turns out the problem is in /bin/ash, presumably linenoise related. stty works fine in sash.
>
> Thanks for the report Helge. Linenoise reads the tty settings at startup, then sets raw mode for the [...]
Thank you @ghaerr,
I'm ready to do the testing any time. It's easy to see the complicated relationship between stty options and linenoise.
Speed changes must work, the other options need to fall in line with the requirements of linenoise. What we're really talking about is the presumed requirements in shell scripts. Such as turning off echo, etc.
I'll leave it to you to figure out what's possible.

-M

Mellvik, 2022-01-16:
Thank you, @ghaerr !

PR #1108 (2022-01-16): [ash] Fix stty baud rate with linenoise and ash
Fixes #1107.

Allows baud rate, character size, stop bits and parity to be adjusted using `stty` with `ash` shell and linenoise.
closed 2022-01-16 by Mellvik