ghaerr/elks #1112: [sash] execl problem

Mellvik, 2022-01-17:
I'm preparing some minor fixes for `sash` for 0.5.0, and ran into a more serious issue:

In line 674 of `sash.c`,  `execl` is invoked to run `/bin/sh -c` with a (presumed) script as argument (after `execvp` failed, indicating that the target is most likely a script). 
Unfortunately, when called this way, `/bin/sh` has no environment and thus no PATH, so it does not find the target script unless a full path is specified. 
This is confusing and obviously erroneous - and probably an easy fix, but instead of a time consuming experimentation into what `exec`-call does what, I'm asking for advice.

Any suggestions, @ghaerr?

```
# net
net: No such file or directory 
# /bin/net
Usage: net [start|stop] [eth|slip|cslip] [baud] [device]
# 
```

Mellvik, 2022-01-17:
Correction, the line in question is 670:

```	
        execvp(argv[0], argv);

	/*
	 * If file has access but didn't execute, guess it's a shell script
	 * and run 'sh -c cmd'.
	 */
	if (errno == ENOEXEC)
		execl("/bin/sh", "sh", "-c", cmd, (char*)0);

	perror(argv[0]);	/* Usually 'No such file or directory'*/
	exit(1);
```

ghaerr, 2022-01-17:
Hello @Mellvik,

> Correction, the line in question is 670:

Strange. The execvp call is supposed to be using the PATH= environment var to find the passed program. It is returning ENOENT, instead of ENOEXEC, so /bin/sh isn't being used to execute `net`. This is handled in the C library rather than the kernel, i'll look into it.

Thank you!

closed 2022-01-18 by ghaerr