ghaerr/elks #1114: [net, slip] zero window stops transmission

Mellvik, 2022-01-18:
OK; so in preparation for the coming `ppp` implementation, some `slip`-testing. 

Admittedly, I haven't used `slip` for ages, and it took some time to get it operational against Linux (raspbian) at the other end. I'll add the setup to the Wiki. Seems like an MTU of ~ 600 is about right - at 9600bps.

Anyway, it seems we're hitting some of the problems we've seen (and fixed) with `ether` before. When doing a `ftp` file transfer into ELKS (ELKS being the originating client, issuing a `get` command, but I don't think that matters), transmission hangs when the `ktcp` advertised window becomes 0, as the trace below shows. At this point, `ktcp` debug says
`tcp: dropping packet, bad seqno: need 4381 got 4380 size 0` (repeated for every retransmit from the other end).
At this point, `ktcp` has already acked 4381 and there are no outstanding packets. The other end continues to ack #1, which is correct. 

@ghaerr - unless you have an immediate clue on this one, I'm going to dive into it later this week.

```
    10.0.3.3.ftp > 10.0.3.15.1025: Flags [P.], cksum 0xebf7 (correct), seq 734:799, ack 86, win 11200, length 65: FTP, length: 65
	150 Opening BINARY mode data connection for 'vi' (69008 bytes).
16:58:45.323117 IP (tos 0x8, ttl 64, id 17808, offset 0, flags [DF], proto TCP (6), length 600)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0x1978 (correct), seq 1:561, ack 1, win 11200, length 560
16:58:45.432649 IP (tos 0x8, ttl 64, id 17809, offset 0, flags [DF], proto TCP (6), length 600)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0x5d64 (correct), seq 561:1121, ack 1, win 11200, length 560
16:58:45.483641 IP (tos 0x0, ttl 64, id 283, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1025 > 10.0.3.3.ftp: Flags [.], cksum 0x8170 (correct), ack 799, win 4315, length 0
16:58:45.534385 IP (tos 0x0, ttl 64, id 284, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1025 > 10.0.3.3.ftp: Flags [.], cksum 0x812f (correct), ack 799, win 4380, length 0
16:58:45.699399 IP (tos 0x8, ttl 64, id 17810, offset 0, flags [DF], proto TCP (6), length 600)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0x54c6 (correct), seq 1121:1681, ack 1, win 11200, length 560
16:58:45.797104 IP (tos 0x8, ttl 64, id 17811, offset 0, flags [DF], proto TCP (6), length 600)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0x493f (correct), seq 1681:2241, ack 1, win 11200, length 560
16:58:46.062692 IP (tos 0x8, ttl 64, id 17812, offset 0, flags [DF], proto TCP (6), length 600)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0x329a (correct), seq 2241:2801, ack 1, win 11200, length 560
16:58:46.130136 IP (tos 0x0, ttl 64, id 285, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1027 > 10.0.3.3.35131: Flags [.], cksum 0x34fe (correct), ack 561, win 3820, length 0
16:58:46.329346 IP (tos 0x8, ttl 64, id 17813, offset 0, flags [DF], proto TCP (6), length 600)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0x107b (correct), seq 2801:3361, ack 1, win 11200, length 560
16:58:46.594916 IP (tos 0x8, ttl 64, id 17814, offset 0, flags [DF], proto TCP (6), length 600)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0xfe7c (correct), seq 3361:3921, ack 1, win 11200, length 560
16:58:46.760767 IP (tos 0x0, ttl 64, id 286, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1027 > 10.0.3.3.35131: Flags [.], cksum 0x34fe (correct), ack 1121, win 3260, length 0 
16:58:46.861608 IP (tos 0x8, ttl 64, id 17815, offset 0, flags [DF], proto TCP (6), length 500)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0xe116 (correct), seq 3921:4381, ack 1, win 11200, length 460
16:58:47.390258 IP (tos 0x0, ttl 64, id 287, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1027 > 10.0.3.3.35131: Flags [.], cksum 0x34fe (correct), ack 1681, win 2700, length 0
16:58:48.024642 IP (tos 0x0, ttl 64, id 288, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1027 > 10.0.3.3.35131: Flags [.], cksum 0x34fe (correct), ack 2241, win 2140, length 0
16:58:48.652713 IP (tos 0x0, ttl 64, id 289, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1027 > 10.0.3.3.35131: Flags [.], cksum 0x34fe (correct), ack 2801, win 1580, length 0
16:58:49.283257 IP (tos 0x0, ttl 64, id 290, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1027 > 10.0.3.3.35131: Flags [.], cksum 0x34fe (correct), ack 3361, win 1020, length 0
16:58:49.920004 IP (tos 0x0, ttl 64, id 291, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1027 > 10.0.3.3.35131: Flags [.], cksum 0x34fe (correct), ack 3921, win 460, length 0
16:58:50.444381 IP (tos 0x0, ttl 64, id 292, offset 0, flags [none], proto TCP (6), length 40)
    10.0.3.15.1027 > 10.0.3.3.35131: Flags [.], cksum 0x34fe (correct), ack 4381, win 0, length 0       <-------------- zero win
16:59:00.912532 IP (tos 0x8, ttl 64, id 17816, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0x093f (correct), ack 1, win 11200, length 0
16:59:21.392541 IP (tos 0x8, ttl 64, id 17817, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0x093f (correct), ack 1, win 11200, length 0
17:00:05.552522 IP (tos 0x8, ttl 64, id 17818, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.3.3.35131 > 10.0.3.15.1027: Flags [.], cksum 0x093f (correct), ack 1, win 11200, length 0
```

ghaerr, 2022-01-18:
> Admittedly, I haven't used slip for ages, and it took some time to get it operational against Linux (raspbian) at the other end. I'll add the setup to the Wiki. Seems like an MTU of ~ 600 is about right - at 9600bps.

Have you seen my `slattach.sh` script? It's in the repo root and was what I used for slip months ago when I got SLIP working. It's meant to be run on the Linux side to configure the remote host. This should probably be updated for what you've done to get RPi working.

> transmission hangs when the ktcp advertised window becomes 0, as the trace below shows. At this point, ktcp debug says
tcp: dropping packet, bad seqno: need 4381 got 4380 size 0 (repeated for every retransmit from the other end).
At this point, ktcp has already acked 4381 and there are no outstanding packets.

Geez. It does sound like we've been here before with ACKs on previously sent packets, but I can't remember off the top of my head exactly what the fix was. This could be that problem again, or another timing-related bug. I suppose this should be fixed before v0.5, though we might push this and PPP issues until after release to avoid another time-consuming cycle. I'm thinking the original problem/fix had something to do with not allowing too many outstanding packets to be sent unacked, which ultimately caused an unrecoverable error loop because the retry packets were somehow no longer saved in the retransmit queue.

You may need to turn (back) on DEBUG_TCP, DEBUG_WINDOW, and DEBUG_RETRANS in order to see what ktcp is doing during this issue.

On another note, I'll be publishing the net configuration enhancement shortly, and I'd like your comments on how SLIP and ETH could both be easily configured using the new net.cfg file.

ghaerr, 2022-01-18:
@Mellvik,

I'm wondering whether there might be an issue with the code in ktcp/tcp.c to drop incoming packets:
```
    if (cb->state != TS_LISTEN && cb->state != TS_SYN_SENT
                               && cb->state != TS_SYN_RECEIVED) {
        if (cb->rcv_nxt != ntohl(tcph->seqnum)) {
            int datalen = iptcp.tcplen - TCP_DATAOFF(iptcp.tcph);

            debug_tune("tcp: dropping packet, bad seqno: need %ld got %ld size %d\n",
                cb->rcv_nxt - cb->irs, ntohl(tcph->seqnum) - cb->irs, datalen);

            if (cb->rcv_nxt != ntohl(tcph->seqnum) + 1) // <--- perhaps comment this line out to force ACK?
                tcp_send_ack(cb);

            return;
        }
```
In the case you described above, cb->rcv_nxt (ELKS receiver) would be 4381 and tcph->sequnum (the sender) 4380. Note after the dropping packet message there is a compare to the equivalent of if (4381 != 4380 + 1) to have ELKS reply with an ACK. I don't quite understand that last bit, and am trying to understand why the two systems are off by just one sequence number. It is possible this has to do with the zero-window interaction between the server "ping" and ktcp being incorrect.

The entire code fragment and dropping packets is made necessary because ktcp doesn't buffer any received packets for reassembly; it can only process packets in strict sequence order. It's almost as if the "ping" from the sender on the zero-window condition of ktcp is off by one for ktcp, and ktcp is dropping it, but saying nothing. This is why perhaps always sending an ACK after the "ping" might resolve this situation?

Thank you!

Mellvik, 2022-01-19:
Thank you @ghaerr,

> 
> Have you seen my slattach.sh script? It's in the repo root and was what I used for slip months ago when I got SLIP working. It's meant to be run on the Linux side to configure the remote host. This should probably be updated for what you've done to get RPi working.
> 
I have not - thanks for pointing me to it. And yes, it may need some adjustments. I'll get back to that. Actually, and unrelated to the script, ping is a very efficient tool to establish a reasonable mtu for the connection. Used almost like the TCP slow start (van jacobsen) algorithm, we increase the packet size till it starts dropping packets, then go backwards a little. Which reminds me that ping is a tool ELKS would benefit from. It's on my list.

> On another note, I'll be publishing the net configuration enhancement shortly, and I'd like your comments on how SLIP and ETH could both be easily configured using the new net.cfg file.
> 
> 
Coming in a day or two.

—M

Mellvik, 2022-01-19:
@ghaerr,
I got around to test the proposed code change, which didn't do the trick. Some ideas below.

Here's the trace - ELKS style:
```
tcp: send 1026->57437 [A] ack 3361 win 1020 urg 0 chk 0 len 20
tcp: recv 57437->1026 [A] seq 3361-3921 ack 1 win 11200 urg 0 chk 0 len 580
tcp: recv data len 560 avail 1020
tcp: ACK seq 3921 len 560
tcp: send 1026->57437 [A] ack 3921 win 460 urg 0 chk 0 len 20
tcp: recv 57437->1026 [A] seq 3921-4381 ack 1 win 11200 urg 0 chk 0 len 480
tcp: recv data len 460 avail 460
tcp: ACK seq 4381 len 460
tcp: send 1026->57437 [A] ack 4381 win 0 urg 0 chk 0 len 20
tcp: recv 57437->1026 [A] ack 1 win 11200 urg 0 chk 0 len 20
tcp: dropping packet, bad seqno: need 4381 got 4380 size 0
tcp: send 1026->57437 [A] ack 4381 win 0 urg 0 chk 0 len 20
tcp: recv 57437->1026 [A] ack 1 win 11200 urg 0 chk 0 len 20
tcp: dropping packet, bad seqno: need 4381 got 4380 size 0
tcp: send 1026->57437 [A] ack 4381 win 0 urg 0 chk 0 len 20
tcp: recv 57437->1026 [A] ack 1 win 11200 urg 0 chk 0 len 20
tcp: dropping packet, bad seqno: need 4381 got 4380 size 0
```

Now, here's some loud  thinking about the issue:
- The packet ktcp is dropping (and reporting), is in fact a 'knock on closed window' (zero length) packet.
- The ktcp message is misleading. The actual received packet is empty, just an ack, and has no sequence numbers. Dropping the packet is correct behaviour - the Ack has been received before.
- The actual problem is somewhere else - It seems to me we have a deadlock: The window is closed (no more buffer space) , why isn't the application getting/reading the data?
- The problem appears only with larger files. Any number of 2000 byte files can be transferred w/o any trouble.
- If a `telnet` session is active (running say `hd /bin/vi`) in parallel with the transfer, there is no hang. Having a `telnet`-session active and idle, then creating some traffic after the hang, does not release it.
- While the ftp session is hung, ktcp is fully operational and accepts connections for both `ftp` and `telnet`. Further, the file that always hangs when fetched via `ftp`, comes in smoothly when fetched with `ftpget`.

I'm about to exonerate ktcp and slip, because of the last two points above. It seems - and I'm going to look at this tomorrow - that the select in `ftp`for some reason does not kick in. Adding a timeout to it will shed more light on the situation.

To be continued.

-M

ghaerr, 2022-01-19:
Hello @Mellvik, thanks for the extensive further testing and analysis. Everything you're saying sounds quite reasonable and likely correct.

> I'm about to exonerate ktcp and slip, because of the last two points above. It seems - and I'm going to look at this tomorrow - that the select in ftpfor some reason does not kick in. 

Interesting. Well, we already know we have an unsolved select problem with telnetd (actually the kernel) that requires it to have a 50ms timeout in order for things not to hang. I think your observation above about "why isn't the application getting/reading the data" could be on target - in that case, ktcp can't do anything as its buffer is full, and we're still dealing with a subtle kernel select problem, already seen in telnetd.

Hopefully you can slightly redesign/rearrange the ftpd main loop to allow for a regular select timeout to see whether this helps. It seems this issue is quite timing dependent, which is why its occurring on SLIP but not ether or ftpget.

> The problem appears only with larger files. Any number of 2000 byte files can be transferred w/o any trouble.

Since ktcp's input buffer is 4380 bytes, this fits with the idea that ktcp is waiting for the application to read buffered data before it can continue.

> If a telnet session is active (running say hd /bin/vi) in parallel with the transfer, there is no hang. 

A few weeks ago I realized that the kernel implementation of /etc/tcpdev, used to read and write to ktcp, isn't well suited to multiple applications waiting for network I/O in the kernel. This will definitely not be easily changed, but could be a contributor to the possible race or deadlock condition still unsolved with multiple applications running. This seems also the same as the telnetd select timeout hack required to keep it working a few weeks ago.

Thank you!

Mellvik, 2022-01-20:
This issue has more surprises to offer, @ghaerr.

It turns out that the timeout hack introduced in the `ftp` client to handle zero length files, is the problem.

What happens is that when the receive select times out for the 4th time, `ftp` assumes we have a zero length file (which will deliver no data) and goes on to wait for the status message on the command channel instead. And there it sits forever, there is no timeout on that one. Until the server times out (900 seconds) and sends a message to that effect.

Interestingly - if the file is small, not the 2k bytes we talked about yesterday, but say 20k, the Transfer Complete message arrives on the command channel before any data whatsoever (another timing issue tanks to the difference in speed between ELKS and the Raspi). So our `ftp` client happily closes up, assuming the file had zero length, and ignores the data.

Again, `ktcp` and `select` are in the clear as far as this particular problem is concerned. While the root cause is in `ktcp` I decided to rewrite parts of the `ftp` client to get around the problem: Eliminate the use of buffered input on the command channel in order to get better control via select. PR coming.

ghaerr, 2022-01-20:
I'm glad you found the problem! I wasn't aware of the extent of the hack that was written to work around a ktcp issue. We should probably fix ktcp rather than add application hacks, especially if the hack ignores the protocol spec?

That said, hacks are sometimes needed. We should probably document them somewhere. Top of my list is the telnetd select timeout, which should not be required, and the funky SO_REUSEADDR option to conserve ktcp memory. I'm sure there are more.

Mellvik, 2022-01-20:
> I'm glad you found the problem! I wasn't aware of the extent of the hack that was written to work around a ktcp issue. We should probably fix ktcp rather than add application hacks, especially if the hack ignores the protocol spec?
> 
No, it doesn't ignore the spec, it just works around the oddities created by the orders of magnitude differences in speed and buffer capacity.. And yes, a fix in ktcp would be perfect, but in the meanwhile I'll devise a new hack to cover both situations. For now.
> That said, hacks are sometimes needed. We should probably document them somewhere. Top of my list is the telnetd select timeout, which should not be required, and the funky SO_REUSEADDR option to conserve ktcp memory. I'm sure there are more.
> 
That's a good idea.  We'll surely forget them otherwise. I suggest a hacks wiki. Seriously. I find it easy to drop into a wiki page and make updates.
> 
Thank you.

--M

ghaerr, 2022-01-22:
> Changes in ftp (the client) fixes the problem

Good news, thanks!

> Then the good news: hundreds of files back and forth via slip and ftp (on physical), and slip has been very stable. Checksum errors, retransmissions etc., nice recovery.

Even better news, I'm very glad to hear that ktcp and slip are still working well after all our changes this last year!

> and removes the use of buffered IO in ftp, which should reduce the heap requirements)

Yes, stdio will malloc 1K bytes every time `fopen` is called, which, if mixed with other mallocs can result in twice the otherwise total requirement if allocation checkerboarding is seen. For fastest data transfers, direct read/write is preferable, as even ELKS' stdio can have lots going on under the hood slowing things down for all it tries to do. I learned this when debugging the recent `sash` ^C fix involving fgets.