ghaerr/elks #1126: 1.2M minix boot image does not boot

Mellvik, 2022-01-29:
The 1.2M Minix image generated by setting the appropriate menuconfig options, no longer boots on physical - see attached sceenshot. It works on QEMU - strange.

It seems the boot process tries to mount the minix image as a FAT fs? 
I haven't tried the actual FAT boot image - will do that asap.

[image]

ghaerr, 2022-01-29:
@Mellvik, that's pretty strange - although it looks like the entire superblock is 0xFF, as can be seen from the FAT: display.

ELKS tries to mount the root filesystem by reading the super block, and I'm guessing that with all FFs, the MINIX FS said "no", and it's trying the next one (FAT) likely because the FAT super block read didn't return an error.

It is possible this has something to do with the variable sector size stuff we've been adding. Can you try a couple of MINIX floppy sizes to see whether they work?

Thank you!

ghaerr, 2022-01-29:
If the problem persists, try setting DEBUG_BIOS and DEBUG_STARTDEF (enable debug at boot, rather than ^P) in include/linuxmt. That will display the BIOS calls in more detail. 

Another thought is to turn off XMS, just in case buffer management is causing problems.

Mellvik, 2022-01-30:
Thank you @ghaerr,
 I did test the 1200 fat image, it's just the same. I also copied the kernel from a working 1440 floppy and still the same. As to XMS, I was running a different 286 machine off of 3.5in floppies a couple of days ago, incidentally with XMS enabled and buffers set to 256, with no problems (apropos - since ELKS correctly reports the cpu as 286, it may make sense with a warning that XMS buffers are not available?)

I'll do more testing later today (shucks, I really needed this image to be operational, in order to dump the contents of the HD off to the Raspi). 
My guess is that the problem is caused by the fact that we're dealing with a 15sectors/track medium. Maybe odd numbers don't work well with the new multisize sector scheme?

More later.

--M

PS: Totally off topic, @ghaerr, but there is a leftover debug message in init that pops up every time init is run from the command line. And yes, I do the weirdest things, such as switching between init-levels when testing slip...

Warning: Sent from a machine with a really creative spelling helper.

Mellvik, 2022-01-30:
Update:
You were right, @ghaerr - it was the XMS buffer config. Removing it and it works fine.

And the reason it worked OK on the other machine was that it actually has XMS memory. A 286 with 6.6M total.
That does not make much sense to me, but I'm sure it does to you. 

Thank you.

--M

closed 2022-01-30 by Mellvik

ghaerr, 2022-01-30:
> And the reason it worked OK on the other machine was that it actually has XMS memory. A 286 with 6.6M total.

Ok, so the working machine has 6M XMS memory? How much XMS memory does the failing machine have?

The current XMS buffer implementation (on Compaq) uses INT 15h to copy memory between main and XMS memory, but ELKS neither checks the copy memory return code, nor tests/asks how much memory is present - it just trusts the config parameters.

> here is a leftover debug message in init that pops up every time init is run from the command line.

Thanks, I'll fix that with the /etc/rc.d directory removal.

>  And yes, I do the weirdest things, such as switching between init-levels when testing slip...

Your real-world actual use of ELKS is greatly helping making it better, thank you!

ghaerr, 2022-01-30:
>  There is a leftover debug message in init that pops up every time init is run from the command line.

I just tested this and can't duplicate: running 'init 3' or 'init 1', 'init 6' etc doesn't display anything extra. What is the message being displayed?

Mellvik, 2022-01-30:
Thank you @ghaerr - 

> 30. jan. 2022 kl. 15:08 skrev Gregory Haerr ***@***.***>:
> 
> 
> And the reason it worked OK on the other machine was that it actually has XMS memory. A 286 with 6.6M total.
> 
> Ok, so the working machine has 6M XMS memory? How much XMS memory does the failing machine have?
> 
That would be zero. It's plain vanilla Compaq Portable III/20M/1.2M with no extras.

One of these days I'll do some testing of XMS on the compaq 286 - to find out what kind of tricks they implement in the BIOS.
I'm noticing no speedup on the machine so I'm assuming XMS buffering isn't working.

—M

ghaerr, 2022-01-30:
> That would be zero. It's plain vanilla Compaq Portable III/20M/1.2M with no extras

When one configures ELKS to use XMS, say 256K of buffers, then all I/O buffers are kept in that memory, and copied back and forth to/from main memory when needed, in order to save main memory use. So, on your machine without XMS memory, the system immediately fails on first disk I/O, since the superblock sector is being kept in non-existent XMS memory - which is (now) obviously returning all 0xFF's when copied back to main memory - thus the display seen above.

I could put in an XMS check after copy, but the kernel would immediately panic anyways, either from that check, or in the VFS root mount you're seeing now. We could add a FAT mount check which would then at least cause the message "VFS root mount failed" instead, then panic.

> One of these days I'll do some testing of XMS on the compaq 286 - to find out what kind of tricks they implement in the BIOS.

I've been reading the (disassembled) Compaq BIOS source code - and BIOS uses protected mode switching to access XMS internally. It is very complicated (as the BIOS itself is copied to high XMS for speed), which is why directly setting unreal mode doesn't work on Compaq: the first call to BIOS messes with any unreal mode CPU cache buffers.

> I'm noticing no speedup on the machine so I'm assuming XMS buffering isn't working

The kernel startup states whether XMS or "ext" (external - meaning external to kernel, i.e. main memory) buffers are being used. With the "spinning disk" animation in the upper right corner of the screen, it is immediately obvious when XMS buffers or disk I/O is being performed - XMS shows no I/O at all (for say ls -lR /), while non-XMS (or pre-loading XMS buffers once time) will take seconds, with the animation running.

Mellvik, 2022-01-30:
> I could put in an XMS check after copy, but the kernel would immediately panic anyways, either from that check, or in the VFS root mount you're seeing now. We could add a FAT mount check which would then at least cause the message "VFS root mount failed" instead, then panic.
> 
Primary focus must be to keep things simple. Maybe use Int15/AH=88 to get the amount of extendend memory would avoid these situations altogether?
> One of these days I'll do some testing of XMS on the compaq 286 - to find out what kind of tricks they implement in the BIOS.
>
> I've been reading the (disassembled) Compaq BIOS source code - and BIOS [...]
I'm unsure as to how much of this actually applies to the 286, in particular the remapping of the bios itself. OTOH - Compaq literally wrote the book on many of these things, and it seems reasonable that their BIOS INT15 services are aligned between the 286 and 386, meaning they do the LOADALL manipulation in the BIOS. 
> I'm noticing no speedup on the machine so I'm assuming XMS buffering isn't working
>
> The kernel startup states whether XMS or "ext" (external - meaning external to kernel, i.e. main memory) buffers [...]
Thanks, I need to take a peek at that. That's what I'm missing when always using serial console and hardly ever see the machine.

Thank you!

--M

ghaerr, 2022-01-30:
> Primary focus must be to keep things simple. Maybe use Int15/AH=88 to get the amount of extendend memory would avoid these situations altogether?

Good idea, for XMS w/INT 15h, I'll add that. Otherwise the memory check has to be made manually, which is supposed to be working with the A20 verify code, which seems may not be working. The A20 verify code only checks a single byte just above 1M, which also is the location of the first XMS buffer.

> it seems reasonable that their BIOS INT15 services are aligned between the 286 and 386, meaning they do the LOADALL manipulation in the BIOS.

No, the 286 (and 386) Compaq BIOS enter normal full protected mode (and then exit, much slower than unreal mode) in order to copy XMS memory. The BIOS disassembly shows the GDT table loads and protected mode startup/exit. There's also a separate INT 15 BIOS call to enter protected mode. The LOADALL mechanism would be much faster, but I doubt it will work on Compaq 286's, since the Compaq 386 BIOS interferes with 80386 unreal mode, which was the first attempt at getting XMS working on my Compaq 386 portable. It crashed as soon as unreal mode was enabled, presumably from the BIOS clock or other interrupt occurring, but I never fully found out why.

> Compaq literally wrote the book on many of these things, and it seems reasonable that their BIOS INT15 services are aligned between the 286 and 386, meaning they do the LOADALL manipulation in the BIOS.

I'm thinking that Compaq's very early implementation of XMS predated the discovery of the LOADALL and unreal mode hacks with the 286 and 386 processors - as they coded directly to the published Intel protected mode specifications. Later, as MSDOS XMS became very popular and needed, the hacks were discovered and quickly found themselves in the HIMEM.SYS and other XMS drivers.

> I'm noticing no speedup on the machine so I'm assuming XMS buffering isn't working

The boot message says "xms: ... 256 xms buffers" when XMS buffering is effective, and "64 ext buffers" when EXT buffers are effective. (the actual buffer counts are set using separate config items). A little more information is given prior for XMS, to help debug/identify A20 line and unreal mode issues. 

I've been meaning to add 286 vs 386+ processor identification routines to ELKS, which would allow XMS kernels without INT 15h compiled in to boot, instead of hanging. However, this is complicated by the fact that the current ELKS CPU identification is buggy, and misidentifies several CPUs (I would like to throw that all out).

Mellvik, 2022-01-31:
Thank you @ghaerr,

again very thorough and educating.

>  There's also a separate INT 15 BIOS call to enter protected mode.

I noticed that one when refreshing my memory on the various INT15 functions (I have this 1100 page book that occasionally makes it out of the bookcase). What ,counght my interest (and yes, we're far off the issue of this thread, but now that we're rolling…) - was that the BIOS call to enter protected mode has a return code. But returning from PM on the 286 is kind of a black art as I understand it, so I'm assuming this applies only to the 386 and later, even though it's not mentioned.
Even more interesting is the INT15-call (AH=88H) to move data to/from memory anywhere above 1M, which seems to imply going to and from PM implicitly. It might be interesting to check out that on a 286, maybe even a 386.

> The LOADALL mechanism would be much faster, but I doubt it will work on Compaq 286's, since the Compaq 386 BIOS interferes with 80386 unreal mode, which was the first attempt at getting XMS working [...]
Thanks, I need to look into how that actually works on my 286. The boot message says exactly that. 256 and all.
> I've been meaning to add 286 vs 386+ processor identification routines to ELKS, which would allow XMS kernels without INT 15h compiled in to boot, instead of hanging. However, this is complicated by [...]
I guess using INT15 to get the processor type would be the opposite of what we need then :-)

—M

ghaerr, 2022-01-31:
> Even more interesting is the INT15-call (AH=88H) to move data to/from memory anywhere above 1M, which seems to imply going to and from PM implicitly. It might be interesting to check out that on a 286, maybe even a 386.

This is the BIOS INT 15h call I've been talking about: it must be used to enable XMS buffers in ELKS on Compaq 386, and definitely uses protected mode to function. I assume it also works on Compaq 286 BIOS's. Do you have a 286 Compaq with XMS memory to test ELKS XMS buffers?

> But returning from PM on the 286 is kind of a black art as I understand it, so I'm assuming this applies only to the 386 and later, even though it's not mentioned.

Almost all PC-compatible 286 systems can return from protected mode using a CPU RESET by using the external keyboard controller, which has a gate, similar to the A20 gate, but is connected to the reset pin. The trick was making it fast. Later, someone figured out how to generate a CPU triple fault which reset the processor, which didn't rely on the slow keyboard controller. The startup path in BIOS then checked a special memory location to determine if this was a cold/warm boot, or instead a 286 protected mode reset.

Bottom line: INT 15 Block memory move is available and must be used on Compaq machines, as unreal mode doesn't work. This was all completed with XMS buffers and tested on my Compaq Portable 386. This mechanism may be slow on 286, but is certainly faster than doing disk I/O, and saves 64K-256K of main memory when XMS buffers (and INT 15 option) is enabled.

**Did you see my previous reply to this question:
> There is a leftover debug message in init that pops up every time init is run from the command line.

I just tested this and can't duplicate: running 'init 3' or 'init 1', 'init 6' etc doesn't display anything extra. What is the message being displayed?

Mellvik, 2022-01-31:
Thank you @ghaerr,
> Even more interesting is the INT15-call (AH=88H) to move data to/from memory anywhere above 1M, which seems to imply going to and from PM implicitly. It might be interesting to check out that on a [...]
Yes, this is the one I was referring to that seemed to run fine with XMS enabled and set to 256 buffers, but I didn't notice any speed improvement. I need to pull that machine back out now that the other one is tested and OKd.
> **Did you see my previous reply to this question:
>
> There is a leftover debug message in init that pops up every time init is run from the command line.
>
> I just tested this and can't duplicate: [...]
Actually, now that I look at it again and just having poked into the XMS stuff, it may be a kernel message, not an init printf, towards the bottom:

# reboot
Direct console, scan kbd 80x25 emulating ANSI (3 virtual consoles)
ttyS0 at 0x3f8, irq 4 is a 16450
ttyS1 at 0x2f8, irq 3 is a 16550A
ttyS2 at 0x3e8, irq 5 is a 16550A
xms: A20 was off now on, using int 15, 256 xms buffers
eth: NE2K at 0x300, irq 12, MAC 00:80:29:ef:d9:51
bioshd: hda BIOS CHS 610,16,63
bioshd: hda  IDE CHS 993,16,63
/dev/hda: 993 cylinders, 16 heads, 63 sectors = 488.7 Mb
/dev/fd0: 80 cylinders, 2 heads, 18 sectors = 1440.0 kb
/dev/fd1: 80 cylinders, 2 heads, 15 sectors = 1200.0 kb
Partitions: hda:(0,1000944)  hda1:(63,997857) 
device_setup: BIOS drive 0x0, root device 0x380
PC/AT class machine, Unknown x86 CPU, 640K base RAM.
ELKS kernel 0.4.0 (52320 text, 12704 ftext, 8032 data, 41856 bss, 15646 heap)
Kernel text at 2d0:0000, ftext f96:0000, data 12b0:0000, top a000:0, 501K free
fd: found valid ELKS disk parameters on /dev/fd0 boot sector
fd: /dev/fd0 probably has 18 sectors, 2 heads, and 80 cylinders
MINIX-fs: mounting unchecked file system, running fsck is recommended.
VFS: Mounted root 0x0380 (minix filesystem).
Running /etc/rc.d/rc.sys script
Starting networking on eth
ktcp -b elks1 10.0.2.2
ktcp: ip 10.0.2.15, gateway 10.0.2.2, netmask 255.255.255.0
ktcp: ethernet 00.80.29.ef.d9.51 mtu 1500
Starting daemons telnetd ftpd 
ELKS built from commit dab1dc72
Tue Jan 18 16:23:41 2022

ELKS 0.4.0

login: root
# init 3
l32_fmemcpyb dst 1
# net stop
Stopping network
# ktcp -d -d -d -b -p cslip -l /dev/ttyS1 -s 9600 10.0.3.15 10.0.3.3 255.255.255.0
ktcp: ip 10.0.3.15, gateway 10.0.3.3, netmask 255.255.255.0
ktcp: cslip /dev/ttyS1 baud 9600 mtu 1024

-M

ghaerr, 2022-01-31:
> l32_fmemcpyb dst 1

Got it (finally), thanks! Long story short, this was left in because a kluge was required in order to get XMS buffer copying to work with an odd transfer size - which I could never get to happen. Now (of course), you've found a way. The Compaq INT 15h BIOS does not accept an odd byte count, only a word (even) count, for memory transfers. This means that we had to come up with a kluge workaround in order to copy a single byte using word block move. Glad to see my code works! I'll remove the display now that it's known working.

Thank you!