ghaerr/elks #1127: Failed to build elks

GNUfan42, 2022-01-31:
I was trying to build elks on my arm64 Android phone and I was following the build instruction. I installed Ubuntu using Linux Deploy. `GCC-IA16` has been built successfully(with a little modification to the build script) and `menuconfig` also works. However I got into trouble at the 5th step, when building the elks itself, where gcc and make printed the following error message: 
```
In file included from _libelf_config.h:167:0,
                 from _libelf.h:34,
                 from elf.c:29:
/home/android/elks/elks/tools/elf2elks/lib/elftoolchain-0.7.1/common/native-elf-format.h:3:20: error: 'unknown' undeclared here (not in a function)
 #define ELFTC_ARCH unknown
                    ^
_libelf_config.h:170:22: note: in expansion of macro 'ELFTC_ARCH'
 #define LIBELF_ARCH  ELFTC_ARCH
                      ^~~~~~~~~~
elf.c:34:18: note: in expansion of macro 'LIBELF_ARCH'
  .libelf_arch  = LIBELF_ARCH,
                  ^~~~~~~~~~~
make[3]: *** [lib/elftoolchain-0.7.1/libelf/libelf.a] Error 1
make[2]: *** [all] Error 2
make[1]: *** [tools] Error 2
make: *** [all] Error 2
```
This seems to be a problem with elf2elks. A script names `native-elf-format` is used to generate `format.h` but there's no arm-related declaration. How do I resolve this issue?

Btw: the `README` file says that `elksemu` only runs on linux-i386 or similar. Would it be helpful if I cross-compile elksemu to i386 then use qemu-user-i386 to run it?

ghaerr, 2022-01-31:
Hello @GNUfan42,

> the README file says that elksemu only runs on linux-i386 or similar. Would it be helpful if I cross-compile elksemu to i386 then use qemu-user-i386 to run it?

Yes. `elksemu` requires Linux 386, but ELKS itself will only run on an x86-compatible processor. So you'd need to run qemu-user-i386 to run and boot ELKS, after compiling it using the cross-compiled ELKS toolchain. While ELKS is quite a bit more portable than it has been over the past two years, many aspects of it are very dependent on a segmented 16-bit architecture.

> I was trying to build elks on my arm64 Android phone and I was following the build instruction.
> This seems to be a problem with elf2elks. 

I'll leave answers to that to the ELKS toolchain maintainer, @tkchia. I don't know whether the toolchain itself has been cross-compiled on ARM, I'm suspecting that to be possibly a problem here.

Thank you!

tkchia, 2022-01-31:
Hello @GNUfan42,

> This seems to be a problem with elf2elks. A script names `native-elf-format` is used to generate `format.h` but there's no arm-related declaration. How do I resolve this issue?

Hmm... indeed, the `native-elf-format` script in the `elftoolchain` sources lacks the needed check.  As a quick fix, you can modify it thus:
```
$1 ~ "Machine:" {
        if (match($0, "Intel.*386")) {
            elfarch = "EM_386";
        } else if (match($0, "MIPS")) {
            elfarch = "EM_MIPS";
        } else if (match($0, ".*[xX]86-64")) {
            elfarch = "EM_X86_64";
        } else if (match($0, "AArch64")) {
            elfarch = "EM_AARCH64";
        } else {
            elfarch = "unknown";
        }
    }
```
(I downloaded a random set of Linux/AArch64 binaries to check the expected `readelf` output for such binaries; so the above should be correct.)

In the meantime I will try to see if there is a newer version of `elftoolchain` that resolves this problem &mdash; or if that fails, write up a patch.

Thank you!

tkchia, 2022-01-31:
Hello @GNUfan42,

> Btw: the `README` file says that `elksemu` only runs on linux-i386 or similar. Would it be helpful if I cross-compile elksemu to i386 then use qemu-user-i386 to run it?

Unfortunately this will not work (yet).  Indeed, `qemu-i386` and `qemu-x86_64` can run a lot of userland Linux/i386 and Linux/x86-64 programs.  However, `elksemu` relies on certain Linux/i386 and Linux/x86-64 syscalls &mdash; `modify_ldt` and `clone` &mdash; which are unimplemented or not completely implemented in the QEMU user mode emulators.  So right now `elksemu` will only work under an actual Linux/i386 or Linux/x86-64 system, or something really close.
```
$ qemu-i386 elksemu/elksemu ./elks-fartext-test 
clone call failed
```

For now, to run ELKS programs in emulation, you will probably need to run an entire ELKS system, under full system emulation.  With QEMU, this means using `qemu-system-i386` or `qemu-system-x86_64`.

Thank you!

GNUfan42, 2022-02-01:
Hi @tkchia,
> Hmm... indeed, the `native-elf-format` script in the `elftoolchain` sources lacks the needed check. As a quick fix, you can modify it thus:
>
> ```
> $1 ~ "Machine:" {
>         if (match($0, [...]
Well, I compiled a random .c file, and this is the output of `readelf` on my machine:
```
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           ARM
  Version:                           0x1
  Entry point address:               0x4e5
  Start of program headers:          52 (bytes into file)
  Start of section headers:          7064 (bytes into file)
  Flags:                             0x5000400, Version5 EABI, hard-float ABI
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         9
  Size of section headers:           40 (bytes)
  Number of section headers:         29
  Section header string table index: 28
```
The "Machine" line somehow is "ARM", not "AArch64". So I changed that, then it compiled successfully, and now I get the image of elks. :-)
Thank you!

tkchia, 2022-02-01:
Hello @GNUfan42,

> So I changed that, then it compiled successfully, and now I get the image of elks. :-)

Cool!  :slightly_smiling_face:  Anyway, for 32-bit ARM binaries (`Machine: ARM`), the constant to set will be `EM_ARM`, rather than `EM_AARCH64`.  Maybe I will write up another small patch to do this.

Thank you!

closed 2022-02-01 by GNUfan42