ghaerr/elks #1131: [ktcp] Application read hangs under certain conditions

Mellvik, 2022-02-02:
For the record: This is and ELKS-to-ELKS issue.

Summary: The application hangs in socket read when there is no more data and the other end has closed the connection.

Here's the scenario:
- ELKS `ftp` client connected to ELKS `ftpd` server - this is one of several (and unrelated) observed problems, possibly the easiest to diagnose.
- The client fetches (`get`) files from the server

The problem:
- If the actual file contains 1024 bytes or less (> 0), the first read delivers the data, the second read never returns. If the file is 1025 bytes, meaning 3 incoming packets, it works fine. Less than 513 bytes (one packet), also hangs in read.
- This applies to active mode only (data connection made from the server to the client). In passive mode (server offers a port for the client to make a data connection to), this works fine, the second read returns 0 as it should. The packet traces are identical in both cases. In both cases the server sends FIN appropriately after delivering the data.
- It has been verified that the hang is in read, not in close.
- Turning on debug changes the behaviour - there is no hang (only tested with packet dump debug, which dramatically changes the timing).
- Retrieving the same size files from a different server (Raspbian) works fine, and looks very different at the packet level since the packet sizes are different.
- 

@ghaerr,
is it a viable idea to change the max transmit packet size on the server side and see what happens?
Which debug settings would be most useful in this scenario?

--M

ghaerr, 2022-02-02:
> is it a viable idea to change the max transmit packet size on the server side and see what happens?

Yes, it couldn't hurt. You could either change TDB_WRITE_MAX in include/linuxmt/tcpdev.h, which limits the amount of data sent to ktcp in a kernel for loop, or, the ftpd server could just send smaller packets by executing a write with less data. (Remember that ktcp never splits sending data packets to smaller values, but always sends the application buffer after being split by the kernel to TDB_WRITE_MAX).

it's probably easier for ftpd to change its write than recompiling the kernel etc. This could be made a variable value with a command line or environment var option for testing pretty easily.

> Which debug settings would be most useful in this scenario?

It sounds like this problem could be a race condition when a FIN is received prior to a read, which I thought we fully fixed a while back. Those messages would still be displayed using ktcp's DEBUG_TCP. Set DEBUG_CLOSE also.I will analyze your trace above, but what we are looking for would be changing the sending packet size to be marginally higher or lower than the so that the last packet arriving might have a FIN sent after an almost empty or almost full packet, to see if this is the FIN bug (again) or not.

Of course, since this is ELKS to ELKS and I presume you've never seen it elsewhere, this could be an new issue with ELKS TCP. Another thought would be trying to get this to occur using a Linux ftp talking to ELKS ftpd. Admittedly, there are quite a few variables by testing the relatively new ftp AND ftpd AND ELKS TCP on both sides!

> Less than 513 bytes (one packet), also hangs in read.

Lets definitely stay with various file sizes all under a packet size. If this is always the case, it might not be too hard to debug.

> Turning on debug changes the behaviour - there is no hang

Geez. Can you get any hang with single packet transfers with debug messages ever?

The state-transition code for read and EOF processing uses SF_CLOSING in elks/net/ipv4/af_inet.c. The socket needs to be set to SS_DISCONNECTED by ktcp before the read will return 0, which is that ftp is looking for, correct? That is, the last read actually returns 0 (EOF) rather than data, correct? If so, then we can look more closely at this scenario between ktcp and the kernel.

Mellvik, 2022-02-02:
Thanks you @ghaerr,
> is it a viable idea to change the max transmit packet size on the server side and see what happens?
>
> Yes, it couldn't hurt. You could either change TDB_WRITE_MAX in include/linuxmt/tcpdev.h, which [...]
Yes, that's good idea, much faster turnaround.
> Which debug settings would be most useful in this scenario?
> 
> It sounds like this problem could be a race condition when a FIN is received prior to a read, which I thought we fully fixed a while back.
> 
It sure looks like it, but that was the zero file size situation IIRC, which works fine in this scenario too.
> Of course, since this is ELKS to ELKS and I presume you've never seen it elsewhere, this could be an new issue with ELKS TCP. Another thought would be trying to get this to occur using a Linux ftp [...]
Indeed! Turned out I had some tuning to do in the FTP client in order to get the much slower transfers to work right (with the timeouts etc.)
> Less than 513 bytes (one packet), also hangs in read.
>
> Lets definitely stay with various file sizes all under a packet size. If this is always the case, it might not be too hard to debug.
>
> Turning [...]
I'll check whether this is still the case now that I've replaced DEBUG_TCPDAT with DEBUG_TCP and DEBUG_CLOSE.
> The state-transition code for read and EOF processing uses SF_CLOSING in elks/net/ipv4/af_inet.c. The socket needs to be set to SS_DISCONNECTED by ktcp before the read will return 0, which is that [...]
Yes, this is correct. 
I figured - if this one turns out to be hard to  catch, we'll just leave it till after 0.5 - now that we have it documented.

—M

Mellvik, 2022-02-03:
Update:
- The problem is definitely timing related, as evidenced by at least 3 different situations, including the removal of DEBUG_TCPDATA, which causes the behaviour to be 'normal', i.e. the problem being evident and consistent. Also, some tests fail some times, runs OK some times.
- There is also a component of state in this. If the previous data transfer had more than 2 packets in it, the next transfer is more likely to succeed, even if it has only 2 packets - and the 2nd of those packets is of reasonable size (more than like 200 bytes). [Seriously ...]
- I said yesterday that zero size files worked as expected. They don't. Transferring a zero size file behaves like the 1 or 2 packet size files, except that the read loop is never entered, so select will timeout and we have 'normal' termination.

Testing varying packet sizes by changing the ftp output buffer, confirms the initial problem report and eliminates max packet size as a component of the problem:

- Small or big packets, the trigger is that if the transfer has one or two data packets, the final read hangs.
- There is some instability in this: Some times - if the 2nd packet is big - it may succeed, say one out of 5 times.

The debug output below (w/some annotations) may be useful - it's like the previous one from this thread, but this time form the client side of things. First a transfer that fails, then one that succeeds. It may be of some value to observe that in the successful case the Transfer Complete (on the cmd channel) pops up after the data have been received, while in the case that fails, it comes before the data. I am however, not sure that this is relevant. On the wire (trace), the sequence of things is exactly the same in the two cases, so this is a scheduling thing in ELKS. Also, this IS the cmd channel, the problem is on the data channel.

```
ftp>                      <---- ftp client freshly started
ktcp: debug on      <----- ^P
get 767b             <---- the ftp command
---> TYPE I
200 Command OK.
PORT: Listening on 10.0.2.16 port 1215
tcp listen: port 1215 sock[00a1a0]
---> PORT 10,0,2,16,4,191
tcp listen: got SYN, cloning sock[00a1a0]
tcpdev checkaccept: sock[00a1a0] searching for newsock[00a1a0]
tcpdev checkaccept: found listen socket[00a1a0] newsocket[000000]
200 PORT command successful.
---> RETR 767b
150 Opening BINARY data connection for 767b (767 bytes).
tcpdev accept: ACCEPT (SYN received before accept) sock[00a1a0], using newsock[0
095c0]
Accepted connection from 10.0.2.15 @ 20
remote: 767b, local: 767b
get: select ctrl
tcpdev release: close socket 00a1a0, state is LISTEN
226 Transfer Complete.
get: got data
get: 767 fd 5
                                     <------------------------ ^C here to abort the hang
tcpdev release: close socket 009890, state is ESTABLISHED
# tcp[009890] setting state to FIN_WAIT_1
tcpdev release: close socket 0095c0, state is ESTABLISHED
tcp[0095c0] setting state to FIN_WAIT_1
tcp[009890] packet in fin_wait_1, fin: 0
tcp[009890] set state CLOSED
tcp[009890] packet in fin_wait_2, fin: 1
tcp[009890] setting state to TIME_WAIT
tcp[0095c0] packet in fin_wait_1, fin: 0
tcp: RST from 10.0.2.15:20->1215
tcp[009890] exit TIME_WAIT state on port 1214 remote 10.0.2.15:21
        <---- second transfer starts here --------->
# ./ftp -d -d -d -d -A 10.0.2.15
cmd line: ip 10.0.2.15 port 21 (default)
BIND: I am 10.0.2.16:1216
Connecting to 10.0.2.15 @ port 21
220 Welcome - ELKS minimal FTP server speaking.
Name (10.0.2.15:ftp): 
---> USER ftp
331 User OK.
Password:
---> PASS 
230 Password OK.
---> SYST
215 UNIX Type: L8 (Linux).
Remote system is Unix/Linux, using binary mode
ftp> get 1200b                    <------ Command
---> TYPE I
200 Command OK.
PORT: Listening on 10.0.2.16 port 1217
tcp listen: port 1217 sock[00a6f0]
---> PORT 10,0,2,16,4,193
tcp listen: got SYN, cloning sock[00a6f0]
tcpdev checkaccept: sock[00a6f0] searching for newsock[00a6f0]
tcpdev checkaccept: found listen socket[00a6f0] newsocket[000000]
200 PORT command successful.
---> RETR 1200b
150 Opening BINARY data connection for 1200b (1200 bytes).
tcpdev accept: ACCEPT (SYN received before accept) sock[00a6f0], using newsock[0
09980]
Accepted connection from 10.0.2.15 @ 20
remote: 1200b, local: 1200b
get: got data
tcpdev release: close socket 00a6f0, state is LISTEN   <----- this is where the first transfer stops
tcp[009980] packet in established, fin: 1, data: 0, setting state to CLOSE_WAIT
get: 1200 fd 5
get: datalen 1200
get: select ctrl
226 Transfer Complete.
ftp> tcpdev release: close socket 009980, state is CLOSE_WAIT
tcp[009980] setting state to LAST_ACK
tcp[009980] packet in last_ack, fin: 0
tcp[009980] set state CLOSED
```

ghaerr, 2022-02-03:
Thanks for the additional information.

> On the wire (trace), the sequence of things is exactly the same in the two cases, so this is a scheduling thing in ELKS

Are you saying that, from what you've seen, the packets on the wire are exactly as they should be? 

It would be interesting to confine the bug to a single packet transfer, say, of a 128-byte file: I assume this only repeats on ELKS-to-ELKS transfers, and almost always fails ELKS-to-ELKS. Does the wire packet trace look identical when sent from an external ftpd (which always works and never fails), versus the buggy ELKS ftpd? It would be interesting to know if the trace is identical, and this is only a timing issue, or whether there is something different in the packets sent between a working rPi ftpd vs buggy ELKS. I would think if the rPi sending transfer never fails, we might find something different in the packet trace than in the failure ELKS-to-ELKS mode.

The above will help sort out whether this is truly an ELKS scheduling issue, or whether there's something in the packet sequence, sent differently be a different ftpd, that is triggering the faulty behavior.

> Transferring a zero size file behaves like the 1 or 2 packet size files, except that the read loop is never entered, so select will timeout and we have 'normal' termination.

Separately, this worries me. It sounds there could be a case where if a data transfer doesn't start in, say 50 ms, then instead of waiting in select (and then read), the select will timeout and read will never be called. How does your code handle the case where the remote end takes a bit longer to start sending? Perhaps the select timeout kluge should be removed entirely for these tests.

Thank you!

Mellvik, 2022-02-04:
Thank you @ghaerr,

> 
> On the wire (trace), the sequence of things is exactly the same in the two cases, so this is a scheduling thing in ELKS
> 
> Are you saying that, from what you've seen, the packets on the wire are exactly as they should be?
> 
Up to the point of the hang, yes.
> It would be interesting to confine the bug to a single packet transfer, say, of a 128-byte file: I assume this only repeats on ELKS-to-ELKS transfers, and almost always fails ELKS-to-ELKS.
> 
For a single placket transfer that would be 'always fail' - in PORT mode, which is part of the mystery, and possibly narrows down the search.
> Does the wire packet trace look identical when sent from an external ftpd (which always works and never fails), versus the buggy ELKS ftpd? It would be interesting to know if the trace is identical, [...]
That's a good idea with the very small (128b) file. I'll collect some screen logs for that.
> The above will help sort out whether this is truly an ELKS scheduling issue, or whether there's something in the packet sequence, sent differently be a different ftpd, that is triggering the faulty [...]
That would be very advantageous indeed. Unfortunately, the select timeout is necessary for some of the odd conditions we encounter with ELKS being so much slower than anything else. The zero size file is one of them.

The select loop does not affect the current situation though. Select catches the 'data available for read' situation and a while-loop is entered to gobble up the data. This is where the hang happens. The first read takes all the data and the second read never returns.
If the file is zero size, select 'gets' neither data nor command replies and times out, which is why we have it in the first place. Even if we got a read in select which returned zero bytes (which is 'normal behaviour' I suppose), we'd need the timeout to catch the fact that the Completed message arrived on the command channel before the select loop started.

I'll be back with more.

-m

Mellvik, 2022-02-04:
@ghaerr,
some interesting observations from the suggested comparison - narrowing down the search for the culprit.

- Upon completing the data transfer, the Raspi (Linux) server immediately sends a FIN (like, the next packet on the wire), while our ELKS ftp server waits till after the 'Transfer Complete' status message before closing (and sending the FIN). This looked suspicious, so I moved the close() in the server code and tested again. As expected, the FIN now immediately follows the final data packet - in most cases. The change had a surprising effect: The problem got worse. The hang now happens if the file occupies 1-3 packets, not 1 or 2. OTOH, we know that the time of arrival of the FIN is part of the problem. 

- The next observation is possibly even more interesting: the Linux server, upon receiving no immediate response to the first FIN (except it being properly acked by ELKS), sends another FIN, which seems to resolve the situation.

So, what seems to be happening is that the client side ktcp ACKs the FIN but fails to use/recognize it, which in the Linux case is being resolved by a second FIN, arriving 0.23 seconds after the first. 

I guess the question then is whether to look for the reason for the miss ('the right thing') or just send another FIN after a small delay ('the easy way out') …

What do you think?

—M

> 4. feb. 2022 kl. 09:51 skrev Helge Skrivervik ***@***.***>:
> 
> Thank you @ghaerr,
> 
>> 
>> On the wire (trace), the sequence of things is exactly the same in the two cases, so this is a scheduling thing in ELKS
>> 
>> Are you saying that, from what you've seen, the packets on the wire are exactly as they should be?
>> 
> Up to the point of the hang, yes.
>> It would be interesting to confine the bug to a single packet transfer, say, of a 128-byte file: I assume this only repeats on ELKS-to-ELKS transfers, and almost always fails ELKS-to-ELKS.
>> 
> For a single placket transfer that would be 'always fail' - in PORT mode, which is part of the mystery, and possibly narrows down the search.
>> Does the wire packet trace look identical when sent from an external ftpd (which always works and never fails), versus the buggy ELKS ftpd? It would be interesting to know if the trace is identical, and this is only a timing issue, or whether there is something different in the packets sent between a working rPi ftpd vs buggy ELKS. I would think if the rPi sending transfer never fails, we might find something different in the packet trace than in the failure ELKS-to-ELKS mode.
>> 
> That's a good idea with the very small (128b) file. I'll collect some screen logs for that.
>> The above will help sort out whether this is truly an ELKS scheduling issue, or whether there's something in the packet sequence, sent differently be a different ftpd, that is triggering the faulty behavior.
>> 
>> Transferring a zero size file behaves like the 1 or 2 packet size files, except that the read loop is never entered, so select will timeout and we have 'normal' termination.
>> 
>> Separately, this worries me. It sounds there could be a case where if a data transfer doesn't start in, say 50 ms, then instead of waiting in select (and then read), the select will timeout and read will never be called. How does your code handle the case where the remote end takes a bit longer to start sending? Perhaps the select timeout kluge should be removed entirely for these tests.
>> 
> That would be very advantageous indeed. Unfortunately, the select timeout is necessary for some of the odd conditions we encounter with ELKS being so much slower than anything else. The zero size file is one of them.
> 
> The select loop does not affect the current situation though. Select catches the 'data available for read' situation and a while-loop is entered to gobble up the data. This is where the hang happens. The first read takes all the data and the second read never returns.
> If the file is zero size, select 'gets' neither data nor command replies and times out, which is why we have it in the first place. Even if we got a read in select which returned zero bytes (which is 'normal behaviour' I suppose), we'd need the timeout to catch the fact that the Completed message arrived on the command channel before the select loop started.
> 
> I'll be back with more.
> 
> -m
> 
>

ghaerr, 2022-02-04:
> So, what seems to be happening is that the client side ktcp ACKs the FIN but fails to use/recognize it, which in the Linux case is being resolved by a second FIN, arriving 0.23 seconds after the first.

I am going to guess that this is exactly the problem - the ELKS client ACKs a FIN but doesn't do anything with it in both cases, but the Raspi server sends another FIN, while ELKS server does not.

I had thought you were saying that the packet traces were exactly the same?

Does this always happen, that is, this is not timing related and always fails, or is this whole thing timing related as well?

Next question is, exactly what is the packet trace at this point - after the final (single data) packet, both servers send a FIN in a separate packet, but the working server sends a FIN again. Why? I'm wondering what causes the working server to resend a FIN? Is there a protocol violation, wrong ACK sequence number, ? The ACK resend in 1/4 seconds seems quite quick.

I'll look at ktcp to see where this might be, I need to know whether the FIN is part of a data packet or separate.

> I guess the question then is whether to look for the reason for the miss ('the right thing') or just send another FIN after a small delay ('the easy way out') …

Not sure yet, but it seems you have likely identified the bug!

Mellvik, 2022-02-04:
> So, what seems to be happening is that the client side ktcp ACKs the FIN but fails to use/recognize it, which in the Linux case is being resolved by a second FIN, arriving 0.23 seconds after the [...]
NO. In the first post I mentioned testing against the Raspi and that the traces were very different because the packet sizes were so different. I did not at the time test very small amounts of data - or notice the extra FIN at the end.
> Does this always happen, that is, this is not timing related and always fails, or is this whole thing timing related as well?
> 
I guess that depends on perspective. It's entirely repeatable except when playing around with a data size that is close to the limit that triggers the error (currently 3 packets). And AFAIK the entire problem is timing related. As mentioned before, putting enough debug prints into the loops and the problem goes away.
> Next question is, exactly what is the packet trace at this point - after the final (single data) packet, both servers send a FIN in a separate packet, but the working server sends a FIN again. Why? [...]
Well, that is a da.. good question. The dive back into the traces reveals that I was wrong, the first FIN is not being acked at all. Really hard to catch since the FIN ack should be the FIN sequence #  + 1, and what I have is an empty ACK packet immediately following the FIN with the FIN seq #. Actually the second ack for the last packet before the FIN. This makes much more sense now. A retransmit after .25 sec isn't unreasonable, a resend would be - as you point out.
So that changes the picture somewhat. The FIN gets lost earlier than I thought. But why doesn't ktcp retransmit the never-acked FIN?

I'll post the trace separately.

Thank you!

—M

Mellvik, 2022-02-04:
Trace referenced above - with some annotations.
The first part is elks-to-elks, which hangs - as noted. The second is ELKS to Raspi.

```
10:15:59.266808 IP (tos 0x0, ttl 64, id 2099, offset 0, flags [none], proto TCP (6), length 62)
    10.0.2.16.1163 > 10.0.2.15.ftp: Flags [P.], cksum 0x802b (correct), seq 32:54, ack 127, win 4380, length 22: FTP, length: 22
	PORT 10,0,2,16,4,140
10:15:59.268349 IP (tos 0x0, ttl 64, id 1977, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp > 10.0.2.16.1163: Flags [.], cksum 0xa078 (correct), ack 54, win 4358, length 0
10:15:59.285127 IP (tos 0x0, ttl 64, id 1978, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp > 10.0.2.16.1163: Flags [.], cksum 0xa062 (correct), ack 54, win 4380, length 0
10:15:59.289727 IP (tos 0x0, ttl 64, id 1979, offset 0, flags [none], proto TCP (6), length 44)
    10.0.2.15.ftp-data > 10.0.2.16.1164: Flags [S], cksum 0x9a07 (correct), seq 532394121, win 4380, options [mss 1460], length 0
10:15:59.304311 IP (tos 0x0, ttl 64, id 2100, offset 0, flags [none], proto TCP (6), length 44)
    10.0.2.16.1164 > 10.0.2.15.ftp-data: Flags [S.], cksum 0x87e1 (correct), seq 574287834, ack 532394122, win 4380, options [mss 1460], length 0
10:15:59.305694 IP (tos 0x0, ttl 64, id 1980, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp-data > 10.0.2.16.1164: Flags [.], cksum 0x9f9e (correct), ack 1, win 4380, length 0
10:15:59.308611 IP (tos 0x0, ttl 64, id 1981, offset 0, flags [none], proto TCP (6), length 70)
    10.0.2.15.ftp > 10.0.2.16.1163: Flags [P.], cksum 0xe354 (correct), seq 127:157, ack 54, win 4380, length 30: FTP, length: 30
	200 PORT command successful.
10:15:59.311589 IP (tos 0x0, ttl 64, id 2101, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.16.1163 > 10.0.2.15.ftp: Flags [.], cksum 0xa062 (correct), ack 157, win 4350, length 0
10:15:59.331797 IP (tos 0x0, ttl 64, id 2102, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.16.1163 > 10.0.2.15.ftp: Flags [.], cksum 0xa044 (correct), ack 157, win 4380, length 0
10:15:59.369668 IP (tos 0x0, ttl 64, id 2103, offset 0, flags [none], proto TCP (6), length 51)
    10.0.2.16.1163 > 10.0.2.15.ftp: Flags [P.], cksum 0x3b23 (correct), seq 54:65, ack 157, win 4380, length 11: FTP, length: 11
	RETR 128b
10:15:59.371150 IP (tos 0x0, ttl 64, id 1982, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp > 10.0.2.16.1163: Flags [.], cksum 0xa044 (correct), ack 65, win 4369, length 0
10:15:59.385080 IP (tos 0x0, ttl 64, id 1983, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp > 10.0.2.16.1163: Flags [.], cksum 0xa039 (correct), ack 65, win 4380, length 0
10:15:59.396551 IP (tos 0x0, ttl 64, id 1984, offset 0, flags [none], proto TCP (6), length 98)
    10.0.2.15.ftp > 10.0.2.16.1163: Flags [P.], cksum 0x03f1 (correct), seq 157:215, ack 65, win 4380, length 58: FTP, length: 58
	150 Opening BINARY data connection for 128b (128 bytes).
10:15:59.399906 IP (tos 0x0, ttl 64, id 2104, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.16.1163 > 10.0.2.15.ftp: Flags [.], cksum 0xa039 (correct), ack 215, win 4322, length 0
10:15:59.400629 IP (tos 0x0, ttl 64, id 1985, offset 0, flags [none], proto TCP (6), length 168)
    10.0.2.15.ftp-data > 10.0.2.16.1164: Flags [P.], cksum 0x1522 (correct), seq 1:129, ack 1, win 4380, length 128
10:15:59.404472 IP (tos 0x0, ttl 64, id 1986, offset 0, flags [none], proto TCP (6), length 64)
    10.0.2.15.ftp > 10.0.2.16.1163: Flags [P.], cksum 0xd109 (correct), seq 215:239, ack 65, win 4380, length 24: FTP, length: 24
	226 Transfer Complete.
10:15:59.405625 IP (tos 0x0, ttl 64, id 2105, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.16.1164 > 10.0.2.15.ftp-data: Flags [.], cksum 0x9f9e (correct), ack 129, win 4252, length 0
10:15:59.410190 IP (tos 0x0, ttl 64, id 2106, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.16.1163 > 10.0.2.15.ftp: Flags [.], cksum 0xa039 (correct), ack 239, win 4298, length 0
10:15:59.412933 IP (tos 0x0, ttl 64, id 2107, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.16.1163 > 10.0.2.15.ftp: Flags [.], cksum 0x9fe7 (correct), ack 239, win 4380, length 0
10:15:59.424291 IP (tos 0x0, ttl 64, id 1987, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp-data > 10.0.2.16.1164: Flags [F.], cksum 0x9f1d (correct), seq 129, ack 1, win 4380, length 0
10:15:59.577040 IP (tos 0x0, ttl 64, id 2108, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.16.1164 > 10.0.2.15.ftp-data: Flags [.], cksum 0x9f1e (correct), ack 129, win 4380, length 0

    -------------------Hangs here - see the time stamps, eventually interrupted by ^C

10:16:06.139315 IP (tos 0x0, ttl 64, id 2109, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.16.1163 > 10.0.2.15.ftp: Flags [F.], cksum 0x9fe6 (correct), seq 65, ack 239, win 4380, length 0
10:16:06.140946 IP (tos 0x0, ttl 64, id 1988, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.ftp > 10.0.2.16.1163: Flags [.], cksum 0x9fe6 (correct), ack 66, win 4380, length 0
10:16:06.142612 IP (tos 0x0, ttl 64, id 2110, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.16.1164 > 10.0.2.15.ftp-data: Flags [F.], cksum 0x9f1d (correct), seq 1, ack 129, win 4380, length 0
10:16:06.149135 IP (tos 0x0, ttl 64, id 1989, offset 0, flags [none], proto TCP (6), length 40)
[... cut ...]
    10.0.2.15.1048 > 10.0.2.2.ftp-data: Flags [.], cksum 0x4646 (correct), ack 130, win 4380, length 0
17:32:44.422024 IP (tos 0x0, ttl 64, id 400, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1048 > 10.0.2.2.ftp-data: Flags [F.], cksum 0x4645 (correct), seq 1, ack 130, win 4380, length 0
17:32:44.422162 IP (tos 0x8, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.ftp-data > 10.0.2.15.1048: Flags [.], cksum 0xe550 (correct), ack 2, win 29200, length 0
```

ghaerr, 2022-02-04:
> So that changes the picture somewhat. The FIN gets lost earlier than I thought. But why doesn't ktcp retransmit the never-acked FIN?

I'm traveling today, so I haven't had time to fully digest what is happening. Are you saying that when ELKS server sends a FIN packet to ELKS client, that that FIN (which is always in a separate packet, correct?) never gets ACKed by the ELKS client/receiver? (Or rather, sometimes does not get ACK'd, which causes this bug?)

Adding to the problem, ELKS server is never retrying the FIN packet. Is that a decent summary of what you're seeing?

> and what I have is an empty ACK packet immediately following the FIN with the FIN seq #. Actually the second ack for the last packet before the FIN. 

So the last ACK'd packet from ELKS client/receiver is just an single ACK packet for the last (only) data packet? And then the FIN is received, not processed, and ftp client hangs, waiting for data? On Raspi the FIN is resent, and thus the read completes (=0 EOF) and works "normally". Is that it?

Thank you!

Mellvik, 2022-02-05:
> So that changes the picture somewhat. The FIN gets lost earlier than I thought. But why doesn't ktcp retransmit the never-acked FIN?
>
> I'm traveling today, so I haven't had time to fully digest what [...]
Yes .... always in a separate packet
> never gets ACKed by the ELKS client/receiver? (Or rather, sometimes does not get ACK'd, which causes this bug?)
> 
> Adding to the problem, ELKS server is never retrying the FIN packet. Is that a decent summary of what you're seeing?
> 
'Decent' is appropriate  @ghaerr. Under the circumstances described (3 or fewer packets in the current setting, 2 or fewer before I moved up the close() in the application) this behavious is repeatable and predictable with one 'anomaly': If the 3rd packet approaches its data size limit, it sometimes work. Which makes sense given the obvious timing dependece of the problem.
> and what I have is an empty ACK packet immediately following the FIN with the FIN seq #. Actually the second ack for the last packet before the FIN.
> 
> So the last ACK'd packet from ELKS client/receiver is just an single ACK packet for the last (only) data packet?
> 
Yes, my mistake. Ktcp's extra acks had me fooled there for a bit. The second ack appears just after the FIN most the time, and I connected them, forgetting that the FIN needs a seq+1 ack.
> And then the FIN is received, not processed, and ftp client hangs, waiting for data?
> 
No and yes: The FIN is on the wire, and - looking at the ELKS client ktcp debug outpu posted earlier on this thread, it's received and partly acted upon (if I'm reading the debug output correctly):

tcpdev release: close socket 00a1a0, state is LISTEN

This is where it stops. OTOH, if the file is larger (more packets), the debug output is like
tcpdev release: close socket 00a6f0, state is LISTEN 
tcp[009980] packet in established, fin: 1, data: 0, setting state to CLOSE_WAIT
So it seems the FIN has been received and processed, but why has it not been acked? Maybe that's a clue …

> On Raspi the FIN is resent, and thus the read completes (=0 EOF) and works "normally". Is that it?
> 
> 
Yes that's it.

Thank you!

—M

ghaerr, 2022-02-06:
> tcpdev release: close socket 00a1a0, state is LISTEN

This is strange. Are you saying that ktcp is closing a listen socket? That would be the socket *prior* to executing accept. In the listen state, a FIN should never be received nor ACKd. I see in the later trace (I presume working), the state is also LISTEN. Does that one work? I might be missing something? The ftp client doesn't execute fork then close on the listen socket, correct?

Is there a chance that a close could be executed on the listen socket, or perhaps the (client) application is executing a close prior or immediately after accept on the proper socket? 

I may be confused here on what is happening.

To answer why a FIN is not ACK'd by the client (ignoring my discussion above), see the following code in tcp_established() in tcp.c:
```
    if (h->flags & TF_FIN) {
        cb->rcv_nxt++;
        debug_close("tcp[%p] packet in established, fin: 1, data: %d, setting state to CLOSE_WAIT\n", cb->sock, datasize);

        cb->state = TS_CLOSE_WAIT;
        cb->time_wait_exp = Now;        /* used for debug output only*/
        debug_tcp("tcp: got FIN with data %d buffer %d\n", datasize, cb->buf_used); // <-- change to printf
        if (cb->bytes_to_push <= 0)    // <--- comment out these two lines
            tcpdev_sock_state(cb, SS_DISCONNECTING);   // <--- comment out
    }

    if (datasize == 0 && ((h->flags & TF_ALL) == TF_ACK))
        return; /* ACK with no data received - so don't answer*/

    cb->rcv_nxt += datasize;
    debug_window("tcp: ACK seq %ld len %d\n", cb->rcv_nxt - cb->irs, datasize); // <--- turn on DEBUG_WINDOW to see if ACK sent
    tcp_send_ack(cb);
}
```
I am wondering whether setting the SS_DISCONNECTING state on the socket might be somehow stopping ktcp from outputting the ACK in the code following, depending on how/when ktcp is scheduled. I can see code where setting SS_DISCONNECTING will wake up the waiting (ftp) client. I'm not sure if ftp client is waiting in select or read at that point. If waiting in a read, it is possible that the ftp client could return from read with -1 and errno=EINTR. 

 Setting DEBUG_WINDOW will show whether an ACK is sent on the received FIN in a data packet.

Mellvik, 2022-02-06:
Thank you @ghaerr,

Remote debugging is hard, and quite frequently communication (the clear type) is even harder...
> tcpdev release: close socket 00a1a0, state is LISTEN
>
> This is strange. Are you saying that ktcp is closing a listen socket? That would be the socket prior to executing accept. In the listen state, a [...]
I'm guilty at creating the confusion. I erroneously connected the close of the listen socket with the (expected) close of the closed accept socket. The 'tcpdev release' message has nothing to do with the problem at hand.
Thus my statement that the FIN is being acted upon was incorrect. It's apparently not, and your analysis is correct. 
Regardless, we're one iteration closer.
> To answer why a FIN is not ACK'd by the client (ignoring my discussion above), see the following code in tcp_established() in tcp.c:
>
>     if (h->flags & TF_FIN) {
>         cb->rcv_nxt++; [...]
I think this was useful. It seems, in the case that hangs, we ever get tho this code at all.
Maybe this trace will be meaningful to you.
In the case that works (and ignoring whatever is happening on the data channel, which is the Tranfer Completed message), it looks like this:

ftp: Accepted connection from 10.0.2.15 @ 20
ftp: remote: 2000b, local: 2000b
ftp: get: got data
tcp: recv data len 500 avail 2880
tcp: ACK seq 2001 len 500        <—— the entire file has been received (2000 bytes)
tcp output: len 0 space 2380 window 2380
tcp[0095c0] packet in established, fin: 1, data: 0, setting state to CLOSE_WAIT
tcp: got FIN with data 0 buffer 2000
tcp: ACK seq 2002 len 0
tcp output: len 0 space 2380 window 2380
tcpdev release: close socket 00b320, state is LISTEN
tcp: recv data len 24 avail 4380   <—----- command channel
tcp: ACK seq 241 len 24
tcp output: len 0 space 4356 window 4356  <— cmd chan ack
ftp: get: 1492 fd 5                <— 1st read in ftp
tcp: dropping packet, bad seqno: need 241 got 217 size 24
tcp output: len 0 space 4356 window 4356
ftp: get: 508 fd 5                 <— 2nd read in ftp
ftp: get: datalen 2000             <- read done, got zero
ftp: get: select ctrl              <- select got data on the cmd chan
ftp: 226 Transfer Complete.        <— cmd chan data (len 24)
tcp: extra ACK seq 241, app read 24 bytes     <— cmd chan ack
ftp>                         <————ftp prompt is back
tcp output: len 0 space 4380 window 4380
tcpdev release: close socket 0095c0, state is CLOSE_WAIT
tcp[0095c0] setting state to LAST_ACK
tcp output: len 0 space 4380 window 4380
tcp[0095c0] packet in last_ack, fin: 0
tcp[0095c0] set state CLOSED

This is the case that hangs - and BTW - it hangs in read, not in select. When select indicates data available, a while loop read data until zero or fail.

ftp: Accepted connection from 10.0.2.15 @ 20
ftp: remote: 999b, local: 999b
tcp: recv data len 24 avail 4380      <——  we've seen this before, from the Raspi FTP server, when the 
				file is small, the 'transfer complete' message on the cmd channel 
				arrives before any data.
ftp: get: got data		<— select found data to read
tcp: ACK seq 346 len 24		<— cmd chan ack
tcp output: len 0 space 4356 window 4356    <— send cmd chan ack
tcpdev release: close socket 00b140, state is LISTEN   <— leftover from the original listen sock
	—— This may be interesting, there is no recv data message and no data ack message (see bolded above) ——
	—— … but there is an ack on the wire, as can be seen from the previous traces.
tcp: extra ACK seq 1000, app read 999 bytes  
ftp: get: 999 fd 5
tcp output: len 0 space 4380 window 4380   <— output the 'extra ack'
——————————— ———————— ^C to abort here

tcpdev release: close socket 00afb0, state is ESTABLISHED
# tcp[00afb0] setting state to FIN_WAIT_1
tcp output: len 0 space 4356 window 4356

# 
tcpdev release: close socket 00b190, state is ESTABLISHED
# tcp[00b190] setting state to FIN_WAIT_1
tcp output: len 0 space 4380 window 4380
tcp[00afb0] packet in fin_wait_1, fin: 0
tcp[00afb0] set state CLOSED
tcp[00afb0] packet in fin_wait_2, fin: 1
tcp[00afb0] setting state to TIME_WAIT
tcp output: len 0 space 4356 window 4356
tcp[00b190] packet in fin_wait_1, fin: 0
tcp: RST from 10.0.2.15:20->1029
tcp[00afb0] exit TIME_WAIT state on port 1027 remote 10.0.2.15:21

toncho11, 2022-02-06:
Actually ... I noticed that in one session ktcp was generating a message like this:

`tcp: refusing packet from ...`

even without transferring a file. I do not know if this helps. It was annoying though because it interferes while you type on the console.

ghaerr, 2022-02-06:
> I think this was useful. It seems, in the case that hangs, we ever get tho this code at all.

Interesting - I have another piece of code for you to check - there is another case where ktcp will drop a packet, which would cause this to happen. See below.

> Maybe this trace will be meaningful to you.

I notice even on the working trace that ktcp drops a packet (with an earlier sequence number). This seems to indicate that the sender may be already retrying and getting a bit out of sync. This probably isn't directly related to the problem at hand, but wanted to point out that whenever I see a dropped packet, I start wondering why.

In tcp.c::tcp_established(), we have the following:
```
    if (cb->unaccepted && (h->flags & TF_FIN)) {
        debug_tcp("tcp: FIN received before accept, dropping packet\n"); // <--- change to printf

        /* We can't change state before accept processing, so drop packet*/
        return;
    }
```
You mention that for short transfers, the command channel gets Transfer Completed before the data channel even gets read. Now, reading the above, I'm wondering whether the FIN is received prior to accept (as was discussed briefly in the above conversation). In the above line, change the debug_tcp to printf, and lets see whether we get the "FIN received before accept, dropping packet" message. If so, then what I think is happening is that ktcp is dropping the packet, but the sender isn't ever resending (for some reason, it should be retrying, but that's another issue).

A possible temp hack fix for the above issue, if this is the cause, would be to add `tcp_send_ack(cb);` before the return in the above code.

Thank you!

Mellvik, 2022-02-07:
Thank you @ghaerr.
> I think this was useful. It seems, in the case that hangs, we ever get tho this code at all.
>
> Interesting - I have another piece of code for you to check - there is another case where ktcp will drop [...]
Understandable, and the observation may be useful. I'm consistently ignoring these because I get them all the time. This is what running on physical floppy based systems does - it's actually good for testing robustness, but admittedly not for my attention. IN this case, it's worth noticing that the dropped packet 'belongs' to the command channel through.
> In tcp.c::tcp_established(), we have the following:
>
>     if (cb->unaccepted && (h->flags & TF_FIN)) {
>         debug_tcp("tcp: FIN received before accept, dropping packet\n"); // <--- change to [...]
I don't think that is the issue this time. I started off with the opposite actually - the distance between the last (only) data packet and the FIN (on the wire) was much smaller when using the Raspi client than elks-to <http://elks.to/>-elks. So I moved the data channel close() statement in the ftp client code closer to the actual write(), which had the (previously reported) effect of changing the 'signature' of the problem from 2 to 3 packets (3 or fewer creates the problem). Not really providing more of a clue than that the problem is time related. (BTW I did make the suggested change above, and got no new output).

The Transfer Complete message does not precede the last (only) data packet on the wire, but does so in ELKS, so it's a ktcp scheduling issue, and neither surprising nor a problem as far as I can see. The server jumps right to the send_reply(cmdchan) after writing the last bytes on the command chan. The change mentioned above moved up the close(datachan) to precede the send_reply(cmdchan).

SO we may not be closer to the actual bug, but have eliminated more alternatives.

—M

ghaerr, 2022-02-07:
> I don't think that is the issue this time.

That could be, the until we change the debug_tcp to printf, we won't know whether this drop packet code is what's causing the problem. There isn't any other code in tcp_established that returns (other than RST received), so if this isn't the issue, then it seems that ktcp isn't receiving the FIN packet at all. 

Earlier, you were saying that we are not even getting to the code in tcp_established which recognizes (sees) the FIN packet. So I'm trying to trace down how it is that the receiving ktcp isn't seeing the FIN that you say is on the wire.

Mellvik, 2022-02-07:
> 7. feb. 2022 kl. 16:37 skrev Gregory Haerr ***@***.***>:
> 
> 
> I don't think that is the issue this time.
> 
> That could be, the until we change the debug_tcp to printf, we won't know whether this drop packet code is what's causing the problem.
> 
Like I said, I tried it anyway and got the expected result. No new message.
> There isn't any other code in tcp_established that returns (other than RST received), so if this isn't the issue, then it seems that ktcp isn't receiving the FIN packet at all.
> 
Or maybe we have a race condition causing the packet to end up in the wrong queue/cb? 
FWIW - just to make sure, I checked of  other network activity would release the hang - like the situation we had with telnet before. It did not. Neither does running a long 'hd /bin/vi' in a telnet window while doing the short file transfer. The latter was sort of surprising since it does change the timing of things significantly.
> Earlier, you were saying that we are not even getting to the code in tcp_established which recognizes (sees) the FIN packet. So I'm trying to trace down how it is that the receiving ktcp isn't seeing the FIN that you say is on the wire.
> 
That's right. And it's really weird (see the tcpdumps). 
I was hoping that the difference between the debug prints in the working case vs the not-working case from yesterday would provide a clue.

BTW, possibly significant and easy to forget - if the data connection is initiated from the client (connect instead of accept) everying works fine. Always.

Thank you!

—M

Mellvik, 2022-02-07:
> 
Thank you @ghaerr,
> When the problem occurs, have any data packets been received after SYN/ACK? Does the client execute a read on that packet and get the data, or hang never getting the first data packet?
> 
It gets the data, it's the 2nd read that hangs waiting for a 0 return.
> I am assuming the FIN packet is always in a separate packet.
> 
yes. 
> The state before ESTABLISHED IS SYN_RECEIVED. This is only setup when the client executes a listen.
> 
> 
Now, here's something possibly interesting in that regard. In the debug output listings I submitted yesterday - one for a working transfer, one for a hung transfer, the sequence of events is kind of odd.
In both cases, the
—> tcpdev accept: ACCEPT (SYN received before accept) …
message comes just before the application (FTP client) says
—> Accepted connection from … etc.
Like this:
——
tcpdev accept: ACCEPT (SYN received before accept) sock[00b140], using newsock[0
0b190]
Accepted connection from 10.0.2.15 @ 20
remote: 999b, local: 999b
tcp: recv data len 24 avail 4380          <—— cmd channel
get: got data
tcp: ACK seq 346 len 24                   <— cmd channel
tcp output: len 0 space 4356 window 4356
tcpdev release: close socket 00b140, state is LISTEN  <—— listen socket, mother of accept socket
tcp: extra ACK seq 1000, app read 999 bytes     <—— original ack far above
get: 999 fd 5                    <—— first read
tcp output: len 0 space 4380 window 4380
                                 <— hang here
——

The interesting part is that
1) In both cases, one or more data packets have been received and ACKed before the 'tcpdev accept' message
2) In the working case, more packets arrive after the tcpdev accept message
3) In the non-working case, everything was received before the tcpdev accept message.

Maybe that observation can narrow down the search somewhat?
> I'm trying to map the state transitions of ktcp to the received packets. If you turn on DEBUG_TCPDEV, that will show the state transitions, which, along with the packets received, may show where this is happening, since it seems it is not in tcp_established.
> 
> 
Yes, let me know if you still need this given the above. 

—M

Mellvik, 2022-02-08:
@ghaerr,
another round on this issue, and with your insistence on looking for a FIN before accept, I found it. 

That's it - FIN before accept has completed – just like you suggested, and I missed it the first time (lame excuse - the debug message came far earlier than I expected).

Anyway, the problem has been located and the next question is how to fix it.

- Ignoring FINs on a connection that hasn't reached established state, seems to make sense.
- OTOH - it seems odd that the state change takes so long. Looking at the packet trace, the SYN/SYN+ACK/ACK are back-to-back, so the connection is really in ESTABLISHED state, but it seems like ktcp takes very long to make that change 'official', and that that is the underlying problem.
- Also, data is being accepted (ACKed and moved into the system buffers) on that connection, and it may be argued that a FIN should be handled similarly. BTW, isn't it odd that data is being accepted before the state change?

- The easy way out is to ignore the problem and make sure the server side retransmits the FIN after a short delay, like the Linux server does. It's actually interesting that that doesn't happen already, maybe that's a different bug altogether.

- It is also possible - as a temporary kluge - to delay the close() in the ftpd server when the file is small. It will slow down multi-file transfers, but not all that much (like a usleep(250000)).

Any thoughts?

—M

> 7. feb. 2022 kl. 18:34 skrev Helge Skrivervik ***@***.***>:
> 
>> 
> Thank you @ghaerr,
>> When the problem occurs, have any data packets been received after SYN/ACK? Does the client execute a read on that packet and get the data, or hang never getting the first data packet?
>> 
> It gets the data, it's the 2nd read that hangs waiting for a 0 return.
>> I am assuming the FIN packet is always in a separate packet.
>> 
> yes. 
>> The state before ESTABLISHED IS SYN_RECEIVED. This is only setup when the client executes a listen.
>> 
>> 
> Now, here's something possibly interesting in that regard. In the debug output listings I submitted yesterday - one for a working transfer, one for a hung transfer, the sequence of events is kind of odd.
> In both cases, the
> —> tcpdev accept: ACCEPT (SYN received before accept) …
> message comes just before the application (FTP client) says
> —> Accepted connection from … etc.
> Like this:
> ——
> tcpdev accept: ACCEPT (SYN received before accept) sock[00b140], using newsock[0
> 0b190]
> Accepted connection from 10.0.2.15 @ 20
> remote: 999b, local: 999b
> tcp: recv data len 24 avail 4380          <—— cmd channel
> get: got data
> tcp: ACK seq 346 len 24                   <— cmd channel
> tcp output: len 0 space 4356 window 4356
> tcpdev release: close socket 00b140, state is LISTEN  <—— listen socket, mother of accept socket
> tcp: extra ACK seq 1000, app read 999 bytes     <—— original ack far above
> get: 999 fd 5                    <—— first read
> tcp output: len 0 space 4380 window 4380
>                                  <— hang here
> ——
> 
> The interesting part is that
> 1) In both cases, one or more data packets have been received and ACKed before the 'tcpdev accept' message
> 2) In the working case, more packets arrive after the tcpdev accept message
> 3) In the non-working case, everything was received before the tcpdev accept message.
> 
> Maybe that observation can narrow down the search somewhat?
>> I'm trying to map the state transitions of ktcp to the received packets. If you turn on DEBUG_TCPDEV, that will show the state transitions, which, along with the packets received, may show where this is happening, since it seems it is not in tcp_established.
>> 
>> 
> Yes, let me know if you still need this given the above. 
> 
> —M
>

ghaerr, 2022-02-08:
Hello @Mellvik,

> FIN before accept has completed – just like you suggested, and I missed it the first time 

Are you saying that you have seen the message "tcp: FIN received before accept, dropping packet", and that is now confirmed that this is the reason the state change later in tcp_established to CLOSE_WAIT (which sets SS_DISCONNECTING to force an EOF read return) does not occur?

BTW, does the receiving application end up reading a single packet (less than 512 bytes), and then hang on the next read (waiting for EOF/FIN), or does it never do a read at all? (I think it should be doing that, since data has been received on the connection.)

Dropping the packet in the established state was added a while ago to fix the accept processing bug found earlier, and like the comment says, we can't change state prior to an accept being executed, because in ELKS, the "new" socket descriptor is created and passed to ktcp at accept time. This could possibly be fixed, but not without rewriting/redesigning the way ktcp and the kernel interface to it work, so I decided to punt instead and drop the packet, forcing a retry.

> it seems odd that the state change takes so long. Looking at the packet trace, the SYN/SYN+ACK/ACK are back-to-back, so the connection is really in ESTABLISHED state, but it seems like ktcp takes very long to make that change 'official', and that that is the underlying problem.

Remember, ktcp is a separate process and so is the application and are subject to scheduling; the application could be in the middle of creating a filename on disk perhaps, and we know this is a timing issue. The TCP <-> TCP connectivity is asynchronous to the applications above it to a degree, except in this case, ktcp has to force a synchronization by prohibiting a state change until accept is executed.

> so the connection is really in ESTABLISHED state

Yes, it actually is set to ESTABLISHED after the SYN is ACKed, and ktcp is storing data on the connection. The problem is that the *next* state change to CLOSE_WAIT after a FIN is seen needs to be delayed until an accept is executed due to the current design.

> BTW, isn't it odd that data is being accepted before the state change?

No, because the state has actually changed to ESTABLISHED.  Like I said, with some redesign this could be fixed, but I don't really want to go down that path (yet), as there'd be lots of retesting to ensure correctness. We could set a flag, for instance, but tcp doesn't use any flags currently... It's that kind of thing.

> The easy way out is to ignore the problem and make sure the server side retransmits the FIN after a short delay, like the Linux server does. 

Well, we already are forcing a retransmit by dropping the packet, which fixed an earlier reported bug. Linux and Rasbpi apparently always retransmit, and that's why it works talking to them.

> It's actually interesting that that doesn't happen already, maybe that's a different bug altogether.

Yes - and that's what we need to fix, since this issue is ELKS to ELKS only, and that's the problem. If you agree with everything above, then we need to switch and look over to the retransmit code, and see why the first FIN packet is not retransmitting. I have already looked at the tcp_output.c code, and can't see why it isn't happening. Turning on DEBUG_RETRANS should help show where the issue is on the sending side.

> as a temporary kluge - to delay the close() in the ftpd server when the file is small.

No, I don't think we should write any application kluges unless we absolutely have to. We went down that road before, and end up in more trouble eventually.

Thank you!

Mellvik, 2022-02-08:
One more thing, @ghaerr,

activating DEBUG_RETRANS on the server side changes the timing sufficiently to eliminate the symptom.

Are there specific retransmit printfs that may be of particular interest?

—M

> 8. feb. 2022 kl. 16:56 skrev Helge Skrivervik ***@***.***>:
> 
> HI @ghaerr,
> 
>> 
>> FIN before accept has completed – just like you suggested, and I missed it the first time
>> 
>> Are you saying that you have seen the message "tcp: FIN received before accept, dropping packet", and that is now confirmed that this is the reason the state change later in tcp_established to CLOSE_WAIT (which sets SS_DISCONNECTING to force an EOF read return) does not occur?
>> 
> Yes!
>> BTW, does the receiving application end up reading a single packet (less than 512 bytes), and then hang on the next read (waiting for EOF/FIN), or does it never do a read at all? (I think it should be doing that, since data has been received on the connection.)
>> 
> It was confirmed during early testing that if the read buffer can hold whatever arrived early (one or 2 or 3 packets depending on the timing and the size of the buffer), the second read would hang forever waiting of a zero return.
>> Dropping the packet in the established state was added a while ago to fix the accept processing bug found earlier, and like the comment says, we can't change state prior to an accept being executed, because in ELKS, the "new" socket descriptor is created and passed to ktcp at accept time. This could possibly be fixed, but not without rewriting/redesigning the way ktcp and the kernel interface to it work, so I decided to punt instead and drop the packet, forcing a retry.
>> 
>> it seems odd that the state change takes so long. Looking at the packet trace, the SYN/SYN+ACK/ACK are back-to-back, so the connection is really in ESTABLISHED state, but it seems like ktcp takes very long to make that change 'official', and that that is the underlying problem.
>> 
>> Remember, ktcp is a separate process and so is the application and are subject to scheduling; the application could be in the middle of creating a filename on disk perhaps, and we know this is a timing issue. The TCP <-> TCP connectivity is asynchronous to the applications above it to a degree, except in this case, ktcp has to force a synchronization by prohibiting a state change until accept is executed.
>> 
>> so the connection is really in ESTABLISHED state
>> 
>> Yes, it actually is set to ESTABLISHED after the SYN is ACKed, and ktcp is storing data on the connection. The problem is that the next state change to CLOSE_WAIT after a FIN is seen needs to be delayed until an accept is executed due to the current design.
>> 
>> BTW, isn't it odd that data is being accepted before the state change?
>> 
>> No, because the state has actually changed to ESTABLISHED. Like I said, with some redesign this could be fixed, but I don't really want to go down that path (yet), as there'd be lots of retesting to ensure correctness. We could set a flag, for instance, but tcp doesn't use any flags currently... It's that kind of thing.
>> 
>> 
> I guess I suspected that based on previous discussions. The asynchronicity is really confusing at times when what's on the wire is sometimes in sync, often not with the debug prints form ktcp...
>> The easy way out is to ignore the problem and make sure the server side retransmits the FIN after a short delay, like the Linux server does.
>> 
>> Well, we already are forcing a retransmit by dropping the packet, which fixed an earlier reported bug. Linux and Rasbpi apparently always retransmit, and that's why it works talking to them.
>> 
>> It's actually interesting that that doesn't happen already, maybe that's a different bug altogether.
>> 
>> Yes - and that's what we need to fix, since this issue is ELKS to ELKS only, and that's the problem. If you agree with everything above,
>> 
> I do!
>> then we need to switch and look over to the retransmit code, and see why the first FIN packet is not retransmitting. I have already looked at the tcp_output.c code, and can't see why it isn't happening. Turning on DEBUG_RETRANS should help show where the issue is on the sending side.
>> 
> OK, I'll take a stab at that next - and clean up all the debugging changes in ftp/ftpd, to get them submitted.
>> as a temporary kluge - to delay the close() in the ftpd server when the file is small.
>> 
>> No, I don't think we should write any application kluges unless we absolutely have to. We went down that road before, and end up in more trouble eventually.
>> 
> We're on the same page.
> 
> 
> Thank you.
> 
> 
> —Mellvik

Mellvik, 2022-02-11:
> Hello @Mellvik,
>
> As I pondered your problem, it occurred to me that I could test ELKS-to-ELKS transfers by using QEMU, so I did. I used "ftp localhost" and it connects and works well with the local ftpd. Then, I purposefully dropped all FIN packets with a modification to tcp.c, to simulate your problem. What I found is that ELKS is not retransmitting the FIN packet, as we thought.
>
> In addition, with another printf in tcp_output.c, I found that ELKS is removing the FIN packet from the retransmit queue too soon. It seems, but I have not fully verified, that we may have an off-by-one error in the business of calculating when to remove a retrans packet. I have modified the code such that a FIN retransmit occurs, and I think this fixes your problem. See the code below in ktcp/tcp_output.c:
>
> /* called every ktcp cycle when tcp_timeruse nonzero - check for expired retrans*/
> void tcp_retrans_expire(void)
> {
>     struct tcp_retrans_list_s *n;
>     int rtt;
>     unsigned int datalen;
>
>     /* avoid running out of memory with excessive retransmits*/
>     if (tcp_retrans_memory > TCP_RETRANS_MAXMEM) {
>         printf("ktcp: retransmit memory over limit (cnt %d, mem %u)\n", tcp_timeruse, tcp_retrans_memory);
>         n = retrans_list;
>         while (n != NULL)
>                 n = rmv_from_retrans(n);
>         return;
>     }
>
>     n = retrans_list;
>     while (n != NULL) {
>         datalen = n->len - TCP_DATAOFF(&n->tcphdr[0]);
> [... 24 lines cut ...]
> I haven't yet confirmed the exact details on the arithmetic, but its not that big of deal, since the retransmit queue will get emptied when the seqno is one greater. The reason this works is that the FIN packet uses up one octet of sequence space, and it is being prematurely removed.
>
> With this change, I am seeing ktcp retransmit the FIN packet, which is then dropped, and this repeats over and over. On your real hardware, there won't be a FIN drop, so everything should just work.
>
> Thank you!

Thank you @ghaerr,

This is fantastic!! Your background CPU seems to be processing really well these days :-)

I'm travelling today and parts of tomorrow, will test as soon as I can. 

Really looking forward to it.

--M

closed 2022-02-11 by ghaerr

Mellvik, 2022-02-13:
Thanks again @ghaerr,
new tests confirm that the problem (the lacking FIN retransmit) is solved. The application (`ftp`) hangs a little (as expected) (ca. 2 seconds according to the packet trace) when the situation arises, but recovers smoothly - the debug output below. I guess the 2s lag becomes our permanent reminder that there is an underlying problem to be fixed at some point.

(Debug output from server side, delivering any collection of test files)
```
Accepted connection from 10.0.2.16:1042.
local: 10.0.2.15, remote: 10.0.2.16, QEMU: 0
tcp retrans: seq 129+0 size 0 rcvwnd 4380 unack 129 rto 64 rtt 14 (RETRY 1 cnt 1
 mem 20)
tcp retrans: seq 129+0 size 0 rcvwnd 4380 unack 129 rto 64 rtt 16 (RETRY 1 cnt 1
 mem 20)
expand wc got 22
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp retrans: seq 129+0 size 0 rcvwnd 4380 unack 129 rto 64 rtt 16 (RETRY 1 cnt 1
 mem 20)
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp retrans: seq 512+0 size 0 rcvwnd 4380 unack 512 rto 64 rtt 16 (RETRY 1 cnt 1
 mem 20)
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp retrans: seq 513+0 size 0 rcvwnd 4380 unack 513 rto 64 rtt 14 (RETRY 1 cnt 1
 mem 20)
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
tcp: port 20 REUSED, freeing previous socket in time_wait
```
The port reused messages don't seem to cause much of a delay.

Test files:
```
ftp> !ls *b
1000b  1001b  1023b  1024b  1025b  1100b  1200b  128b   1459b  1999b  2000b  
2001b  511b   512b   513b   599b   600b   601b   767b   768b   769b   999b   

ftp>
```

Thank you @ghaerr,

--M

Mellvik, 2022-02-13:
> 13. feb. 2022 kl. 19:14 skrev Gregory Haerr ***@***.***>:
> 
> 
> Thanks for confirming this now works, @Mellvik <https://github.com/Mellvik>.
> 
> I guess the 2s lag becomes our permanent reminder that there is an underlying problem to be fixed at some point.
> 
> It's on my list, but there's a lot of stuff ahead of it at this point. And the retrans from non-ELKS seems to be quicker, right?
> 
Yes, the Raspi waits about .23 seconds. But let's keep it this way for now. If it becomes unbearable, let's discuss it …
Actually, when I change the ftp client back to before the debugging, the file size triggering the error is < 513 bytes, and the occurrence quite rare.
> tcp retrans: seq 129+0 size 0 rcvwnd 4380 unack 129 rto 64 rtt 16
> tcp: port 20 REUSED, freeing previous socket in time_wait
> 
> It could be time to move these messages to debug-output only? I still hesitate a bit on the retrans, but perhaps it should be made available on the ^P default only.
> 
I'll say we keep them for now, at least the retransmit message. I'm getting a lot of them when doing ELKS-to-ELKS transfers, much more than seems reasonable, something I'd like to look closer at that real soon. 

Moving the REUSED message to ^P is OK.

Thank you!

-M

PR #1145 (2022-02-11): [net] Fix premature removal of FIN packets from retransmit queue
Fixes #1131.

@Mellvik, the problem wasn't solved by changing the SEQ_LEQ to SEQ_LT, but instead the retransmit expiry code didn't add an extra sequence number for the FIN packet, causing it to be dropped out of the retransmit queue too early, and thus never retransmitting.

Also added code to count more cases of dropped packets for `netstat` reporting, and cleaned up the displayed "unack" debug value to be the unack sequence number rather than the previous amount of unacked data.