ghaerr/elks #1141: Problem starting telnetd

toncho11, 2022-02-06:
I am unable to run the telnetd. It can not bind to a port. 

I use ktcp to start networking and it works if I telnet to my Linux server from ELKS, but the opposite does not work. 

Which port it is 23 or 2323?

Mellvik, 2022-02-06:
Hi toncho11,
Are you on qemu or physical?
If telnetd cannot bind (to port 23), there may be another telnetd running. Use ps and netstat to check it out.

-M

ghaerr, 2022-02-06:
> I use ktcp to start networking

Use `net start` and `net stop` to start and stop networking. Use `ps` to see what's running. Use `netstat` to see what daemons are listening and on what ports. Edit `/etc/net.cfg` to configure networking.

> I am unable to run the telnetd. It says:
telnetd: bind error (may already be running)

That's definitely because telnetd is already running.

ghaerr, 2022-02-06:
Hello @toncho11,

I'm replying here from your cross-post in https://github.com/jbruchon/elks/issues/1131#issuecomment-1030882615 to try to keep things from getting too messy:

> Actually ... I noticed that in one session ktcp was generating a message like this:

> tcp: refusing packet from ...

> even without transferring a file. I do not know if this helps. It was annoying though because it interferes while you type on the console.
> Or maybe it was OK because I started using wget and not the ftp program ...

I notice that the picture you posted (below) happens to show a capital "R" on Refusing packet... 
[image]

The current repo uses a lower-case "r" in the latest version, this was changed a while back; this seemingly means you are running old code, which could very well be the reason that networking isn't working for you. Can you check that your repo is up-to-date? (You can also look at elkscmd/ktcp/tcp.c to and grep for "refusing" to ensure you are running later ktcp code.)

And to answer the question of why the refused packet: this is displayed by ktcp when a TCP data packet arrives for which there is no active TCP control block, that is, connection. This could happen, for example, when a connection is reset. But I want to make sure we're all running the same code before debugging any further on this. Also, running "ktcp" directly (that is, not using "net start/stop" is not recommended any longer.

Thank you!

toncho11, 2022-02-07:
So for telnetd:

- I did not have /etc/net.cfg at all after install with sys (one has to add it manually)
- I started using net.cfg and `net start` instead of ktcp

And now it works. The error message was confusing though.

toncho11, 2022-02-07:
For "R" problem.

In my source it is correct.
I am not using a single image though.
I use hd32-fat.img to extract ELKS commands and to copy them on my real pc hdd.

ghaerr, 2022-02-07:
> And now it works. 

Great. Perhaps in v0.6 we can add an ability to copy files required for networking using `sys`. We can't do this by default since they won't all fit on a 360K floppy.

> The error message was confusing though.

"telnetd: bind error (may already be running)"... telnetd was already running, which was the problem, yes?

Should the error message say: "telnetd: bind error, exiting (may already be running)." or something like that? It used to just say "bind error".

> I use hd32-fat.img to extract ELKS commands and to copy them on to my real pc hdd.
Anyway I made some changes and now I am getting lowercase "refusing".

All the more reason to get v0.5 out, which will produce a standard set of application binaries.

Thank you!

toncho11, 2022-02-07:
So it seems everything is OK in general.
My problems were related to missing binaries, /var and /etc files. This is because you need to complement your installation manually after `sys` install.
One thing is that all 3 services are started by default. And if one of them does not start and somebody tries to access it then you get the refuse packet message every time. Normally they are started and that is why it is OK. Still it can be annoying because it interrupts you while typing on a very slow machine.

closed 2022-02-07 by toncho11

toncho11, 2022-02-07:
I actually updated the wiki pointing out that the sys command does a minimalistic installation that needs to be complemented to work properly.

ghaerr, 2022-02-07:
> I actually updated the wiki pointing out that the sys command does a minimalistic installation that needs to be complemented to work properly.

Thanks. Hopefully in v0.6 we will find a way to determine the size of the sys target disk and automatically add all the files that can be contained on that size target. This could be made available as an option to add more system files, but they won't all fit on smaller disks. Another option would be to enable executable compression, which would allow for 1/3 more files to be held on smaller disks.

Thank you!