ghaerr/elks #1144: Writing hd32 images to SD card from Windows

toncho11, 2022-02-08:
Hi,

What program do you use to write these:

hd32-fat.img
hd32mbr-fat.img 
?

I tried Rufus and Win32DiskImager, but they fail on real hardware.  The Mbr version reads the kernel, but halts on `4!`.
Does anyone has some experience?

tyama501, 2022-02-08:
Does the SD card emulate real ide hard drive?

I have never used these images with real hardware but I think it won't  work with floppy emulator.

ghaerr, 2022-02-08:
> Does the SD card emulate real ide hard drive?

I don't think so, it's just a memory chip. The floppy emulator emulates a hardware floppy, but that can be made to work with an ELKS hard drive image, if that's what you're looking for.

> I have never used these images with real hardware but I think it won't work with floppy emulator.

The ELKS images have the CHS value stamped into the boot sector, and I think the floppy emulators will allow almost any CHS (as they just multiply the values together to get an LBA). However, I haven't actually tried to boot a hard disk image from SD card/floppy emulator.

BTW, "4!" above means that the boot loader thinks it found a Linux kernel, but that the magic number in the a.out header for the kernel was wrong. This usually means the image is in the wrong sector (the SD image is messed up or incorrect).

tyama501, 2022-02-08:
Oh ok, I will try hd32mbr-fat.img with SD card and PC/AT.

Mellvik, 2022-02-08:
If the SD card (or CF card) is plugged into an IDE adapter it looks like an ide drive to the machine.

Older CF cards actually have CHS values programmed in, the CF interface is identical to IDE.

I'm using this combination in my main ELKS machine.

—M

tyama501, 2022-02-08:
I wrote hd32mbr-fat.img of ELKS 0.4.0 to SD Card by Rufus and tried with PC/AT(Core 2 Duo) machine.

I showed as the picture and I don't know what to do.
[image]

This machine can boot floppy ELKS.

ghaerr, 2022-02-08:
> It showed as the picture and I don't know what to do.

Type 'f' for floppy boot and '1' to '4' to boot MBR partition!

tyama501, 2022-02-08:
I typed "4".

It seems the message loops and not booting.

ghaerr, 2022-02-08:
If nothing happens after typing 4, that means that MBR partition is not valid (no entry, likely).

ghaerr, 2022-02-08:
You might also try writing ELKS 0.5.0 version (just posted under Releases), as well as hd32-fat.img, the non-partitioned version.

tyama501, 2022-02-08:
I mean I typed 1 to 4 and the message repeats continuously.

Well I think this adaptor won't work.

tyama501, 2022-02-08:
disk error for 'f'.

It is just SD reader.

Sorry I don't have time. I need to go sleep. It is almost morning.

toncho11, 2022-02-08:
I am using XT-IDE + Micro SD card. 
The hardware is not the problem, because it works with the sys command with the same configuration. 
The problem is rather that no Windows application places the image correctly in the beginning of the disk ...

Mellvik, 2022-02-08:
Maybe you can use cygwin to get more direct access to the devices under Windows.

--M

ghaerr, 2022-02-08:
> The problem is rather that no Windows application places the image correctly in the beginning of the disk ...

A friend of mine recommends https://sourceforge.net/projects/win32diskimager/ or Etcher: https://www.balena.io/etcher/.

toncho11, 2022-02-08:
So I used:

-dd for Windows 0.6 beta
- Rufus
- Balena Etcher
- Win32diskimager

Nothing worked! It says error 4 or it can not load beyond the MBR manager when slecting 12,3,4. Or device ubootable.

ghaerr, 2022-02-08:
> Nothing worked! 

Wow, but you can take an image using dd from Linux and it works on same hardware? I wonder if the .img file itself on Windows is bungled.

toncho11, 2022-02-08:
[usbimage.zip](https://github.com/jbruchon/elks/files/8027659/usbimage.zip)

This is the hd32-fat.img read from the USB with DD for Windows.

toncho11, 2022-02-08:
Ok ... is it possible that these are minix and not fat eve if it says fat?

toncho11, 2022-02-08:
So maybe just the file system type is wrong in the images.

ghaerr, 2022-02-08:
Hello @toncho11,

> This is the hd32-fat.img read from the USB with DD for Windows.

I ran this under QEMU using "... -hda usbimage.img" and it booted ELKS v0.5.0 fine. It's a FAT HD image with no MBR (compatible or identical to hd32-fat.img):
[image]

toncho11, 2022-02-08:
OK even using dd on Linux it does not work - fat mbr version.
It gives an error 4! after loading the kernel.
So there is a problem with the fat images.

toncho11, 2022-02-08:
This is the thing qemu is probing and avoiding the problem while on real hardware the fat images can not load beyond reading the kernel.

ghaerr, 2022-02-08:
> So there is a problem with the fat images.

Possibly - I just tested the FAT hd32-fat.img and hd32mbr-fat.img on QEMU and they both boot. But apparently not on your (or other?) hardware?

@Mellvik, could you try booting an ELKS v.0.5.0 FAT image (floppy, and/or hard disk) on real hardware? We may need to try both the "flat" as well as "mbr" versions. They both work fine on QEMU.

toncho11, 2022-02-08:
This is the thing qemu is probing and avoiding the problem while on real hardware the fat images can not load beyond reading the kernel.
Why is the disk detected two times "BIOS" and "IDE". Maybe qemu is providing one of these compared to a real system.

ghaerr, 2022-02-08:
> Why is the disk detected two times "BIOS" and "IDE". Maybe qemu is providing one of these compared to a real system.

That's something else entirely, and has been there a while. @Mellvik added it to query the IDE interface directly in the case(s) that the BIOS returned the incorrect CHS values. That won't be the reason.

toncho11, 2022-02-08:
And is the partition type correct indeed?

toncho11, 2022-02-08:
Device     Boot Start   End Sectors Size Id Type
/dev/sdb1  *       63 63566   63504  31M  6 FAT16
$ sudo dd if=hd32mbr-fat.img of=/dev/sdb
63567+0 records in
63567+0 records out
32546304 bytes (33 MB, 31 MiB) copied, 6,54643 s, 5,0 MB/s
$

ghaerr, 2022-02-08:
> And is the partition type correct indeed?

I'm getting a bit mixed up now... the first image was a non-partition (non-MBR) image. Now you're talking about hd32mbr-fat.img which is a partitioned image, with MBR...  We should debug one at a time. The screenshot I sent over shows a non-partitioned boot from a non-partitioned image (hd32-fat.img).

> sudo dd if=hd32mbr-fat.img of=/dev/sdb

What is this supposed to show?

toncho11, 2022-02-08:
I am testing both .
Even on Linux  I think there is no way to write these images and boot from them.

ghaerr, 2022-02-08:
> I am testing both .

You might test a MINIX image, to see whether that works, so we can determine whether MINIX works and FAT doesn't or possible neither do on your hardware.

> Even on Linux I think there is no way to write these images and boot from them.

The 'dd' command should work. For a hard disk image of either MINIX or FAT format, the hd32mbr-xxx.img should boot and show an "MBR F1234>" prompt. If this all fails, perhaps go back a step and write a floppy to an SD card (you're using a floppy emulator, correct?)

toncho11, 2022-02-09:
I tried the minix image (no MBR) and it did not load. 
It said0

```
ELKS .......Linux found ................. 4!
Press key
```

My system is Amstrad 1640 with 8086 CPU, 640 kb memory.
When the sys command is used it works. 
So the sys command is either positioning the image correctly or doing an extra step that allows it to boot.

Mellvik, 2022-02-09:
FWIW -

I just created and booted the first 0.5. fd1440 image. It boots normally on physical. (80286 this time).

At the time I have no means to test the hd images on physical (except wiping out my HD, which is inconvenient).

Also confirming (out of context but still) that XMS works fine on this machine (which has 6M RAM):

Direct console, scan kbd 80x25 emulating ANSI (3 virtual consoles)
ttyS0 at 0x3f8, irq 4 is a 16450
xms: A20 was off now on, using int 15, 256 xms buffers
eth: NE2K at 0x300, irq 12, MAC 00:40:c7:21:d7:9f
bioshd: hda BIOS CHS 979,5,17
/dev/hda: 979 cylinders, 5 heads, 17 sectors = 40.6 Mb
/dev/fd0: 80 cylinders, 2 heads, 18 sectors = 1440.0 kb
Partitions: hda:(0,83215)  hda1:(1,6374)  hda2:(6375,1275)  hda3:(7650,41565)  hda4:(49215,34000) 
device_setup: BIOS drive 0x0, root device 0x380
PC/AT class machine, Intel 80286 CPU, 640K base RAM.
…

-M

toncho11, 2022-02-09:
Byt @tyama501 already tested on more modern machine.
Maybe the geometry provided by these cards does not match the geometry for which the ELKS image is configured. Then qemu and sys command both adapt to the correct geometry needed.

toncho11, 2022-02-09:
Hmm and what is the main difference between a floppy image and a hdd image ?

Mellvik, 2022-02-09:
The floppy image is created for a specific floppy type, with well known CHS values. It will always work.

So is the HD image too, but there is no predictability as to the physical CHS values. They differ from one product to the next. If the image copied to a disk with different geometry than assumed when the image was created, it may not work. 

I'd like to say 'will never work', but it's not true. Very old systems use direct CHS addressing, newer use LBA addressing.
In the latter case it may work, in the former it will not - unless there happen to be a CHS value match.

—M

tyama501, 2022-02-09:
How about taking diff by binary editor for the bootable image created by " sync "  command and the unbootable image to see what is the differences are in the first sector.

Mellvik, 2022-02-09:
Yes, you can do that. Now, sync does not create anything, it just flushes the cache. 

So if you'd like to compare images across systems (or on the same system - I'm doing this all the time to verify that I have the same version of a program), you use the sum program. Generate a checksum of the files you'd like to compare. The first number you get is the checksum, the second is the size in blocks which may be different from one system to the next and doesn't matter. If the checksums are different you have files that differ at least one bit.

Then - if there is a difference, you use the cmp program to find out where the first difference is (byte #). Then you can use hd or xxd to look at that particular place in the files.

—M

toncho11, 2022-02-09:
That is why @ghaerr did the sys command (proposed by me) indeed ... It allows HDD installations. 

OK so:

1) We need to say in the documentation and the release page that the HD images are only for virtual machines.
 
2) Is it possible to configure the HDD geometry of an image before outputting it. I can check the geometry of my sdcard and then produce the correct image, right?

ghaerr, 2022-02-09:
> Hmm and what is the main difference between a floppy image and a hdd image ?

For this conversation, and ignoring MBR (i.e. using flat not mbr images), the difference is the CHS values, which are stored in the boot sector of the image. Also, the CHS values are stored in different locations in the boot sector, depending on whether the boot image is in MINIX or FAT format.

Since @toncho11 is saying that he can create an bootable image using `sys`, but not when using a pre-built .img file, it is highly likely that the problem is that the CHS settings are incorrect.

The `sys` command, which runs on ELKS, is actually a shell script. Take a look at it. It first runs the `makeboot` program, which copies the boot sector from the ELKS boot drive to the new drive, then queries the destination CHS values from the BIOS driver and writes them in the appropriate (different) place in the destination boot sector, depending on whether the destination filesystem is FAT or MINIX. `sys` then mounts the destination, and copies files to it. The heavy lifting is done by makeboot.
```
Usage: makeboot [-M][-F] /dev/{fd0,fd1,hda1,hda2,etc}
```
@toncho11, I suggest boot up on your Amstradt, with a fresh hd32-fat.img file installed on the HDD. Then, run `makeboot` (for hd32mbr-fat.img) or `makeboot -F` (for hd32-fat.img) and then see whether the HDD boots. It should.

I suggest looking at the makeboot source, which is at elkscmd/sys_utils/makeboot.c. This will help a lot in understanding exactly what goes in to make a bootable disk. It is detailed, but straightforward. Start with main() and work upwards/backwards.

When `makeboot` runs, it displays the CHS values for both the boot volume, as well as the destination. The destination  values can be used in `make menuconfig` to build a custom image that will boot on your machine. (When booting from floppy with a hard drive, the hard drive CHS values are also displayed in the boot messages).

Here's an example running `makeboot` from a floppy booted from /dev/fd0 on a non-partitioned hard drive on /dev/hda (-F required since non-partitioned):
[image]
Note the destination CHS values displayed by makeboot (in this case 63,16,63 - the ELKS default for hard drives), as well as the ELKS boot (top line).

ghaerr, 2022-02-09:
Hello @toncho11,

> We need to say in the documentation and the release page that the HD images are only for virtual machines.

No, we (or you) should probably write several paragraphs about how to get a hard disk up and running on ELKS, after trying a boot floppy first, then using `sys`, etc as described above.

I think the problem you are having has to do with the age of your BIOS. As @Mellvik explained, newer BIOSes are built around LBA (logical block addressing) versus the original CHS (cylinder head sector). A newer BIOS will end up just multiplying together the C, H and S values to come up with an LBA (0-based linear address for entire disk) which ends up working. The software emulators use this method as well, since the image is actually a file that allows for linear addressing.

Thus, for very old systems, a proper CHS is required, as there is no translation in the BIOS or the hard disk drive itself.

> Is it possible to configure the HDD geometry of an image before outputting it. 

Yes, `make menuconfig` allows to set the individual C, H, S parameters. However, the "standard" distribution images, which work on most newer systems, are set to 63, 16, 63 by default.

> I can check the geometry of my sdcard and then produce the correct image, right?

Hopefully `makeboot` as described above will work. A potential issue is that it just queries the BIOS for the CHS values that are displayed, but, given that the HDD in fact works with that BIOS, this should also work.

Thank you!

ghaerr, 2022-02-09:
> How about taking diff by binary editor for the bootable image created by " sys " command and the unbootable image to see what is the differences are in the first sector.

For instructional purposes, what would help would be a text file with a visual map of where the various entries are in a boot sector. I'll add that to my list. 

A quick summary is that ELKS treats MINIX and FAT volumes differently. For FAT volumes, the CHS parameters are stored in the [BIOS Parameter Block (BPB)](https://en.wikipedia.org/wiki/BIOS_parameter_block).

 For MINIX volumes, we invented the ELKS Parameter Block, which stores the CHS values. This non-standard invention is stored in the upper portion of the boot block, and is recognized by a special ELKS signature. It is always produced by our tools and is read by our bioshd.c driver. However, if not found, then a disk probe (for floppies only) is performed to guess CHS. For hard drives, the BIOS is called to return the CHS parameters using BIOS INT 13h AH=08, but on very early BIOSes, this function is not present.

The elkscmd/sys_util/makeboot.c program, used to create bootable volumes from scratch, lays out all this detail in one place for those that are interested.

toncho11, 2022-02-09:
So after a lot of testing here are the results:

1) Writing MBR image on SD card, booting from floppy and [makeboot /dev/hda1](url)
OK
2) Writing non-MBR image on SD card, booting from floppy and [makeboot -F /dev/hda1](url)
No
3) Compiling with changed geometry CHS (in my case 1024/255/63)
No
4) 3) Compiling with changed geometry CHS (in my case 1024/255/63) + makeboot
No

ghaerr, 2022-02-09:
> So after a lot of testing here are the results:

Interesting. So it seems that the only image that will work is an MBR image, using makeboot. The default `make menuconfig` options don't work, but I think that's because they are not creating MBR images, only flat images, correct?

Are you using "cd image; make images" to make the hd32mbr-fat and hd32-fat images, or only using the default image created from make menuconfig?

toncho11, 2022-02-09:
default image created from make menuconfig

toncho11, 2022-02-09:
I can not "cd image; make images" 
makeboot -F /dev/hda does not work, it says you need to specify hda1 (error message)

ghaerr, 2022-02-09:
> makeboot -F /dev/hda does not work, it says you need to specify hda1 (error message)

On my screenshot above, I show it working. Does your system show a partitioned hda1 at boot (mine does not). This error message is displayed when trying to write an HDD with a MBR boot sector with a flat image and/or boot sector. They must match.

> I can not "cd image; make images"

The image directory is directly off the elks source root directory:
```
zorg:elks-gh greg$ pwd
/Users/greg/net/elks-gh
zorg:elks-gh greg$ cd image
zorg:image greg$ make images
echo CONFIG_APPS_360K=y		> Config
echo CONFIG_IMG_FD360=y		>> Config
echo CONFIG_IMG_MINIX=y		>> Config
echo CONFIG_IMG_DEV=y		>> Config
echo CONFIG_IMG_BOOT=y		>> Config
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f Make.image "CONFIG=/Users/greg/net/elks-gh/image/Config" NAME=fd360-minix
```

toncho11, 2022-02-09:
Actually every time I tried both.

It says:
`Makefile:4: *** TOPDIR is not defined, ELKS not configured yet.  Stop.`
on `make images`

ghaerr, 2022-02-09:
> It says:
> Makefile:4: *** TOPDIR is not defined, ELKS not configured yet. Stop.

It will say that if you are not setup for compilation, I thought you were compiling ELKS when using the normal "make" to create everything after "make menuconfig". You should be executing ".   env.sh" before your regular make as well as this one. Does "make" work at top level, it would be giving this same error message if not configured?

toncho11, 2022-02-09:
I made this page: https://github.com/jbruchon/elks/wiki/Installing-HD-image-on-physical-media
It will be updated once we see that other methods also work.

toncho11, 2022-02-09:
This number changes every time you plug the USB card reader.  It increments.

ghaerr, 2022-02-09:
> This number changes every time you plug the USB card reader. It increments.

I get that, just saying that your example doesn't use the same number as what was shown within it, so it's confusing.

toncho11, 2022-02-09:
Corrected. I also added the alternative method and a link to it from the main documentation.

closed 2022-02-09 by toncho11

Mellvik, 2022-02-09:
That’s a distinct possibility.
Make sure you use the ide CHS detection, not the bios detection. It was created for this specific purpose. 

M

toncho11, 2022-02-10:
@Mellvik Can you please elaborate on the CHS IDE detection? This is in the case when getting the CHS and setting them in the menu config to compile to a suitable image file?

Mellvik, 2022-02-10:
Hi @toncho11,
Take a look at #742 for details.
The code - if configured in - queries the IDE drive for CHS parameters, which is generally more reliable than using BIOS values. Some (actually many) BIOSes have a predefined set of drives to choose from when configuring a drive, so the only way to tell the kernel the real values (except compiling them in) is to ask the drive.

--M

toncho11, 2022-02-10:
1) This is configure in: Kernel & hardware -> Drivers -> Block Device Drivers ->  IDE hard drive CHS probe in the menu config?

2) And the output should be: `bioshd: hda IDE CHS ...` ?

I think I saw it already and it was the same as the "BIOS" reported one: 1024/255/63 ...

Mellvik, 2022-02-10:
Yes, then you're all set. 

The boot message is your verification that ELKS is using the 'real' values from the (IDE) drive.

Here's what I'm getting on one of the COMPAQs - which has a CF card as the HD:

xms: A20 was off now on, using int 15, 256 xms buffers
eth: NE2K at 0x300, irq 12, MAC 00:80:29:ef:d9:51
bioshd: hda BIOS CHS 610,16,63
bioshd: hda  IDE CHS 993,16,63
/dev/hda: 993 cylinders, 16 heads, 63 sectors = 488.7 Mb
/dev/fd0: 80 cylinders, 2 heads, 18 sectors = 1440.0 kb
/dev/fd1: 80 cylinders, 2 heads, 15 sectors = 1200.0 kb

—M

toncho11, 2022-02-10:
So if they are different the kernel decides to use the IDE ones?

And I think "IDE hard drive CHS probe" is enabled by default, right ?

Mellvik, 2022-02-10:
That's a 'yes' on both accounts, @toncho11.

To be pedantic, ELKS just reports and ignores the BIOS values if the IDE values are present.

—M

Mellvik, 2022-02-10:
> I have updated: https://github.com/jbruchon/elks/wiki/Installing-HD-image-on-physical-media <https://github.com/jbruchon/elks/wiki/Installing-HD-image-on-physical-media>

Thank you @toncho11,

this is good.

It might be worth noting that PCs before the AT had MFM drives, IDE wasn't invented yet.
On systems that old, it is likely that the IDE probing will have to be configured out in order to get the kernel to boot.

—M