ghaerr/elks #1154: Quick question: Is ELKS real-time OS?

toncho11, 2022-02-14:
Hi,

Can ELKS be considered real-time OS? Usually that depends on the kernel scheduler.

Still the kernel code is small (compared to regular Linux) and there are not many processes running when the system boots up. So in general it should be pretty responsive, even "real-time" ?

Mellvik, 2022-02-14:
Hi @toncho11,
I'm sure @ghaerr will elaborate on this, but ELKS is definitely not a real time OS. There is no guarantee that an interrupt or a process will be serviced within a certain time frame. And - since ELKS is using BIOS-services for a number of tasks (such as floppy and disk I/O), real time is off the table.

There have been many discussions about the benefits and drawbacks of this BIOS dependability over the years, and the conclusion has always been (this far) that the portability provided by the BIOS, which is 'hiding' peculiarities of the actual system from the OS, is extremely attractive. 

--M

toncho11, 2022-02-14:
And BIOS adds a considerable unpredictable delay?

Mellvik, 2022-02-14:
'Unpredictable' yes, 'considerable' – you never know because it's unpredictable.

The thing is, you don't control interrupts when handing over control to the BIOS. And you never know how the BIOS does things. There are thousands of them, all different on the inside, (almost) alike on the outside (API).

-M

toncho11, 2022-02-14:
Interesting. I always though the BIOS calls are rather short segments of code that should not take too much time. But then even back then developers preferred to access the hardware directly (graphics for example).

toncho11, 2022-02-15:
But if it is on a fast CPU you can set it to 5ms to switch between tasks instead of currently 10?

OK. I see:
- BIOS interrupts that can take long time
- slow switching between tasks
- no priorities in the kernel

I was thinking for some real world use of ELKS when combined with some hardware like for example drone control over wifi using this board: [FabGL ESP32 board](https://www.youtube.com/watch?v=O9AP4vZHfjI&ab_channel=FabrizioDiVittorio) that has an IBM PC emulator.

ssieb, 2022-02-15:
If you have an ESP32, it already comes with an RTOS.  Why slow it down with an emulator?

closed 2022-02-15 by toncho11