ghaerr/elks #1155: [system] Practical experiences

Mellvik, 2022-02-14:
With recent vast improvements in stability, performance and usability, some of us use ELKS differently - and more. In my case that means 2 physical systems with ethernet and (primarily) serial access. Thus some practical experiences and suggestions.

When starting off with a fresh copy from the repository (no local adaptions), the procedure after first boot is as follows:
- Edit `/bootopts` to set console, run level, console speed, networking, buffers and now sync value. Ethernet defaults are fine.
- Edit `/etc/inittab` to get the serial speed right - for ttyS0 and ttyS1.
- Edit `/etc/net.cfg` to set the right IP address, possibly to eliminate `httpd` from the default startup and (possibly) set or remove the debug option from/to `ftpd`.
- Edit `/etc/mount.cfg` to
 - ... add the a-option to `fsck` (don't want the boot process to stop/wait regardless ...)
 - ... Enable checking of the root floppy (and other minix filesystems, I don't have that yet though)
 - ... Enable mounting of the DOS partition of the HD

All good, these are all settings to adapt to the local requirements.  And yes, I could have my own set of config files ready when building the images. However, recent development has changed those files about as frequently as other code ...

IOW, there are a few things that would have made life a lot easier:
- A LOCALIP (or HOSTNAME) variable in /bootopts would get me going w/o editing more than one file (other adjustments are less critical).
- Having `init` somehow pick up the console speed setting in /bootopts would be very helpfull.
- Having a FSCK variable in /bootopts that 1) tell mount.cfg to fsck root and 2) contain the fsck options for that particular run.

My 2 cents ...

--M

ghaerr, 2022-02-14:
Thanks @Mellvik, I'll think about this a bit more. 

It sounds like having localip= and a baud= serial port speed in /bootopts would solve most of your problems other than mount checking. Can a single baud rate suffice for all serial ports speed defaults? This would allow the /bin/getty speed parameter to be made optional in /etc/inittab and getty would just use the default speed.

A possible problem with using localip= (rather than LOCALIP=) is that `net start` would only work at boot, not later, unless we kept the environment variable around for the entire ELKS uptime (i.e. use LOCALIP=). I've been trying hard not to require lots of special long-term environment variables, which is why the previous HOSTNAME= was removed.

> add the a-option to fsck (don't want the boot process to stop/wait regardless ...)

The fsck source says that -a isn't implemented, so we actually need to test that more before getting ahead of ourselves. Have you played with it much? 

The next big step is allowing FAT volumes to boot and use /bootopts, as otherwise MSDOS/FAT boot users can't use (m)any of the above features, even now. I have a plan for that, which involves loading /bootopts in the "setup" code, after the boot code but before the kernel runs. The FAT boot sector is almost completely out of space.

Mellvik, 2022-02-14:
Thank you @ghaerr.

> 14. feb. 2022 kl. 18:10 skrev Gregory Haerr ***@***.***>:
> 
> 
> Thanks @Mellvik <https://github.com/Mellvik>, I'll think about this a bit more.
> 
> It sounds like having localip= and a baud= serial port speed in /bootopts would solve most of your problems other than mount checking.
> 
It would.
> Can a single baud rate suffice for all serial ports speed defaults? This would allow the /bin/getty speed parameter to be made optional in /etc/inittab and getty would just use the default speed.
> 
As usual (?) I'm a little weary about changing something and standardized as inittab, but I do not object to this. I was thinking more along the lines of init detecting that /dev/console is /dev/tty-something and use the speed from /bootopts instead of the inittab-speed. I think it's good to keep the ability to have inittab set different speeds for non-console ports. The problem we're solving is that the speed for the serial console is set in two different places.
> A possible problem with using localip= (rather than LOCALIP=) is that net start would only work at boot, not later, unless we kept the environment variable around for the entire ELKS uptime (i.e. [...]
I get that - and agree with the idea. Still, maybe HOSTNAME is such an important parameter that it's worth keeping in a global env variable?
> add the a-option to fsck (don't want the boot process to stop/wait regardless ...)
> 
> The fsck source says that -a isn't implemented, so we actually need to test that more before getting ahead of ourselves. Have you played with it much?
> 
No, not much. I started to use it a few weeks back when I enabled fsck at boot on /dev/fd0 and got annoyed by questions from it. At first I added a -y option, old habit ('answer yes to all questions'), then discovered it is named -a, and since then I haven't had the problem. Apparently I haven't had dirty file systems either :-)
> The next big step is allowing FAT volumes to boot and use /bootopts, as otherwise MSDOS/FAT boot users can't use (m)any of the above features, even now. I have a plan for that, which involves [...]
I'm getting the problem – and I'm beginning to think that a minimal (and potentially invisible) minix FS to bootstrap FAT root would be nice. Like most Linux systems do these days (although not invisible). Many FAT disks skip the entire first cylinder, leaving ample space for a significant size boot 'partition' (which need not be in the partition table, but identified by a magic number and partition table values indicating that the space is actually available …)
Maybe not very compatible, but then again it would potentially solve a bunch of problems. Or just create new ones????

—M

ghaerr, 2022-02-14:
> As usual (?) I'm a little weary about changing something and standardized as inittab, but I do not object to this. I was thinking more along the lines of init detecting that /dev/console is /dev/tty-something and use the speed from /bootopts instead of the inittab-speed.

We wouldn't do any of that. /bin/getty takes an [optional] parameter for baud rate. We'd just remove it for ttyS0 and ttyS1, that's all. Any other hardcoded value for baud rate could be used at any time, if desired, in /etc/inittab.

> The problem we're solving is that the speed for the serial console is set in two different places.

Yes - all we need is to set a reasonable default (in the driver itself?) or /bootopts, and remove the optional /bin/getty baud rate parameter and all will work and remain compatible.

> Still, maybe HOSTNAME is such an important parameter that it's worth keeping in a global env variable?

Agreed, especially to make this work. I am thinking (following your lead) that using LOCALIP would be better, since HOSTNAME should probably be reserved for actually setting a hostname, not an IP address (we'll deal with that later).

> Apparently I haven't had dirty file systems either :-)

Why don't you add testing fsck -a etc to your list, and we'll table modification to mount.cfg until after. I think I can get the LOCALIP= and baud/BAUD= figured out before then.

> Maybe not very compatible, but then again it would potentially solve a bunch of problems. Or just create new ones????

No need for all that. The setup.S code can handle reading the sector, written in ASM, instead of executing in the boot code, like MINIX does now, FAT boot will execute this code in setup.S and /bootopts will be loaded, without worrying about FAT extra cylinders, more boot blocks, etc etc.

Mellvik, 2022-02-15:
Thank you @ghaerr,
> 
> Apparently I haven't had dirty file systems either :-)
> 
> Why don't you add testing fsck -a etc to your list, and we'll table modification to mount.cfg until after. 
> 
> 
I'll do that. 

—M

Mellvik, 2022-02-26:
While keeping this issue open pending more `fsck` testing,
here are some other minor issues that keep popping up.

- `chmem`: Why does it refuse to open files not in `/bin`?
```
# ls -l sash
-rwxr-xr-x  1 root     root        36400  Feb 26 13:58 sash
# pwd
/root
# chmem ./sash sash /root/sash /bin/sash
 TEXT  FTEXT   DATA    BSS   HEAP  STACK  TOTDATA  TOTAL
Can't open ./sash
Can't open sash
Can't open /root/sash
31056      0   5152   6560   6144   3072    20928  51984 /bin/sash
# cd /bin
# chmem sash
 TEXT  FTEXT   DATA    BSS   HEAP  STACK  TOTDATA  TOTAL
31056      0   5152   6560   6144   3072    20928  51984 sash
# 
```
- It seems to me `execvp` (in this case called from  `sash`) should be returning a more informative error code:
```
# chmem /root/sash
sys_brk(77) fail: brk 45dd over by 1 bytes
chmem: No such file or directory 
# 
```
- Also, it seems a small mystery why this particular call to `chmem` cause a sbrk error, while other `chmem` commands don't.
- `ls` would benefit from more heap, I suggest 16k. I'm running into trouble with large FAT directories.
- `init`- actually boot calling init, doesn't notice if the run level is specified several times in `/bootopts`, adds them all to the initial` command line. No big deal, but which one takes precedence (If they're different, which they're not in this case)?  
```
# ps
  PID   GRP  TTY USER STAT CSEG DSEG  HEAP   FREE   SIZE COMMAND
    1     0      root    S 32f1 3494  3072   2018  12608 /bin/init 3 3 
   21    21    1 root    S 66a5 3605     0   1976  13072 /bin/getty /dev/tty1 
   22    22   S0 root    S 3bc4 4702  1196   8832  66656 -/bin/sh 
   50     0      root    Z    0    0 
```

-M

Mellvik, 2022-03-14:
@ghaerr, I keep using this thread for issues that pop up while using ELKS in practice.

And - FWIW - the addition of manpages is becoming very helpful: Not having to go back to the sources to refresh the command options all the time (around my neck of the woods we call it 'teflon memory', nothing sticks ...). Since the man pages don't fit on a floppy, which is still the easiest way to ensure a consistent system, I've set up man pages on the HD and using MANPATH to get to them. And a procedure with `urlput` (`ftpput`) in the works to keep them up to date.

Back to the issue: Is there any chance you can move a man page for `meminfo` up your list? Options (are there any?) and what the labels mean?  Like - there are 2 SIZE columns, what are they? And like - there is a TYPE SEG and there is a SEG column? I think understanding the memeinfo output would be a significant contribution to understanding how ELKS works.

--M

ghaerr, 2022-03-14:
> Is there any chance you can move a man page for meminfo up your list? 

Sure.

>  the addition of manpages is becoming very helpful: Not having to go back to the sources to refresh the command options all the time

Agreed. Remember that not all pages are entirely accurate (yet) - when the pages were first imported from MINIX 3, a quick review of most of the pages was done, to check formatting, and non-existent commands and command options deleted. Not all existing command options were checked for implementation accuracy, and I'm sure there are a quite a few undocumented command options. FYI!

Mellvik, 2022-04-07:
More practical issues:
- `mkdir` in `mount.cfg` should use `-p` to get rid of the error messages
- `fsck -a` in `mount.cfg` seems fine so far, but we do need a `-y` option to make `fsck` non-interactive (assume `yes` to all questions).
- `reboot` badly needs a -f option to force reboot even if there are hung processes on a mounted FS.

ghaerr, 2022-04-07:
Hello @Mellvik,

> mkdir in mount.cfg should use -p to get rid of the error messages

Good idea, will do.

> fsck -a in mount.cfg seems fine so far, but we do need a -y option to make fsck non-interactive 

Looking at the source, -a should automatically accept all asked repair questions. Does this not work, or are you suggesting that -a should not display the question and just automatically repair?

> reboot badly needs a -f option to force reboot even if there are hung processes on a mounted FS.

Hmmm... I presume the reboot program may be hanging in the 'sync' call prior to the reboot system call. That would require -f to not sync, which could be dangerous. Can reboot be ^C interrupted during one of these hangs? If so, then perhaps an alarm() around the sync for a couple seconds would solve this.

Thank you!

Mellvik, 2022-04-07:
HI @ghaerr,

> 
> fsck -a in mount.cfg seems fine so far, but we do need a -y option to make fsck non-interactive
> 
> Looking at the source, -a should automatically accept all asked repair questions. Does this not work, or are you suggesting that -a should not display the question and just automatically repair?
> 
Sorry, I missed the -a option, need to test that.
> reboot badly needs a -f option to force reboot even if there are hung processes on a mounted FS.
> 
> Hmmm... I presume the reboot program may be hanging in the 'sync' call prior to the reboot system call.
> 
Hmm, I don't think so. I can do a sync just before the reboot. 
Reboot complains about cannot umount, which is correct, umount will fail if run manually too because there is a hung process with open files on  some file system.
What is needed is an option to report, then ignore the fact that there is a stuck mount. Like Linux and BSD reboot I presume.

Thank you.

M;

Mellvik, 2022-04-07:
> Ok I get it now - reboot is attempting an "umount /" after syncing, and complains about the unmount not being possible. The unmount isn't hanging, nor are there "hung" processes - the root filesystem just has a file descriptor open for writing on it and thus can't be unmounted (which actually is a remount read-only).
> So, we can add a '-f' option that ignores the error return on the unmount and proceeds to reboot anyways. I'll add that along with mkdir -p in the next PR.

Short update - 

The '-a' option to fsck works fine, mea culpa.

mkdir -p works fine from the command line but hangs the mount.cfg script. I haven't looked at the code, presumably some stdio dependency …

—M;

ghaerr, 2022-04-07:
>  I think -a should be the default on mount so it can be omitted on the command line. -t is available if we need to make a mount fs-specific. Automatic is the default on Linux IIRC.

Agreed, I'll make the change for mount -a to be the default.

> The '-a' option to fsck works fine, mea culpa.

Ok, no action required for fsck.

> mkdir -p works fine from the command line but hangs the mount.cfg script. I haven't looked at the code, presumably some stdio dependency …

Added to my list for fixing with the others brought up today.

Mellvik, 2022-04-29:
Need `hostname` 

Running sometimes 3 ELKS machines concurrently and having 6 or more serial + telnet windows open, the need for a `hostname` command or at least a `HOSTNAME` variable becomes acute (an easy way to get the host name into the shell prompt).
What do you think @ghaerr - a file in /etc and a command or just keep it in /bootopts?

-M

ghaerr, 2022-04-29:
Hello @Mellvik,

We can definitely do something, and I can see your dilemma. Amazing how you're using ELKS serial and TCP connectivity, and the systems are stable. I'm definitely interested in seeing some `uptime` output!

The implementation of hostname should reflect its needs; here are some of my questions:
- How is the IP address current being set, LOCALIP= in /bootopts? What about /etc/host, is the IP address (and hostname) being changed there also, or not?
- Will the hostname ever need to be changed dynamically without rebooting?
- Do we need a separate `hostname` command (to set hostname), or just a way to see current network settings, like `net status`? If HOSTNAME env variable is used, it won't be easy to reset dynamically.

Thank you!

ghaerr, 2022-04-29:
> So, setting hostname in /bootopts for now would be good, and there is no need (now) to be able to change while running.

Great. As you know, any capitalized environment variable set in /bootopts will be automatically passed to /bin/sh and other applications, so setting HOSTNAME=elks3 for instance will work now.

> Keeping the local/hostspecific changes in one place makes life easy when there are many machines and one master image.

Agreed. We're still keeping (sample) hostnames and IP numbers in /etc/hosts, although that file is only used by `in_gethostbyname()`. It may possibly incorrectly duplicate an /etc/hosts host or IP value, and `in_gethostbyname()` won't currently translate an environment variable HOSTNAME= to localhost to get the IP address. I'm wondering about the real usefulness of /etc/hosts, unless it also is updated with each machine's values.

> I think it's perfectly reasonable to do this in stages, adapt to whatever needs pop up.

Lets start with using HOSTNAME= in /bootopts and find out what else needs to be changed.

Thank you!

Mellvik, 2022-04-29:
Thanks @ghaerr,

could you include that in a PR (I'll start using it locally immediately) - the name(s) I would use would be 'elks15', 'elks16' to make it easy to remember the IP addresses.

Also, it may make sense to include the ash prompt update in the PR ???

—M

ghaerr, 2022-04-29:
> could you include that in a PR (I'll start using it locally immediately) - the name(s) I would use would be 'elks15', 'elks16' to make it easy to remember the IP addresses.

I'm not really following you - just add your local HOSTNAME= line to /bootopts for each machine, no PR needed. The other machine names are not included in /bootopts. Are you suggesting that your local machine names should be included in /etc/hosts? There are three machines, you've only indicated two machine names... what is needed is a better setup mechanism, rather than coding /etc/hosts specifically in the master branch. But if that's what you want, please post your preferred /etc/hosts and I'll add that in my next PR for you.

> Also, it may make sense to include the ash prompt update in the PR ???

No, I don't think ash should automatically do that, but instead be handled by some uncommented mechanism in /etc/profile or /root/.profile. 

It seems what is needed here is a method of patching a filesystem to one's specific desires directly after ELKS installation. This could possibly be done by writing a shell script that resides in elkscmd/rootfs_template/root or possibly a subdirectory, which you quickly edit, then execute once on the target system after boot (or on the make machine that creates the system image). The script would copy various files to their locations on each system or image. Note that files that are placed in elkscmd/rootfs_template/root will not be modified by git when pulling from master or upstream master, so this could work quite well. To start this, we need a complete list of files to be modified, and the values within them changed. The `cp` or `sed` commands could be used to effect the changes.

Mellvik, 2022-05-05:
@ghaerr,

testing of `tar` has been generally successful, this is great. We now have a fully functional `tar` on ELKS, which was sorely needed for file/image transfers.
There is one snag that you may want to have a look at:
```
elks15# tar cf - bin | (cd /mnt/tmp; tar xvfp - )
Tar: tape blocksize error
elks15# 
```
Also, I suggest removing the word 'tape' from all messages, as seem to have been done for most other versions of `tar`.

Completely unrelated, `cat` does not understand ´-´ as a command line argument indicating stdin. Seemingly superfluous, but in a long chain of commands, using the single dash makes shell code more readable.

Thank you.

M

ghaerr, 2022-05-05:
Hello @Mellvik,

Thanks for your testing! Did you manage to test any sym or hard links, and is the date/time and file permissions processing on extraction using the -p option OK?

> tar cf - bin | (cd /mnt/tmp; tar xvfp - )
Tar: tape blocksize error

Looking at the source, it seems that tar will complain if `read` doesn't return a multiple of 512 bytes, when requesting blocksize bytes. Blocksize starts at 512 and can get as big as 10k. The problem is, for kernel memory reasons, a standard pipe buffer is now only 80 bytes, so each read is likely returning 80 rather than 512 bytes. This will have to be specially handled in tar by requesting more reads in a loop to handle this ELKS-specific situation.

I've added this and the other two issues to my list.

Thank you!

Mellvik, 2022-05-05:
> 5. mai 2022 kl. 16:57 skrev Gregory Haerr ***@***.***>:
> 
> 
> Hello @Mellvik <https://github.com/Mellvik>,
> 
> Thanks for your testing! Did you manage to test any sym or hard links, and is the date/time and file permissions processing on extraction using the -p option OK?
> 
> 
Hi @ghaerr,
yes - I ran most of the test I could think of, create on minix, extract on FAT and minix, hard & symbolic links, dates preserved - all look good. I have not tested 'follow symbolic links' if it's there ...

Thank you.

—M

ghaerr, 2022-05-05:
> I ran most of the test I could think of, create on minix, extract on FAT and minix, hard & symbolic links, dates preserved - all look good.

Great, thank you!

>  I have not tested 'follow symbolic links' if it's there ...

That's the -h flag. I'm sure it also works, as it just uses lstat rather than stat when opening files.

We'll say it all works now, with the #1283 fix.

Mellvik, 2022-05-05:
Agreed, thank you!

--M

Mellvik, 2022-06-03:
Just recently I decided it was time to take the big step ( :-) ) and move from floppy based rootfs to disk.

That should be easy, just copy the files from the floppy to the hd partition, change a statement in /bootopts (still booting of of floppy) and off we go. Except there is no way to copy the `/dev` directory. And `tar` doesn't support devices (yet). So the devices need to be created manually.

This script is modelled after the BSD MAKEDEV script, which does a lot more, but which - when run w/o arguments - builds a default `/dev` directory.

Adding this to ELKS will make the system more self-sufficient, and I suggest it be placed in the /dev directory (out of the way, not easily run accidentally) just like BSD. 

```
# cat /MAKEDEV 
mknod hda b 3 0
mknod hda0  b 3 0
mknod hda1  b 3 1
mknod hda2  b 3 2
mknod hda3  b 3 3
mknod hda4  b 3 4
mknod hda5  b 3 5
mknod hda6  b 3 6
mknod hda7  b 3 7
mknod hda8  b 3 8
mknod hdb   b 3 32
mknod hdb1  b 3 33
mknod hdb2  b 3 34
mknod hdb3  b 3 35
mknod hdb4  b 3 36
mknod hdc   b 3 64
mknod hdc1  b 3 65
mknod hdd   b 3 96
mknod fd0   b 3 128
mknod fd1   b 3 160
mknod ptyp0  c 2 8
mknod ptyp1  c 2 9
mknod ptyp2  c 2 10
mknod ptyp3  c 2 11
mknod tty1   c 4 0
mknod tty2  c 4 1
mknod tty3  c 4 2
mknod ttyp0  c 4 8
mknod ttyp1  c 4 9
mknod ttyp2  c 4 10
mknod ttyp3  c 4 11
mknod tty    c 4 255; chmod 0666 dev/tty
mknod console c 4 254; chmod 0600 dev/console
mknod ttyS0 c 4 64
mknod ttyS1 c 4 65
mknod ttyS2 c 4 66
mknod ttyS3 c 4 67
mknod lp  c 6 0
mknod tcpdev c 8 0
mknod eth c 9 0
mknod kmem c 1 2
mknod mem c 1 1
mknod null c 1 3
mknod ssd b 2 0
mknod rd0 b 1 0
mknod rd1 b 1 1
mknod zero c 1 5
```

ghaerr, 2022-06-03:
> I decided it was time to take the big step ( :-) ) and move from floppy based rootfs to disk.
> That should be easy, just copy the files from the floppy to the hd partition,
> Except there is no way to copy the /dev directory.

Creating another bootable rootfs is the job of the `sys` script: you should be able to run `sys -M /dev/hda` to create an MBR-bootable system (-F for flat HD without MBR). This script also creates the /dev entries, although not as many as you've included. I'll add your more complete list into /bin/sys.

> And tar doesn't support devices (yet). So the devices need to be created manually.

For some reason I thought it did. I'll add that to the list also...

ghaerr, 2022-06-03:
The `sys` script was originally written to copy a minimal (like 360k) system onto a floppy, although the -M and -F options allow creating a minimal HD system. Since then, additional important files in /etc and possibly other directories have been added, which aren't copied. 

There is no option to duplicate the boot media entirely, and some definition might be in order for exactly what that should do. I'm going to look into updating this. There is also the problem of using the existing /mnt directory for copying, which may already be in use.

Mellvik, 2022-06-03:
Thanks @ghaerr,

I didn't know sys worked with MBR and partitions, great!

Can it make an MBR partition bootable too?

Tar: Seems we had about the same reaction. I went back to the 4.3 sources and was surprised it was missing there too. Anyway, a convenience whenever there is time to add it.

Different topic altogether: 
While working on the 3c509 driver I ran into a problem with request_irq(): It returned EBUSY all the time. Looking at irq.c it seems the irq_action table is not initialized, so testing for zero is unreliable. Adding an initializer fixed my problem, but it still seems weird that this hasn't popped up before. IOW - I keep thinking that an unknown memory leak could be the problem. OTOH why would the initializer fix it?

Any thoughts?

-M

Mellvik, 2022-06-03:
> 
> Can it make an MBR partition bootable too?
> 
> Yes - that was the primary function, although sys hands that off to the makeboot executable to automatically make that happen. That's what the -M and -F flags are for. I'm trying to make this automatic, but it seems our shell doesn't support variable substitution of the ${a:5:8} variety, unfortunately.
> 
Ok, thanks, I'll try it. Apropos ash, I was trying to generalize the setup script by cheching whether an argument is numeric. You know of a way to do that?
> Looking at irq.c it seems the irq_action table is not initialized, so testing for zero is unreliable.
> 
> No - any non-initialized (BSS) area of the kernel or C programs is initialized to zero.
> 
Great, then it's a memory leak for sure. Thanks!

-M

ghaerr, 2022-06-03:
> Apropos ash, I was trying to generalize the setup script by cheching whether an argument is numeric. You know of a way to do that?

Hmmm, not a great way, but if you only need to check if the value is greater than 0 (which would be true only for numbers, not strings), then you could use:
```
if test $1 -gt 0; then...
```
Or, if you want to check for greater than or equal to 0, you could use:
```
if test $1 -gt 0 -o $1 = "0"; then...
```
Both of these will fail for negative numbers, or "hex" numbers like 0x300.

ghaerr, 2022-06-09:
Thanks for the `sys` testing, @Mellvik.

> Should check if the target file systems is already mounted.

I'm not sure how to do that yet in shell script..., other than some very fancy mount output piped through sed. We can't guarantee we have those programs on 360k floppies, so they can't be used with sys.

> Assumes /mnt is available, it may not be.

If we assume that the boot disk is writable, we could create a special mount directory for running sys, and then remove it afterwards. This should fix the problem, since many users may have /mnt in use.

> The ELKS MBR boot manager does not recognize Minix partitions (see below).

I'll look into this - are you saying that sys always fails when /dev/hda1 is MINIX, rather than FAT partition?

> Of course I'd like the script to fully populate the new root, maybe a -f (for 'full') would make sense, and just copy everything (in /etc, /bin, /lib, /var, /root, /tmp ) and create all devices?

A full system copy with -f shouldn't be too hard, although we should perhaps leave that to `cp -R`. That way we don't have to wonder about which directories are directly under root (except /dev). cp -R should work now except it will likely have problems copying /dev (would you like to test that)?. I'm wondering whether just keeping sys simple to create a new system with /dev devices, then using a fixed cp -R, so that it ignores /dev would be the best idea, what do you think?

> minor bug in mount - it displays the info message twice if the is an error:

Yes - unfortunately the fd: message is the kernel probe message, which is the output of floppy probing that occurs whenever the block device is opened, which is twice in this case. I'll look into it.

Thank you!

Mellvik, 2022-06-09:
> Should check if the target file systems is already mounted.
> 
> I'm not sure how to do that yet in shell script..., other than some very fancy mount output piped through sed. We can't guarantee we have those programs on 360k floppies, so they can't be used with sys.
> 
It may not have to be all that fancy, although the current mount w/o arguments doesn't lend itself well, in particular because it takes forever to determine the free space on a FAT volume. How about changing mount so that it outputs like BSD when no options are present - very simple, very fast, and easy to get the last column in a script:

# uname -a
BSD pdp11 2.11 2.11 BSD UNIX #11: Sep 9 16:28:47 PDT 2019       ***@***.***:/usr/src/sys/PIDP11  pdp11
# mount
/dev/ra0a on /
/dev/ra0g on /usr
/dev/ra0h on /home
# 

and then add an option to get the df-like output with free space we're getting today? Maybe even add a hardlink named df to the bin directory to assist?
> Assumes /mnt is available, it may not be.
> 
> If we assume that the boot disk is writable, we could create a special mount directory for running sys, and then remove it afterwards. This should fix the problem, since many users may have /mnt in use.
> 
Agreed.
> The ELKS MBR boot manager does not recognize Minix partitions (see below).
> 
> I'll look into this - are you saying that sys always fails when /dev/hda1 is MINIX, rather than FAT partition?
> 
No, I don't have a system with a Minix partition as hda1 available. The layout of the disk is FAT, minix, minix, minix.
And it's not sys that fails, it's the MBR boot. I can enter any # at the console prompt and it reports 'no bootable partition' (I don't remember the exact message). Except for the FAT partition which is #1. And presumably F(loppy) which I didn't try.
> Of course I'd like the script to fully populate the new root, maybe a -f (for 'full') would make sense, and just copy everything (in /etc, /bin, /lib, /var, /root, /tmp ) and create all devices?
>
> A [...]
I just tested cp -R with /dev, it works. I didn't expect that, this is great - and agreed, this is the way to go.
> minor bug in mount - it displays the info message twice if the is an error:
> 
> Yes - unfortunately the fd: message is the kernel probe message, which is the output of floppy probing that occurs whenever the block device is opened, which is twice in this case. I'll look into it.
> 
Minor annoyance. Low priority.
> 

Thank you!!

—M

ghaerr, 2022-06-09:
> Should check if the target file systems is already mounted.
> How about changing mount so that it outputs like BSD when no options are present 

I don't like that option because then the default will be to display little information, just for the sake of an oft-unused sys command.

Suggest adding "if mount ... || exit 1" on the initial mount. That shell script code will attempt to mount the destination, and if the mount is unsuccessful, sys will exit. The preceding mount error message will say "... already mounted". Try that and see if that solves this problem. I can then add it to the other fixes for sys.

> although the current mount w/o arguments doesn't lend itself well, in particular because it takes forever to determine the free space on a FAT volume. 

Forgot that it takes forever on large disks... this is currently the only way to determine the free space on a FAT filesystem. We could default to no free space check, and add a -f option to compute free space on FAT, if you really think that the wait is not worth it?

> The ELKS MBR boot manager does not recognize Minix partitions (see below).
>  I can enter any # at the console prompt and it reports 'no bootable partition' 

OK - I now understand. The big problem is that the sys command (through makeboot) does NOT create a partition table that reflects the partitions on the drive, the -M option is ONLY useful to stamp a predefined, single partition MBR on the HD, which will overwrite the previous MBR. This was implemented for those that had no MBR at all, and was not meant to be used to prepare a partition table. Fdisk should used for that.

Mellvik, 2022-06-09:
@ghaerr,

> Suggest adding "if mount ... || exit 1" on the initial mount. That shell script code will attempt to mount the destination, and if the mount is unsuccessful, sys will exit. The preceding mount error [...]
Sure, will do.
> although the current mount w/o arguments doesn't lend itself well, in particular because it takes forever to determine the free space on a FAT volume.
>
> Forgot that it takes forever on large disks... [...]
YEs, I like that. The wait is more than long enough to qualify this. And as suggested before, mount could then recognize 'df' as a progname and activate -f...
> The ELKS MBR boot manager does not recognize Minix partitions (see below).
> I can enter any # at the console prompt and it reports 'no bootable partition'
>
> OK - I now understand. The big problem is [...]
Well, now it's my turn to not understand. The MBR on my disk has 4 partitions, all are still there after running makeboot, and the MBR boot seems to be asking which one to boot from. If it can only boot from #1 regardless, that  makes little sense or am I missing something? Or are you saying I need disk to change the ACTIVE partition?

—M

ghaerr, 2022-06-09:
> The MBR on my disk has 4 partitions, all are still there after running makeboot, and the MBR boot seems to be asking which one to boot from.

Hmmm, maybe I'm incorrect and makeboot is keeping the original partition table, which would be nice. I haven't had time to (re)check the source code on that. I'm pretty sure it hard-codes partition #1 though (although it may look at the destination /dev/hdaX parameter to determine which to make active).

> Or are you saying I need disk to change the ACTIVE partition?

Possibly. Can you check a little further with FDISK to see what it says about the partition table after sys -M?

Mellvik, 2022-06-09:
> YEs, I like that. The wait is more than long enough to qualify this. And as suggested before, mount could then recognize 'df' as a progname and activate -f…

Actually, I looked again at the current ´df´ - it's fine (format and all) except that it only lists the root-mounted fs.

—M

ghaerr, 2022-06-16:
@Mellvik - everything requested in this issue has been fixed. I'm trying to clean up some older issues, so I'm closing this now. Please reopen another "Practical Experiences" if you have more experiences you'd like to share in the future, thank you!

closed 2022-06-16 by ghaerr