ghaerr/elks #1166: Question: Minix or FAT?

southern-computer-geek, 2022-02-19:
I see you have floppy images with FAT and Minix filesystems. I was wondering if there was any benefit to using one over the other, or are they equally good? Please forgive my ignorance.

ghaerr, 2022-02-19:
Hello @southern-computer-geek,

The Minix filesystem implements UNIX-style permissions (owner, group, world), as well as hard and sym-links, which in general are an advantage. The FAT filesystem does not implement any of these. However, compatibility with FAT filesystems is important, and some users have lots of DOS floppies etc lying around and/or can't easily create non-FAT images.

In general, if you can easily use a Minix ELKS image for first time use for booting, I'd probably recommend that, as it was built with that in mind.

Thank you!

Mellvik, 2022-02-19:
Hello @southern-computer-geek,

I'd like to add to @ghaerr's info (as a heavy user of ELKS on physical hardware) that the only real advantage of FAT is the file system size. IIRC ELKS can handle a FAT file system up to 2GB, while a Minix filesystem maxes out at 64M. Also, the Minix filesystem is significantly faster than FAT.

So - like @ghaerr, I'd strongly recommend you start with Minix fs. You can then create and mount larger FAT filesystems on the Minix root as needed. That's what I'm doing all the time.

Welcome to ELKS, it's great.

—Mellvik

southern-computer-geek, 2022-02-19:
I get it now. Thanks. 👍

closed 2022-02-23 by ghaerr