ghaerr/elks #1168: System boot does not continue until I press enter at xms

ConiKost, 2022-02-20:
**Description**
When I do boot ELKS 0.5.0, system stops at the following output:
```xms: A20 was off```

Boot continues just fine, when I do press enter. After pressing enter, the line changes to:
```xms: A20 was off now off, xms disabled, A20 error,64 ext buffers

**Configuration**
- Version (commit or release): 0.5.0
- Emulator or HW platform: NuXT (No EMS present)

**How to reproduce ?**
Boot ELKS 0.5.0

ghaerr, 2022-02-20:
Hello @ConiKost,

I think this problem is due to my having created the v0.5.0 distribution with the XMS buffer option enabled by default (which I now realize is an error).

Looking further, I see that when XMS is enabled, instead of checking for a 386 processor first (or INT 15h support) which would disable XMS, ELKS tries to test the A20 gate first, which is hanging up on your system. The reason for the hangup is that A20 line support is actually implemented on most systems using the keyboard controller, which in your case is hanging the system for a bit, which after pressing return, ends up testing for 386 processor, disabling XMS, and using EXT buffers.

In less technical language, this is an ELKS bug and two things should be done: on your system, recompile with XMS buffers configured off, and ELKS should not try testing the A20 line unless a 386 processor is present. (And we should probably not include XMS option set in the distribution floppies, anyway).

Thank you!

ConiKost, 2022-02-20:
Hello @ghaerr,
thank you for the detailed analysis :-)

> [..] on your system, recompile with XMS buffers configured off

I do wonder, if there is any way to automatically continue boot, even, when XMS is enabled in kernel, but not present?
But yes, I can confirm, after I disabled XMS and recompiled ELKS, boot does not stop anymore.

ghaerr, 2022-02-20:
> if there is any way to automatically continue boot, even, when XMS is enabled in kernel, but not present?

Yes, in general it tries to do that now, and worked (after having to hit return). My fix will be to check for 386 CPU first, and not perform the A20 check, and thus continue with normal boot, XMS disabled.

ConiKost, 2022-02-20:
Great, thank you :-) Happy to test, when ready.

Mellvik, 2022-02-20:
My 2 cents on this one:
Checking on CPU=386 first sounds like it's going to exclude the 286, which is not good …

—M

ghaerr, 2022-02-20:
> Checking on CPU=386 first sounds like it's going to exclude the 286, which is not good …

Good point, thanks. The check will have to be: "if INT 15h/1fh not enabled, AND not 386 CPU", then don't check A20 and disable XMS, otherwise check A20 and if A20 working, then enable XMS.

ghaerr, 2022-02-20:
Hello @ConiKost,

I've posted PR #1171 which should fix this problem. Please test, thank you!

ConiKost, 2022-02-20:
Hello @ghaerr 

> I've posted PR https://github.com/jbruchon/elks/pull/1171 which should fix this problem. Please test, thank you!

looks good! I applied against master and recompiled ELKS. Boot does now continue and does not stop.

> xms: disabled, required 386, 64 ext buffers, 65536 ram.

Just a small nit pick: Wouldn't be a better something like ```requires 286 or newer```. AFAIK, 286 is also supported?

ghaerr, 2022-02-20:
> Wouldn't be a better something like requires 286 or newer. AFAIK, 286 is also supported?

No, unreal mode requires a 32-bit (386+ CPU). 80286 is only supported if the system BIOS implements INT 15h.

PR #1171 (2022-02-20): [kernel] Check for 386 CPU before enabling A20 for XMS buffers
Should fix #1168.
closed 2022-02-20 by ghaerr