ghaerr/elks #1169: Toolchain build fails

ConiKost, 2022-02-20:
**Description**
I am trying to build ELKS toolchain on my Gentoo. It looks like, due newer? GCC, it fails. IIUC, -Werror=maybe-uninitialized seems the main cause, being set by default now?

Note: Using my Ubuntu 20.04 LTS vm works fine, but it uses older GCC.

```
/bin/sh ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../../binutils-src/bfd  -DBINDIR='"/home/conikost/elks/cross/bin"' -I. -I../../binutils-src/bfd -I../../binutils-src/bfd/../include  -DHAVE_i386_elf32_vec -DHAVE_i386_elks_vec -DHAVE_i386_msdos_vec -DHAVE_i386_aout_vec -DHAVE_elf32_le_vec -DHAVE_elf32_be_vec -DHAVE_plugin_vec   -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wstack-usage=262144 -Werror -I../../binutils-src/bfd/../zlib -g -O2 -MT plugin.lo -MD -MP -MF .deps/plugin.Tpo -c -o plugin.lo ../../binutils-src/bfd/plugin.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../binutils-src/bfd -DBINDIR=\"/home/conikost/elks/cross/bin\" -I. -I../../binutils-src/bfd -I../../binutils-src/bfd/../include -DHAVE_i386_elf32_vec -DHAVE_i386_elks_vec -DHAVE_i386_msdos_vec -DHAVE_i386_aout_vec -DHAVE_elf32_le_vec -DHAVE_elf32_be_vec -DHAVE_plugin_vec -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wstack-usage=262144 -Werror -I../../binutils-src/bfd/../zlib -g -O2 -MT plugin.lo -MD -MP -MF .deps/plugin.Tpo -c ../../binutils-src/bfd/plugin.c -o plugin.o
In function 'load_plugin',
    inlined from 'bfd_plugin_object_p' at ../../binutils-src/bfd/plugin.c:424:53:
../../binutils-src/bfd/plugin.c:403:20: error: 'valid_plugin' may be used uninitialized [-Werror=maybe-uninitialized]
  403 |         has_plugin = valid_plugin;
      |         ~~~~~~~~~~~^~~~~~~~~~~~~~
../../binutils-src/bfd/plugin.c: In function 'bfd_plugin_object_p':
../../binutils-src/bfd/plugin.c:397:11: note: 'valid_plugin' declared here
  397 |       int valid_plugin;
      |           ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make[5]: *** [Makefile:1602: plugin.lo] Error 1
make[5]: Leaving directory '/home/conikost/elks/cross/build/binutils-build/bfd'
make[4]: *** [Makefile:1645: all-recursive] Error 1
make[4]: Leaving directory '/home/conikost/elks/cross/build/binutils-build/bfd'
make[3]: *** [Makefile:1170: all] Error 2
make[3]: Leaving directory '/home/conikost/elks/cross/build/binutils-build/bfd'
make[2]: *** [Makefile:2691: all-bfd] Error 2
make[2]: Leaving directory '/home/conikost/elks/cross/build/binutils-build'
make[1]: *** [Makefile:849: all] Error 2
make[1]: Leaving directory '/home/conikost/elks/cross/build/binutils-build'
make: *** [Makefile:54: /home/conikost/elks/cross/build/.binutils.build] Error 2
make: Leaving directory '/home/conikost/elks/tools'
Build script has terminated with error 1
```

```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-pc-linux-gnu/11.2.1/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /var/tmp/portage/sys-devel/gcc-11.2.1_p20220115/work/gcc-11-20220115/configure --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/gcc-bin/11.2.1 --includedir=/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/include --datadir=/usr/share/gcc-data/x86_64-pc-linux-gnu/11.2.1 --mandir=/usr/share/gcc-data/x86_64-pc-linux-gnu/11.2.1/man --infodir=/usr/share/gcc-data/x86_64-pc-linux-gnu/11.2.1/info --with-gxx-include-dir=/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/include/g++-v11 --with-python-dir=/share/gcc-data/x86_64-pc-linux-gnu/11.2.1/python --enable-languages=c,c++,jit,fortran --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --disable-libunwind-exceptions --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo 11.2.1_p20220115 p4' --disable-esp --enable-libstdcxx-time --with-build-config=bootstrap-lto --disable-libstdcxx-pch --enable-host-shared --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --enable-multilib --with-multilib-list=m32,m64 --disable-fixed-point --enable-targets=all --enable-libgomp --disable-libssp --disable-libada --disable-cet --disable-systemtap --disable-valgrind-annotations --enable-vtable-verify --with-zstd --enable-lto --with-isl --disable-isl-version-check --enable-default-pie --enable-default-ssp
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.1 20220115 (Gentoo 11.2.1_p20220115 p4)
```

ghaerr, 2022-02-20:
Hello @ConiKost, hello @tkchia,

Thanks for the report. I'll ask @tkchia, our compiler maintainer, what he thinks about this.

Thank you!

ConiKost, 2022-02-20:
Just a note:

I can workaround the problem by:
```CFLAGS="${CFLAGS} -Wno-error=maybe-uninitialized" ./build.sh```

But this results in a new problem:

```
g++ -DHAVE_CONFIG_H -I. -I../../binutils-src/gold  -I../../binutils-src/gold -I../../binutils-src/gold/../include -I../../binutils-src/gold/../elfcpp -DLOCALEDIR="\"/home/conikost/elks/cross/share/locale\"" -DBINDIR="\"/home/conikost/elks/cross/bin\"" -DTOOLBINDIR="\"/home/conikost/elks/cross/ia16-elf/bin\"" -DTOOLLIBDIR="\"/home/conikost/elks/cross/ia16-elf/lib\""   -W -Wall    -Wstack-usage=262144 -Werror -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -frandom-seed=dwp.o -I../../binutils-src/gold/../zlib -pthread -g -O2 -MT dwp.o -MD -MP -MF .deps/dwp.Tpo -c -o dwp.o ../../binutils-src/gold/dwp.cc
mv -f .deps/dwp.Tpo .deps/dwp.Po
g++ -W -Wall    -Wstack-usage=262144 -Werror -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -frandom-seed=dwp -I../../binutils-src/gold/../zlib -pthread -g -O2  -static-libstdc++ -static-libgcc  -o dwp dwp.o libgold.a ../libiberty/libiberty.a     -L./../zlib -lz
g++ -DHAVE_CONFIG_H -I. -I../../binutils-src/gold  -I../../binutils-src/gold -I../../binutils-src/gold/../include -I../../binutils-src/gold/../elfcpp -DLOCALEDIR="\"/home/conikost/elks/cross/share/locale\"" -DBINDIR="\"/home/conikost/elks/cross/bin\"" -DTOOLBINDIR="\"/home/conikost/elks/cross/ia16-elf/bin\"" -DTOOLLIBDIR="\"/home/conikost/elks/cross/ia16-elf/lib\""   -W -Wall    -Wstack-usage=262144 -Werror -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -frandom-seed=main.o -I../../binutils-src/gold/../zlib -pthread -g -O2 -MT main.o -MD -MP -MF .deps/main.Tpo -c -o main.o ../../binutils-src/gold/main.cc
../../binutils-src/gold/main.cc: In Funktion »int main(int, char**)«:
../../binutils-src/gold/main.cc:301:35: Fehler: »mallinfo mallinfo()« ist veraltet [-Werror=deprecated-declarations]
  301 |       struct mallinfo m = mallinfo();
      |                           ~~~~~~~~^~
In Datei, eingebunden von ../../binutils-src/gold/main.cc:35:
/usr/include/malloc.h:113:24: Anmerkung: hier deklariert
  113 | extern struct mallinfo mallinfo (void) __THROW __MALLOC_DEPRECATED;
      |                        ^~~~~~~~
cc1plus: Alle Warnungen werden als Fehler behandelt
make[5]: *** [Makefile:1133: main.o] Fehler 1
make[5]: Verzeichnis „/home/conikost/elks/cross/build/binutils-build/gold“ wird verlassen
make[4]: *** [Makefile:1156: all-recursive] Fehler 1
make[4]: Verzeichnis „/home/conikost/elks/cross/build/binutils-build/gold“ wird verlassen
make[3]: *** [Makefile:896: all] Fehler 2
make[3]: Verzeichnis „/home/conikost/elks/cross/build/binutils-build/gold“ wird verlassen
make[2]: *** [Makefile:6044: all-gold] Fehler 2
make[2]: Verzeichnis „/home/conikost/elks/cross/build/binutils-build“ wird verlassen
make[1]: *** [Makefile:849: all] Fehler 2
make[1]: Verzeichnis „/home/conikost/elks/cross/build/binutils-build“ wird verlassen
make: *** [Makefile:54: /home/conikost/elks/cross/build/.binutils.build] Fehler 2
make: Verzeichnis „/home/conikost/elks/tools“ wird verlassen
Build script has terminated with error 1
````

But I am _not_ able to workaround this by ```CFLAGS="${CFLAGS} -Wno-error=deprecated-declarations -Wno-error=maybe-uninitialized" ./build.sh```

tkchia, 2022-02-20:
Hello @ConiKost, hello @ghaerr,

Yes, it looks like the slightly old version of Binutils used by ELKS is triggering compiler warnings in GCC 11.x.  And `binutils-ia16` is compiled with warnings treated as errors (`-Werror`) on default.

One way to work around this is to pass a `--disable-werror` switch to Binutils's `configure` script.  To work around this, you can probably modify the `tools/Makefile` to insert this option under the `$(BUILDDIR)/.binutils.build` rule &mdash; this should work:

> `cd $(BUILDDIR)/binutils-build && ../binutils-src/configure --target=ia16-elf` ... `--disable-sim --disable-nls` **`--disable-werror`**

Thank you!

ConiKost, 2022-02-20:
Hi @tkchia,
thank you!

> Yes, it looks like the slightly old version of Binutils used by ELKS is triggering compiler warnings in GCC 11.x. And binutils-ia16 is compiled with warnings treated as errors (-Werror) on default.

@ghaerr Maybe ELKS repo could be updated to a newer toolchain snapshot then?

> One way to work around this is to pass a --disable-werror switch to Binutils's configure script. You can probably modify the tools/Makefile to insert this option under the $(BUILDDIR)/.binutils.build rule — this should work:

Thank you, it worked! I was able to compile the toolchain now.

ghaerr, 2022-02-20:
> Maybe ELKS repo could be updated to a newer toolchain snapshot then?

Yes, but we try to keep that to a minimum when contributions are heavy, so other developers don't have to recompile their toolchain.

ConiKost, 2022-02-20:
> Yes, but we try to keep that to a minimum when contributions are heavy, so other developers don't have to recompile their toolchain.

I understand. But thb, I need only about 35 minutes to rebuild toolchain :-) Anyway, I bumped locally to newest version and it works now without any modification. Compilation worked and boot from created floppy image too. But I guess, you have more and better ways to test? Or do you want me to fill a PR?

Heres the diff of my used versions:
```
diff --git a/tools/Makefile b/tools/Makefile
index 8c62836f..042d1e19 100644
--- a/tools/Makefile
+++ b/tools/Makefile
@@ -23,7 +23,7 @@ BUILDDIR=$(CROSSDIR)/build

 # BINUTILS for IA16

-BINUTILS_VER=164dc55d3d9488a487d39c2e7f3f8cadf6dc12f5
+BINUTILS_VER=7d32a05e75cb19749708881a609ee96bd6bfb107
 BINUTILS_DIST=binutils-ia16-$(BINUTILS_VER)

 $(DISTDIR)/$(BINUTILS_DIST).tar.gz:
@@ -50,7 +50,7 @@ $(BUILDDIR)/.binutils.src: $(DISTDIR)/$(BINUTILS_DIST).tar.gz
 $(BUILDDIR)/.binutils.build: $(BUILDDIR)/.binutils.src
        cd $(BUILDDIR) && rm -rf binutils-build
        mkdir -p $(BUILDDIR)/binutils-build
-       cd $(BUILDDIR)/binutils-build && ../binutils-src/configure --target=ia16-elf --prefix="$(CROSSDIR)" --enable-ld=default --enable-gold=yes --enable-targets=ia16-elf --enable-x86-hpa-segelf=yes --disable-gdb --disable-libdecnumber --disable-readline --disable-sim --disable-nls --disable-werror
+       cd $(BUILDDIR)/binutils-build && ../binutils-src/configure --target=ia16-elf --prefix="$(CROSSDIR)" --enable-ld=default --enable-gold=yes --enable-targets=ia16-elf --enable-x86-hpa-segelf=yes --disable-gdb --disable-libdecnumber --disable-readline --disable-sim --disable-nls
        $(MAKE) -C $(BUILDDIR)/binutils-build $(PARALLEL)
        touch $(BUILDDIR)/.binutils.build

@@ -62,11 +62,11 @@ all:: $(CROSSDIR)/.binutils.install

 # GCC prerequisites: GNU MP, MPFR, and MPC libraries

-GMP_VER=6.1.2
+GMP_VER=6.2.1
 GMP_DIST=gmp-$(GMP_VER)
-MPFR_VER=3.1.5
+MPFR_VER=4.1.0
 MPFR_DIST=mpfr-$(MPFR_VER)
-MPC_VER=1.0.3
+MPC_VER=1.2.1
 MPC_DIST=mpc-$(MPC_VER)

 $(DISTDIR)/$(GMP_DIST).tar.bz2:
@@ -97,7 +97,7 @@ $(DISTDIR)/$(MPC_DIST).tar.gz:

 # GCC for IA16

-GCC_VER=17bd8a491ca31f8fd867b1f1a380cbbc5ef53b07
+GCC_VER=f0d0b1182e1a864292af39c45a94bcd627663aac
 GCC_DIST=gcc-ia16-$(GCC_VER)

 $(DISTDIR)/$(GCC_DIST).tar.gz:
@@ -154,7 +154,7 @@ all:: $(CROSSDIR)/.gcc.install

 # EMU86

-EMU86_VER=f99937262b2566d3d20339e5b200e8483cdec3a2
+EMU86_VER=9844dd0c63875a03339f7b6310b937e39b8e4796
 EMU86_DIST=emu86-$(EMU86_VER)

 $(DISTDIR)/$(EMU86_DIST).tar.gz:
diff --git a/tools/build.sh b/tools/build.sh
index 184a91c9..46eacdda 100755
```

closed 2022-02-24 by ghaerr